fwupd/fwupd #5871: Add tpm2.0-tools to dependencies on OpenSUSE

Jackaed, 2023-06-02:
Possibly the wrong place to list this bug, but on OpenSUSE Tumbleweed fwupd does not install tpm2.0-tools as a dependency. This led me on a bit of a wild goose chase as to why my TPM was not being detected up by fwupd. If possible, please add this to the list of dependencies.

superm1, 2023-06-02:
> Possibly the wrong place to list this bug, but on OpenSUSE Tumbleweed fwupd does not install tpm2.0-tools as a dependency. This led me on a bit of a wild goose chase as to why my TPM was not being detected up by fwupd. If possible, please add this to the list of dependencies.

The library is needed at runtime but the actual tools aren't. What went wrong with them not installed? Could you provide some verbose daemon logs attached to the bug for both cases?

hughsie, 2023-06-02:
I think you need to file this in the suse bug tracker too fwiw. Thanks.

Jackaed, 2023-06-03:
Having removed tpm2.0-tools fwupd continued to recognise my tpm, however reinstalling it and then removing tpm2.0-tools with --clean-deps also uninstalled the following packages: `libtss2-fapi1 libtss2-tcti-device0 libtss2-tctildr0 libtss2-tcti-tabrmd0 system-user-tss tpm2.0-abrmd tpm2.0-tools tpm2-0-tss`.

Having done this, when running `sudo fwupdtool security` I get the following output:

```
Loading…                 [******                                 ]15:13:23.043 FuEngine             failed to add device usb:01:00:03: failed to get static tool info from device: USB error on device 05e3:0610 : Pipe error [-9]
Loading…                 [************************               ]15:13:24.336 FuEngine             failed to add device /sys/devices/platform/STM0151:00/tpm/tpm0: failed to initialize TPM library
Loading…                 [************************************** ]
Host Security ID: HSI:0! (v1.8.12)

HSI-1
✔ Fused platform:                Locked
✔ Supported CPU:                 Valid
✔ UEFI platform key:             Valid
✔ UEFI secure boot:              Enabled
✘ TPM v2.0:                      Not found

HSI-2
✔ BIOS rollback protection:      Enabled
✔ IOMMU:                         Enabled
✔ Platform debugging:            Locked
✔ SPI write protection:          Enabled

HSI-3
✔ Pre-boot DMA protection:       Enabled
✔ SPI replay protection:         Enabled
✔ Suspend-to-idle:               Enabled
✔ Suspend-to-ram:                Disabled

HSI-4
✔ Encrypted RAM:                 Encrypted
✔ Processor rollback protection: Enabled

Runtime Suffix -!
✔ Linux kernel:                  Untainted
✔ fwupd plugins:                 Untainted
✘ Linux kernel lockdown:         Disabled
✘ Linux swap:                    Unencrypted

This system has a low HSI security level.
 » https://fwupd.github.io/hsi.html#low-security-level

This system has HSI runtime issues.
 » https://fwupd.github.io/hsi.html#hsi-runtime-suffix
```

I'm unsure as to which package it is, but one or several of the dependencies that got installed with `tpm2.0-tools` should also be a dependency of fwupd.

The output I get with tpm2.0-tools (and dependencies) installed is:

```
Loading…                 [******                                 ]15:16:42.434 FuEngine             failed to add device usb:01:00:03: failed to get static tool info from device: USB error on device 05e3:0610 : Pipe error [-9]
Loading…                 [************************************** ]
Host Security ID: HSI:4! (v1.8.12)

HSI-1
✔ Fused platform:                Locked
✔ Supported CPU:                 Valid
✔ TPM empty PCRs:                Valid
✔ TPM v2.0:                      Found
✔ UEFI platform key:             Valid
✔ UEFI secure boot:              Enabled

HSI-2
✔ BIOS rollback protection:      Enabled
✔ IOMMU:                         Enabled
✔ Platform debugging:            Locked
✔ SPI write protection:          Enabled
✔ TPM PCR0 reconstruction:       Valid

HSI-3
✔ Pre-boot DMA protection:       Enabled
✔ SPI replay protection:         Enabled
✔ Suspend-to-idle:               Enabled
✔ Suspend-to-ram:                Disabled

HSI-4
✔ Encrypted RAM:                 Encrypted
✔ Processor rollback protection: Enabled

Runtime Suffix -!
✔ Linux kernel:                  Untainted
✔ fwupd plugins:                 Untainted
✘ Linux kernel lockdown:         Disabled
✘ Linux swap:                    Unencrypted

This system has HSI runtime issues.
 » https://fwupd.github.io/hsi.html#hsi-runtime-suffix
```

Note that the issue it finds with the sub port in both instances is unrelated.

superm1, 2023-06-03:
Thanks for the details, you'll need to add these into your distro bug tracker.

This is expected behavior from upstream fwupd.

closed 2023-06-03 by superm1