fwupd/fwupd #5877: Failed to update metadata for lvfs

iyanmv, 2023-06-05:
**Describe the bug**

When running `fwupdmgr refresh` I get the following error:

```
Failed to update metadata for lvfs: checksum failure: failed to verify data, expected b032f42970cb96a7ebe2b371c36f1a01568de9f686617487dc19ed24e3b76d6c and got 4fd797106833dfa69d1bafd8769f2decc82301cb9f1c6038a863265adfbd4f9d
```

I tried with `--force` and `--verbose` flags and I get:

```
$ fwupdmgr refresh --force --verbose
(fwupdmgr:207307): GLib-DEBUG: 12:52:39.244: setenv()/putenv() are not thread-safe and should not be used after threads are created
(pkttyagent:207309): GLib-GIO-DEBUG: 12:52:39.247: Using cross-namespace EXTERNAL authentication (this will deadlock if server is GDBus < 2.73.3)
(fwupdmgr:207307): GLib-GIO-DEBUG: 12:52:39.258: _g_io_module_get_default: Found default implementation dconf (DConfSettingsBackend) for ‘gsettings-backend’
(fwupdmgr:207307): dconf-DEBUG: 12:52:39.258: watch_fast: "/system/proxy/" (establishing: 0, active: 0)
(fwupdmgr:207307): dconf-DEBUG: 12:52:39.259: watch_fast: "/system/proxy/http/" (establishing: 0, active: 0)
(fwupdmgr:207307): dconf-DEBUG: 12:52:39.259: watch_fast: "/system/proxy/https/" (establishing: 0, active: 0)
(fwupdmgr:207307): dconf-DEBUG: 12:52:39.259: watch_fast: "/system/proxy/ftp/" (establishing: 0, active: 0)
(fwupdmgr:207307): dconf-DEBUG: 12:52:39.259: watch_fast: "/system/proxy/socks/" (establishing: 0, active: 0)
(fwupdmgr:207307): dconf-DEBUG: 12:52:39.259: unwatch_fast: "/system/proxy/" (active: 0, establishing: 1)
(fwupdmgr:207307): dconf-DEBUG: 12:52:39.259: unwatch_fast: "/system/proxy/http/" (active: 0, establishing: 1)
(fwupdmgr:207307): dconf-DEBUG: 12:52:39.259: unwatch_fast: "/system/proxy/https/" (active: 0, establishing: 1)
(fwupdmgr:207307): dconf-DEBUG: 12:52:39.259: unwatch_fast: "/system/proxy/ftp/" (active: 0, establishing: 1)
(fwupdmgr:207307): dconf-DEBUG: 12:52:39.259: unwatch_fast: "/system/proxy/socks/" (active: 0, establishing: 1)
(fwupdmgr:207307): GLib-GIO-DEBUG: 12:52:39.259: Using cross-namespace EXTERNAL authentication (this will deadlock if server is GDBus < 2.73.3)
(fwupdmgr:207307): dconf-DEBUG: 12:52:39.260: watch_established: "/system/proxy/" (establishing: 0)
(fwupdmgr:207307): dconf-DEBUG: 12:52:39.260: watch_established: "/system/proxy/http/" (establishing: 0)
(fwupdmgr:207307): dconf-DEBUG: 12:52:39.260: watch_established: "/system/proxy/https/" (establishing: 0)
(fwupdmgr:207307): dconf-DEBUG: 12:52:39.260: watch_established: "/system/proxy/ftp/" (establishing: 0)
(fwupdmgr:207307): dconf-DEBUG: 12:52:39.260: watch_established: "/system/proxy/socks/" (establishing: 0)
(fwupdmgr:207307): GLib-GIO-DEBUG: 12:52:39.270: _g_io_module_get_default: Found default implementation libproxy (GLibproxyResolver) for ‘gio-proxy-resolver’
(fwupdmgr:207307): GLib-GIO-DEBUG: 12:52:39.270: Using cross-namespace EXTERNAL authentication (this will deadlock if server is GDBus < 2.73.3)
(fwupdmgr:207307): Fwupd-DEBUG: 12:52:39.271: Emitting ::status-changed() [idle]
Updating lvfs
Fwupd-INFO: 12:52:39.274: downloading https://cdn.fwupd.org/downloads/firmware.xml.xz.jcat
(fwupdmgr:207307): Fwupd-DEBUG: 12:52:39.280: Emitting ::status-changed() [downloading]
Downloading…             [    \                                  ]Fwupd-INFO: 12:52:39.407: download progress: 100%
(fwupdmgr:207307): Fwupd-DEBUG: 12:52:39.407: Emitting ::status-changed() [idle]
Fwupd-INFO: 12:52:39.407: status-code was 200
Fwupd-INFO: 12:52:39.408: changing metadata URI from https://cdn.fwupd.org/downloads/firmware.xml.xz to https://cdn.fwupd.org/downloads/firmware-06368-stable.xml.xz
Fwupd-INFO: 12:52:39.408: downloading https://cdn.fwupd.org/downloads/firmware-06368-stable.xml.xz
(fwupdmgr:207307): Fwupd-DEBUG: 12:52:39.409: Emitting ::status-changed() [downloading]
Downloading…             [***************************************]Fwupd-INFO: 12:52:39.499: download progress: 1%
Downloading…             [                                       ]Fwupd-INFO: 12:52:39.504: download progress: 3%
Downloading…             [*                                      ]Fwupd-INFO: 12:52:39.511: download progress: 5%
Downloading…             [*                                      ]Fwupd-INFO: 12:52:39.515: download progress: 8%
Downloading…             [***                                    ]Fwupd-INFO: 12:52:39.519: download progress: 10%
Downloading…             [***                                    ]Fwupd-INFO: 12:52:39.522: download progress: 12%
Downloading…             [****                                   ]Fwupd-INFO: 12:52:39.524: download progress: 15%
Downloading…             [*****                                  ]Fwupd-INFO: 12:52:39.526: download progress: 17%
Downloading…             [******                                 ]Fwupd-INFO: 12:52:39.527: download progress: 21%
Downloading…             [********                               ]Fwupd-INFO: 12:52:39.531: download progress: 22%
Downloading…             [********                               ]Fwupd-INFO: 12:52:39.535: download progress: 24%
Downloading…             [*********                              ]Fwupd-INFO: 12:52:39.535: download progress: 28%
Downloading…             [**********                             ]Fwupd-INFO: 12:52:39.537: download progress: 31%
Downloading…             [************                           ]Fwupd-INFO: 12:52:39.538: download progress: 35%
Downloading…             [*************                          ]Fwupd-INFO: 12:52:39.540: download progress: 38%
Downloading…             [**************                         ]Fwupd-INFO: 12:52:39.543: download progress: 46%
Downloading…             [*****************                      ]Fwupd-INFO: 12:52:39.543: download progress: 48%
Downloading…             [******************                     ]Fwupd-INFO: 12:52:39.546: download progress: 50%
Downloading…             [*******************                    ]Fwupd-INFO: 12:52:39.548: download progress: 55%
Downloading…             [*********************                  ]Fwupd-INFO: 12:52:39.549: download progress: 57%
Downloading…             [**********************                 ]Fwupd-INFO: 12:52:39.552: download progress: 64%
Downloading…             [************************               ]Fwupd-INFO: 12:52:39.556: download progress: 83%
Downloading…             [********************************       ]Fwupd-INFO: 12:52:39.560: download progress: 100%
(fwupdmgr:207307): Fwupd-DEBUG: 12:52:39.560: Emitting ::status-changed() [idle]
Fwupd-INFO: 12:52:39.560: status-code was 200

Failed to update metadata for lvfs: checksum failure: failed to verify data, expected b032f42970cb96a7ebe2b371c36f1a01568de9f686617487dc19ed24e3b76d6c and got 4fd797106833dfa69d1bafd8769f2decc82301cb9f1c6038a863265adfbd4f9d
```

However, if I manually download that file with `curl` and pass it to `sha256sum` I get the correct hash.

```shell
$ curl -s https://cdn.fwupd.org/downloads/firmware-06368-stable.xml.xz | sha256sum
b032f42970cb96a7ebe2b371c36f1a01568de9f686617487dc19ed24e3b76d6c  -
```

**fwupd version information**

```shell
$ fwupdmgr --version
compile   org.freedesktop.fwupd         1.9.1
compile   com.hughsie.libxmlb           0.3.11
compile   com.hughsie.libjcat           0.1.13
runtime   org.freedesktop.fwupd-efi     1.4
compile   org.freedesktop.gusb          0.4.5
runtime   com.hughsie.libjcat           0.1.13
runtime   com.dell.libsmbios            2.4
runtime   org.freedesktop.gusb          0.4.5
runtime   org.kernel                    6.3.5-arch1-1
runtime   org.freedesktop.fwupd         1.9.1
```

Please note how you installed it (`apt`, `dnf`, `pacman`, source, etc): `pacman`

**Additional questions**

- Operating system and version: Arch Linux

iyanmv, 2023-06-05:
```shell
wget --quiet -O - https://cdn.fwupd.org/downloads/firmware-06368-stable.xml.xz | sha256sum
b032f42970cb96a7ebe2b371c36f1a01568de9f686617487dc19ed24e3b76d6c  -
```

hughsie, 2023-06-05:
> got 4fd797106833dfa69d1bafd8769f2decc82301cb9f1c6038a863265adfbd4f9d

Any idea what that is? I'm guessing it was some kind of transitory problem? Could it have been a captive html portal page or something?

iyanmv, 2023-06-05:
I have no idea what it is. Where is `fwupd` caching downloaded files so I can have a look?

hughsie, 2023-06-05:
If you use `fwupdmgr download https://cdn.fwupd.org/downloads/firmware-06368-stable.xml.xz` what's the SHA 256?

iyanmv, 2023-06-05:
I cannot reproduce the issue anymore... weird. I get the right hash with `fwupdmgr download` and `wget`, and `fwupdmgr refresh --force` does not complain anymore. I didn't update or reboot the laptop. The only difference is the network I'm connected (at work it was complaining, now at home it works). I will close, and reopen if it happens again in the future. I will try again tomorrow the `fwupdmgr download` command.