fwupd/fwupd #5878: Microsoft Surface TBT4 dock  PD component (TI PD controller TPS65988)

alebrati, 2023-06-05:
issue:
PD not detected (TI contgroller TPS65988).
we had a session on this with Denis (Colabora). had tried to add in quirk but no luck.
sending this bug ID to Denis as well.
[microsoft.log](https://github.com/fwupd/fwupd/files/11652740/microsoft.log)

localhost /home # fwupdmgr get-devices
Google Kano
?
??Display controller:
?     Device ID:          5792b48846ce271fab11c4a545f7a3df0d36e00a
?     Current version:    04
?     Vendor:             PCI:0x8086
?     GUIDs:              7114c807-404e-558e-89a4-120d80d7cbdb ? PCI\VEN_8086&DEV_A7A1
?                         6b281974-60e8-5d60-a3c8-71d9a1cddead ? PCI\VEN_8086&DEV_A7A1&REV_04
?                         5a5df648-815a-57fb-9973-825a267e6407 ? PCI\VEN_8086&DEV_A7A1&SUBSYS_8086A7A1
?                         2ea862b5-8d06-5a16-89f6-361419fc5064 ? PCI\VEN_8086&DEV_A7A1&SUBSYS_8086A7A1&REV_04
?                         a2f16d8e-7fc7-5e42-9fa2-bb6f49121eef ? PCI\VEN_8086&DEV_A7A1&REV_00
?                         ddc9daf8-24b2-55ea-b00e-126d9d5104e4 ? PCI\VEN_8086&DEV_A7A1&SUBSYS_8086A7A1&REV_00
?     Device Flags:       ? Internal device
?                         ? Cryptographic hash verification is available
?
??Integrated Webcam?:
?     Device ID:          08d460be0f1f9f128413f816022a6439e0078018
?     Summary:            Fake webcam
?     Current version:    1.2.2
?     Minimum Version:    1.2.0
?     Bootloader Version: 0.1.2
?     Vendor:             ACME Corp. (USB:0x046D)
?     GUID:               b585990a-003e-5270-89d5-3705a17f9a43
?     Device Flags:       ? Updatable
?                         ? System requires external power source
?                         ? Supported on remote server
?                         ? Cryptographic hash verification is available
?                         ? Unsigned Payload
?
??Surface Thunderbolt? 4 Dock:
?     Device ID:          6d37dc27fe4d0248e2e74209ac3682427aa9a8b8
?     Current version:    40.84
?     Vendor:             Microsoft (THUNDERBOLT:0x017C, TBT:0x017C)
?     GUIDs:              7a7f9900-3d89-50a1-9d32-ce7787d0b1f7 ? THUNDERBOLT\VEN_017C&DEV_0856
?                         aeeb8c74-7f8b-5cfc-b336-e34befeee3d0 ? THUNDERBOLT\VEN_017C&DEV_0856&REV_00
?                         2ad4a29b-92b8-5231-b184-4340d24d453a ? TBT-017c0856
?                         2385d8aa-d2c5-5e79-a5bd-b13482ef88e7 ? TBT-017c0856-controller0-1
?     Device Flags:       ? Updatable
?                         ? System requires external power source
?                         ? Device stages updates
?                         ? Signed Payload
?
??USB3.0 Hub:
?     Device ID:          cf294bf55b333004beb7c41f952c1838c23e1f4a
?     Current version:    40.84
?     Vendor:             Intel Corporation. (USB:0x8087)
?     GUIDs:              b3f8b518-ebdd-5969-9b44-6b096eff5b9b ? USB\VID_8087&PID_0B40
?                         27f4ed60-ed53-52a7-ae28-ea87d9a23302 ? USB\VID_8087&PID_0B40&REV_1234
?                         c6e7a14f-f13c-55e5-99de-d59461db1d27 ? USB\VID_8087&PID_0B40&HUB_0000
?                         2ad4a29b-92b8-5231-b184-4340d24d453a ? TBT-017c0856
?     Device Flags:       ? Updatable
?                         ? Signed Payload
?
??USB4 Retimer:
      Device ID:          16d35c510618116e7899ae45e62605572872e713
      Summary:            A physical layer protocol-aware, software-transparent extension device that forms two separate electrical link segments
      Current version:    10.00
      Vendor:             Intel (THUNDERBOLT:0x8087)
      GUIDs:              e6f2e4ac-f76a-5fc4-b36a-6f9cb548689e ? THUNDERBOLT\VEN_8087&DEV_15EE
                          f00257c8-0e09-5a32-bbb3-f49748ae6cc5 ? THUNDERBOLT\VEN_8087&DEV_15EE&REV_00
                          6d71ac25-b368-5e7b-ac62-32df365fa640 ? TBT-808715ee-retimer0-0:1.1
      Device Flags:       ? Internal device
                          ? Updatable
                          ? Device stages updates

alebrati, 2023-06-05:
[
[fwupdtool_get_devices_verbos.txt](https://github.com/fwupd/fwupd/files/11653126/fwupdtool_get_devices_verbos.txt)
[lsusb.txt](https://github.com/fwupd/fwupd/files/11653127/lsusb.txt)
](url)

alebrati, 2023-06-06:
dmesg -w dettach and attach logs:

[ 2942.412118] usb 3-1: USB disconnect, device number 17
[ 2942.412129] usb 3-1.4: USB disconnect, device number 18
[ 2942.412133] usb 3-1.4.5: USB disconnect, device number 21
[ 2942.425822] usb 2-1: USB disconnect, device number 8
[ 2942.425826] usb 2-1.4: USB disconnect, device number 9
[ 2942.479246] usb 3-1.5: USB disconnect, device number 19
[ 2942.835208] thunderbolt 0000:00:0d.2: control channel starting...
[ 2942.835215] thunderbolt 0000:00:0d.2: starting TX ring 0
[ 2942.835220] thunderbolt 0000:00:0d.2: enabling interrupt at register 0x38200 bit 0 (0x0 -> 0x1)
[ 2942.835224] thunderbolt 0000:00:0d.2: starting RX ring 0
[ 2942.835228] thunderbolt 0000:00:0d.2: enabling interrupt at register 0x38200 bit 12 (0x1 -> 0x1001)
[ 2942.836367] thunderbolt 0000:00:0d.2: ICM rtd3 veto=0x00000000
[ 2943.152537] pcieport 0000:02:00.0: can't change power state from D3cold to D0 (config space inaccessible)
[ 2943.152574] pcieport 0000:00:07.0: pciehp: Slot(0): Card not present
[ 2943.152592] pcieport 0000:03:04.0: can't change power state from D3cold to D0 (config space inaccessible)
[ 2943.152611] pcieport 0000:03:03.0: can't change power state from D3cold to D0 (config space inaccessible)
[ 2943.152625] pcieport 0000:03:02.0: can't change power state from D3cold to D0 (config space inaccessible)
[ 2943.152637] pcieport 0000:03:01.0: can't change power state from D3cold to D0 (config space inaccessible)
[ 2943.152649] pcieport 0000:03:00.0: can't change power state from D3cold to D0 (config space inaccessible)
[ 2943.156426] pci_bus 0000:04: busn_res: [bus 04] is released
[ 2943.160154] pci 0000:03:00.0: Removing from iommu group 18
[ 2943.160245] pci_bus 0000:05: busn_res: [bus 05-11] is released
[ 2943.160453] pci 0000:03:01.0: Removing from iommu group 19
[ 2943.160528] pci_bus 0000:12: busn_res: [bus 12-1e] is released
[ 2943.160610] pci 0000:03:02.0: Removing from iommu group 20
[ 2943.160672] pci_bus 0000:1f: busn_res: [bus 1f-2b] is released
[ 2943.160805] pci 0000:03:03.0: Removing from iommu group 21
[ 2943.160862] pci_bus 0000:2c: busn_res: [bus 2c] is released
[ 2943.160965] pci 0000:03:04.0: Removing from iommu group 22
[ 2943.161009] pci_bus 0000:03: busn_res: [bus 03-2c] is released
[ 2943.161215] pci 0000:02:00.0: Removing from iommu group 17
[ 2943.344780] thunderbolt 0-1: device disconnected
[ 2944.848550] usb 2-1.4.4: Disable of device-initiated U2 failed.
[ 2944.848667] cdc_ncm 2-1.4.4:1.0 enxd8e2dfc69ce2: unregister 'cdc_ncm' usb-0000:00:0d.0-1.4.4, CDC NCM
[ 2944.877142] usb 2-1.4.4: USB disconnect, device number 10
[ 2949.039105] thunderbolt 0000:00:0d.2: current switch config:
[ 2949.039113] thunderbolt 0000:00:0d.2:  USB4 Switch: 8087:b26 (Revision: 3, TB Version: 32)
[ 2949.039119] thunderbolt 0000:00:0d.2:   Max Port Number: 19
[ 2949.039121] thunderbolt 0000:00:0d.2:   Config:
[ 2949.039123] thunderbolt 0000:00:0d.2:    Upstream Port Number: 1 Depth: 1 Route String: 0x1 Enabled: 1, PlugEventsDelay: 254ms
[ 2949.039127] thunderbolt 0000:00:0d.2:    unknown1: 0x0 unknown4: 0x0
[ 2949.097346] thunderbolt 0000:00:0d.2: 1: reading drom (length: 0xc1)
[ 2949.119943] thunderbolt 0000:00:0d.2: 1: DROM version: 1
[ 2949.155864] thunderbolt 0000:00:0d.2: 1: uid: 0x80879fee89708f00
[ 2949.182106] thunderbolt 0000:00:0d.2:  Port 1: 8086:b26 (Revision: 3, TB Version: 1, Type: Port (0x1))
[ 2949.182118] thunderbolt 0000:00:0d.2:   Max hop id (in/out): 19/19
[ 2949.182121] thunderbolt 0000:00:0d.2:   Max counters: 16
[ 2949.182123] thunderbolt 0000:00:0d.2:   NFC Credits: 0xc7800000
[ 2949.182125] thunderbolt 0000:00:0d.2:   Credits (total/control): 120/2
[ 2949.213631] thunderbolt 0000:00:0d.2:  Port 2: 8086:b26 (Revision: 3, TB Version: 1, Type: Port (0x1))
[ 2949.213642] thunderbolt 0000:00:0d.2:   Max hop id (in/out): 19/19
[ 2949.213645] thunderbolt 0000:00:0d.2:   Max counters: 16
[ 2949.213647] thunderbolt 0000:00:0d.2:   NFC Credits: 0x80000000
[ 2949.213650] thunderbolt 0000:00:0d.2:   Credits (total/control): 0/2
[ 2949.239911] thunderbolt 0000:00:0d.2:  Port 3: 8086:b26 (Revision: 3, TB Version: 1, Type: Port (0x1))
[ 2949.239923] thunderbolt 0000:00:0d.2:   Max hop id (in/out): 19/19
[ 2949.239926] thunderbolt 0000:00:0d.2:   Max counters: 16
[ 2949.239928] thunderbolt 0000:00:0d.2:   NFC Credits: 0x83c00000
[ 2949.239930] thunderbolt 0000:00:0d.2:   Credits (total/control): 60/2
[ 2949.268358] pcieport 0000:00:07.0: pciehp: Slot(0): Card present
[ 2949.268363] pcieport 0000:00:07.0: pciehp: Slot(0): Link Up
[ 2949.271376] thunderbolt 0000:00:0d.2:  Port 4: 8086:b26 (Revision: 3, TB Version: 1, Type: Port (0x1))
[ 2949.271387] thunderbolt 0000:00:0d.2:   Max hop id (in/out): 19/19
[ 2949.271390] thunderbolt 0000:00:0d.2:   Max counters: 16
[ 2949.271392] thunderbolt 0000:00:0d.2:   NFC Credits: 0x83c00000
[ 2949.271394] thunderbolt 0000:00:0d.2:   Credits (total/control): 60/2
[ 2949.297588] thunderbolt 0000:00:0d.2:  Port 5: 8086:b26 (Revision: 3, TB Version: 1, Type: Port (0x1))
[ 2949.297600] thunderbolt 0000:00:0d.2:   Max hop id (in/out): 19/19
[... 275 lines cut ...]
[ 2950.280492] usb 3-1.5: new full-speed USB device number 24 using xhci_hcd
[ 2950.387399] usb 3-1.5: New USB device found, idVendor=045e, idProduct=0855, bcdDevice= 1.01
[ 2950.387409] usb 3-1.5: New USB device strings: Mfr=1, Product=2, SerialNumber=0
[ 2950.387413] usb 3-1.5: Product: Surface Dock 3
[ 2950.387415] usb 3-1.5: Manufacturer: Microsoft
[... cut ...]
[ 2951.393013] usb 3-1.4.5: Product: Surface Thunderbolt 4 Dock Audio
[ 2951.393016] usb 3-1.4.5: Manufacturer: Microsoft
[ 2951.393019] usb 3-1.4.5: SerialNumber: 000000000000
[ 2951.445652] input: Microsoft Surface Thunderbolt 4 Dock Audio as /devices/pci0000:00/0000:00:14.0/usb3/3-1/3-1.4/3-1.4.5/3-1.4.5:1.3/0003:045E:085B.000D/input/input33
[ 2951.505077] hid-generic 0003:045E:085B.000D: input,hidraw4: USB HID v1.11 Device [Microsoft Surface Thunderbolt 4 Dock Audio] on usb-0000:00:14.0-1.4.5/input3

alebrati, 2023-06-06:
boltctl output:
```
● Microsoft Surface Thunderbolt(TM) 4 Dock #2
   ├─ type:          peripheral
   ├─ name:          Surface Thunderbolt(TM) 4 Dock
   ├─ vendor:        Microsoft
   ├─ uuid:          008f7089-ee9f-8780-ffff-ffffffffffff
   ├─ generation:    USB4
   ├─ status:        authorized
   │  ├─ domain:     00701d49-b5d1-8780-ffff-ffffffffffff
   │  ├─ rx speed:   40 Gb/s = 2 lanes * 20 Gb/s
   │  ├─ tx speed:   40 Gb/s = 2 lanes * 20 Gb/s
   │  └─ authflags:  none
   ├─ authorized:    Tue 06 Jun 2023 09:19:15
   ├─ connected:     Tue 06 Jun 2023 09:19:15
   └─ stored:        Tue 06 Jun 2023 08:30:34
      ├─ policy:     iommu
      └─ key:        no

```

d4s, 2023-06-06:
From the log above I see that not DMC, nor PD are detected properly.
So some questions:
- did DMC/PD are supported in Linux?
- is it possible to achieve information how to access TPS65988?

closed 2024-11-11 by hughsie