fwupd/fwupd #5879: Belkin dock TBT4 fails to upgrade Thunderbolt NVM (intel-usb4 plugin)

alebrati, 2023-06-05:
same dock mentioned here : https://github.com/fwupd/fwupd/issues/5829
while we install cab file of existing nvm version its ending with an error. seems we try to send restart to device but its not happening.
attach the files sent to Denis(Colabora) for internal checks.
[upgrade_failure.txt](https://github.com/fwupd/fwupd/files/11652761/upgrade_failure.txt)

alebrati, 2023-06-05:
localhost /home # fwupdmgr --version
runtime   org.freedesktop.fwupd         1.8.12
compile   com.hughsie.libxmlb           0.3.11
compile   org.freedesktop.gusb          0.4.5
compile   com.hughsie.libjcat           0.1.8
runtime   org.kernel                    5.10.175-22069-g9e86d6fc3943
runtime   org.freedesktop.gusb          0.4.5
compile   org.freedesktop.fwupd         1.8.12

d4s, 2023-06-05:
Looks like the reset doesn't work as expected for this hub.
As far as I understood no visible signs of reset as well.

PS Works well for Gatkex reference board with the same `intel-usb4` plugin.

alebrati, 2023-06-12:
what are the next steps on this one please? do we need to update vendor?

d4s, 2023-06-12:
@alebrati yes please, it would be helpful if you can contact the vendor of the dock and check with him how the dock should be restarted automatically after the flash or we have to send any additional commands.

alebrati, 2023-07-18:
@d4s what we do with this bug please ? seems vendor does not have any data on this : 

From: 研发部-罗锴 <kai.luo@ce-link.com> 
Sent: Tuesday, June 13, 2023 03:51
To: Lebrati, Arie <arie.lebrati@intel.com>; Lan, Rex <rex.lan@intel.com>; Liu, Alvin C <alvin.c.liu@intel.com>
Cc: Denis Pynkin <denis.pynkin@collabora.com>; Leung, Benson <bleung@google.com>; Rao, Abhijeet <abhijeet.rao@intel.com>; Zou, Richard <richang.zou@ce-link.com>
Subject: Re: Belkin International Thunderbolt 4 5-in-1 Core Hub

Hi Arie
   The upgrade tools are downloaded from Intel's official website，and the design also follows the reference design completely.
At present, we only know how to update NVM on windows.
     Thanks！

alebrati, 2023-07-18:
humm i cant send it its not mine..you think you can contact kai.luo@ce-link.com for that ?

hughsie, 2024-11-11:
Lets assume this is fixed in 2.0.x

closed 2024-11-11 by hughsie

superm1, 2024-11-11:
> Lets assume this is fixed in 2.0.x

Actually I'm not sure. I tried to update a Dell dock the other day with this plugin and it never updated.