fwupd/fwupd #5896: Quectel eg25-g community firmware branch is listed as affected by CVE-2021-31698 but it's not (also new firmware releases available)

DevenBL, 2023-06-12:
**Describe the bug**
The Quectel eg25-g community firmware branch is not affected by CVE-2021-31698 but all versions of firmware are listed as affected

**Steps to Reproduce**
View the issues section of any firmware release for the Quectel eg25-g community firmware 

**Expected behavior**
Have it not listed as an issue 

**fwupd version information**
Please provide the version of the daemon and client. 

Gnome firmware 43.2
```shell
fwupdmgr --version
runtime   org.freedesktop.fwupd         1.8.12
compile   com.hughsie.libxmlb           0.3.10
compile   org.freedesktop.gusb          0.3.10
compile   com.hughsie.libjcat           0.1.9
runtime   org.kernel                    6.1-rockchip
compile   org.freedesktop.fwupd         1.8.12
runtime   org.freedesktop.gusb          0.3.10

```

Please note how you installed it (`apt`, `dnf`, `pacman`, source, etc): Comes preinstalled with the distro, updated via `apt`

<details>

<summary>**fwupd device information**</summary>

Please provide the output of the fwupd devices recognized in your system.

```shell
fwupdmgr get-devices --show-all-devices
├─QUECTEL Mobile Broadband Module:
│     Device ID:          
│     Summary:            Quectel EG25-G modem
│     Current version:    0.7.2
│     Vendor:             Qualcomm (USB:0x2C7C)
│     Release Branch:     community
│     Problems:           • Device requires AC power to be connected
│     *i am not listing serials*
│     Device Flags:       • System requires external power source
│                         • Supported on remote server
│                         • Updatable
│                         • Unsigned Payload

```

</details>

**Additional questions**

- Operating system and version: Mobian/bookworm and Mobian/unstable
- Have you tried rebooting? Yes
- Is this a regression? No?

[image]
the bug exists in stock userspace and the community firmware replaces the stock userspace

**New firmware versions are available**
Version 0.7.3 and 0.7.4 of the community firmware have been released: https://github.com/the-modem-distro/pinephone_modem_sdk
The newest available version in fwupd is 0.7.2.

hughsie, 2023-06-12:
The firmware does *fix* the CVE, but you're correct that the branch does not *contain* the security fix. I think the firmware is labelled appropriately as I don't think say "this fixes the CVE if you're using the other branch". I think asking @biktorgj to upload the latest firmware is probably the right thing to do. Sorry that's not more helpful.

closed 2023-06-12 by hughsie

DevenBL, 2023-06-14:
So your keeping the cve label on the community firmware (biktorgj's) that is unaffected by that cve? Even if the cve is supposed to only apply to the quectel proprietary firmware?

Biktorgj, 2023-06-15:
The label was set by me, because at the time it was the only available firmware you could flash to the modem that would mitigate that issue.

Quectel seems to have fixed it in `EG25GGBR07A08M2G_30.006.30.006` version and up (tested against 006 and 007) , *but*, Quectel doesn't provide an official way to flash their original firmware, you have to either
a) Look through their forums to find a working link to a version newer than what you have
b) Search for random version numbers until you find something remotely similar
c) Go to my recovery repo (which seems to be what most people do, either for the Pinephone or for standalone models of the eg25)

Quectel hasn't provided any firmware for the eg25 in LVFS, and they still ship modems with version 01.002 of their firmware, so the only semi-automated way of fixing that cve is effectively flash the community firmware, hence why the CVE is still mentioned. @hughsie should I remove it from the metadata on new builds?

On the firmware versions available on LVFS, that's on me, haven't really had much time for anything lately and 0.7.4 may have some bugs I didn't want to risk force-pushing to users, that's why it's not there. 0.7.5 will be though (when it's ready)

hughsie, 2023-06-15:
> @hughsie should I remove it from the metadata on new builds?

Yup, the CVE should only be listed on the first build that fixes it; if there are 5 updates pending then the release notes and list of CVEs gets folded together. If you list it on every build then it does funny things to the stats as we assume the latest build with the CVE "really fixed it this time".