fwupd/fwupd #5898: 1.9.2: build fails in generate-dbus-interface.py script

kloczek, 2023-06-12:
Looks like something is wrong.
```console
[tkloczko@pers-jacek x86_64-redhat-linux-gnu]$ ninja
[76/243] Generating src/fwupd-generate-dbus-interface with a custom command
FAILED: src/org.freedesktop.fwupd.xml
/usr/bin/python3 ../generate-build/generate-dbus-interface.py ../src/org.freedesktop.fwupd.xml src/org.freedesktop.fwupd.xml
Traceback (most recent call last):
  File "../generate-build/generate-dbus-interface.py", line 42, in <module>
    ET.indent(tree, space=" ", level=0)
AttributeError: module 'xml.etree.ElementTree' has no attribute 'indent'
```
I'm using libxml2 2.10.4.

hughsie, 2023-06-12:
What python are you using? You need 3.9 at least.

kloczek, 2023-06-12:
I'm using 3.8.17 (still it is not EOSed).
Nothing on meson run warned that python >=3.9 needs to be used.

closed 2023-06-12 by hughsie