fwupd/fwupd #5906: UEFI/ME update failures on Lenovo X1 Gen 11h

thwalker3, 2023-06-14:
**Describe the bug**
Attempts to upgrade UEFI and/or ME firmware (I've tried both together as well as separately) fail during the following POST with (please excuse typos, transcribing from a photo):

```
fwupd-efi version 1.4
WARNING: QueryCapsuleCapabilities failed, assuming EFIResetWarm: Unsupported
WARNING: Could not apply capsule update: Not Found
WARNING: Could not apply capsules: Not Found
Reset System
```

**Steps to Reproduce**
```
$ sudo fwupdmgr update
Devices with no available firmware updates: 
 • ELAN067B:00 04F3:31F8
 • Integrated Camera
 • Prometheus IOTA Config
 • UEFI Device Firmware
 • UEFI Device Firmware
 • UEFI Device Firmware
 • UEFI Device Firmware
 • UEFI Device Firmware
 • UEFI Device Firmware
 • UEFI Device Firmware
 • UEFI Device Firmware
 • UEFI Device Firmware
 • UEFI Device Firmware
 • UEFI Device Firmware
 • UEFI Device Firmware
 • UEFI Device Firmware
 • UEFI Device Firmware
╔══════════════════════════════════════════════════════════════════════════════╗
║ Upgrade Intel Management Engine from 1.25.2091 to 1.25.2124?                 ║
╠══════════════════════════════════════════════════════════════════════════════╣
║ RPL Initial Release                                                          ║
║                                                                              ║
║ Version 16.1.25.2124 (LVFS: 1.25.2124)                                       ║
║                                                                              ║
║ This is an Initial stable LVFS Release for ThinkPad X1 Yoga Gen 8 and        ║
║ ThinkPad X1 Carbon Gen 11                                                    ║
║                                                                              ║
║ 21HMCTO1WW must remain plugged into a power source for the duration of the   ║
║ update to avoid damage.                                                      ║
╚══════════════════════════════════════════════════════════════════════════════╝
Perform operation? [Y|n]: y
Waiting…                 [***************************************]
Successfully installed firmware
Do not turn off your computer or remove the AC adapter while the update is in progress.
Update failure is a known issue, visit this URL for more information: https://github.com/fwupd/fwupd/wiki/LVFS-Triaged-Issue:-Failed-to-run-update-on-reboot
Successfully uploaded 1 report
Devices with the latest available firmware version:
 • KXG8AZNV1T02 LA KIOXIA
 • Prometheus
╔══════════════════════════════════════════════════════════════════════════════╗
║ Upgrade System Firmware from 0.1.12 to 0.1.13?                               ║
╠══════════════════════════════════════════════════════════════════════════════╣
║ Lenovo System Firmware Version 1.13                                          ║
║                                                                              ║
║ Problem fixes                                                                ║
║                                                                              ║
║ • Fixed an issue where USB-C/Thunderbolt port and charging function          ║
║ become unavailable rarely.                                                   ║
║                                                                              ║
║ 21HMCTO1WW must remain plugged into a power source for the duration of the   ║
║ update to avoid damage.                                                      ║
╚══════════════════════════════════════════════════════════════════════════════╝
Perform operation? [Y|n]: y
Updating System Firmware…[   -                                   ]
Waiting…                 [***************************************]
[... 647 lines cut ...]
├── System.map-6.3.5-200.fc38.x86_64
├── System.map-6.3.6-200.fc38.x86_64
├── System.map-6.3.7-200.fc38.x86_64
├── vmlinuz-0-rescue-e5edbcbd7b3e4c6d8ad276960a0769d0
├── vmlinuz-6.3.5-200.fc38.x86_64
├── vmlinuz-6.3.6-200.fc38.x86_64
└── vmlinuz-6.3.7-200.fc38.x86_64

18 directories, 108 files
```

**Additional questions**

- Operating system and version:
Up to date Fedora 38
- Have you tried rebooting?
Yes.
- Is this a regression?
I've never been able to update this device, but the same disk image on my prior X1 gen7 worked without issue.
- Are you using an NVMe disk?
Yes 
- Is secure boot enabled?
I've tried both with and without Secure Boot enabled.  The EFI binary loads, it just fails.
- Is this a Lenovo system with 'Boot Order Lock' turned on in the BIOS?
Lenovo, yes.  Boot Order Lock, No.  Nor is WindowsUEFIFirmwareUpdate disabled.

thwalker3, 2023-06-14:
@mrhpearson

mrhpearson, 2023-06-15:
I'll check with the FW team - internal ticket is  LO-2516

We had the previous ticket (https://github.com/fwupd/fwupd/issues/5612) where we confirmed the partition needed to be GPT formatted and have the ESP flag set. Can we confirm that's the case here just in case?

thwalker3, 2023-06-15:
Doh, now I feel like an idiot. Somehow I managed to set up the disk on the new laptop (not a fresh install) without ESP set. I'm actually kind of surprised it booted, but I guess direct EFI boot entries will do that.

As soon as I flipped the flag, things work and both UEFI and ME updated without a problem.

Sorry for the noise...

hughsie, 2023-06-15:
@mrhpearson okay, we really need to do a better job enforcing ESP. Do you know if this is a new requirement? Older firmwares used to load the capsule from the BDP too. If you can give me a list of hardware I can add a quirk to fwupd.

mrhpearson, 2023-06-15:
@thwalker3 - all good. Thanks for confirming that fixes it :)

@hughsie - No idea when it became a requirement but it's part of the UEFI spec isn't it that you have at least one ESP partition? I suspect you're going to see it on all platforms going forward...which would be a nightmare to maintain as a list so not sure that scales.

How about doing a warning? A "hey - your EFI partition does not have ESP set, you should probably do .... or you may see FW update issues".

Are we the only vendor that has this as a requirement? I do wonder if it's part of the MS secured-core stuff that is enforcing it maybe?

superm1, 2023-06-20:
> @hughsie - No idea when it became a requirement but it's part of the UEFI spec isn't it that you have at least one ESP partition? I suspect you're going to see it on all platforms going forward...which would be a nightmare to maintain as a list so not sure that scales.

Many platforms don't enforce a partition type being an ESP type and will happily take a FAT32 partition to boot from.

dspyrhsu, 2024-10-12:
Hi there, 

sorry to bring this up again, but I am experiencing the exact same issue described in the original report on a Lenovo Thinkpad P71. I have read through a few issues with similar descriptions and therefore I have made sure that the ESP-flag is set and the BIOS settings should allow , but that still gives me the exact same error message @thwalker3 got upon rebooting and trying to apply the updates, namely:

<pre>fwupd-efi version 1.4
WARNING: QueryCapsuleCapabilities failed, assuming EFIResetWarm: Unsupported
WARNING: Could not apply capsule update: Not Found
WARNING: Could not apply capsules: Not Found
Reset System</pre>

Just to be on the safe side, let's show my attempt:

<pre>sudo fwupdmgr update
Devices with no available firmware updates: 
 • MZVLW256HEHP-000L7
 • SSD 960 EVO 500GB
 • ST2000LM015-2E8174
 • ST2000LM015-2E8174
 • UEFI Device Firmware
╔══════════════════════════════════════════════════════════════════════════════╗
║ Upgrade Intel Management Engine from 182.10.1196 to 184.96.4657?             ║
╠══════════════════════════════════════════════════════════════════════════════╣
║ • 8 Intel Platform Update                                                    ║
║                                                                              ║
║ Version 11.8.96.4657 (LVFS: 184.96.4657)                                     ║
║                                                                              ║
║ Problem Fixes Mitigated the following security vulnerabilities under         ║
║ issues. Please see fixed issues for details.                                 ║
║                                                                              ║
║ 20HK0000GE must remain plugged into a power source for the duration of the   ║
║ update to avoid damage.                                                      ║
╚══════════════════════════════════════════════════════════════════════════════╝
Perform operation? [Y|n]:  
Waiting…                 [***************************************]]Less than one minute remaining…
Successfully installed firmware
Do not turn off your computer or remove the AC adapter while the update is in progress.
╔══════════════════════════════════════════════════════════════════════════════╗
║ Upgrade System Firmware from 0.1.8 to 0.1.24?                                ║
╠══════════════════════════════════════════════════════════════════════════════╣
║ This update fixes the following important problems:                          ║
║                                                                              ║
║ • Update includes a security fix.                                            ║
║ • Fixed an issue where BIOS POST might display the error message "Boot       ║
║ Manager recover from an error."                                              ║
║                                                                              ║
║ This update also adds the following features:                                ║
║                                                                              ║
║ • Updated the Diagnostics module to version 04.00.001.                       ║
║ • Add Oculus device support.                                                 ║
║                                                                              ║
║ 20HK0000GE must remain plugged into a power source for the duration of the   ║
║ update to avoid damage.                                                      ║
╚══════════════════════════════════════════════════════════════════════════════╝
Perform operation? [Y|n]: 
Updating System Firmware…[  -                                    ]
Waiting…                 [***************************************]]
Successfully installed firmware
Do not turn off your computer or remove the AC adapter while the update is in progress.
Devices with the latest available firmware version:
 • Unifying Receiver
An update requires a reboot to complete. Restart now? [y|N]: y
</pre>

version info:

<pre>fwupdmgr --version
compile   org.freedesktop.fwupd         1.9.24
compile   com.hughsie.libxmlb           0.3.18
compile   com.hughsie.libjcat           0.2.0
runtime   org.freedesktop.fwupd-efi     1.4
compile   org.freedesktop.gusb          0.4.8
runtime   com.hughsie.libxmlb           0.3.x
runtime   com.hughsie.libjcat           0.2.0
runtime   org.freedesktop.gusb          0.4.8
runtime   org.freedesktop.fwupd         1.9.24
runtime   org.kernel                    6.8.0-45-generic</pre>

As to the **Additional questions** (all answers more or less the same as above ...)

* Operating system and version:
Latest Ubuntu 24.4
* Have you tried rebooting?
Yes.
* Is this a regression?
I've never been able to update this device.
* Are you using an NVMe disk?
Yes
* Is secure boot enabled?
I've tried both with and without Secure Boot enabled. The EFI binary loads, it just fails.
* Is this a Lenovo system with 'Boot Order Lock' turned on in the BIOS?
Lenovo, yes. Boot Order Lock, No. Nor is WindowsUEFIFirmwareUpdate disabled.

So, I am at a loss and any help would be greatly appreciated!

hughsie, 2024-10-12:
> EDIT: I did read on a bit after posting my comment below and it turned out that having an MBR partition layout did prevent the update to work

Yup, fwupd 2.0.0 actually checks for this -- I was about to suggest it.