fwupd/fwupd #5907: Jabra Evolve 65 not detected

rodehoed, 2023-06-15:
**Describe the bug**
I have a Jabra Evolve 65 which is not getting detected with get-devices. As stated in the release notes the Jabra Evolve line should be supported in 1.9.2. 

**Steps to Reproduce**
- Plugin USB cable in headset
- run _fwupdmgr get-devices_

**Expected behavior**
The device should be shown

**fwupd version information**

```
compile   org.freedesktop.fwupd         1.9.2
compile   com.hughsie.libxmlb           0.3.11
compile   com.hughsie.libjcat           0.1.13
runtime   org.freedesktop.fwupd-efi     1.4
compile   org.freedesktop.gusb          0.4.5
runtime   com.hughsie.libjcat           0.1.13
runtime   org.freedesktop.gusb          0.4.5
runtime   org.freedesktop.fwupd         1.9.2
runtime   org.kernel                    6.3.5-200.fc38.x86_64
```
RPM packages:
```
fwupd-efi-1.4-1.fc38.x86_64
fwupd-plugin-flashrom-1.9.2-1.fc38.x86_64
fwupd-plugin-modem-manager-1.9.2-1.fc38.x86_64
fwupd-plugin-uefi-capsule-data-1.9.2-1.fc38.x86_64
fwupd-1.9.2-1.fc38.x86_64
```

fwupdmgr get-plugins Shows me that _jabra_ and _jabra_gnp_ are enabled.

<details>

<summary>**fwupd device information**</summary>

ASUSTeK COMPUTER INC. M11AD
│
├─Core™ i5-4460S CPU @ 2.90GHz:
│     Device ID:          4bde70ba4e39b28f9eab1628f9dd6e6244c03027
│     Current version:    0x00000028
│     Vendor:             Intel
│     GUIDs:              b9a2dd81-159e-5537-a7db-e7101d164d3f ← cpu
│                         30249f37-d140-5d3e-9319-186b1bd5cac3 ← CPUID\PRO_0&FAM_06
│                         dd2cdaeb-2b08-599d-aaec-5a8374add8d6 ← CPUID\PRO_0&FAM_06&MOD_3C
│                         112c1ba3-2c29-5ab2-bd72-ea65bd10dd67 ← CPUID\PRO_0&FAM_06&MOD_3C&STP_3
│     Device Flags:       • Internal device
│   
├─Lexa PRO [Radeon 540/540X/550/550X / RX 540X/550/550X]:
│     Device ID:          ce4c74a5188d5b9cdb1e72ed32dad2d313c1c999
│     Current version:    113-TIC32700-100
│     Vendor:             Advanced Micro Devices, Inc. [AMD/ATI] (PCI:0x1002, PCI:0x8086)
│     GUIDs:              4daaf54e-4446-5b25-8979-7fc017051118 ← PCI\VEN_1002&DEV_699F
│                         86aee031-c0ce-5ac8-9570-3fec122d4893 ← PCI\VEN_1002&DEV_699F&REV_C7
│                         e2fba49b-b9f0-5b6e-a582-323c23e1f79f ← PCI\VEN_1002&DEV_699F&SUBSYS_18495001
│                         9c86f1bb-13b6-54da-a0dd-9f2e66f0faa9 ← PCI\VEN_1002&DEV_699F&SUBSYS_18495001&REV_C7
│                         5006e997-5789-5967-9f7e-14c1d06716b8 ← PCI\VEN_8086&DEV_0C01
│                         39497364-2d1a-52aa-894b-7db96fe00ac6 ← PCI\VEN_8086&DEV_0C01&REV_06
│                         c92c3bbf-617d-549b-8784-34f9b06009f9 ← PCI\VEN_8086&DEV_0C01&SUBSYS_10438534
│                         d2ddd13a-d819-57eb-ac3d-efb93cc026e6 ← PCI\VEN_8086&DEV_0C01&SUBSYS_10438534&REV_06
│     Device Flags:       • Internal device
│                         • Cryptographic hash verification is available
│   
├─SSD 860 PRO 512GB:
│     Device ID:          602b0a6cc821d155208724f0e22f8d111542b74c
│     Summary:            ATA drive
│     Current version:    RVM01B6Q
│     Vendor:             Samsung (ATA:0x144D, OUI:002538)
│     Serial Number:      S42YNF0M200230L
│     GUIDs:              32680750-2bdd-5df9-9656-a4231b8d3db4 ← IDE\Samsung_SSD_860_PRO_512GB_______________RVM01B6Q
│                         d3abf858-ab4e-58b1-a8b5-f7ba5b4ba0a9 ← IDE\0Samsung_SSD_860_PRO_512GB_______________
│                         1c99189e-7c54-57df-b60e-b659821142bc ← Samsung SSD 860 PRO 512GB
│     Device Flags:       • Internal device
│                         • Updatable
│                         • System requires external power source
│                         • Needs a reboot after installation
│                         • Device is usable for the duration of the update
│   
├─System Firmware:
│ │   Device ID:          a45df35ac0e948ee180fe216a5f703f32dda163f
│ │   Summary:            UEFI ESRT device
│ │   Current version:    31
│ │   Minimum Version:    31
│ │   Vendor:             ASUSTeK COMPUTER INC. (DMI:American Megatrends Inc.)
│ │   Update State:       Success
│ │   GUIDs:              7039436b-6acf-433b-86a1-368ec2ef7e1f
│ │                       230c8b18-8d9b-53ec-838b-6cfc0383493a ← main-system-firmware
│ │   Device Flags:       • Internal device
│ │                       • Updatable
│ │                       • System requires external power source
│ │                       • Needs a reboot after installation
│ │                       • Device is usable for the duration of the update
│ │ 
│ ├─UEFI Platform Key:
│ │     Device ID:        6924110cde4fa051bfdc600a60620dc7aa9d3c6a
│ │     Summary:          ASUSTeK MotherBoard PK Certificate
│ │     GUID:             54ef26b1-94d9-51d5-9e46-91732769a3c2 ← UEFI\CRT_6F52EF400A8634497D726A7A7D469FCA5905BADD
│ │   
│ └─UEFI dbx:
│       Device ID:        362301da643102b9f38477387e2193e57abaa590
│       Summary:          UEFI revocation database
│       Current version:  217
│       Minimum Version:  217
│       Vendor:           UEFI:Linux Foundation
│       Install Duration: 1 second
│       GUIDs:            fda6234b-adcb-5105-8515-9af647d29775 ← UEFI\CRT_D7F66BE77CEF858C174BF4338A99263C8795B74E02026411F5F532F716AE3263
│                         f8ff0d50-c757-5dc3-951a-39d86e16f419 ← UEFI\CRT_D7F66BE77CEF858C174BF4338A99263C8795B74E02026411F5F532F716AE3263&ARCH_X64
│                         c6682ade-b5ec-57c4-b687-676351208742 ← UEFI\CRT_A1117F516A32CEFCBA3F2D1ACE10A87972FD6BBE8FE0D0B996E09E65D802A503
│                         f8ba2887-9411-5c36-9cee-88995bb39731 ← UEFI\CRT_A1117F516A32CEFCBA3F2D1ACE10A87972FD6BBE8FE0D0B996E09E65D802A503&ARCH_X64
│                         7d5759e5-9aa0-5f0c-abd6-7439bb11b9f6 ← UEFI\CRT_ED1FE72CB9CA31C9AF5B757AFCD733323D675825032E6CED7FE1AE9EB767998C
│                         0c7691e1-b6f2-5d71-bc9c-aabee364c916 ← UEFI\CRT_ED1FE72CB9CA31C9AF5B757AFCD733323D675825032E6CED7FE1AE9EB767998C&ARCH_X64
│       Device Flags:     • Internal device
│                         • Updatable
│                         • Supported on remote server
│                         • Needs a reboot after installation
│                         • Device is usable for the duration of the update
│                         • Only version upgrades are allowed
│                         • Signed Payload
│     
└─WDC WD1003FZEX-00K3CA0:
      Device ID:          0a8c36d4c09c803cd6b5861e443fb7a41a20cbe6
      Summary:            ATA drive
      Current version:    01.01A01
      Vendor:             Western Digital (ATA:0x101C, OUI:0014ee)
      Serial Number:      WD-WCC6Y4XEAS8N
      GUIDs:              29becef2-c76b-5c16-888c-bb0fe0f08fc3 ← IDE\WDC_WD1003FZEX-00K3CA0__________________01.01A01
                          803a2186-47ea-53ca-89b5-e05e6319c567 ← IDE\0WDC_WD1003FZEX-00K3CA0__________________
                          99210704-cd04-5eb3-828b-f1558f516b8d ← WDC WD1003FZEX-00K3CA0
      Device Flags:       • Internal device
                          • Updatable
                          • System requires external power source
                          • Needs a reboot after installation
                          • Device is usable for the duration of the update
/

</details>

lspci:
`Bus 001 Device 007: ID 0b0e:030c GN Netcom Jabra Evolve 65`

dmesg:
```
715886.858948] usb 1-3: new full-speed USB device number 7 using xhci_hcd
[715886.995613] usb 1-3: New USB device found, idVendor=0b0e, idProduct=030c, bcdDevice= 2.97
[715886.995628] usb 1-3: New USB device strings: Mfr=0, Product=2, SerialNumber=3
[715886.995634] usb 1-3: Product: Jabra Evolve 65
[715886.995638] usb 1-3: SerialNumber: 50C2ED5666D3
[715887.052074] input: Jabra Evolve 65 as /devices/pci0000:00/0000:00:14.0/usb1/1-3/1-3:1.3/0003:0B0E:030C.0005/input/input18
[715887.054813] mc: Linux media interface: v0.10
[715887.088870] usb 1-3: 1:1: cannot set freq 48000 to ep 0x3
[715887.104200] jabra 0003:0B0E:030C.0005: input,hiddev98,hidraw4: USB HID v1.11 Device [Jabra Evolve 65] on usb-0000:00:14.0-3/input3
[715887.166738] usbcore: registered new interface driver snd-usb-audio
```

**Additional questions**

- Operating system and version: Fedora 38
- Have you tried rebooting? Yes
- Is this a regression? No

ljrk0, 2023-06-15:
Having the same issue, I noticed that the PID in `lsusb` is in yours and mine system 0x030C, while the quirks file seems to expect 0x24FC to 0x24FE.  OTOH the docs say:

> The device usually presents in runtime mode, but on detach re-enumerates with a different USB VID and PID in DFU APP mode. The device is then further detached by the dfu plugin.
> 
> On DFU attach the device again re-enumerates back to the Jabra runtime mode.
> 
> For this reason the REPLUG_MATCH_GUID internal device flag is used so that the bootloader and runtime modes are treated as the same device.

So maybe these are exactly those runtime mode PIDs (although the VID is the same, unlike what's described in the doc)?  But then I'm unsure how to "trigger" the DFU detach.

hughsie, 2023-06-15:
Ah, be careful; the `jabra` plugin is the one for the SPEAK devices. The one you want for EVOLVE is `jabra-gnp`, e.g. https://github.com/fwupd/fwupd/blob/main/plugins/jabra-gnp/jabra-gnp.quirk

rodehoed, 2023-06-15:
> Ah, be careful; the `jabra` plugin is the one for the SPEAK devices. The one you want for EVOLVE is `jabra-gnp`, e.g. https://github.com/fwupd/fwupd/blob/main/plugins/jabra-gnp/jabra-gnp.quirk

Maybe it's just an end-user issue of using fwupd incorrect. But both plugins jabra en jabra-gnp are enabled.

hughsie, 2023-06-15:
It looks like the PID is indeed missing. Is that the `Jabra Evolve 65 SE Stereo` you have there? Is that different from the `Jabra Evolve 65`?

If so, can you try adding the PID in a quirk file located in /var/lib/fwupd/quirks.d/ and see if fwupd enumerates it correctly?

ljrk0, 2023-06-15:
I had a look at the model number and it's a HSC018W or `Jabra Evolve 65 MS Stereo`: https://www.jabra.com/supportpages/jabra-evolve-65#/#6599-823-309

```
$ cat /var/lib/fwupd/quirks.d/jabra.quirk
[USB\VID_0B0E&PID_030C]
Plugin = jabra_gnp
```

When piping the output through grep I noticed errors related to the USB transfer that are otherwise hidden:
```
# fwupdtool get-devices | grep -i jabra
10:57:31.897 FuEngine             failed to add device usb:03:00:06:04:04:02:01: failed after 3 retries: failed to read from device: device sent more data than requested
10:57:21.470 FuEngine             failed to add device usb:03:00:07: failed after 3 retries: failed to read from device: transfer timed out
```
Plugging devices out and in confirms that they are related to the Link (first error) and Headset (second error).  They happen even when the above quirk is not added.

Looking at the `journalctl` output of `fwupd.service` shows these errors as well (I should have checked there earlier, my bad).

hughsie, 2023-06-15:
I've asked Jabra to comment here.

rodehoed, 2023-06-15:
@hughsie I made an quirk file: cat /var/lib/fwupd/quirks.d/jabra.quirk:

```
[USB\VID_0B0E&PID_030C]
Plugin = jabra_gnp
```

When I run fwupd again I see:
```11:21:03.491 FuEngine             failed to add device usb:01:00:07: failed after 3 retries: failed to read from device: device sent more data than requested```
When I remove the quirk file I do not see this error. 

I don't know where this usb:01:00:07 comes from, because it's not in the lsusb overview.

hughsie, 2023-06-15:
> I don't know where this usb:01:00:07 comes from, because it's not in the lsusb overview.

It's the USB logical hierarchy that fwupd uses. You could try `sudo fwupdtool get-devices --plugins jabra_gnp --verbose --verbose`

gdpcastro, 2023-06-15:
Hi, this one is a bit simple to answer, it just has an unfortunate answer:
The devices in question use a different protocol that has not been added (yet) to fwupd. The simplest way I can put it is that not all of our products use the same protocol, and on top of that, not all devices in the same "line" (Engage, Evolve, Speak, etc.) necessarily use the same protocol. So the Evolve 65 SE (second edition) is supported, while the Evolve 65 isn't.

I'm sorry about any confusion caused

hughsie, 2023-06-15:
Okay, many thanks @gdpcastro for the wisdom -- I'll close this for now and we can link it when (if?) we get support for the older devices. Thanks.

closed 2023-06-15 by hughsie