fwupd/fwupd #5913: fwupdtool: DFU image write progress is not properly updated

MarekPieta, 2023-06-19:
**Describe the bug**
The DFU image write progress bar does not indicate image transfer progress properly, it is displayed as full `Writing` progress bar all the time:
`Writing?                [******************************************************************************************************]`

**Steps to Reproduce**
Trigger firmware image download and update using the following command:
`sudo ./build/src/fwupdtool upgrade --plugin-whitelist=nordic-hid`

The issue does not replicate when updating from the file:
`sudo ./build/src/fwupdtool install fw.cab --plugin-whitelist=nordic-hid`
In that case, the `Write` progress bar starts around the middle, but then it updates properly.

The issue does not replicate in fwupdmgr CLI built from the same commit hash as the fwupdtool (the progress bar is properly updated in case of fwupdmgr).

**Expected behavior**
The progress bar should reflect DFU image write progress. 

**fwupd version information**

fwupd built locally from source on the following commit hash: `957f18f649e5a6677a4d49cd338f35d78b07e506`.

```
fwupdmgr --version

compile   org.freedesktop.fwupd         1.9.3
compile   com.hughsie.libxmlb           0.3.6
compile   com.hughsie.libjcat           0.1.9
compile   org.freedesktop.gusb          0.3.10
runtime   com.dell.libsmbios            2.4
runtime   org.freedesktop.gusb          0.3.10
runtime   org.kernel                    5.15.0-73-generic
runtime   org.freedesktop.fwupd         1.9.3
```

**Additional questions**

Operating system: `Ubuntu 22.04.2 LTS`

MarekPieta, 2023-06-19:
Hi @hughsie,
It seems that it's a generic issue related to the fwupdtool (it does not seem specific to the `nordic-hid` plugin). Do you know what could cause it?

hughsie, 2023-06-20:
> The issue does not replicate in fwupdmgr CLI built from the same commit hash as the fwupdtool

Right, this is somewhat expected. in `fwupdmgr` we've already loaded all the data files, enumerated and tagged all the devices, and everything is ready to go in a predictable and measured way. When we're using `fwupdtool` we have to use a gross hack to "guess" how long the engine startup is going to take in relation to actually updating the device: https://github.com/fwupd/fwupd/blob/main/src/fu-tool.c#L1237

I suppose `fu_util_update()` is even worse; it doesn't try to guess :) That would be harder to fix, and considering `fwupdtool` is really a debugging utility I'm not sure it's worth it. I'd really advise against using `fwupdtool` unless you're debugging.

> but the prompt state Writing? instead of Writing....

This is probably because the locale is something like `C` rather than something sensible like `en_GB.UTF-8`.

MarekPieta, 2023-06-21:
Hi @hughsie
Thanks for clarification, now I fully understood why we "start from 50%" when updating from file. :)
I also noticed that proper fix for that one may be a bit tricky since `fu_util_update` uses `fu_util_install_release -> fu_util_install` which leads to silently reusing the `priv->progress`. Introducing a proper progress chain here may be tricky (since fwupdtool is only a debug tool maybe it's actually not necessary...).

Still, maybe we could at least reset the progress before reusing it? That would ensure user would see the progress bar that updates itself. It could also prevent side-effects - some time ago I also saw logs related to progress percentage drop: `percentage should not go down from 100 to 85: 0) ../src/fu-tool.c:1238`.
Please take a look at the linked PR and let me know if the fix is acceptable.

closed 2023-06-21 by hughsie