fwupd/fwupd #5928: Emulated Wacom Intuos USB device fails to upgrade on s390x

superm1, 2023-06-21:
As part of bringing newer fwupd 1.9.1 into Ubuntu 23.10, autopkgtest is failing on s390x.

```
158s Restarting device…: 1%
158s Writing…: 1%
158s Decompressing…: 1%
158s Decompressing…           [                                       ]Decompressing…           [                                       ]Restarting device…: 1%
158s Erasing…: 1%
158s Writing…: 1%
158s 
158s Wacom Intuos BT-M: [31m[1mfailed after 10 retries: failed to SetReport: no matching event for ControlTransfer:Direction=0x01,RequestType=0x01,Recipient=0x01,Request=0x09,Value=0x03db,Idx=0x0000,Data=2wgAyYINBw==,Length=0x7[0m
158s Downloading…: 1%
160s Downloading…             [                                       ]Downloading…: 100%
163s Downloading…             [ -                                     ]Downloading…             [ -                                     ]Downloading…: 1%
```

This is unique to s390x, so it makes me wonder if we have an endianness problem with emulation or with this plugin.

Full log: https://autopkgtest.ubuntu.com/results/autopkgtest-mantic/mantic/s390x/f/fwupd/20230619_091825_be245@/log.gz

hughsie, 2023-06-22:
The only slightly maybe unsafe thing I can think is:

		bd->addr[0] = (addr >> 16) & 0xff;
		bd->addr[1] = (addr >> 8) & 0xff;
		bd->addr[2] = addr & 0xff;

It looks like all the other emulations pass. It's failing in fu_wac_module_bluetooth_write_firmware() in the writing phase. Unfortunately using a s390x fwupd with a systemd-nspawn doesn't work very well -- and USB devices don't work either.

If I come up with a PR are you able to test on the s390x system? https://github.com/fwupd/fwupd/pull/5929 for example?

superm1, 2023-06-22:
No I don't have one so I can't easily test on a real system, will need to make the round trip into a proper Ubuntu package in the archive and autopkgtest.  I'm fine doing that though.

hughsie, 2024-08-16:
@superm1 still broken? If so I can fire up a s390s fedora vm.

closed 2024-11-11 by hughsie