fwupd/fwupd #5937: How to disable 10 seconds delay when no firmware is found?

nonZero, 2023-06-25:
Dell Latitde 5511, Arch Linux,
On **every** boot a "WARNING: No updates to process, exiting in 10 seconds" message appears (for 10 seconds...).
How can this message be disabled?

**fwupd version information**
```text
compile   org.freedesktop.fwupd         1.9.2
compile   com.hughsie.libxmlb           0.3.11
compile   com.hughsie.libjcat           0.1.13
runtime   org.freedesktop.fwupd-efi     1.4
compile   org.freedesktop.gusb          0.4.6
runtime   com.hughsie.libjcat           0.1.14
runtime   com.dell.libsmbios            2.4
runtime   org.freedesktop.gusb          0.4.6
runtime   org.kernel                    6.3.9-arch1-1
runtime   org.freedesktop.fwupd         1.9.2
```

superm1, 2023-06-25:
This message comes from fwupd-efi.
Please attach everything requested by the UEFI bug template.

nonZero, 2023-06-25:
> This message comes from fwupd-efi. Please attach everything requested by the UEFI bug template.

Thanks! opened #5938

closed 2023-06-25 by hughsie