fwupd/fwupd #5938: Disable message: "WARNING: No updates to process, exiting in 10 seconds"

nonZero, 2023-06-25:
**Describe the bug**
On every boot a "WARNING: No updates to process, exiting in 10 seconds" message appears (for 10 seconds...).

**Steps to Reproduce**
Turn on my Dell Latitde 5511, with Arch Linux.

**Expected behavior**
No 10 seconds delay.

**fwupd version information**
Please provide the version of the daemon and client.

```text
compile   org.freedesktop.fwupd         1.9.2
compile   com.hughsie.libxmlb           0.3.11
compile   com.hughsie.libjcat           0.1.13
runtime   org.freedesktop.fwupd-efi     1.4
compile   org.freedesktop.gusb          0.4.6
runtime   com.hughsie.libjcat           0.1.14
runtime   com.dell.libsmbios            2.4
runtime   org.freedesktop.gusb          0.4.6
runtime   org.kernel                    6.3.9-arch1-1
runtime   org.freedesktop.fwupd         1.9.2
```

`pacman -Q | grep fwupd`:
```text
fwupd 1.9.2-2
fwupd-efi 1.4-1
```

<summary>**fwupd device information**</summary>
Please provide the output of the fwupd devices recognized in your system.

```text
Dell Inc. Latitude 5511
│
├─WD19:
│ │   Device ID:          [redacted]
│ │   Summary:            High performance dock
│ │   Current version:    01.01.00.03
│ │   Vendor:             Dell Inc. (USB:0x413C)
│ │   Install Duration:   1 minute
│ │   Serial Number:      [redacted]
│ │   GUID:               cd357cf1-40b2-5d87-b8df-bb2dd82774aa ← USB\VID_413C&PID_B06E&hub&embedded
│ │   Device Flags:       • Updatable
│ │                       • Supported on remote server
│ │                       • Device stages updates
│ │                       • Device can recover flash failures
│ │                       • Device is usable for the duration of the update
│ │                       • Signed Payload
│ │
│ ├─Package level of Dell dock:
│ │     Device ID:        [redacted]
│ │     Summary:          A representation of dock update status
│ │     Current version:  01.00.25.01
│ │     Vendor:           Dell Inc. (USB:0x413C)
│ │     Install Duration: 5 seconds
│ │     GUIDs:            d8927ff5-a5b2-5618-848b-8e8bfb75b66f
│ │                       8ceeeffd-51b6-580c-9b75-69143227aff8 ← USB\VID_413C&PID_B06E&hub&status
│ │     Device Flags:     • Updatable
│ │                       • Supported on remote server
│ │                       • Device can recover flash failures
│ │                       • Device is usable for the duration of the update
│ │                       • Unsigned Payload
│ │
│ ├─RTS5413 in Dell dock:
│ │     Device ID:        [redacted]
│ │     Summary:          USB 3.1 Generation 1 Hub
│ │     Current version:  01.21
[... 265 lines cut ...]
│       └── systemd-bootx64.efi
├── initramfs-linux-fallback.img
├── initramfs-linux.img
├── initramfs-linux-lts-fallback.img
├── initramfs-linux-lts.img
├── loader
│   ├── entries
│   │   ├── arch.conf
│   │   └── arch-lts.conf
│   └── loader.conf
├── vmlinuz-linux
└── vmlinuz-linux-lts

13 directories, 18 files

```

**Additional questions**

- Operating system and version: `Arch linux (Linux 6.3.9-arch1-1 #1 SMP PREEMPT_DYNAMIC Wed, 21 Jun 2023 20:46:20 +0000 x86_64 GNU/Linux)`
- Have you tried rebooting? :-)
- Is this a regression? not sure
- Are you using an NVMe disk? yes
- Is secure boot enabled? `bootctl status` says: `Secure Boot: disabled (unknown)`
- Is this a Lenovo system with 'Boot Order Lock' turned on in the BIOS? No.

hughsie, 2023-06-26:
> System UEFI configuration
>    efibootmgr -v
> Does this info contains identifying data?

Usually not. The efibootmgr output is literally what we need to triage this issue.

rustyx, 2023-09-02:
The 10-second wait is quite long and annoying.
Can it be reduced to 0 seconds and the message to just "No updates to process, exiting." ? Or at least be made configurable?

hughsie, 2023-09-02:
@rustyx what does `efibootmgr -v` say?

paddylandau, 2023-09-06:
I've been having this message for a long time. I first asked about it on [Ubuntu Forums](https://ubuntuforums.org/showthread.php?t=2481085), thinking that it was something to do with Ubuntu, but obviously it isn't!

The very few references to this message that I have found makes me suspect that it's something to do with Dell's firmware, as only Dell seems to be referenced — but this might be a coincidence.

(I have a Dell OptiPlex 5480, running Ubuntu 22.04.)

The exact message reads:

```
fwupd-efi version 1.4
WARNING: No updates to process, exiting in 10 seconds.
```

For clarity, this message appears before the Grub menu is displayed.

Curiously, this happens only when I boot from my internal drive. When I boot from an external drive (also running Ubuntu), the message doesn't appear. I have no clue why this could be the case.

```
$ fwupdmgr --version
runtime   org.freedesktop.fwupd         1.7.9
runtime   com.dell.libsmbios            2.4
compile   org.freedesktop.gusb          0.3.10
runtime   org.kernel                    6.2.0-32-generic
compile   com.hughsie.libjcat           0.1.9
compile   org.freedesktop.fwupd         1.7.9
runtime   org.freedesktop.gusb          0.3.10
```

```
$ efibootmgr --version
version 17
```

Secure boot: Enabled

Output for `fwupdmgr --get-devices` is [in Pastebin](https://pastebin.com/pSCJkaYN).

This is a frustrating message, because not having updates to install doesn't require a ten-second WARNING delay.

hughsie, 2023-09-06:
@paddylandau it's `efibootmgr -v` not `efibootmgr --version`

paddylandau, 2023-09-06:
> @paddylandau it's `efibootmgr -v` not `efibootmgr --version`

Oops, sorry! I confused the two short versions.

```
$ efibootmgr --verbose
BootCurrent: 0001
Timeout: 1 seconds
BootOrder: 0000,0008,0009,0001
Boot0000* ubuntu	HD(5,GPT,83910bd1-d8b7-41dc-b2fa-7c48caa2dc42,0xe8cec000,0x10c000)/File(\EFI\ubuntu\shimx64.efi)
Boot0001* Linux Firmware Updater	HD(1,GPT,03be18b6-3a45-423e-8c45-3a2470a37a6a,0x800,0x100000)/File(\EFI\ubuntu\shimx64.efi)\.f.w.u.p.d.x.6.4...e.f.i...
Boot0003* Onboard NIC(IPV4)	PciRoot(0x0)/Pci(0x1f,0x6)/MAC(a4bb6d485278,0)/IPv4(0.0.0.00.0.0.0,0,0)..BO
Boot0004* Onboard NIC(IPV6)	PciRoot(0x0)/Pci(0x1f,0x6)/MAC(a4bb6d485278,0)/IPv6([::]:<->[::]:,0,0)..BO
Boot0008* Onboard NIC(IPV4)	PciRoot(0x0)/Pci(0x1f,0x6)/MAC(a4bb6d485278,0)/IPv4(0.0.0.00.0.0.0,0,0)..BO
Boot0009* Onboard NIC(IPV6)	PciRoot(0x0)/Pci(0x1f,0x6)/MAC(a4bb6d485278,0)/IPv6([::]:<->[::]:,0,0)..BO
```

hughsie, 2023-09-06:
So, that's a bit confusing -- what's supposed to happen is we boot 0000,0008,0009 then 0001 -- but it appears we're booting 0001 and then falling back to ???? which eventually gets to 0000. This very much looks like a firmware bug -- and I'm hesitant to give advice as I don't want to break anything more. If it was my system I'd delete the Boot0001 entry in the system firmware screen and say a little prayer.

paddylandau, 2023-09-06:
> So, that's a bit confusing -- what's supposed to happen is we boot 0000,0008,0009 then 0001 -- but it appears we're booting 0001 and then falling back to ???? which eventually gets to 0000. This [...]

On reading the output with your explanation in mind, it seems that 0001 is the firmware updater.

Firmware updates come in automatically with `fwupdmgr refresh`. When that happens and I restart the machine, the machine automatically reboots into the firmware updater, which applies the firmware updates (I don't have to do a thing; it's 100% automatic). Once that's complete, it reboots into Grub. (Grub leads to the LUKS password prompt, which then leads to Ubuntu.)

Perhaps that's why 0001 takes priority over 0000?

So, maybe what's happening is that 0001 starts in order to process any waiting updates; if there isn't an update, fwupd shows a redundant WARNING (instead of just shrugging its metaphorical shoulders) before proceeding to Grub?

> If it was my system I'd delete the Boot0001 entry in the system firmware screen and say a little prayer.

Hmm, that might be a little risky for me! I'm not technically proficient. Perhaps I'd best leave things alone. If 0001 is the firmware updater, this might prevent me from getting new firmware updates. In any case, I wouldn't how to find the system firmware screen! (Unless you the BIOS?)

hughsie, 2023-09-06:
> it seems that 0001 is the firmware updater

Agree.

> Perhaps that's why 0001 takes priority over 0000?

When you schedule an update we set up BootNext -- which says "only boot 0001 just once" -- and it seems for some reason the firmware you have is doing that every time -- even tho BootNext seems unset in efiboomgr.

> this might prevent me from getting new firmware updates.

Nah, fwupd fixes things up before rebooting so that it's correct. You can delete that one freely.

> In any case, I wouldn't how to find the system firmware screen! (Unless you mean the BIOS?)

Yup, the BIOS screen.

hughsie, 2023-09-06:
> What is the most simple way to safely delete 0001

sudo efibootmgr --delete-bootnum --bootnum 0001

paddylandau, 2023-09-06:
Thank you, @hughsie. To clarify, because I don't want to break anything (as I say, I'm not technically proficient)…

When I reboot and ask for the boot options, BIOS provides me with four main choices as follows (it provides other options such as diagnostics and BIOS setup). I don't know what the asterisk means at the start of the first three lines.
```
*ubuntu
*Onboard NIC(IPV4)
*Onboard NIC(IPV6)
 Linux Firmware Updater
```
When I go into the BIOS settings, those same four items are in the Boot Sequence in the same order.

When I view the BIOS settings for Linux Firmware Updater, it shows the following. I note that these match what `efibootmgr --verbose` showed.
```
Boot Option Name: Linux Firmware Updater
File System List: HD(1,GPT,03BE18B6-3A45-423E-8C45-3A2470A37A6A)
File Name: \EFI\ubuntu\shim64.efi
```
So, to confirm, it's OK to delete the Linux Firmware Updater entry from the BIOS?

I guess, if it doesn't work, it will suffice to manually add back that entry?

paddylandau, 2023-09-06:
I went into the BIOS and deleted the Linux Firmware Updater entry. I restarted and checked the BIOS again, and curiously it has added another entry.
```
ubuntu
UEFI: PM981a NVMe Samsung 512GB
Onboard NIC(IPV4)
Onboard NIC(IPV6)
```
Also, on rebooting…
```
$ efibootmgr --verbose
BootCurrent: 0001
Timeout: 1 seconds
BootOrder: 0001,0000,0008,0009
Boot0000* ubuntu	HD(5,GPT,83910bd1-d8b7-41dc-b2fa-7c48caa2dc42,0xe8cec000,0x10c000)/File(\EFI\ubuntu\shimx64.efi)
Boot0001* ubuntu	HD(1,GPT,03be18b6-3a45-423e-8c45-3a2470a37a6a,0x800,0x100000)/File(\EFI\UBUNTU\SHIMX64.EFI)
Boot0003* Onboard NIC(IPV4)	PciRoot(0x0)/Pci(0x1f,0x6)/MAC(a4bb6d485278,0)/IPv4(0.0.0.00.0.0.0,0,0)..BO
Boot0004* Onboard NIC(IPV6)	PciRoot(0x0)/Pci(0x1f,0x6)/MAC(a4bb6d485278,0)/IPv6([::]:<->[::]:,0,0)..BO
Boot0008* Onboard NIC(IPV4)	PciRoot(0x0)/Pci(0x1f,0x6)/MAC(a4bb6d485278,0)/IPv4(0.0.0.00.0.0.0,0,0)..BO
Boot0009* Onboard NIC(IPV6)	PciRoot(0x0)/Pci(0x1f,0x6)/MAC(a4bb6d485278,0)/IPv6([::]:<->[::]:,0,0)..BO
```
And…

The WARNING message has disappeared!

So, **thank you**, this seems to have worked.

We'll see what happens when Dell next submits a firmware update. Who knows when that will be!

rustyx, 2023-09-07:
> @rustyx what does `efibootmgr -v` say?

```
$ efibootmgr -v
BootNext: 0001
BootCurrent: 0000
Timeout: 0 seconds
BootOrder: 0000,0003,001E,001C,001D,001F,0020,0021,0022,0023,0024,0025,0026,0001
Boot0000* ubuntu        HD(1,GPT,c0d9fdce-4025-4307-adc5-3d3082e7cee7,0x800,0x82000)/File(\EFI\ubuntu\shimx64.efi)
Boot0001* Linux-Firmware-Updater        HD(1,GPT,c0d9fdce-4025-4307-adc5-3d3082e7cee7,0x800,0x82000)/File(\EFI\ubuntu\fwupdx64.efi)
Boot0003* Windows Boot Manager  HD(1,GPT,c0d9fdce-4025-4307-adc5-3d3082e7cee7,0x800,0x82000)/File(\EFI\Microsoft\Boot\bootmgfw.efi)WINDOWS.........x...B.C.D.O.B.J.E.C.T.=.{.9.d.e.a.8.6.2.c.-.5.c.d.d.-.4.e.7.0.-.a.c.c.1.-.f.3.2.b.3.4.4.d.4.7.9.5.}...4................
Boot0010  Setup FvFile(721c8b66-426c-4e86-8e99-3457c46ab0b9)
Boot0011  Boot Menu     FvFile(126a762d-5758-4fca-8531-201a7f57f850)
Boot0012  Diagnostic Splash Screen      FvFile(a7d8d9a6-6ab0-4aeb-ad9d-163e59a7a380)
Boot0013  Lenovo Diagnostics    FvFile(3f7e615b-0d45-4f80-88dc-26b234958560)
Boot0014  Asset Information     FvFile(da465b87-a26f-4c12-b78a-0361428fa026)
Boot0015  Regulatory Information        FvFile(478c92a0-2622-42b7-a65d-5894169e4d24)
Boot0016  ThinkShield secure wipe       FvFile(3593a0d5-bd52-43a0-808e-cbff5ece2477)
Boot0017  ThinkShield Passwordless Power-On Device Manager      FvFile(08448b41-7f83-49be-82a7-0e84790ab133)
Boot0018  Wi-Fi Configuration   FvFile(d3aaff0f-cb22-4792-896c-802c2e9383ba)-.A.p.p...
Boot0019  Intel(R) MEBx FvFile(29a70110-7762-4211-ae88-fab19b7665be)
Boot001A  Startup Interrupt Menu        FvFile(f46ee6f4-4785-43a3-923d-7f786c3c8479)
Boot001B  Rescue and Recovery   FvFile(665d3f60-ad3e-4cad-8e26-db46eee9f1b5)
Boot001C* USB CD        VenMsg(bc7838d2-0f82-4d60-8316-c068ee79d25b,86701296aa5a7848b66cd49dd3ba6a55)
Boot001D  USB FDD       VenMsg(bc7838d2-0f82-4d60-8316-c068ee79d25b,6ff015a28830b543a8b8641009461e49)
Boot001E* NVMe0 VenMsg(bc7838d2-0f82-4d60-8316-c068ee79d25b,001c199932d94c4eae9aa0b6e98eb8a400)
Boot001F* NVMe1 VenMsg(bc7838d2-0f82-4d60-8316-c068ee79d25b,001c199932d94c4eae9aa0b6e98eb8a401)
Boot0020  ATA HDD0      VenMsg(bc7838d2-0f82-4d60-8316-c068ee79d25b,91af625956449f41a7b91f4f892ab0f600)
Boot0021* USB HDD       VenMsg(bc7838d2-0f82-4d60-8316-c068ee79d25b,33e821aaaf33bc4789bd419f88c50803)
Boot0022  PXE BOOT      VenMsg(bc7838d2-0f82-4d60-8316-c068ee79d25b,78a84aaf2b2afc4ea79cf5cc8f3d3803)
Boot0023  LENOVO CLOUD  VenMsg(bc7838d2-0f82-4d60-8316-c068ee79d25b,ad38ccbbf7edf04d959cf42aa74d3650)/Uri(https://download.lenovo.com/pccbbs/cdeploy/efi/boot.efi)
Boot0024  ON-PREMISE    VenMsg(bc7838d2-0f82-4d60-8316-c068ee79d25b,ad38ccbbf7edf04d959cf42aa74d3650)/Uri()
Boot0025  Other CD      VenMsg(bc7838d2-0f82-4d60-8316-c068ee79d25b,aea2090adfde214e8b3a5e471856a35400)
Boot0026  Other HDD     VenMsg(bc7838d2-0f82-4d60-8316-c068ee79d25b,ca88c2349e7ae947beeb43038a5aeae700)
Boot0027* IDER BOOT CDROM       PciRoot(0x0)/Pci(0x14,0x0)/USB(11,1)
Boot0028* IDER BOOT Floppy      PciRoot(0x0)/Pci(0x14,0x0)/USB(11,0)
Boot0029* ATA HDD       VenMsg(bc7838d2-0f82-4d60-8316-c068ee79d25b,91af625956449f41a7b91f4f892ab0f6)
Boot002A* ATAPI CD      VenMsg(bc7838d2-0f82-4d60-8316-c068ee79d25b,aea2090adfde214e8b3a5e471856a354)
```

Every time my laptop boots I get the 10-second fwupd warning message, after that Grub starts normally.

As a temporary workaround I have `sudo rm /boot/efi/EFI/ubuntu/fwupdx64.efi` in my startup script, which is obviously far from ideal.

Laptop is Lenovo ThinkPad X1 Extreme Gen 5, if that matters.

hughsie, 2023-09-07:
@rustyx it seems your firmware isn't clearing BootNext for some reason -- it's only supposed to be used once. I'm guessing `sudo efibootmgr --delete-bootnext` fixes things?

rustyx, 2023-09-07:
> @rustyx it seems your firmware isn't clearing BootNext for some reason -- it's only supposed to be used once. I'm guessing `sudo efibootmgr --delete-bootnext` fixes things?

It does... ok so I can change my workaround to that one... but perhaps `fwupdx64` could clear the BootNext automatically as a workaround for the firmware, especially when there is no update available?

hughsie, 2023-09-07:
@superm1 how do you feel about deleting BootNext if it exists on first-boot after doing a firmware update?

superm1, 2023-09-07:
> @superm1 how do you feel about deleting BootNext if it exists on first-boot after doing a firmware update?

Queazy. It's obviously a firmware bug if it's not used.

But it's also a worthwhile cleanup, just like we delete old capsules from the ESP at startup (iirc).

I think we would also need to log something into the fwupd log that such a bug was detected.

paddylandau, 2023-09-10:
@hughsie — Update:

Yesterday, as luck would have it, Dell released a firmware update.

I'm pleased to report that everything worked automatically as it should, and the WARNING message remains absent, as it should.

So, everything is working, thank you!

closed 2023-10-13 by hughsie