fwupd/fwupd #5958: How is Linux runtime firmware matched to a device?

merge, 2023-06-30:
**Describe the question**
AFAIU, (optional) firmware that applies to a "machine" like a laptop can use `<firmware type="runtime">filename.elf</firmware>` for it to be installed to `/lib/firmware`. Linux can then load it at runtime. I'd like to understand how the hardware-matching works in such a case (without GUID?). Are there examples? Can I (as a vendor) *in addition* somehow provide the GUID of the processor (or SoC) in such a case?

thank you very much

hughsie, 2023-06-30:
I'm a bit confused -- fwupd only deals with firmware flashed to devices -- it doesn't handle anything in /lib/firmware at all.

merge, 2023-06-30:
that's what I wasn't sure about - that's also an answer, thanks!

background: fwupd would provide a nice update-mechanism for situations where firmware is free software (and so doesn't need to be in "nonfree" distribution packages like Debian's) but creating a "normal" (Debian) package would be a lot of work because of non-packaged build-tools or toolchains (even if free software).

closed 2023-06-30 by merge

superm1, 2023-06-30:
> that's what I wasn't sure about - that's also an answer, thanks!
>
> background: fwupd would provide a nice update-mechanism for situations where firmware is free software (and so doesn't need to be in [...]

Yes. you can specify in the update metadata in the cab that the license is a free software license and include links to URLs as well.