fwupd/fwupd #5970: Checksum validation error at Satellite repo sync

zahnwiegebiss, 2023-07-05:
**Describe the bug**
When I try to sync the repo into RedHat Satellite I get the following error:

RH Satellite:
```
A file located at the url https://cdn.fwupd.org/downloads/firmware.xml.gz.jcat failed validation due to checksum.
```
Foreman-Katello:
```
A file located at the url https://cdn.fwupd.org/downloads/firmware-05131-testing.xml.gz failed validation due to checksum. Expected '7b81279ebf85191cb56a9e26884ce1d002514d07f9458ca8857e1b135c114812', Actual 'be68240689e10fdcc49a8c68202dc113bcdb48d402f3774cceaa62c7122c95ef
```

Here's the documentation I followed:
https://access.redhat.com/articles/5215571

**Steps to Reproduce**
Add filerepo to Satellite and klick sync

**Expected behavior**
Files should sync into Satellite repo.

**fwupd version information**
Please provide the version of the daemon and client.

I'm using Satellite GUI or hammer command:
```
hammer repository synchronize --product-id 244 --name lvfs --organization-id 1
[............................................................................................................................................] [100%]
No content added.
Total steps: 1128/7715
--------------------------------
Associating Content: 0/0
Downloading Artifacts: 2/2
Downloading Metadata: 1/1
Parsing Metadata Lines: 1125/7712
Error: A file located at the url https://cdn.fwupd.org/downloads/firmware-06600-stable.xml.gz failed validation due to checksum.
```

**Additional questions**

- Operating system and version:
 REHL8/CentOS Stream 8
- Have you tried rebooting?
 Yes

hughsie, 2023-07-05:
> A file located at the url https://cdn.fwupd.org/downloads/firmware.xml.gz.jcat failed validation due to checksum.

That's not unexpected -- the firmware.xml.gz.jcat file regenerates every few minutes, and so if you start the PULP sync it's highly plausible it's changed before the download has completed. If you run this again it should sync as the remote is mostly populated already.

> A file located at the url https://cdn.fwupd.org/downloads/firmware-05131-testing.xml.gz failed validation

This is less expected and more of a worry. I'll debug this a bit now.

zahnwiegebiss, 2023-07-05:
Hi hughsie,
I tried it several times now in a very short period (less than 5 min) and I got the following results:
```
A file located at the url https://cdn.fwupd.org/downloads/firmware-testing.xml.gz failed validation due to checksum. Expected '7b81279ebf85191cb56a9e26884ce1d002514d07f9458ca8857e1b135c114812', Actual 'be68240689e10fdcc49a8c68202dc113bcdb48d402f3774cceaa62c7122c95ef'
```
```
A file located at the url https://cdn.fwupd.org/downloads/firmware-05131-testing.xml.gz failed validation due to checksum. Expected '7b81279ebf85191cb56a9e26884ce1d002514d07f9458ca8857e1b135c114812', Actual 'be68240689e10fdcc49a8c68202dc113bcdb48d402f3774cceaa62c7122c95ef'
```
```
A file located at the url https://cdn.fwupd.org/downloads/firmware-testing.xml.gz failed validation due to checksum. Expected '7b81279ebf85191cb56a9e26884ce1d002514d07f9458ca8857e1b135c114812', Actual 'be68240689e10fdcc49a8c68202dc113bcdb48d402f3774cceaa62c7122c95ef'
```
```
A file located at the url https://cdn.fwupd.org/downloads/firmware-05131-testing.xml.gz failed validation due to checksum. Expected '7b81279ebf85191cb56a9e26884ce1d002514d07f9458ca8857e1b135c114812', Actual 'be68240689e10fdcc49a8c68202dc113bcdb48d402f3774cceaa62c7122c95ef'
```
```
A file located at the url https://cdn.fwupd.org/downloads/firmware-testing.xml.gz failed validation due to checksum. Expected '7b81279ebf85191cb56a9e26884ce1d002514d07f9458ca8857e1b135c114812', Actual 'be68240689e10fdcc49a8c68202dc113bcdb48d402f3774cceaa62c7122c95ef
```
```
A file located at the url https://cdn.fwupd.org/downloads/firmware-testing.xml.gz failed validation due to checksum. Expected '7b81279ebf85191cb56a9e26884ce1d002514d07f9458ca8857e1b135c114812', Actual 'be68240689e10fdcc49a8c68202dc113bcdb48d402f3774cceaa62c7122c95ef'
```
```
A file located at the url https://cdn.fwupd.org/downloads/firmware-05131-testing.xml.gz failed validation due to checksum. Expected '7b81279ebf85191cb56a9e26884ce1d002514d07f9458ca8857e1b135c114812', Actual 'be68240689e10fdcc49a8c68202dc113bcdb48d402f3774cceaa62c7122c95ef'
```
```
A file located at the url https://cdn.fwupd.org/downloads/firmware-05131-testing.xml.gz failed validation due to checksum. Expected '7b81279ebf85191cb56a9e26884ce1d002514d07f9458ca8857e1b135c114812', Actual 'be68240689e10fdcc49a8c68202dc113bcdb48d402f3774cceaa62c7122c95ef'
```
These errors are now from my private Katello/Foreman because I'm not allowed to send stuff of my company's satellite.

So it's switching all the time.

I'm kind of blind, because I cannot see the content of https://cdn.fwupd.org/downloads. And I'm also not able to see which file has the wrong checksum.
Just tell me, if I can assist with sth.

hughsie, 2023-07-05:
So I'm a bit confused myself now:

    wget https://cdn.fwupd.org/downloads/firmware-05131-testing.xml.gz
    $ sha256sum firmware-05131-testing.xml.gz 
    7b81279ebf85191cb56a9e26884ce1d002514d07f9458ca8857e1b135c114812  firmware-05131-testing.xml.gz

Do you get the same?

hughsie, 2023-07-05:
and PULP_MANIFEST i see:

    firmware-05131-testing.xml.gz,7b81279ebf85191cb56a9e26884ce1d002514d07f9458ca8857e1b135c114812,156022

sooo I'm a bit confused where be68240689e10fdcc49a8c68202dc113bcdb48d402f3774cceaa62c7122c95ef is coming from. Can you copy the downloaded firmware-05131-testing.xml.gz file with be68240689e10fdcc49a8c68202dc113bcdb48d402f3774cceaa62c7122c95ef off the server and have a look? i.e. is it a zero-byte file or a gz-decompressed version of itself perhaps? it's weird it's not happening with the .xz version.

zahnwiegebiss, 2023-07-05:
Sadly there's no file on the server.
My PULP_MANIFEST was completely empty.

Just for fun I uploaded the the files, that I downloaded manually, into the Foreman repo and now they appear in the PULP_MANIFEST.

Here I now can see, where the sum is coming from:
```
firmware-testing.xml.gz,be68240689e10fdcc49a8c68202dc113bcdb48d402f3774cceaa62c7122c95ef,1035782
firmware-05131-testing.xml.gz,be68240689e10fdcc49a8c68202dc113bcdb48d402f3774cceaa62c7122c95ef,1035782
```

hughsie, 2023-07-05:
I wonder if the CDN is serving an outdated copy -- it's not on the CDN PURGE list. I've added it now and manually purged -- can you try again please?

zahnwiegebiss, 2023-07-05:
OK, the error is the same but now we have a different number and an additional file:
```
A file located at the url https://cdn.fwupd.org/downloads/firmware.xml.gz failed validation due to checksum. Expected '79d572b2b48fc3861d4c2a7ea5f2a2a4b8560b838f3529e03143de9392a141c7', Actual '55ae58e328b6cd8de191d622cfba21974c92514f5d00b4a28401654ebce508a1'
```
```
A file located at the url https://cdn.fwupd.org/downloads/firmware-06600-stable.xml.gz failed validation due to checksum. Expected '79d572b2b48fc3861d4c2a7ea5f2a2a4b8560b838f3529e03143de9392a141c7', Actual '55ae58e328b6cd8de191d622cfba21974c92514f5d00b4a28401654ebce508a1'
```
This is really kind of weird!

hughsie, 2023-07-06:
Weird indeed, because it's all too complicated. I've done this: https://lists.linuxfoundation.org/pipermail/lvfs-announce/2023-July/000081.html

I'll deploy in a few hours.

zahnwiegebiss, 2023-07-06:
OK, cool.
Then I'll wait till tomorrow and check then again, if it works.

zahnwiegebiss, 2023-07-09:
Hi,
sorry for the late answer.

I still have the issue:
```
No content added.
Total steps: 1013/7744
--------------------------------
Associating Content: 0/0
Downloading Artifacts: 9/9
Downloading Metadata: 1/1
Parsing Metadata Lines: 1003/7734
```
```
A file located at the url https://cdn.fwupd.org/downloads/firmware-05134-testing.xml.gz failed validation due to checksum. Expected '01bf5051b7f46ffceeb6f32833c9adb1c344946e10f236ed656dfdcf0c6c2012', Actual '01b312eb5088c8a9ccfc7eb6f088c9f3fe19addbd7482b609807902d0b0c7798'
```

Could it be that Foreman/Satellite is doing som strange magic?

Just for fun I downloaded the Files with weget, imported the files manually to the Foreman repo and had a look at the PULP_MANIFEST.
When I did this, the checksums matched to the upstream ones.

I clud raise a Red Hat Case for Satellite on Tuesday, when I'm back at work.

hughsie, 2023-07-09:
If it helps, this isn't the first case we've heard about where satellite checksums are wrong. If you use sync-pulp it seems to always work.

hughsie, 2023-07-13:
Thanks @zahnwiegebiss -- can you share the RH case ID or BZ number? I'm also rhughes_at_redhat.com if that helps.

zahnwiegebiss, 2023-07-13:
sure @hughsie , here's the Case ID: 03561960

orbit80, 2023-07-18:
Hey guys 
im facing the same issue , trying to sync a openshift4 registry via satellite 6.11 , ending with this failed validation due to checksum.
also opened a RH case , would you kindly share the solution offered by RH so far, if any ?

hughsie, 2023-07-18:
There's no solution so far. I'm also trying to assist, but I'm having problems understanding where the downloading actually gets done -- I have no familiarity with the satellite project. Do you know if satellite is using foreman to actually sync the PULP_MANIFEST repos?

orbit80, 2023-07-18:
as far as i know , it uses forman.
according to the error , i guess its indeed foreman how syncs the PULP_MANIFEST repos : 
A file located at the url https://registry.redhat.io/v2/openshift4/apb-base/blobs/sha256:d29f4421bfc41cf93cd17550f1dfdd6440c02b94d379466f8dbc87457fc8f44c failed validation due to checksum. (Katello::Errors::Pulp3Error)

hughsie, 2023-07-18:
100% not an expert in this web stuff, but one hunch I have is that the content type might be being handled wrongly. e.g. see:

    curl https://cdn.fwupd.org/downloads/firmware-05131-testing.xml.gz --output firmware-05131-testing.xml.gz -v
    ...
    < content-type: text/xml; charset=utf-8
    < server: gunicorn
    < content-encoding: gzip
    < content-disposition: inline; filename=firmware-05131-testing.xml.gz

Maybe that should be `application/gzip` not `text/xml`? It's being set when using https://fwupd.org/downloads/firmware-05131-testing.xml.gz -- i.e. no CDN so it's probably the webapp or gunicorn that's doing it (if it's indeed not expected). Running the LVFS locally without gunicorn I also get

    < Server: Werkzeug/2.3.6 Python/3.11.4
    < Content-Encoding: gzip
    < Content-Disposition: inline; filename=firmware.xml.gz
    < Content-Type: text/xml; charset=utf-8

...so I wonder if foreman is detecting the content type and doing something slightly odd.

> where be68240689e10fdcc49a8c68202dc113bcdb48d402f3774cceaa62c7122c95ef is coming from

    $ sha256sum firmware-05131-testing.xml.gz 
    7b81279ebf85191cb56a9e26884ce1d002514d07f9458ca8857e1b135c114812  firmware-05131-testing.xml.gz
    $ gunzip firmware-05131-testing.xml.gz
    $ sha256sum firmware-05131-testing.xml
    be68240689e10fdcc49a8c68202dc113bcdb48d402f3774cceaa62c7122c95ef  firmware-05131-testing.xml

I think that means foreman is auto-ungzipping the downloaded files before adding it to the repo. Looking at the various RFCs it seems flask is doing exactly the right thing setting `content-type: text/xml` e.g. https://stackoverflow.com/questions/23600229/what-content-type-header-to-use-when-serving-gzipped-files

I guess one workaround might be to switch the LVFS to send `application/octet-stream` for all archives, although that sounds bad at worst, and I feel like it might break other software that **is** using `Accept-Encoding` correctly. 

I think this might be https://github.com/pulp/pulp-smash/issues/1309 -- or the same fix needed in a different project perhaps.

ggainey, 2023-07-20:
> I think this might be [pulp/pulp-smash#1309](https://github.com/pulp/pulp-smash/issues/1309) -- or the same fix needed in a different project perhaps.

That isn't a "fix" - that is pulp-smash working around a borked upstream-webserver-configuration, in a way that will work until it's used against a working webserver that uses that header the way the Rules Of The Internet say it should be used, and then pulp-smash will be broken again except in the other direction. See https://pulp.plan.io/issues/9224#note-4 for an exhaustive set of references.

hughsie, 2023-07-20:
I've deployed https://gitlab.com/fwupd/lvfs-website/-/commit/4280f8d06a3690498f7988020c0794910ec37f60 to the LVFS to unstick this customer which we can revert when the aiohttp fix is applied to Pulp.

dralley, 2023-07-20:
>It's literally the default webserver configuration

This particular misconfiguration is rather famous for having been incredibly common due to being in the default configuration of httpd from like 2009-2011.   It's still wrong.

> notably no other client http library auto-decompresses content like aiohttp does.

Right, because those other client libraries are being "pragmatic" about how many webservers exist in the wild that are misconfigured this way, and aiohttp refuses to be pragmatic and instead tries to follow the specs religiously.  This has been a source of great annoyance at times, but they're not wrong.

Convincing people of this is difficult because Chrome, Firefox, wget, etc. have all carried a bunch of extra, non-standard heuristics to work around this, (and those heuristics have at time lead to breaking other things i.e. they aren't harmless).

 We're at a point now where this misconfiguration is finally not very common anymore, so it's easier if everyone just follows the specs in the first place.  

If you're serving a .gz file, and don't want it to be automatically decompressed, it shouldn't have a `Content-Encoding: gzip`, because that will tell the client that "I, the server, have encoded this response in a certain way (and you ought to reverse it)".

hughsie, 2023-07-20:
> Right, because those other client libraries are being "pragmatic" about how many webservers exist in the wild

So why was https://github.com/pulp/pulp-smash/pull/1310 merged?

hughsie, 2023-07-20:
>  so the sync should work again later this day, right?

I've purged the CDNs, so it should work now.

zahnwiegebiss, 2023-07-20:
Sadly I still get the errors:
```
No content added.
Total steps: 1512/7890
--------------------------------
Associating Content: 0/0
Downloading Artifacts: 8/8
Downloading Metadata: 1/1
Parsing Metadata Lines: 1503/7881
```
```
A file located at the url https://cdn.fwupd.org/downloads/firmware-testing.xml.gz failed validation due to checksum. Expected 'a6ae4bc292e042dfe2e92163dd81a7fdf2acb89ac39d8035ace002f1dc8cd4c2', Actual '7e99c7a0a95ae4038615dc5ee71163c040365d798e1cf4514972425562356ce9'
```

hughsie, 2023-07-20:
Maybe I didn't invalidate enough. What does curl -v say?

hughsie, 2023-07-20:
Ohh hang on -- that's the testing metadata -- that's a different issue. I forgot we included that. Fix incoming.

hughsie, 2023-07-20:
https://gitlab.com/fwupd/lvfs-website/-/commit/ae913adb9fe2578e293534b49b0b860b9010ecc2 is that fix -- I'll deploy now which will take ~10 mins.

hughsie, 2023-07-20:
@zahnwiegebiss Okay, can you try now?

zahnwiegebiss, 2023-07-21:
sorry, I already left the PC yesterday.

sadly it's still not working:
```
A file located at the url https://cdn.fwupd.org/downloads/firmware-testing.xml.gz.asc failed validation due to checksum. Expected 'aced6e8b99b246bb11c16bc961b872b1edc46a31da7a5badd0a2491dda1e6786', Actual '659e57ea825126c869d3fd1ee8c4057accd902c852180c39fa91db19f4b53b7f'
```
here's the wget output:
```
wget -v https://cdn.fwupd.org/downloads/firmware-testing.xml.gz.asc
--2023-07-21 20:20:10--  https://cdn.fwupd.org/downloads/firmware-testing.xml.gz.asc
Resolving cdn.fwupd.org (cdn.fwupd.org)... 199.232.190.49, 2a04:4e42:6f::561
Connecting to cdn.fwupd.org (cdn.fwupd.org)|199.232.190.49|:443... connected.
HTTP request sent, awaiting response... 200 OK
Length: 488 [application/octet-stream]
Saving to: ‘firmware-testing.xml.gz.asc’

firmware-testing.xml.gz.asc   100%[=================================================>]     488  --.-KB/s    in 0s

2023-07-21 20:20:11 (30.7 MB/s) - ‘firmware-testing.xml.gz.asc’ saved [488/488]
```
and here's the sha256sum output:
```
sha256sum firmware-testing.xml.gz.asc
659e57ea825126c869d3fd1ee8c4057accd902c852180c39fa91db19f4b53b7f  firmware-testing.xml.gz.asc
```

hughsie, 2023-07-21:
The CDN is funky. Try syncing from  `https://fwupd.org/downloads` i.e. remove the `cdn.`

zahnwiegebiss, 2023-07-21:
this time it looks better, I get a different error message:
```
No content added.
Total steps: 2052/8436
--------------------------------
Associating Content: 500/500
Downloading Artifacts: 48/48
Downloading Metadata: 1/1
Parsing Metadata Lines: 1503/7887
```
```
A file located at the url https://fwupd.org/downloads/0d5f74ce2e8d24697569fe4e04c86e463fedb9705db59db711263a60b3a0c07b-Signed_1152921505696408660.cab failed validation due to size. Expected '25503905', Actual '25503904'
```
It is starting to download something but then Satellite is rolling everything back...

hughsie, 2023-07-21:
I've merged https://gitlab.com/fwupd/lvfs-website/-/commit/e2f9758b3380e5c701bc1c3bfc42740c9dbfa167 -- and I'll run that in a few minutes.

hughsie, 2023-07-21:
[image]

I've just regenerated the testing remote -- which has then re-created the PULP_MANIFEST with the correct values. Can you try again please.

zahnwiegebiss, 2023-07-21:
I just started a sync, but it will take some time.
My Foreman just started its syncplan and is syncing all the repos and queued fwupd tat the end. Due to weak hardware this will take some time :-(

I'll repost when sync has finished.

zahnwiegebiss, 2023-07-22:
Good news!
It's working:
```
Added Files: 7887
Total steps: 16657/16657
--------------------------------
Associating Content: 7887/7887
Downloading Artifacts: 882/882
Downloading Metadata: 1/1
Parsing Metadata Lines: 7887/7887
Un-Associating Content: 0/0
```
Thanks for your effort!

hughsie, 2023-07-22:
Great! I'll close this upstream issue and we can continue the pulp onebon the RH tracker. Thanks for all your patience while we worked this out. All good bugs have three root causes! 🙂

closed 2023-07-22 by hughsie

adampankow, 2023-08-25:
@hughsie I'm a bit late to the party, but I did just want to confirm that the resolutions here seem to have fixed the similar issues I reported in #5849. Thank you for all that had their hand in resolving this tricky issue. :+1: