fwupd/fwupd #5975: ThinkPad T480 Thunderbolt controller not found

xvitaly, 2023-07-07:
**Describe the bug**
Trying to install T480 Thunderbolt controller update, but fwupd fails.

**Steps to Reproduce**
```
$ fwupdmgr get-upgrades
Devices with no available firmware updates: 
 • MZVLW256HEHP-000L7
 • TPM
 • Thunderbolt Controller
 • Thunderbolt Controller
 • Thunderbolt Controller
 • Thunderbolt host controller
Devices with the latest available firmware version:
 • Embedded Controller
 • Intel Management Engine
 • System Firmware
```

```
$ wget https://fwupd.org/downloads/391c6be40422dc1d253f5a6b9bdf0c122ddc5f61f9fcb3ba4e61cf42a7c6f7bc-Lenovo-ThinkPad-T480-SystemFirmware-23.cab
$ fwupdmgr install ~/391c6be40422dc1d253f5a6b9bdf0c122ddc5f61f9fcb3ba4e61cf42a7c6f7bc-Lenovo-ThinkPad-T480-SystemFirmware-23.cab
Decompressing…           [ -                                     ]
No supported devices found
```

**Expected behavior**
The update should be installed.

**fwupd version information**
Please provide the version of the daemon and client.

```shell
$ fwupdmgr --version
compile   org.freedesktop.fwupd         1.8.14
compile   com.hughsie.libxmlb           0.3.11
compile   com.hughsie.libjcat           0.1.13
runtime   org.freedesktop.fwupd-efi     1.4
compile   org.freedesktop.gusb          0.4.5
runtime   com.hughsie.libjcat           0.1.13
runtime   com.dell.libsmbios            2.4
runtime   org.freedesktop.gusb          0.4.5
runtime   org.freedesktop.fwupd         1.8.14
runtime   org.kernel                    6.2.10-200.fc37.x86_64
```

Please note how you installed it (`apt`, `dnf`, `pacman`, source, etc):

Installed from Fedora 37 repository.

<summary>**fwupd device information**</summary>

Please provide the output of the fwupd devices recognized in your system.

```shell
LENOVO 20L50007RT
│
├─Core™ i7-8550U CPU @ 1.80GHz:
│     Device ID:          4bde70ba4e39b28f9eab1628f9dd6e6244c03027
│     Current version:    0x000000f0
│     Vendor:             Intel
│     GUIDs:              b9a2dd81-159e-5537-a7db-e7101d164d3f ← cpu
│                         30249f37-d140-5d3e-9319-186b1bd5cac3 ← CPUID\PRO_0&FAM_06
│                         561403e8-143a-5071-ab09-bf5e4c146983 ← CPUID\PRO_0&FAM_06&MOD_8E
│                         54bdc429-0179-5019-9cb6-8a343324c97c ← CPUID\PRO_0&FAM_06&MOD_8E&STP_A
│     Device Flags:       • Internal device
│   
├─Embedded Controller:
│     Device ID:          2292ae5236790b47884e37cf162dcf23bfcd1c60
│     Summary:            UEFI ESRT device
[... 252 lines cut ...]
├── initramfs-0-rescue-d81e28cb17d24c1e9176a83c32f34d23.img
├── initramfs-6.2.10-200.fc37.x86_64.img
├── initramfs-6.2.9-200.fc37.x86_64.img
├── loader
│   └── entries  [error opening dir]
├── lost+found  [error opening dir]
├── symvers-6.2.10-200.fc37.x86_64.gz -> /lib/modules/6.2.10-200.fc37.x86_64/symvers.gz
├── symvers-6.2.9-200.fc37.x86_64.gz -> /lib/modules/6.2.9-200.fc37.x86_64/symvers.gz
├── System.map-6.2.10-200.fc37.x86_64
├── System.map-6.2.9-200.fc37.x86_64
├── vmlinuz-0-rescue-d81e28cb17d24c1e9176a83c32f34d23
├── vmlinuz-6.2.10-200.fc37.x86_64
└── vmlinuz-6.2.9-200.fc37.x86_64

6 directories, 12 files
```

**Additional questions**

- Operating system and version: Fedora 37
- Have you tried rebooting? Yes
- Is this a regression? Maybe
- Are you using an NVMe disk? yes
- Is secure boot enabled? Yes
- Is this a Lenovo system with 'Boot Order Lock' turned on in the BIOS? No

hughsie, 2023-07-07:
@anttih can you try https://github.com/fwupd/fwupd/pull/5976 please? If it helps I can do a Fedora RPM.

anttih, 2023-07-08:
@hughsie I quickly tried but can't install it due to some of the [Nix patches](https://github.com/NixOS/nixpkgs/blob/nixos-unstable/pkgs/os-specific/linux/firmware/fwupd/default.nix#L140) not working for the `master` version of fwupd. I'll see if I can make a workaround.

anttih, 2023-07-08:
I backported the patch and it works! I was able to update the Thunderbolt firmware from 14.0 -> 23.0.

```
├─Thunderbolt host controller:
│     Device ID:          d3c506436c4bb90704c5837177b7da616901aabd
│     Summary:            Unmatched performance for high-speed I/O
│     Current version:    23.00
│     Vendor:             Lenovo (THUNDERBOLT:0x0109, TBT:0x0109)
│     Update State:       Success
│     GUIDs:              6d4d771c-0579-504d-92a1-737180c7b360 ← THUNDERBOLT\VEN_0109&DEV_1707
│                         8487e4cb-fe6b-5cb6-ad88-46535fb110e1 ← THUNDERBOLT\VEN_0109&DEV_1707&REV_00
│                         fe328c2d-5300-597b-a21d-991116db8ab6 ← TBT-01091707-native
│                         15a5ae1d-0ec9-5013-9221-cd6dd753fcb0 ← TBT-01091707-native-controller0-0
│
```

closed 2023-07-08 by hughsie

xvitaly, 2023-07-08:
@hughsie Can you build a Fedora RPM scratch build with this patch? I can test.

hughsie, 2023-07-08:
Can you try https://copr.fedorainfracloud.org/coprs/rhughes/fwupd/ please

mrhpearson, 2023-07-08:
Thanks to all for their patience while we tracked this down and mega-thanks to @hughsie for figuring out the fix - very awesome!

xvitaly, 2023-07-08:
> Can you try https://copr.fedorainfracloud.org/coprs/rhughes/fwupd/ please

It works. Thanks.

hughsie, 2023-07-08:
Excellent, thanks. I've backported it to 1_8_X too.

xvitaly, 2023-07-15:
@hughsie Can you push 1.8.17 to Fedora 37 repositories please?

hughsie, 2023-07-15:
Mea culpa; building now. I'll submit the update in ~1h.

xvitaly, 2023-07-15:
> Mea culpa; building now. I'll submit the update in ~1h.

Thank you.

hughsie, 2023-07-15:
https://bodhi.fedoraproject.org/updates/FEDORA-2023-ed2793b658

TheMrRandomDude, 2023-07-31:
Hey people. What if fwupdmgr get-devices or get-upgrades (after refresh), (tried both as sudo and without,) isn't finding "Thunderbolt Controller" at all. None in list. I did apt update & upgrade beforehand too.

christal87, 2024-01-21:
I've came across this closed issue today while updating  the tbolt fw on my T480. Debian 12.4.0 has an older version (1.8.12) packed, which I suspect was affected. So I cloned the main branch, set up a test virtualenv as per the readme, shut down the older systemd service as a temporary measure and updated the fw successfully from 14 to 23. I'd like to leave this here to anyone as a guideline. Thanks guys for your hard work!

va1entin, 2024-07-15:
I kept having this issue on my T480 with fwupd `1.9.21` on Arch with TB assist mode and secure boot disabled in BIOS. I also reset the BIOS, making sure TB assist mode and secure boot were still disabled, to no avail: No "Thunderbolt host controller" device would be visible whatever I tried.

I managed to fix it by setting Thunderbolt `Security level` to `User Authorization` in BIOS as per @xvitaly's screenshot from earlier. It was set to `Display Port and USB` on my machine for some reason - even after BIOS reset.
After changing the security level the Thunderbolt host controller appeared in fwupdmgr and I was able to update FW from 21 to 23.

Hope this might help someone in the future!

> > [@xvitaly](https://github.com/xvitaly) , this GUID seems the assist mode is still enabled, the FW team requested to enter F10 to exit BIOS after disabled assist mode. Could you check this? thanks
> 
> [image]