fwupd/fwupd #5979: xps 17 9720 finding no updates

droidben, 2023-07-09:
**Describe the bug**
I have a Dell xps17 9720 running Ubuntu 22.04.2 with uefi and secureboot enabled. UEFI capsule updates are enabled in the BIOS. despite this, fwupd failes to update my firmwares. I have tried manually setting the `OverrideESPMountPoint` to `/boot/efi`, which didn't help. Yes, udisk is installed. My boot-disk is ZFS.

**Steps to Reproduce**
```
$ sudo fwupd.fwupdmgr refresh --force 
WARNING: This package has not been validated, it may not work properly.
Updating lvfs
Downloading…             [***************************************]
Successfully downloaded new metadata: 0 local devices supported
$ sudo fwupd.fwupdmgr get-updates 
WARNING: This package has not been validated, it may not work properly.
Devices with no available firmware updates: 
 • VEN 04F3:00 04F3:311C
 • Fingerprint Sensor
 • Integrated Webcam HD
 • SSD 980 PRO 1TB
 • SSD 980 PRO 2TB
 • System Firmware
 • TPS DMC Family
 • TPS6598X PD#0
 • TPS6598X PD#1
 • TS4
 • UEFI Device Firmware
 • UEFI dbx
 • USB3.0 Hub
 • USB4 Retimer
 • USB4 Retimer
No updatable devices
```

**Expected behavior**
It should find some updates, since I know, my device isn't up2date.

**fwupd version information**
I'm using the latest stable snap version of fwupd

```
$ sudo fwupd.fwupdmgr --version
compile   org.freedesktop.fwupd         1.8.10
compile   com.hughsie.libjcat           0.1.9
runtime   org.freedesktop.fwupd-efi     1.0
compile   org.freedesktop.gusb          0.3.10
runtime   com.dell.libsmbios            2.4
runtime   org.freedesktop.gusb          0.3.10
runtime   org.freedesktop.fwupd         1.8.10
runtime   org.kernel                    5.19.0-46-generic
```

Please note how you installed it (`apt`, `dnf`, `pacman`, source, etc):

<details>

<summary>**fwupd device information**</summary>

Please provide the output of the fwupd devices recognized in your system.

```shell
Dell Inc. XPS 17 9720
│
├─VEN 04F3:00 04F3:311C:
│     Device ID:          cc9c1b47c3126a46379087db3bf18fa324f317ae
│     Summary:            Touchpad
│     Current version:    0x000c
│     Bootloader Version: 0x0001
│     Vendor:             ELAN Microelectronics (HIDRAW:0x04F3)
│     GUIDs:              46598f8d-5777-50c4-8da9-cda0cc8bbc26 ← HIDRAW\VEN_04F3&DEV_311C
│                         9b6fef1a-77aa-565c-9983-9da2dd2c932b ← HIDRAW\VEN_04F3&DEV_311C&REV_00
│                         d00322f0-ff1f-5458-9c64-8a771fb0b72e ← HIDRAW\VEN_04F3&DEV_311C&MOD_0000
│                         4f9fe0ae-f26e-5c1d-8a4e-a7506fb02f96 ← ELANTP\ICTYPE_13
[... 1105 lines cut ...]
├── initrd.img -> initrd.img-5.19.0-46-generic
├── initrd.img-5.19.0-45-generic
├── initrd.img-5.19.0-46-generic
├── initrd.img.old -> initrd.img-5.19.0-45-generic
├── memtest86+.bin
├── memtest86+.elf
├── memtest86+_multiboot.bin
├── System.map-5.19.0-45-generic
├── System.map-5.19.0-46-generic
├── vmlinuz -> vmlinuz-5.19.0-46-generic
├── vmlinuz-5.19.0-45-generic
├── vmlinuz-5.19.0-46-generic
└── vmlinuz.old -> vmlinuz-5.19.0-45-generic

60 directories, 743 files
```

**Additional questions**

- Operating system and version: Ubuntu 22.04.2
- Have you tried rebooting? yes
- Is this a regression?
- Are you using an NVMe disk? yes
- Is secure boot enabled? yes
- Is this a Lenovo system with 'Boot Order Lock' turned on in the BIOS? no

superm1, 2023-07-09:
I don't see any problems above. Are you sure your device has updates available at LVFS?

hughsie, 2023-07-09:
> XPS 17 9720

That has lots of updates -- something else is going on. @droidben can you get the output of `fwupd.fwupdtool get-devices --verbose --plugins uefi-capsule` please.

droidben, 2023-07-10:
> I don't see any problems above. Are you sure your device has updates available at LVFS?

Yes, I checked, there are updates. Including BIOS-Updates (system-firmware)

Is it possible that this has to do with the ZFS disk?

> output of `fwupd.fwupdtool get-devices --verbose --plugins uefi-capsule` please.

```
00:26:54.524 FuDebug              Verbose debugging enabled (on console 1)
00:26:54.525 FuMain               locked /run/lock/fwupdtool
Loading…                 [-                                      ]00:26:54.528 FuConfig             trying to load config values from /root/snap/fwupd/4600/etc/fwupd/daemon.conf
00:26:54.528 FuCommon             reading /root/snap/fwupd/4600/etc/fwupd/daemon.conf with 2494 bytes
00:26:54.528 FuConfig             trying to load config values from /root/snap/fwupd/4600/var/etc/fwupd/daemon.conf
00:26:54.528 FuConfig             using autodetected max archive size 4.3 GB
00:26:54.528 GLib-GIO             _g_io_module_get_default: Found default implementation local (GLocalVfs) for ‘gio-vfs’
Loading…                 [ -                                     ]00:26:54.528 FuRemoteList         loading /snap/fwupd/4600/share/fwupd/metainfo
00:26:54.529 XbSilo               attempting to load /root/snap/fwupd/4600/var/cache/fwupd/metainfo.xmlb
00:26:54.529 XbSilo               file: a455acda-24b5-a564-2588-bd2dbe3d99e6, current:a455acda-24b5-a564-2588-bd2dbe3d99e6, cached: (null)
00:26:54.529 XbSilo               loading silo with file contents
00:26:54.529 FuRemoteList         loading remote from /root/snap/fwupd/4600/etc/fwupd/remotes.d/vendor.conf
00:26:54.529 FuRemoteList         loading remote from /root/snap/fwupd/4600/etc/fwupd/remotes.d/dell-esrt.conf
00:26:54.529 FuRemoteList         loading remote from /root/snap/fwupd/4600/etc/fwupd/remotes.d/vendor-directory.conf
00:26:54.529 FuRemoteList         loading remote from /root/snap/fwupd/4600/etc/fwupd/remotes.d/lvfs-testing.conf
00:26:54.529 FuRemoteList         loading remote from /root/snap/fwupd/4600/etc/fwupd/remotes.d/lvfs.conf
00:26:54.529 FuRemoteList         ordering lvfs-testing=lvfs+1
00:26:54.529 FuRemoteList         ignoring unfound remote fwupd
Loading…                 [                                       ]00:26:54.529 Jcat                 ignoring GPG-KEY-Linux-Vendor-Firmware-Service as not PKCS-7 certificate
00:26:54.529 Jcat                 ignoring GPG-KEY-Linux-Foundation-Firmware as not PKCS-7 certificate
00:26:54.529 Jcat                 trying to load certificate from /root/snap/fwupd/4600/etc/pki/fwupd/LVFS-CA.pem
00:26:54.529 Jcat                 reading /root/snap/fwupd/4600/etc/pki/fwupd/LVFS-CA.pem with 1679 bytes
00:26:54.529 Jcat                 loaded 1 certificates
00:26:54.529 Jcat                 ignoring GPG-KEY-Linux-Vendor-Firmware-Service as not PKCS-7 certificate
00:26:54.529 Jcat                 trying to load certificate from /root/snap/fwupd/4600/etc/pki/fwupd-metadata/LVFS-CA.pem
00:26:54.529 Jcat                 reading /root/snap/fwupd/4600/etc/pki/fwupd-metadata/LVFS-CA.pem with 1679 bytes
00:26:54.529 Jcat                 loaded 1 certificates
00:26:54.529 Jcat                 ignoring GPG-KEY-Linux-Foundation-Metadata as not PKCS-7 certificate
00:26:54.529 Jcat                 reading /root/snap/fwupd/4600/var/lib/fwupd/pki/secret.key with 2455 bytes
00:26:54.529 Jcat                 reading /root/snap/fwupd/4600/var/lib/fwupd/pki/client.pem with 1383 bytes
00:26:54.529 FuEngine             client certificate now exists: nothing to do
Loading…                 [                                       ]00:26:54.529 FuHistory            trying to open database '/root/snap/fwupd/4600/var/lib/fwupd/pending.db'
00:26:54.529 FuHistory            got schema version of 8
Loading…                 [*                                      ]00:26:54.530 FuPlugin             load(/snap/fwupd/4600/lib/x86_64-linux-gnu/fwupd-1.8.10/libfu_plugin_modem_manager.so)
00:26:54.533 FuPluginUefiCapsule  failed to get fwupd-efi runtime version: searching /snap/fwupd/4600/libexec/fwupd/efi/fwupdx64.efi.signed: needle of 0x24 bytes was not found in haystack of 0xfd68 bytes
00:26:54.533 FuContext            added udev subsystem watch of platform
00:26:54.533 FuContext            added udev subsystem watch of block
00:26:54.533 FuContext            added udev subsystem watch of pci
00:26:54.533 GLib                 setenv()/putenv() are not thread-safe and should not be used after threads are created
00:26:54.533 FuPluginDell         Using libsmbios 2.4
00:26:54.533 FuContext            added udev subsystem watch of i2c
00:26:54.533 FuContext            added udev subsystem watch of i2c-dev
00:26:54.533 FuContext            added udev subsystem watch of hidraw
00:26:54.533 FuContext            added udev subsystem watch of gpio
00:26:54.533 FuContext            added udev subsystem watch of mei
00:26:54.533 FuContext            added udev subsystem watch of iommu
00:26:54.534 FuContext            added udev subsystem watch of video4linux
00:26:54.534 FuContext            added udev subsystem watch of msr
00:26:54.534 FuContext            added udev subsystem watch of mtd
00:26:54.534 FuContext            added udev subsystem watch of nvme
00:26:54.534 FuContext            added udev subsystem watch of drm
00:26:54.534 FuContext            added udev subsystem watch of drm_dp_aux_dev
00:26:54.534 FuContext            added udev subsystem watch of serio
00:26:54.534 FuPluginTest         init
[... cut ...]
  Version:              1065
  VersionLowest:        1065
  VersionFormat:        number
  VersionRaw:           0x00000429
  VersionLowestRaw:     0x00000429
  Created:              2023-07-10
  UpdateState:          success

00:26:55.536 FuPluginTest         destroy
00:26:55.537 FuProgress           raw timing data was { 0.078, 0.000, 0.021, 0.001, 0.011, 0.003 } -- steps were set as [ 63 1 8 1 26 1 ] but should have been [ 69 0 18 1 10 2 ] at ../src/plugins/uefi-capsule/fu-uefi-capsule-plugin.c:906
```

superm1, 2023-07-10:
```
00:26:55.290 FuEngine             failed to set release for component: no HWIDs matched 133b2bf2-76c7-5637-9deb-76c432e1324b
00:26:55.290 FuEngine             failed to set release for component: no HWIDs matched 133b2bf2-76c7-5637-9deb-76c432e1324b
00:26:55.290 FuEngine             failed to set release for component: no HWIDs matched 133b2bf2-76c7-5637-9deb-76c432e1324b
00:26:55.290 FuEngine             failed to set release for component: no HWIDs matched 133b2bf2-76c7-5637-9deb-76c432e1324b
00:26:55.290 FuEngine             failed to set release for component: no HWIDs matched 133b2bf2-76c7-5637-9deb-76c432e1324b
00:26:55.290 FuEngine             failed to set release for component: no HWIDs matched 133b2bf2-76c7-5637-9deb-76c432e1324b
00:26:55.291 FuEngine             failed to set release for component: no HWIDs matched 133b2bf2-76c7-5637-9deb-76c432e1324b
```
Is it possible that this system shares a firmware binary with the Precision variant?  If so; then the HWIDs at LVFS may only specify the Precision versoin.