fwupd/fwupd #5980: Suspend-to-idle Disabled in red color

64542426, 2023-07-09:
**Describe the bug**
 Suspend-to-idle is disabled, but appears in red color and ✘ sign.
`✘ Suspend-to-idle:               Disabled`

**Steps to Reproduce**
I run the command "fwupdmgr security" from Konsole as a user/root.

**Expected behavior**
It should have ✔sign with green color status as is it disabled.

**fwupd version information**
Please provide the version of the daemon and client.

```shell
fwupdmgr --version
compile   org.freedesktop.fwupd         1.8.16
compile   com.hughsie.libxmlb           0.3.11
compile   com.hughsie.libjcat           0.1.14
runtime   org.freedesktop.fwupd-efi     1.3
compile   org.freedesktop.gusb          0.3.10
runtime   com.hughsie.libjcat           0.1.14
runtime   com.dell.libsmbios            2.4
runtime   org.freedesktop.gusb          0.3.10
runtime   org.kernel                    6.3.9-1-default
runtime   org.freedesktop.fwupd         1.8.16
```

Please note how you installed it (`apt`, `dnf`, `pacman`, source, etc):
Installed from openSUSE Tumbleweed repository.
`sudo zypper in fwupd`

<summary>**fwupd device information**</summary>

Please provide the output of the fwupd devices recognized in your system.

```shell
Host Security ID: HSI:2! (v1.8.16)

HSI-1
✔ CSME manufacturing mode:       Locked: https://fwupd.github.io/libfwupdplugin/hsi.html#org.fwupd.hsi.Mei.ManufacturingMode
✔ CSME override:                 Locked: https://fwupd.github.io/libfwupdplugin/hsi.html#org.fwupd.hsi.Mei.OverrideStrap
✔ CSME v0:14.1.67.2046:          Valid: https://fwupd.github.io/libfwupdplugin/hsi.html#org.fwupd.hsi.Mei.Version
✔ MEI key manifest:              Valid: https://fwupd.github.io/libfwupdplugin/hsi.html#org.fwupd.hsi.Mei.KeyManifest
✔ Platform debugging:            Disabled: https://fwupd.github.io/libfwupdplugin/hsi.html#org.fwupd.hsi.PlatformDebugEnabled
✔ SPI BIOS region:               Locked: https://fwupd.github.io/libfwupdplugin/hsi.html#org.fwupd.hsi.Spi.SmmBwp
✔ SPI lock:                      Enabled: https://fwupd.github.io/libfwupdplugin/hsi.html#org.fwupd.hsi.Spi.Ble
✔ SPI write:                     Disabled: https://fwupd.github.io/libfwupdplugin/hsi.html#org.fwupd.hsi.Spi.Bioswe
✔ Supported CPU:                 Valid: https://fwupd.github.io/libfwupdplugin/hsi.html#org.fwupd.hsi.SupportedCpu
✔ TPM empty PCRs:                Valid: https://fwupd.github.io/libfwupdplugin/hsi.html#org.fwupd.hsi.Tpm.EmptyPcr
✔ TPM v2.0:                      Found: https://fwupd.github.io/libfwupdplugin/hsi.html#org.fwupd.hsi.Tpm.Version20
✔ UEFI platform key:             Valid: https://fwupd.github.io/libfwupdplugin/hsi.html#org.fwupd.hsi.Uefi.Pk
✔ UEFI secure boot:              Enabled: https://fwupd.github.io/libfwupdplugin/hsi.html#org.fwupd.hsi.Uefi.SecureBoot

HSI-2
✔ BIOS rollback protection:      Enabled: https://fwupd.github.io/libfwupdplugin/hsi.html#org.fwupd.hsi.Bios.RollbackProtection
✔ IOMMU:                         Enabled: https://fwupd.github.io/libfwupdplugin/hsi.html#org.fwupd.hsi.Iommu
✔ Intel BootGuard:               Enabled: https://fwupd.github.io/libfwupdplugin/hsi.html#org.fwupd.hsi.IntelBootguard.Enabled
✔ Intel BootGuard ACM protected: Valid: https://fwupd.github.io/libfwupdplugin/hsi.html#org.fwupd.hsi.IntelBootguard.Acm
✔ Intel BootGuard OTP fuse:      Valid: https://fwupd.github.io/libfwupdplugin/hsi.html#org.fwupd.hsi.IntelBootguard.Otp
✔ Intel BootGuard verified boot: Valid: https://fwupd.github.io/libfwupdplugin/hsi.html#org.fwupd.hsi.IntelBootguard.Verified
✔ Platform debugging:            Locked: https://fwupd.github.io/libfwupdplugin/hsi.html#org.fwupd.hsi.PlatformDebugLocked
✔ TPM PCR0 reconstruction:       Valid: https://fwupd.github.io/libfwupdplugin/hsi.html#org.fwupd.hsi.Tpm.ReconstructionPcr0

HSI-3
✔ Intel BootGuard error policy:  Valid: https://fwupd.github.io/libfwupdplugin/hsi.html#org.fwupd.hsi.IntelBootguard.Policy
✔ Pre-boot DMA protection:       Enabled: https://fwupd.github.io/libfwupdplugin/hsi.html#org.fwupd.hsi.PrebootDma
✔ Suspend-to-ram:                Disabled: https://fwupd.github.io/libfwupdplugin/hsi.html#org.fwupd.hsi.SuspendToRam
✘ Intel CET Enabled:             Not supported: https://fwupd.github.io/libfwupdplugin/hsi.html#org.fwupd.hsi.IntelCet.Enabled
✘ Suspend-to-idle:               Disabled: https://fwupd.github.io/libfwupdplugin/hsi.html#org.fwupd.hsi.SuspendToIdle

HSI-4
✔ Intel SMAP:                    Enabled: https://fwupd.github.io/libfwupdplugin/hsi.html#org.fwupd.hsi.IntelSmap
✘ Encrypted RAM:                 Not supported: https://fwupd.github.io/libfwupdplugin/hsi.html#org.fwupd.hsi.EncryptedRam

Runtime Suffix -!
✔ Linux kernel:                  Untainted: https://fwupd.github.io/libfwupdplugin/hsi.html#org.fwupd.hsi.Kernel.Tainted
✔ Linux swap:                    Encrypted: https://fwupd.github.io/libfwupdplugin/hsi.html#org.fwupd.hsi.Kernel.Swap
✔ fwupd plugins:                 Untainted: https://fwupd.github.io/libfwupdplugin/hsi.html#org.fwupd.hsi.Fwupd.Plugins
✘ Linux kernel lockdown:         Disabled: https://fwupd.github.io/libfwupdplugin/hsi.html#org.fwupd.hsi.Kernel.Lockdown

This system has HSI runtime issues.
 » https://fwupd.github.io/hsi.html#hsi-runtime-suffix

Host Security Events
  2023-06-10 11:40:40:  ✔ Suspend-to-ram changed: Enabled → Disabled
  2023-06-10 10:47:48:  ✔ TPM v2.0 changed: Not found → Found
  2023-06-10 10:47:48:  ✔ IOMMU device protection enabled
```

</details>

**Additional questions**

- Operating system and version:
```
--- Software ---  
OS: openSUSE Tumbleweed 
KDE Plasma: 5.27.6 
KDE Frameworks: 5.107.0 
Qt: 5.15.10 
Kernel: 6.3.9-1-default 
Compositor: wayland 
 
--- Hardware --- 
CPU: Intel(R) Core(TM) i7-10700 CPU @ 2.90GHz 
RAM: 30.9 GB 
GPU: Mesa Intel(R) UHD Graphics 630
Video memory: 31661MB 
```

- Have you tried rebooting?
Too many times, rebooting doesn't change anything...
- Is this a regression?
I think so, I'm not sure it's an application or distro package issue.
Another user [complain](https://gitlab.gnome.org/GNOME/gnome-control-center/-/issues/2149) about same matter on Gnome GitLab for OS Fedora.

superm1, 2023-07-09:
Can you please share your FADT/FACP?

64542426, 2023-07-09:
superm1, please let me know how to show them. 
I am running OpenSUSE TW on a DELL Desktop MT OptiPlex-5080 that haven't noted above.

superm1, 2023-07-09:
/sys/firmware/acpi/tables

64542426, 2023-07-09:
Is it right?

ls /sys/firmware/acpi/tables
`APIC  ASF!  BGRT  BOOT  DBG2  DBGP  DMAR  DSDT  FACP  FACS  FIDT  FPDT  HPET  LPIT  MCFG  NHLT  SSDT1  SSDT2  SSDT3  SSDT4  SSDT5  SSDT6  SSDT7  SSDT8  TPM2  data  dynamic`

I am not very familiar with this, I apologize.

superm1, 2023-07-09:
Copy the FACP one from that directory

64542426, 2023-07-09:
Copied, but i can't read within FACP with text editor Kate/nano. I changed ownership from root to my username.

hughsie, 2023-07-09:
can you copy the binary blob into this issue please.

superm1, 2023-07-09:
You can decode it using the tool iasl. Or you can attach it to this bug, but you'll probably need to compress it for GitHub to let you.

64542426, 2023-07-09:
I zipped FACP since it won't allow me to upload directly. 
[FACP.zip](https://github.com/fwupd/fwupd/files/11995802/FACP.zip)

hughsie, 2023-07-09:
@superm1 i make flags 0x3C6A5 i.e. bit 21 LOW_POWER_S0_IDLE_CAPABLE unset... that's somewhat unexpected, no?

hughsie, 2023-07-09:
@64542426 what hardware is this please? i.e. what laptop/desktop model.

superm1, 2023-07-09:
Well so I guess that means this machine ships with s3 mode and OEM doesn't offer a way to do hardware assisted s2idle mode.

The whole reason for this check in fwupd is specifically machines that can do both modes to make sure the kernel picked the correct (safer) default policy.

Maybe it doesn't make sense to include this check but rather just drop it?

hughsie, 2023-07-09:
Maybe it makes sense only to run if s3 is enabled?

superm1, 2023-07-09:
> Maybe it makes sense only to run if s3 is enabled?

Well if s3 is enabled you'll fail the s3 check...

64542426, 2023-07-10:
[OptiPlex-5080](https://dl.dell.com/topicspdf/optiplex-5080-desktop_owners-manual6_en-us.pdf) it is my desktop.

Hardware specs
CPU: Intel(R) Core(TM) i7-10700 CPU @ 2.90GHz
GPU: Mesa Intel(R) UHD Graphics 630

AlexLJordan, 2024-02-28:
I also have this output on my Lenovo Thinkpad L380.
In addition to that suspend to ram enabled is red too.
Is there something I can do about this?

hughsie, 2024-02-29:
See https://fwupd.github.io/libfwupdplugin/hsi.html#suspend-to-ram-disabled for more info.

genkieagle, 2024-07-31:
Hello, author of this ticket opened a year ago here.
I am coming back to this "Suspend-to-Idle" matter that confuses me pretty much.

Susdend-to-Idle must be Enabled or Disabled?

According to [hsi.html#org.fwupd.hsi.SuspendToIdle](https://fwupd.github.io/libfwupdplugin/hsi.html#org.fwupd.hsi.SuspendToIdle) says it must be not-enabled to success,
```
enabled: deep sleep enabled (failure)
not-valid: could not determine the default (failure)
not-enabled: suspend-to-idle being used (success)
```
While saw users that report enable as success
[image]

However, i have it Disabled as you can see and it reported as "✘".
```
$ fwupd
Host Security ID: HSI:2 (v1.9.22)

HSI-1
✔ BIOS firmware updates:         Enabled
✔ MEI key manifest:              Valid
✔ csme manufacturing mode:       Locked
✔ csme override:                 Locked
✔ csme v0:14.1.72.2287:          Valid
✔ Platform debugging:            Disabled
✔ SPI write:                     Disabled
✔ SPI lock:                      Enabled
✔ SPI BIOS region:               Locked
✔ Supported CPU:                 Valid
✔ TPM empty PCRs:                Valid
✔ TPM v2.0:                      Found
✔ UEFI bootservice variables:    Locked
✔ UEFI platform key:             Valid
✔ UEFI secure boot:              Enabled

HSI-2
✔ BIOS rollback protection:      Enabled
✔ Intel BootGuard ACM protected: Valid
✔ Intel BootGuard:               Enabled
✔ Intel BootGuard OTP fuse:      Valid
✔ Intel BootGuard verified boot: Valid
✔ Intel GDS mitigation:          Enabled
✔ IOMMU:                         Enabled
✔ Platform debugging:            Locked
✔ TPM PCR0 reconstruction:       Valid

HSI-3
✔ Intel BootGuard error policy:  Valid
✔ Pre-boot DMA protection:       Enabled
✔ Suspend-to-ram:                Disabled
✘ CET Platform:                  Not supported
✘ Suspend-to-idle:               Disabled

HSI-4
✔ SMAP:                          Enabled
✘ Encrypted RAM:                 Not supported

Runtime Suffix -!
✔ fwupd plugins:                 Untainted
✔ Linux kernel lockdown:         Enabled
✔ Linux swap:                    Encrypted
✔ Linux kernel:                  Untainted

Host Security Events
  2024-07-31 08:40:45:  ✔ Intel GDS mitigation changed: Invalid → Enabled
  2024-07-31 08:40:45:  ✔ Platform debugging changed: Unlocked → Locked
  2024-03-16 09:40:18:  ✘ Intel GDS mitigation changed: Enabled → Invalid
  2024-03-16 09:40:18:  ✘ Platform debugging changed: Locked → Unlocked
  2024-03-16 09:26:15:  ✔ TPM v2.0 changed: Not found → Found
  2024-03-16 08:58:20:  ✔ Platform debugging changed: Not supported → Disabled
  2024-03-16 08:58:20:  ✔ IOMMU device protection enabled
  2024-03-16 08:58:20:  ✔ Platform debugging changed: Not supported → Locked
```
Don't know with which conclusion you reached above, but a year later i am in the same situation.

I discussed with a user in r/[openSUSE](https://www.reddit.com/r/openSUSE/comments/1ec7i8p/enable_suspendtoidle/), we think that something is wrong with fwupd that it doesn't read it properly system firmware.

Please may we come back to this issue again? what to do, how it should be and so on?

My desktop is still [OptiPlex-5080 MT](https://dl.dell.com/topicspdf/optiplex-5080-desktop_owners-manual6_en-us.pdf)
```
$ fwupdmgr --version
compile   com.hughsie.libxmlb           0.3.18
compile   com.hughsie.libjcat           0.2.1
compile   org.freedesktop.fwupd         1.9.22
runtime   org.freedesktop.fwupd-efi     1.3
compile   org.freedesktop.gusb          0.4.9
runtime   com.hughsie.libxmlb           0.3.x
runtime   com.hughsie.libjcat           0.2.1
runtime   org.freedesktop.gusb          0.4.9
runtime   org.kernel                    6.10.2-1-default
runtime   org.freedesktop.fwupd         1.9.22
```
Distro Info
```
Operating System: openSUSE Tumbleweed 20240730
KDE Plasma Version: 6.1.3
KDE Frameworks Version: 6.4.0
Qt Version: 6.7.2
Kernel Version: 6.10.2-1-default (64-bit)
Graphics Platform: Wayland
Processors: 8 × Intel® Core™ i7-10700 CPU @ 2.90GHz
Memory: 30.9 GiB of RAM
Graphics Processor: Mesa Intel® UHD Graphics 630
```

I am open to other questions or requests if you have. Thank you.

superm1, 2024-08-02:
A passing system firmware configuration is like this:
```
✔ Suspend-to-idle:               Enabled
✔ Suspend-to-ram:                Disabled
```
I think we have a documentation problem here.

https://github.com/fwupd/fwupd/pull/7566

genkieagle, 2024-08-03:
I'm glad to read that it will be improved. 

What should I do with my Suspend-to-idle status that show disabled?

Is there any command or action that i can enable it?

superm1, 2024-08-03:
Some vendors offer BIOS settings that can improve it, but if yours doesn't there is nothing else you can do.

genkieagle, 2024-08-03:
I don't have an option in the BIOS, I have checked it very well. 

Thank you very much for your help and efforts.

Please close this ticket.

closed 2024-08-05 by hughsie

hfc2x, 2025-02-08:
I think this issue needs to be reopened, because I have exactly the same problem as OP, but on a desktop computer with the `s2idle` setting enabled in BIOS, yet `fwupdmgr` not being able to see it.

Motherboard is an MSI MAG Tomahawk B550, and BIOS (UEFI) version is 7C91vAH (latest stable). Attached is screenshot demonstrating that "[Modern Standby](https://learn.microsoft.com/en-us/windows-hardware/design/device-experiences/modern-standby)" is enabled:

[image]

Full text for both settings is as follows:
-  S3/Modern Standby Support: `Modern Standby Enable`
-  Modern Standby Type: `Modern Standby + S0i2 + S0i3`

Possible options for "S3/Modern Standby Support" in BIOS are `Disabled`, `Modern Standby Enable`, and `S3 Enable`.

Possible options for "Modern Standby Type" are `Modern Standby + S0i2`, `Modern Standby + S0i3`, and `Modern Standby + S0i2 + S0i3`. 

I'm running Arch Linux with the 6.13.1-arch2-1 kernel (most current at the time of writing). Set `mem_sleep_default=s2idle` kernel parameter, and the output of `cat /sys/power/mem_sleep` shows `[s2idle] deep`. However, `fwupdmgr` still reports the following:

```
HSI-3
✔ CET Platform:                  Supported
✔ Suspend-to-ram:                Disabled
✘ SPI replay protection:         Not supported
✘ Pre-boot DMA protection:       Disabled
✘ Suspend-to-idle:               Disabled
```

This is clearly untrue, as it is 100% enabled in BIOS settings, as seen in screenshot above. None of the "Modern Standby" options change what `fwupdmgr` reports.

Output of `fwupdmgr --version` is:
```
compile   info.libusb                   1.0.27
compile   org.freedesktop.fwupd         2.0.5
compile   com.hughsie.libxmlb           0.3.21
compile   org.freedesktop.Passim        0.1.8
compile   com.hughsie.libjcat           0.2.2
runtime   org.freedesktop.fwupd-efi     1.7
runtime   com.hughsie.libxmlb           0.3.21
runtime   com.hughsie.libjcat           0.2.2
runtime   org.kernel                    6.13.1-arch2-1
runtime   org.freedesktop.fwupd         2.0.5
```

superm1, 2025-02-08:
Please attach the report generated from a test attempt with this script.

https://gitlab.freedesktop.org/drm/amd/-/blob/master/scripts/amd_s2idle.py

hfc2x, 2025-02-09:
Nevermind, it seems like MSI screwed up, and my NVMe drives are not compatible with `s2idle`:

[image]