fwupd/fwupd #5985: Does not discover ESP at standard location

xeruf, 2023-07-11:
**Describe the bug**
ESP is mounted at `/boot/efi` and udisks2 running, but ESP is not found.

**Steps to Reproduce**
```sh
❯ sudo /usr/bin/fwupdtool esp-list --verbose
06:55:18.695 FuDebug              verbose to info (on console 1)
06:55:18.695 FuEngine             starting fwupd 1.9.2…
No ESP or BDP found
❯ fwupdmgr refresh
WARNING: UEFI ESP partition not detected or configured
See https://github.com/fwupd/fwupd/wiki/PluginFlag:esp-not-found for more information.
WARNING: Will measure elements of system integrity around an update
See https://github.com/fwupd/fwupd/wiki/PluginFlag:measure-system-integrity for more information.
Firmware metadata last refresh: 25 minutes ago. Use --force to refresh again.
❯ sudo systemctl status udisks2
● udisks2.service - Disk Manager
     Loaded: loaded (/usr/lib/systemd/system/udisks2.service; disabled; preset: disabled)
     Active: active (running) since Tue 2023-07-11 07:25:57 CEST; 1h 29min ago
       Docs: man:udisks(8)
   Main PID: 1016 (udisksd)
      Tasks: 6 (limit: 38412)
     Memory: 8.5M
        CPU: 546ms
     CGroup: /system.slice/udisks2.service
             └─1016 /usr/lib/udisks2/udisksd

Jul 11 07:25:57 elephant.arch systemd[1]: Starting Disk Manager...
Jul 11 07:25:57 elephant.arch udisksd[1016]: udisks daemon version 2.9.4 starting
Jul 11 07:25:57 elephant.arch udisksd[1016]: Acquired the name org.freedesktop.UDisks2 on the system message bus
Jul 11 07:25:57 elephant.arch systemd[1]: Started Disk Manager.
❯ lsblk --fs | grep boot
└─nvme1n1p4 vfat   FAT16 efi               44BC-9FF8                             209.4M     9% /boot/efi
❯ tree -L 2 /boot/efi
/boot/efi
├── EFI
│   ├── BOOT
│   └── fedora
├── System
│   └── Library
└── mach_kernel

6 directories, 1 file
```

**Expected behavior**
ESP is detected

**fwupd version information**
Please provide the version of the daemon and client.

```shell
❯ fwupdmgr --version
compile   org.freedesktop.fwupd         1.9.2
compile   com.hughsie.libxmlb           0.3.11
compile   com.hughsie.libjcat           0.1.13
runtime   org.freedesktop.fwupd-efi     1.4
compile   org.freedesktop.gusb          0.4.6
runtime   com.hughsie.libjcat           0.1.14
runtime   com.dell.libsmbios            2.4
runtime   org.freedesktop.gusb          0.4.6
runtime   org.kernel                    6.4.1-arch2-1
runtime   org.freedesktop.fwupd         1.9.2

```

Please note how you installed it (`apt`, `dnf`, `pacman`, source, etc): pacman

**Additional questions**

- Operating system and version: Arch
- Have you tried rebooting? yes
- Is this a regression? i don't think so

hughsie, 2023-07-11:
What does `sudo fwupdtool esp-list -v -v` say please?

tigeritest, 2023-07-22:
Hello, I've been struggling with a similar problem - ESP is no longer detected by fwupd. I haven't noticed this problem until I reinstalled Arch on a new drive in the same configuration as before (GRUB2, UEFI boot, no secure boot, installed udisks2) on a new SSD about a month ago.
Distro: Arch Linux, installed fwupd via pacman, tried rebooting
```bash
[user@device ~]$ ls /boot
EFI            grub                          initramfs-linux.img
amd-ucode.img  initramfs-linux-fallback.img  vmlinuz-linux
[user@device ~]$ ls /boot/efi
GRUB
[user@device ~]$ fwupdmgr refresh --force
WARNING: UEFI ESP partition not detected or configured
See https://github.com/fwupd/fwupd/wiki/PluginFlag:esp-not-found for more information.
Updating lvfs
Downloading…             [        |                              ]
Successfully downloaded new metadata: 2 local devices supported
[user@device ~]$ sudo systemctl status udisks2
● udisks2.service - Disk Manager
     Loaded: loaded (/usr/lib/systemd/system/udisks2.service; disabled; preset: disable>
     Active: active (running) since Sat 2023-07-22 13:47:49 MSK; 2min 24s ago
       Docs: man:udisks(8)
   Main PID: 976 (udisksd)
      Tasks: 7 (limit: 28580)
     Memory: 8.4M
        CPU: 49ms
     CGroup: /system.slice/udisks2.service
             └─976 /usr/lib/udisks2/udisksd

Jul 22 13:47:49 device systemd[1]: Starting Disk Manager...
Jul 22 13:47:49 device udisksd[976]: udisks daemon version 2.9.4 starting
Jul 22 13:47:49 device systemd[1]: Started Disk Manager.
Jul 22 13:47:49 device udisksd[976]: Acquired the name org.freedesktop.UDisks2 on the>
[user@device ~]$ lsblk --fs | grep boot
├─nvme0n1p1 vfat   FAT32           1BBB-818E                             382.7M    25% /mnt/boot
│                                                                                      /boot
[user@device ~]$ fwupdmgr --version
compile   org.freedesktop.fwupd         1.9.3
compile   com.hughsie.libxmlb           0.3.11
compile   com.hughsie.libjcat           0.1.14
runtime   org.freedesktop.fwupd-efi     1.4
compile   org.freedesktop.gusb          0.4.6
runtime   com.hughsie.libjcat           0.1.14
runtime   com.dell.libsmbios            2.4
runtime   org.freedesktop.gusb          0.4.6
runtime   org.kernel                    6.4.3-arch1-2
runtime   org.freedesktop.fwupd         1.9.3

[user@device ~]$ sudo fwupdtool esp-list -v -v
10:52:27.409 FuDebug              verbose to debug (on console 1)
10:52:27.409 FuEngine             starting fwupd 1.9.3…
10:52:27.414 FuVolume             device /org/freedesktop/UDisks2/block_devices/nvme0n1p2, type: 0fc63daf-8483-4772-8e79-3d69d8477de4, internal: 1, fs: ext4
10:52:27.415 FuVolume             device /org/freedesktop/UDisks2/block_devices/nvme0n1, type: (null), internal: 1, fs:
10:52:27.415 FuVolume             device /org/freedesktop/UDisks2/block_devices/mmcblk0p1, type: 0x0c, internal: 0, fs: vfat
10:52:27.416 FuVolume             device /org/freedesktop/UDisks2/block_devices/nvme0n1p1, type: 0fc63daf-8483-4772-8e79-3d69d8477de4, internal: 1, fs: vfat
10:52:27.417 FuVolume             device /org/freedesktop/UDisks2/block_devices/mmcblk0p2, type: 0xe0, internal: 0, fs:
10:52:27.418 FuVolume             device /org/freedesktop/UDisks2/block_devices/mmcblk0, type: (null), internal: 0, fs:
10:52:27.418 FuContext            no volumes of type c12a7328-f81f-11d2-ba4b-00a0c93ec93b
10:52:27.422 FuVolume             device /org/freedesktop/UDisks2/block_devices/nvme0n1p2, type: 0fc63daf-8483-4772-8e79-3d69d8477de4, internal: 1, fs: ext4
10:52:27.422 FuVolume             device /org/freedesktop/UDisks2/block_devices/nvme0n1, type: (null), internal: 1, fs:
10:52:27.423 FuVolume             device /org/freedesktop/UDisks2/block_devices/mmcblk0p1, type: 0x0c, internal: 0, fs: vfat
10:52:27.424 FuVolume             device /org/freedesktop/UDisks2/block_devices/nvme0n1p1, type: 0fc63daf-8483-4772-8e79-3d69d8477de4, internal: 1, fs: vfat
10:52:27.425 FuVolume             device /org/freedesktop/UDisks2/block_devices/mmcblk0p2, type: 0xe0, internal: 0, fs:
10:52:27.426 FuVolume             device /org/freedesktop/UDisks2/block_devices/mmcblk0, type: (null), internal: 0, fs:
10:52:27.426 FuContext            no volumes of type ebd0a0a2-b9e5-4433-87c0-68b6b72699c7
No ESP or BDP found
```

hughsie, 2023-07-22:
> No ESP or BDP found

/mnt/boot isn't an ESP or BDP -- fwupd is doing exactly the correct thing. EVen if you got fwupd to recognise that it's an ESP the firmware wouldn't and the update would probably fail.

tigeritest, 2023-07-22:
Wow, that's a fast reply. Thank you :)

> /mnt/boot isn't an ESP or BDP

I thought the /boot entry under lsblk should also mean something, but either way, this is the output when EFI partition is not mounted in /mnt:
```bash
[user@device ~]$ lsblk --fs | grep boot
├─nvme0n1p1 vfat   FAT32           1BBB-818E                             382.7M    25% /boot
[user@device ~]$ fwupdmgr refresh --force
WARNING: UEFI ESP partition may not be set up correctly
See https://github.com/fwupd/fwupd/wiki/PluginFlag:esp-not-valid for more information.
Updating lvfs
Downloading…             [       \                               ]
Successfully downloaded new metadata: 2 local devices supported
```

I stumbled into a solution for my device in one of the closed issues: https://github.com/fwupd/fwupd/issues/5612#issuecomment-1462222169
Running `parted /dev/nvme0n1 set 1 esp on` then rebooting made fwupdmgr stop complaining about the EFI partition. To verify that it was what helped, running `parted /dev/nvme0n1 set 1 esp off` and rebooting breaks everything again. So I guess that was my solution, hope I didn't derail this issue too much.

12c76dc, 2023-10-19:
Hello,
I'm having the same issue. My ESP is mounted at `/boot`, but it is not recognized as ESP:
```[adomas@nebula Downloads]$ sudo fwupdtool esp-list -v -v
11:16:25.697 FuDebug              verbose to debug (on console 1)
11:16:25.697 FuEngine             starting fwupd 1.9.6…
11:16:25.711 FuVolume             device /org/freedesktop/UDisks2/block_devices/sdb1, type: 0fc63daf-8483-4772-8e79-3d69d8477de4, internal: 0, fs: ext4
11:16:25.713 FuVolume             device /org/freedesktop/UDisks2/block_devices/sda, type: (null), internal: 0, fs: 
11:16:25.715 FuVolume             device /org/freedesktop/UDisks2/block_devices/nvme2n1p1, type: 0fc63daf-8483-4772-8e79-3d69d8477de4, internal: 1, fs: ext4
11:16:25.716 FuVolume             device /org/freedesktop/UDisks2/block_devices/nvme0n1, type: (null), internal: 1, fs: 
11:16:25.718 FuVolume             device /org/freedesktop/UDisks2/block_devices/nvme2n1, type: (null), internal: 1, fs: 
11:16:25.719 FuVolume             device /org/freedesktop/UDisks2/block_devices/nvme1n1p2, type: 0fc63daf-8483-4772-8e79-3d69d8477de4, internal: 1, fs: ext4
11:16:25.721 FuVolume             device /org/freedesktop/UDisks2/block_devices/nvme0n1p1, type: 0fc63daf-8483-4772-8e79-3d69d8477de4, internal: 1, fs: ext4
11:16:25.723 FuVolume             device /org/freedesktop/UDisks2/block_devices/sdb, type: (null), internal: 0, fs: 
11:16:25.725 FuVolume             device /org/freedesktop/UDisks2/block_devices/sda1, type: 0fc63daf-8483-4772-8e79-3d69d8477de4, internal: 0, fs: ext4
11:16:25.726 FuVolume             device /org/freedesktop/UDisks2/block_devices/nvme1n1p1, type: 0fc63daf-8483-4772-8e79-3d69d8477de4, internal: 1, fs: vfat
11:16:25.728 FuVolume             device /org/freedesktop/UDisks2/block_devices/nvme1n1, type: (null), internal: 1, fs: 
11:16:25.728 FuContext            no volumes of type c12a7328-f81f-11d2-ba4b-00a0c93ec93b
11:16:25.742 FuVolume             device /org/freedesktop/UDisks2/block_devices/sdb1, type: 0fc63daf-8483-4772-8e79-3d69d8477de4, internal: 0, fs: ext4
11:16:25.744 FuVolume             device /org/freedesktop/UDisks2/block_devices/sda, type: (null), internal: 0, fs: 
11:16:25.746 FuVolume             device /org/freedesktop/UDisks2/block_devices/nvme2n1p1, type: 0fc63daf-8483-4772-8e79-3d69d8477de4, internal: 1, fs: ext4
11:16:25.747 FuVolume             device /org/freedesktop/UDisks2/block_devices/nvme0n1, type: (null), internal: 1, fs: 
11:16:25.749 FuVolume             device /org/freedesktop/UDisks2/block_devices/nvme2n1, type: (null), internal: 1, fs: 
11:16:25.751 FuVolume             device /org/freedesktop/UDisks2/block_devices/nvme1n1p2, type: 0fc63daf-8483-4772-8e79-3d69d8477de4, internal: 1, fs: ext4
11:16:25.752 FuVolume             device /org/freedesktop/UDisks2/block_devices/nvme0n1p1, type: 0fc63daf-8483-4772-8e79-3d69d8477de4, internal: 1, fs: ext4
11:16:25.754 FuVolume             device /org/freedesktop/UDisks2/block_devices/sdb, type: (null), internal: 0, fs: 
11:16:25.756 FuVolume             device /org/freedesktop/UDisks2/block_devices/sda1, type: 0fc63daf-8483-4772-8e79-3d69d8477de4, internal: 0, fs: ext4
11:16:25.757 FuVolume             device /org/freedesktop/UDisks2/block_devices/nvme1n1p1, type: 0fc63daf-8483-4772-8e79-3d69d8477de4, internal: 1, fs: vfat
11:16:25.759 FuVolume             device /org/freedesktop/UDisks2/block_devices/nvme1n1, type: (null), internal: 1, fs: 
11:16:25.759 FuContext            no volumes of type ebd0a0a2-b9e5-4433-87c0-68b6b72699c7
No ESP or BDP found
```
Any idea how to fix this?

closed 2024-11-11 by hughsie

TobiKuik, 2025-04-26:
Setting the 'esp' flag in gparted on my 1 GB /boot partition solved my issue.

superm1, 2025-04-26:
> Setting the 'esp' flag in gparted on my 1 GB /boot partition solved my issue.

Thanks for confirming. Please file a bug with the installer that got you a broken setup in the first place.