fwupd/fwupd #5994: 1.9.3 snap shows error for fwupd.fwupdtool

superm1, 2023-07-12:
**Describe the bug**
I installed the fwupd 1.9.3 confined snap and ran `sudo fwupd.fwupdtool install-blob foo.bin`
I see the following error:

```
14:55:29.364 FuEngine             failed to add device /sys/devices/virtual/msr/msr0: failed to add device using on msr: failed to open /dev/cpu/0/msr: Operation not permitted
```

I believe that this may be caused by confinement not able to access an MSR normally used to enumerate security things on some systems.

**Steps to Reproduce**
```
sudo apt remove fwupd
sudo snap install fwupd
sudo fwupd.fwupdtool install-blob foo.bin
```

**Expected behavior**
No errors

**fwupd version information**
```
compile   org.freedesktop.fwupd         1.9.3
compile   com.hughsie.libxmlb           0.3.6
compile   com.hughsie.libjcat           0.1.9
runtime   org.freedesktop.fwupd-efi     1.4
compile   org.freedesktop.gusb          0.3.10
runtime   com.dell.libsmbios            2.4
runtime   org.freedesktop.gusb          0.3.10
runtime   org.kernel                    6.5.0-rc1-00006-g6bd1e1abc9af
runtime   org.freedesktop.fwupd         1.9.3
```

superm1, 2023-07-12:
@valentindavid can you check if this is missing on `snapd` side perhaps?

valentindavid, 2023-07-12:
It should be there. Do you get any apparmor error in the journal? I thought apparmor profiles gave EACCES rather than EPERM. But maybe I am wrong.

superm1, 2023-07-12:
```
[ 1460.037326] audit: type=1326 audit(1689174562.066:13): auid=1000 uid=0 gid=0 ses=4 subj=kernel pid=5648 comm="fwupdtool" exe="/snap/fwupd/5039/bin/fwupdtool" sig=0 arch=c000003e syscall=314 compat=0 ip=0x7f660c09ba3d code=0x50000
```

valentindavid, 2023-07-12:
Error with syscall 314 can be ignored. That has probably no relation to the EPERM error opening `/dev/cpu/0/msr` you are getting.

superm1, 2023-07-12:
I did double check; this issue doesn't happen outside of snap on the same system.

superm1, 2023-07-12:
[strace_fwupdtool.txt](https://github.com/fwupd/fwupd/files/12033948/strace_fwupdtool.txt)

superm1, 2023-07-12:
```
[pid 2816566] openat(AT_FDCWD, "/dev/cpu/0/msr", O_RDWR) = -1 EPERM (Operation not permitted)
[pid 2816566] write(2, "22:50:14.472 ", 1322:50:14.472 ) = 13
[pid 2816566] write(2, "FuEngine             ", 21FuEngine             ) = 21
[pid 2816566] write(2, "failed to add device /sys/device"..., 142failed to add device /sys/devices/virtual/msr/msr0: failed to add device using on msr: failed to open /dev/cpu/0/msr: Operation not permitted
) = 142
```

valentindavid, 2023-07-13:
Looking at the kernel, we must be missing `CAP_SYS_RAWIO`.

superm1, 2023-07-13:
But isn't that already in the fwupd interface?

https://github.com/snapcore/snapd/blob/392589b1f7af4cf6f1be8ca214e8a64e32d46929/interfaces/builtin/fwupd.go#L70

superm1, 2024-01-16:
I can't reproduce this anymore with snapd 2.61.1.

closed 2024-01-16 by superm1