fwupd/fwupd #6006: Fwupdmgr stuck on adding `(null)` device after an update

Gusted, 2023-07-15:
**Describe the bug**
After I updated the UEFI dbx database, every time `fwupdmgr` or `fwupdtool` is executed and initializes everything, it takes considerable time to start responding. Running `fwupdtool -v get-upgrades` (same when `fwupdmgr -v` is runned) yields that it stops at:

```
18:37:24.721 FuDeviceList         ::added 6b5a06645e27c4e94b6e67d5185c46d11ec9ad67 [(null)]
```

Then after 2.5 minutes:

```
18:40:02.504 FuDeviceList         ::added c6a80ac3a22083423992a3cb15018989f37834d6 [TPM]
```

I did updates before on this laptop with fwupd, and this hasn't happened before.

**Steps to Reproduce**
1. Update UEFI dbx database (This might just be unrelated).
2. Any command that "loads" devices? Is taking significant time.

**Expected behavior**
That either there's no `null` device or that it doesn't take considerable time to "load".

**fwupd version information**

```shell
-> % sudo fwupdtool --version
compile   org.freedesktop.fwupd         1.9.3
compile   com.hughsie.libxmlb           0.3.11
compile   com.hughsie.libjcat           0.1.14
runtime   org.freedesktop.fwupd-efi     1.4
compile   org.freedesktop.gusb          0.4.6
runtime   com.hughsie.libjcat           0.1.14
runtime   com.dell.libsmbios            2.4
runtime   org.freedesktop.gusb          0.4.6
runtime   org.kernel                    6.4.3-arch1-1
runtime   org.freedesktop.fwupd         1.9.3
```

Please note how you installed it (`apt`, `dnf`, `pacman`, source, etc): Installed trough pacman.

<details>

<summary><b>fwupd device information</b></summary>

Please provide the output of the fwupd devices recognized in your system.

```shell
-> % sudo fwupdtool get-devices --show-all-devices 
Loading?                 [************************************** ]
Framework Laptop (12th Gen Intel Core)
?
??Unknown Device:
?     Device ID:          6b5a06645e27c4e94b6e67d5185c46d11ec9ad67
?     GUID:               c96751f4-16e8-59ea-975d-69d3dac88d7d ? GPIO\ID_INTC1055:00
?   
??12th Gen Intel Core? i5-1240P:
? ?   Device ID:          4bde70ba4e39b28f9eab1628f9dd6e6244c03027
? ?   Current version:    0x0000042a
? ?   Vendor:             Intel
? ?   GUIDs:              b9a2dd81-159e-5537-a7db-e7101d164d3f ? cpu
? ?                       30249f37-d140-5d3e-9319-186b1bd5cac3 ? CPUID\PRO_0&FAM_06
? ?                       ab855c04-4ff6-54af-8a8a-d8193daa0cd8 ? CPUID\PRO_0&FAM_06&MOD_9A
? ?                       3ebbde86-d03e-549a-a8fd-02ebf9aa537a ? CPUID\PRO_0&FAM_06&MOD_9A&STP_3
? ?   Device Flags:       ? Internal device
? ? 
? ??USB4 host controller:
? ?     Device ID:        4baaf6f99af729b7d6a370fc98b8ec9a56d5452e
? ?     Summary:          Unmatched performance for high-speed I/O
? ?     Vendor:           INTEL (THUNDERBOLT:0x8087)
? ?     Problems:         ? Device requires AC power to be connected
[... 159 lines cut ...]
?   
??WD BLACK SN770 500GB:
      Device ID:          71b677ca0f1bc2c5b804fa1d59e52064ce589293
      Summary:            NVM Express solid state drive
      Current version:    731030WD
      Vendor:             Sandisk Corp (NVME:0x15B7)
      Serial Number:      2205GC444008
      Problems:           ? Device requires AC power to be connected
      GUIDs:              1524d43d-ed91-5130-8cb6-8b8478508bae ? NVME\VEN_15B7&DEV_5017
                          87cfda90-ce08-52c3-9bb5-0e0718b7e57e ? NVME\VEN_15B7&DEV_5017&SUBSYS_15B75017
                          008bacd9-329d-58e9-a092-c45170fa96b6 ? WD_BLACK SN770 500GB
      Device Flags:       ? Internal device
                          ? System requires external power source
                          ? Needs a reboot after installation
                          ? Device is usable for the duration of the update
                          ? Updatable
```

</details>

**Additional questions**

- Operating system and version: Linux 6.4.3-arch1-1
- Have you tried rebooting? Yes
- Is this a regression? Don't think so, downgraded to fwupd 1.8.15 then tried again and it once again stalls on the `null` device.

hughsie, 2023-07-15:
Can you attach the output of `sudo fwupdtool get-devices -vv` please.

Gusted, 2023-07-15:
> Can you attach the output of `sudo fwupdtool get-devices -vv` please.

Yes, of course! It got stuck at `20:01:13.504`.

[output.txt](https://github.com/fwupd/fwupd/files/12062702/output.txt)

hughsie, 2023-07-15:
The engine even detects this is weird itself:

    ../fwupd-1.9.3/src/fu-engine.c:8214:backend-coldplug [167956.99ms]:
        ../fwupd-1.9.3/src/fu-engine.c:8177:udev [167653.92ms]:
            ../fwupd-1.9.3/src/fu-engine.c:8158:add-devices [167597.22ms]:
                ../fwupd-1.9.3/src/fu-engine.c:7746:/sys/devices/platform/NTC0702:00/tpm/tpm0 [166907.06ms]:
                    ../fwupd-1.9.3/src/fu-engine.c:7713:query-possible-plugins [166906.97ms]:
                        ../fwupd-1.9.3/libfwupdplugin/fu-plugin.c:1536:tpm [166906.94ms]:
                            :open [162640.12ms]

so opening the TPM device took a long time. Is tpm2-tss installed and working correctly working too? Does `sudo tpm2_pcrread` also take a long time to read?

Gusted, 2023-07-15:
> Is tpm2-tss installed

Yes it is.

>  and working correctly working too?

I'm not entirely sure how to check for that, the userspace tools seems to be working (even though also significantly slow). So just some general information: I've had seen for a while a message about: `Firmware Bug: TPM interrupt not working, polling instead` from the kernel, but it seems like that hasn't been popping up anymore from the kernel, so that might be related to TPM being slow?

<details>

```
-> % sudo dmesg | grep -i "tpm"
doas (gusted@gusted-laptop) password: 
[    0.000000] efi: ACPI=0x43afe000 ACPI 2.0=0x43afe014 TPMFinalLog=0x43986000 SMBIOS=0x3f086000 MEMATTR=0x38b99018 ESRT=0x38dd8218 RNG=0x43a3df18 INITRD=0x38c83c18 TPMEventLog=0x38b82018 
[    0.004775] ACPI: SSDT 0x0000000043ADC000 00077B (v02 INSYDE Tpm2Tabl 00001000 INTL 20200717)
[    0.004777] ACPI: TPM2 0x0000000043ADB000 00004C (v04 INSYDE ADL-P-M  00000002 ACPI 00040000)
[    0.004826] ACPI: Reserving TPM2 table memory at [mem 0x43adb000-0x43adb04b]
[    1.292431] tpm_tis NTC0702:00: 2.0 TPM (device-id 0xFC, rev-id 1)
[    7.352285] systemd[1]: systemd 253.6-2-arch running in system mode (+PAM +AUDIT -SELINUX -APPARMOR -IMA +SMACK +SECCOMP +GCRYPT +GNUTLS +OPENSSL +ACL +BLKID +CURL +ELFUTILS +FIDO2 +IDN2 -IDN +IPTC +KMOD +LIBCRYPTSETUP +LIBFDISK +PCRE2 -PWQUALITY +P11KIT -QRENCODE +TPM2 +BZIP2 +LZ4 +XZ +ZLIB +ZSTD +BPF_FRAMEWORK +XKBCOMMON +UTMP -SYSVINIT default-hierarchy=unified)
[    7.383157] systemd[1]: TPM2 PCR Barrier (initrd) was skipped because of an unmet condition check (ConditionPathExists=/sys/firmware/efi/efivars/StubPcrKernelImage-4a67b082-0a4c-41cf-b6c7-440b29bb8c4f).
[   31.649552] systemd[1]: systemd 253.6-2-arch running in system mode (+PAM +AUDIT -SELINUX -APPARMOR -IMA +SMACK +SECCOMP +GCRYPT +GNUTLS +OPENSSL +ACL +BLKID +CURL +ELFUTILS +FIDO2 +IDN2 -IDN +IPTC +KMOD +LIBCRYPTSETUP +LIBFDISK +PCRE2 -PWQUALITY +P11KIT -QRENCODE +TPM2 +BZIP2 +LZ4 +XZ +ZLIB +ZSTD +BPF_FRAMEWORK +XKBCOMMON +UTMP -SYSVINIT default-hierarchy=unified)
[   32.182894] systemd[1]: TPM2 PCR Machine ID Measurement was skipped because of an unmet condition check (ConditionPathExists=/sys/firmware/efi/efivars/StubPcrKernelImage-4a67b082-0a4c-41cf-b6c7-440b29bb8c4f).
[  805.492902] tpm tpm0: tpm_try_transmit: send(): error -62
```

</details>

> Does sudo tpm2_pcrread also take a long time to read?

Yup, took about 8 minutes:
`sudo tpm2_pcrread  0.01s user 0.01s system 0% cpu 8:09.46 total`

Gusted, 2023-07-15:
Alright, I booted `linux-lts` to see if that kernel mentioned have the mentioned TPM polling workaround, and it did. And surprise TPM just worked flawlessly:
```
sudo tpm2_pcrread  0.01s user 0.01s system 41% cpu 0.042 total
```

I assume this means it's rather a kernel bug and I would have to report it to the Linux kernel?

Gusted, 2023-07-15:
It seems to be that it was [already reported](https://bugzilla.kernel.org/show_bug.cgi?id=217631) about the TPM issues and matches what I'm seeing. I guess that really concludes this bug then, thank you for pointing me into the right direction! Weird that fwupd was the one noticing this 😅. Feel free to close the issue, I don't think there's anything actionable for fwupd about this issue.

closed 2023-07-15 by superm1