fwupd/fwupd #6012: UEFI update just reboots after printing "fwupd-efi version 1.4"

yejingchen, 2023-07-18:
**Describe the bug**
My Laptop is [HP EliteBook 845 14 inch G9 Notebook PC](https://fwupd.org/lvfs/devices/com.hp.laptop.system.U82.firmware). After using `fwupdmgr update` to update system firmware, and accept the reboot prompt, the system reboots, prints `fwupd-efi version 1.4`, but after <5 seconds the system reboots again into Arch Linux, without actually updating the firmware.

After booting into Arch, `fwupdmgr get-history` says `failed to run update on reboot`.

**Steps to Reproduce**
- Run `fwupdmgr update`, accept the System Firmware update from 0x01010c00 to 0x01060000
- Wait for it to download and setup, accept the reboot prompt
- Laptop reboots, displays HP logo
- Logo disappears, prints a line: `fwupd-efi version 1.4`, then hangs for a few seconds
- Screen goes black, reboots into Linux

**Expected behavior**
After printin `fwupd-efi version 1.4`, the update continues, then the HP firmware update screen shows up and handles the rest.

**fwupd version information**
Please provide the version of the daemon and client.

```shell
% fwupdmgr --version
compile   org.freedesktop.fwupd         1.9.3
compile   com.hughsie.libxmlb           0.3.11
compile   com.hughsie.libjcat           0.1.14
runtime   org.freedesktop.fwupd-efi     1.4
compile   org.freedesktop.gusb          0.4.6
runtime   com.hughsie.libjcat           0.1.14
runtime   com.dell.libsmbios            2.4
runtime   org.freedesktop.gusb          0.4.6
runtime   org.kernel                    6.4.3-arch1-2
runtime   org.freedesktop.fwupd         1.9.3
```

Please note how you installed it (`apt`, `dnf`, `pacman`, source, etc):
Arch Linux packages using pacman:

```
fwupd 1.9.3-1
fwupd-efi 1.4-1
```

<details>

<summary>**fwupd device information**</summary>

Please provide the output of the fwupd devices recognized in your system.

```shell
% LANGUAGE= LANG=C.UTF-8 fwupdmgr get-devices --show-all-devices
HP HP EliteBook 845 14 inch G9 Notebook PC
│
├─ELAN0767:00 04F3:31EC:
│     Device ID:          903986b371e59b3d3dcf12a32c7f339fbb56a29f
│     Summary:            Touchpad
│     Current version:    0x0007
│     Bootloader Version: 0x0001
│     Vendor:             ELAN Microelectronics (HIDRAW:0x04F3)
│     GUIDs:              51b98b32-da4e-5171-9310-3c2859f6a6fb ← HIDRAW\VEN_04F3&DEV_31EC
│                         d421b6c5-038e-517a-a525-ce89bcc2032d ← HIDRAW\VEN_04F3&DEV_31EC&MOD_001D
│                         15e96dd6-4f5e-5d56-a991-2e63e82d1bd2 ← ELANTP\ICTYPE_11
│                         fc637c95-0376-54a4-9c37-c5b077e71327 ← ELANTP\ICTYPE_11&MOD_001D
│                         5830ebc2-7952-5ca5-a0c1-569243b6f7ea ← ELANTP\ICTYPE_11&MOD_001D&DRIVER_HID
│     Device Flags:       • Internal device
│                         • Updatable
│   
├─AMD Ryzen 7 PRO 6850HS with Radeon Graphics:
│ │   Device ID:          4bde70ba4e39b28f9eab1628f9dd6e6244c03027
│ │   Current version:    0x0a404102
│ │   Vendor:             Advanced Micro Devices, Inc.
│ │   GUIDs:              b9a2dd81-159e-5537-a7db-e7101d164d3f ← cpu
│ │                       22f9ecf4-588d-5c0a-8326-6ebff3655c6d ← CPUID\PRO_0&FAM_19
[... 247 lines cut ...]
└── fwupdx64.efi

2 directories, 2 files
```

**Additional questions**

- Operating system and version: 
Arch Linux

- Have you tried rebooting?
Yes

- Is this a regression?
Maybe, worked a year ago using fwupd 1.8.1 or 1.8.2 and fwupd-efi 1.3

- Are you using an NVMe disk?
Yes

- Is secure boot enabled?
Yes, self signed PK, etc and fwupdx64.efi.
I tried `fwupdmgr update` again after disabling Secure Boot, but result is the same `fwupd-efi version 1.4` line and rebooting back into Linux.

- Is this a Lenovo system with 'Boot Order Lock' turned on in the BIOS?
No, it is an HP laptop, and I couldn't find similar settings.

superm1, 2023-07-18:
To me this looks like a BIOS bug then.  @hughsie agree?

Thinking about a workaround for a similar BIOS bug, you can try to make the directory structure and touch `/EFI/boot/bootx64.efi`.  Contents don't matter, just try to touch the file and see if it helps.

hughsie, 2023-07-27:
@yejingchen can you try with https://github.com/fwupd/fwupd/pull/6028 applied to your fwupd please? If that works I'll respin a new fwupd tarball.

yejingchen, 2023-07-27:
> @yejingchen can you try with #6028 applied to your fwupd please? If that works I'll respin a new fwupd tarball.

fdisk says my SSD sector size is indeed 512 bytes, should I just try it?

```
% sudo LANGUAGE= LANG=C.UTF-8 fdisk -l /dev/nvme0n1
Disk /dev/nvme0n1: 476.94 GiB, 512110190592 bytes, 1000215216 sectors
Disk model: WD PC SN810 SDCPNRY-512G-1006           
Units: sectors of 1 * 512 = 512 bytes
Sector size (logical/physical): 512 bytes / 512 bytes
I/O size (minimum/optimal): 512 bytes / 512 bytes
Disklabel type: gpt
```

hughsie, 2023-07-27:
If it's 512 bytes then I don't think that PR is going to help much.

yejingchen, 2024-03-10:
As HP released a new firmware version 0x1070000 I decided to try this again, only to hit the same result.

However, I was able to reinstall the current version running `fwupdmgr --allow-reinstall install U82_01011200.cab` after downloading from https://fwupd.org/lvfs/devices/com.hp.laptop.system.U82.firmware, and I can go beyond the `fwupd-efi version 1.4` reboot to the updating screen, so I guess the reboot after `fwupd-efi version 1.4` is the intended behavior, not fwupd's fault.

So I am closing this as probably only HP can do something with the firmware update process :(

closed 2024-03-10 by yejingchen