fwupd/fwupd #6017: fwupd test local .cab occur: Transaction  timed out

Fanice-luo, 2023-07-20:
Describe the bug
I was trying pack .cab to test fwupd update the firmware for T99W373, but it occur: Transaction timed out. 

Steps to Reproduce
1.fwupdate get-devices
2.sudo fwupdmgr install XXX
jbd@jbd-ThinkPad-X1-Nano-Gen-2:~/shawn/cab/T99w373-105b-ap010$ sudo fwupdmgr install 20230713-t99w373.fde.f0.3.0.1.1.cab -v
(fwupdmgr:14349): GLib-DEBUG: 10:47:35.824: setenv()/putenv() are not thread-safe and should not be used after threads are created
(fwupdmgr:14349): GLib-GIO-DEBUG: 10:47:35.832: _g_io_module_get_default: Found default implementation dconf (DConfSettingsBackend) for ‘gsettings-backend’
(fwupdmgr:14349): dconf-DEBUG: 10:47:35.832: watch_fast: "/system/proxy/" (establishing: 0, active: 0)
(fwupdmgr:14349): dconf-DEBUG: 10:47:35.833: watch_fast: "/system/proxy/http/" (establishing: 0, active: 0)
(fwupdmgr:14349): dconf-DEBUG: 10:47:35.833: watch_fast: "/system/proxy/https/" (establishing: 0, active: 0)
(fwupdmgr:14349): dconf-DEBUG: 10:47:35.833: watch_fast: "/system/proxy/ftp/" (establishing: 0, active: 0)
(fwupdmgr:14349): dconf-DEBUG: 10:47:35.833: watch_fast: "/system/proxy/socks/" (establishing: 0, active: 0)
(fwupdmgr:14349): dconf-DEBUG: 10:47:35.833: unwatch_fast: "/system/proxy/" (active: 0, establishing: 1)
(fwupdmgr:14349): dconf-DEBUG: 10:47:35.833: unwatch_fast: "/system/proxy/http/" (active: 0, establishing: 1)
(fwupdmgr:14349): dconf-DEBUG: 10:47:35.833: unwatch_fast: "/system/proxy/https/" (active: 0, establishing: 1)
(fwupdmgr:14349): dconf-DEBUG: 10:47:35.833: unwatch_fast: "/system/proxy/ftp/" (active: 0, establishing: 1)
(fwupdmgr:14349): dconf-DEBUG: 10:47:35.833: unwatch_fast: "/system/proxy/socks/" (active: 0, establishing: 1)
(fwupdmgr:14349): GLib-DEBUG: 10:47:35.833: posix_spawn avoided (fd close requested) 
(fwupdmgr:14349): GLib-GIO-DEBUG: 10:47:35.834: _g_io_module_get_default: Found default implementation libproxy (GLibproxyResolver) for ‘gio-proxy-resolver’
(fwupdmgr:14349): dconf-DEBUG: 10:47:35.839: watch_established: "/system/proxy/" (establishing: 0)
(fwupdmgr:14349): dconf-DEBUG: 10:47:35.840: watch_established: "/system/proxy/http/" (establishing: 0)
(fwupdmgr:14349): dconf-DEBUG: 10:47:35.840: watch_established: "/system/proxy/https/" (establishing: 0)
(fwupdmgr:14349): dconf-DEBUG: 10:47:35.840: watch_established: "/system/proxy/ftp/" (establishing: 0)
(fwupdmgr:14349): dconf-DEBUG: 10:47:35.840: watch_established: "/system/proxy/socks/" (establishing: 0)
(fwupdmgr:14349): Fwupd-DEBUG: 10:47:35.840: Emitting ::status-changed() [idle]
WARNING: This package has not been validated, it may not work properly.
(fwupdmgr:14349): Fwupd-DEBUG: 10:47:35.927: Emitting ::status-changed() [decompressing]
Decompressing…           [                    \                  ](fwupdmgr:14349): Fwupd-DEBUG: 10:47:36.693: Emitting ::status-changed() [idle]
Decompressing…           [***************************************]
(fwupdmgr:14349): Fwupd-DEBUG: 10:47:36.694: Emitting ::status-changed() [waiting-for-auth]
Authenticating…          [                    \                  ]
(fwupdmgr:14349): Fwupd-DEBUG: 10:47:36.696: Emitting ::status-changed() [idle]
Authenticating…          [***************************************]
(fwupdmgr:14349): Fwupd-DEBUG: 10:47:36.718: Emitting ::status-changed() [device-restart]
Restarting device…       [                           |           ](fwupdmgr:14349): Fwupd-DEBUG: 10:47:37.023: Emitting ::status-changed() [device-write]
Writing…                 [                                       ](fwupdmgr:14349): Fwupd-DEBUG: 10:47:37.023: Emitting ::status-changed() [decompressing]
Decompressing…           [                                       ](fwupdmgr:14349): Fwupd-DEBUG: 10:47:44.923: Emitting ::status-changed() [device-write]
Writing…                 [******************                     ] 7 minutes remaining…(fwupdmgr:14349): Fwupd-DEBUG: 10:54:15.937: Emitting ::device-changed(bba54f76de66958e5d70964b6ddd2d1e6c7dae9e)
(fwupdmgr:14349): Fwupd-DEBUG: 10:54:37.054: Emitting ::device-changed(bba54f76de66958e5d70964b6ddd2d1e6c7dae9e)
(fwupdmgr:14349): Fwupd-DEBUG: 10:54:37.054: Emitting ::changed()
(fwupdmgr:14349): Fwupd-DEBUG: 10:54:37.087: Emitting ::device-changed(bba54f76de66958e5d70964b6ddd2d1e6c7dae9e)
(fwupdmgr:14349): Fwupd-DEBUG: 10:54:37.094: Emitting ::device-changed(bba54f76de66958e5d70964b6ddd2d1e6c7dae9e)
Transaction timed out

Best Wishes, Fanice

hughsie, 2023-07-20:
> Transaction timed out

I think this is a message from D-Bus -- what distro is this please?

Fanice-luo, 2023-07-20:
fwupd version is: 1.7.5
Now debug find fu-mm-device.c: #define FU_MM_DEVICE_REMOVE_DELAY_REPROBE 120000 /* ms */ and #define MAX_WAIT_TIME_SECS 150 /* s */    will cause the length of time that "Transaction timed out" will occur.

Best Wishes,Fanice

hughsie, 2023-07-20:
Does upping the value of MAX_WAIT_TIME_SECS make things work? i.e. `#define MAX_WAIT_TIME_SECS 300 /* s */`

SlarkXiao, 2023-07-28:
Hi Hughsie,
  Thanks for your reply. It don't work by upping the value of MAX_WAIT_TIME_SECS .
  I want to add more details about this issue.
1. Device use QDU to do the download.
2. There is problem to esitime the upgrade percentage progress.  The function fu_progress_set_percentage_full() in fu_mbim_qdu_updater_file_write_ready() reach to 99 , the percentage was updated by fu_progress_child_percentage_changed_cb(), no more than 46%. So the writing status only reach to about 46(more or less), device would have a reboot since the download is finished actually. 
3. T99W373 would always reboot twice to upgrade FW. This is expected for this device since fw image is too big to upgrade. Module would enter into a special mode(lets' say recovery mode) to do upgrade by first reboot, after that finished, device would have a second reboot. So how can we avoid this in fwupd side?

Fanice please help provide the debug log .

Fanice-luo, 2023-07-28:
Hi Hugsie,
I add some g_print to debug for fwupd1.7.5.Here are some the debug log as follow:

03:20:16:0642 FuPluginMm           [ota.bin] MD5 matched
Writing…: 1%
111#Fanice# *********fu_mbim_qdu_updater_set_update_session start!!
222#Fanice# *********fu_mbim_qdu_updater_session_ready start!!
03:20:17:0023 FuPluginMm           [/dev/wwan0mbim0] Successfully request modem to update session
222#Fanice# *********fu_mbim_qdu_updater_session_ready end!!
333#Fanice# *********fu_mbim_qdu_updater_file_open_ready start!!
8888**********Fanice [/dev/wwan0mbim0] Successfully request modem to update file open
9999**********Fanice [/dev/wwan0mbim0] Successfully request modem to update file open
333#Fanice# *********fu_mbim_qdu_updater_file_open_ready end!!
444#Fanice# *********fu_mbim_qdu_updater_file_write_ready start!!
####Fanice fu_progress_set_percentage percentage: 0
####Fanice fu_progress_set_percentage priv->percentage: 0
####Fanice fu_progress_set_percentage_full (gsize)ctx->chunk_sent: 1
####Fanice fu_progress_set_percentage_full (gsize)ctx->chunks->len: 1188
444#Fanice# *********fu_mbim_qdu_updater_file_write_ready start!!
####Fanice fu_progress_set_percentage percentage: 0
####Fanice fu_progress_set_percentage priv->percentage: 0
####Fanice fu_progress_set_percentage_full (gsize)ctx->chunk_sent: 2
####Fanice fu_progress_set_percentage_full (gsize)ctx->chunks->len: 1188
Writing…: 47%
####Fanice fu_progress_set_percentage_full (gsize)ctx->chunk_sent: 1187
####Fanice fu_progress_set_percentage_full (gsize)ctx->chunks->len: 1188
03:21:42:0375 Mbim                 [/dev/wwan0mbim0] number of consecutive timeouts: 1
444#Fanice# *********fu_mbim_qdu_updater_file_write_ready start!!
03:21:42:0375 FuPluginMm           error: operation failed: Fanice MBIM Transaction timed out
#shawn# fu_mbim_qdu_updater_write error!!
#shawn# *********g_propagate_error error: 2
#shawn# *********g_propagate_error ctx.error: Fanice MBIM Transaction timed out
#shawn# *********g_propagate_error end

Best Wishes,Fanice

Fanice-luo, 2023-08-03:
Hi Hugsie,
  Please ignore the last log.Here is the new log.
[SDX65_fwupdtool_log.pdf](https://github.com/fwupd/fwupd/files/12247651/SDX65_fwupdtool_log.pdf)

hughsie, 2023-08-14:
> T99W373 would always reboot twice to upgrade FW. This is expected for this device since fw image is too big to upgrade. Module would enter into a special mode(lets' say recovery mode) to do upgrade by first reboot, after that finished, device would have a second reboot.

I'm not sure how the plugin code would cope with this. Would it be possible to send me a T99W373 and some firmware so I can test locally?

hughsie, 2023-08-17:
> Could you share a address for shipment

Yup, but not here :) Can you email richard_at_hughsie_dot_com and I'll reply with my address. Thanks.

If you do a PR making just the 10 second timeout to be 20 seconds we can merge that ASAP.

SlarkXiao, 2023-08-17:
Hi Richard,
Actually, we need to modify 2 places at least. 
One is we talked in above, another is the FU_MM_DEVICE_REMOVE_DELAY_REPROBE need to update to 180 seconds or 200 seconds. But this was tested with fwupd 1.7.5. since my OS is Ubuntu 20.04 and I got a problem to upgrade to 1.9.3. So still don't know the result of latest version.
 Even though fwupd report upgrade successfully, the modem manager can't work well as it report sim-missing. Suppose it has a problem when checking device since device reboot twice.
So, do you think should we make PR ignore above potential issues?

hughsie, 2023-08-17:
lets do the timeout PR first -- that seems safe to include in the pending release

SlarkXiao, 2023-08-17:
'Only 10->20 'or 'both 10->20 and 120000->180000' ?

hughsie, 2023-08-17:
the latter

SlarkXiao, 2023-08-21:
Hi Hughsie,
  Does the pull request would be integrated into the stable release? Or only the latest release?

hughsie, 2023-08-21:
> Does the pull request would be integrated into the stable release

I've just pushed it into the 1_8_X branch too -- but I'm not expecting to do a 1.8.x release for quite a few weeks I'm afraid.

SlarkXiao, 2023-08-21:
Nice to hear that it has been pushed into stable release branch. Thanks!

SlarkXiao, 2023-08-26:
Hi Hughsie,
Issue still reproduced with modified 1.7.5(1/6) and 1.9.4(3/5).
Could you help do a cehck in your side? I tried increase the timeout of write_file_qdu to 30 or 60, issue still could be reproduced. Actually, I find the response has been received from QXDM log. So is there any other issue exist?
I will share 2 cab files to you later. Please help do a check if possible

SlarkXiao, 2023-08-26:
[0826_1750_1.7.5_passed.log](https://github.com/fwupd/fwupd/files/12445098/0826_1750_1.7.5_passed.log)
[0826_1754_1.7.5_failed.log](https://github.com/fwupd/fwupd/files/12445099/0826_1754_1.7.5_failed.log)
[0826_1810_1.9.4_passed.log](https://github.com/fwupd/fwupd/files/12445100/0826_1810_1.9.4_passed.log)
[0826_1820_1.9.4_failed.log](https://github.com/fwupd/fwupd/files/12445101/0826_1820_1.9.4_failed.log)

SlarkXiao, 2023-08-30:
Hi Hughsie,
This timed out issue seems caused by the value of "/power/auosuspend_delay_ms" not enough.
For last fu_mbim_qdu_updater_file_write_ready, we needs 12~16s to get the response. Once we send last MBIM write request, device may enter into suspend state after 10000 ms which was set in fu_mm_device_write_firmware_mbim_qdu(). This would block the response to return.
So we need to increase the delay time from 10000ms to 20000ms to sync with the time out of MBIM command. After this, the new transaction timed out issue gone.
We'll create a pull request later.

hughsie, 2023-08-30:
Good detective work. I'll test with your new PR.

Fanice-luo, 2023-09-07:
Hi Hughsie,
Since we changed the autosuspend_delay_ms from 10000ms to 20000ms, we want to restore settings back for this item.
But there is an error "cat: autosuspend_delay_ms: Input/output error" when "cat /sys/bus/mhi/devices/mhi0/power/autosuspend_delay_ms" , even read it after executing fu_mm_device_writeln().
I tried to use fu_io_channel_read_bytes() to get values but also report "failed to read 20: Bad file descriptor".
This issue mainly comes from driver side or something else, but may I know how to get the default settings or current settings in fwupd side? It's better to restore it back to default settings for power saving.

SlarkXiao, 2023-09-07:
After internal checking, the previous path "/sys/bus/mhi/devices/mhi0/power/autosuspend_delay_ms" should be wrong.
Correct path is "/sys/devices/pci0000:00/0000:00:1c.0/0000:08:00.0/power/autosuspend_dealay_ms" in my PC.
And we found default settings is 2000ms by cat command.
So question is:
1. Why fu_io_channel_read_bytes() can't get value?
2. We shall restore 2000ms back after upgrade. Do you agree?

hughsie, 2023-09-25:
> Why fu_io_channel_read_bytes() can't get value?

Is the sysfs path kernel version dependant?

> We shall restore 2000ms back after upgrade. Do you agree?

Yes, I think that's a good idea.

hughsie, 2024-11-11:
I believe we fixed this, closing.

closed 2024-11-11 by hughsie