fwupd/fwupd #6019: Can't update system firmware

PhilDevProg, 2023-07-21:
**Describe the bug**
I have a Lenovo ThinkPad L13 Yoga Gen 4 (AMD) with Fedora 38 and its system firmware has an available update. See `sudo fwupdmgr update`:

> Devices with no available firmware updates: 
 ? ELAN067F:00 04F3:3209
 ? Integrated Camera
 ? MZAL4512HBLU-00BL1
 ? Prometheus IOTA Config
 ? UEFI Device Firmware
 ? UEFI Device Firmware
 ? UEFI Device Firmware
 ? UEFI Device Firmware
 ? UEFI Device Firmware
 ? UEFI Device Firmware
 ? UEFI Device Firmware
 ? UEFI Device Firmware
Devices with the latest available firmware version:
 ? Prometheus
????????????????????????????????????????????????????????????????????????????????
? Upgrade System Firmware from 0.1.6 to 0.1.16?                                ?
????????????????????????????????????????????????????????????????????????????????
? Lenovo ThinkPad L13YogaGen3 System Firmware:                                 ?
?                                                                              ?
? *Add support for ECFW WUFU/LVFS function                                     ?
?                                                                              ?
? 21FSS00100 must remain plugged into a power source for the duration of the   ?
? update to avoid damage.                                                      ?
????????????????????????????????????????????????????????????????????????????????

When it restarts, it displays `fwupd-efi version 1.4` and `Reset system` and then switches to the Lenovo firmware update progress bar but immediatly (with 0%) my laptop restarts and cancels the update.

`sudo fwupdmgr get-history`:

```

LENOVO 21FSS00100
?
??Prometheus:
? ?   Device ID:          23ec719b6aabc2d2dac5176c232f0da7a21881b0
? ?   Previous version:   10.01.3273255
? ?   Update State:       Success
? ?   Last modified:      2023-07-11 14:57
? ?   GUID:               659f7e45-8d45-528d-b3c7-0695eed055f6
? ?   Device Flags:       ? Updatable
? ?                       ? Supported on remote server
? ?                       ? Cryptographic hash verification is available
? ?                       ? Signed Payload
? ? 
? ??Prometheus Fingerprint Reader Update:
?       New version:      10.01.3478575
?       Remote ID:        lvfs
?       Release ID:       10696
?       Summary:          Firmware for the Synaptics Prometheus Fingerprint Reader device
?       License:          Proprietary
?       Size:             425.5?kB
?       Created:          2021-06-09
?       Urgency:          Medium
?       Vendor:           Synaptics
?       Duration:         2 seconds
?       Release Flags:    ? Trusted metadata
?       Description:      
?       New features and enhancements:
?       
?       ? Fix an authentication issue on FOP project.
?     
??UEFI dbx:
? ?   Device ID:          362301da643102b9f38477387e2193e57abaa590
? ?   Previous version:   190
? ?   Update State:       Success
? ?   Last modified:      2023-07-11 14:57
[... 474 lines cut ...]
│   │   └── unicode.pf2
│   ├── grub.cfg
│   └── grubenv
├── initramfs-0-rescue-89e0cfaccc9a4e5fa04073c7bb26b088.img
├── initramfs-6.3.12-200.fc38.x86_64.img
├── loader
│   └── entries
│       ├── 89e0cfaccc9a4e5fa04073c7bb26b088-0-rescue.conf
│       └── 89e0cfaccc9a4e5fa04073c7bb26b088-6.3.12-200.fc38.x86_64.conf
├── lost+found
├── symvers-6.3.12-200.fc38.x86_64.gz -> /lib/modules/6.3.12-200.fc38.x86_64/symvers.gz
├── System.map-6.3.12-200.fc38.x86_64
├── vmlinuz-0-rescue-89e0cfaccc9a4e5fa04073c7bb26b088
└── vmlinuz-6.3.12-200.fc38.x86_64

14 directories, 32 files```

**Additional questions**

- Operating system and version: Fedora 38
- Have you tried rebooting? Yes.
- Is this a regression? Maybe?
- Are you using an NVMe disk? Yes.
- Is secure boot enabled? Yes.
- Is this a Lenovo system with 'Boot Order Lock' turned on in the BIOS? No.

PhilDevProg, 2023-07-22:
Looks like this firmware update is the first update that enables LVFS function... Looks like I have to update with Windows this time.

closed 2023-07-22 by PhilDevProg

SemperRoto, 2023-09-29:
Why was this closed? I have the same problem at the moment. Is there a solution?