fwupd/fwupd #6020: Unable to install fwupd - Ubuntu 20 04

mailsanchu, 2023-07-21:
svarkey@svarkey-Precision-7760:~$ sudo snap install fwupd --classic
error: cannot perform the following tasks:
- Mount snap "fwupd" (5039) (installation not allowed by "shutdown" plug rule of interface "shutdown" for "fwupd" snap)
svarkey@svarkey-Precision-7760:~$ uname -a
Linux svarkey-Precision-7760 5.15.0-76-generic #83~20.04.1-Ubuntu SMP Wed Jun 21 20:23:31 UTC 2023 x86_64 x86_64 x86_64 GNU/Linux

superm1, 2023-07-22:
It shouldn't be classic.

mailsanchu, 2023-07-22:
svarkey@svarkey-Precision-7760:~$ sudo snap install fwupd --classic
```
[sudo] password for svarkey: 
error: cannot perform the following tasks:
- Mount snap "fwupd" (5039) (installation not allowed by "shutdown" plug rule of interface "shutdown" for "fwupd" snap)
```
svarkey@svarkey-Precision-7760:~$ sudo snap install fwupd
```
error: cannot perform the following tasks:
- Mount snap "fwupd" (5039) (installation not allowed by "shutdown" plug rule of interface "shutdown" for "fwupd" snap)
```
svarkey@svarkey-Precision-7760:~$ sudo snap install fwupd --edge
```
error: cannot perform the following tasks:
- Mount snap "fwupd" (5062) (installation not allowed by "shutdown" plug rule of interface "shutdown" for "fwupd" snap)
```
svarkey@svarkey-Precision-7760:~$ sudo snap install fwupd --stable
```
error: cannot perform the following tasks:
- Mount snap "fwupd" (5039) (installation not allowed by "shutdown" plug rule of interface "shutdown" for "fwupd" snap)
svarkey@svarkey-Precision-7760:~$ 
```

mailsanchu, 2023-07-22:
All of them are giving me the same error

valentindavid, 2023-07-25:
This is needed since https://github.com/fwupd/fwupd/pull/5513

I have opened a request on the store: https://forum.snapcraft.io/t/add-shutdown-to-fwupd/36108

CloudExt, 2023-07-28:
Hi,

My Ubuntu 22.04 has the same error:

```
syna@syna-MS-7978:~$ sudo snap install fwupd --classic
error: cannot perform the following tasks:
- Mount snap "fwupd" (5039) (installation not allowed by "shutdown" plug rule of interface "shutdown" for "fwupd" snap)
``` 

```
syna@syna-MS-7978:~$ lsb_release -a
No LSB modules are available.
Distributor ID:	Ubuntu
Description:	Ubuntu 22.04.2 LTS
Release:	22.04
Codename:	jammy
```

mistr, 2023-07-28:
> Hi,
> 
> My Ubuntu 22.04 has the same error:

Me too.

kirisakow, 2023-07-31:
> It shouldn't be classic.

And yet @superm1, that's what the [wiki](https://github.com/fwupd/fwupd/wiki/fwupd-snap) says:

> To install the snap:
> 
>     # snap install fwupd --classic

superm1, 2023-07-31:
> > It shouldn't be classic.
> 
> And yet @superm1, that's what the [wiki](https://github.com/fwupd/fwupd/wiki/fwupd-snap) says:
> 
> > To install the snap:
> > ```
> > # snap install fwupd --classic
> > ```

It's a mistake that it's not updated yet.  I'll fix it.

superm1, 2023-08-02:
Snap store just granted shutdown permission. Try again now.

mailsanchu, 2023-08-02:
It is working now. Thanks

closed 2023-08-02 by mailsanchu