fwupd/fwupd #6022: Failed to connect to daemon: Error calling StartServiceByName for org.freedesktop.fwupd: Failed to activate service 'org.freedesktop.fwupd': timed out (service_start_timeout=25000ms)

vishnumur777, 2023-07-22:
**Describe the bug**
when I tried to run fwupdmgr it shows the error:-

Failed to connect to daemon: Error calling StartServiceByName for org.freedesktop.fwupd: Failed to activate service 'org.freedesktop.fwupd': timed out (service_start_timeout=25000ms)

**Steps to Reproduce**

```shell sudo systemctl status fwupd```
fwupd.service - Firmware update daemon
     Loaded: loaded (/usr/lib/systemd/system/fwupd.service; static)
     Active: failed (Result: timeout) since Sat 2023-07-22 19:07:01 IST; 18min ago
       Docs: https://fwupd.org/
    Process: 3756 ExecStart=/usr/lib/fwupd/fwupd (code=killed, signal=TERM)
   Main PID: 3756 (code=killed, signal=TERM)
        CPU: 732ms

Jul 22 19:03:27 varun-vostro3401 systemd[1]: Starting Firmware update daemon...
Jul 22 19:03:54 varun-vostro3401 fwupd[3756]: 13:33:54.147 FuEngine             failed to add device /sys/devices/pci0000:00/0000:00:1f.0: failed to mmap spibar: operation not permitted

Jul 22 19:06:27 varun-vostro3401 systemd[1]: fwupd.service: start operation timed out. Terminating.
Jul 22 19:07:01 varun-vostro3401 systemd[1]: fwupd.service: Failed with result 'timeout'.
Jul 22 19:07:01 varun-vostro3401 systemd[1]: Failed to start Firmware update daemon.

**Expected behavior**
A clear and concise description of what you expected to happen.

**fwupd version information**
Please provide the version of the daemon and client.

```shell
sudo fwupdtool --version
```
Loading…                 [************************************** ]14:00:52.639 FuEngine             failed to update history database: device ID a45df35ac0e948ee180fe216a5f703f32dda163f was not found
14:00:52.639 FuEngine             failed to update history database: device ID 362301da643102b9f38477387e2193e57abaa590 was not found

compile   org.freedesktop.fwupd         1.9.3
compile   com.hughsie.libxmlb           0.3.11
compile   com.hughsie.libjcat           0.1.14
compile   org.freedesktop.gusb          0.4.6
runtime   com.hughsie.libjcat           0.1.14
runtime   com.dell.libsmbios            2.4
runtime   org.freedesktop.gusb          0.4.6
runtime   org.freedesktop.fwupd         1.9.3
runtime   org.kernel                    6.4.4-arch1-1

Please note how you installed it (`apt`, `dnf`, `pacman`, source, etc): `pacman`

<details>

<summary>**fwupd device information**</summary>

Please provide the output of the fwupd devices recognized in your system.

```shell
fwupdmgr get-devices --show-all-devices
```
Failed to connect to daemon: Error calling StartServiceByName for org.freedesktop.fwupd: Failed to activate service 'org.freedesktop.fwupd': timed out (service_start_timeout=25000ms)

</details>

**Additional questions**

- Operating system and version: Arch Linux [Endeavour OS]
- Have you tried rebooting? `yes'
- Is this a regression? no

hughsie, 2023-07-24:
I don't know why it's refusing to start but using `-Dplugin_intel_spi=false` will at least disable Intel SPI support. What does `sudo /usr/libexec/fwupd/fwupd -vv` say?

kamiccolo, 2023-09-20:
Having similar issue on ArchLinux. Running `fwdupdmgr` second time in a row - makes it run.

systemd status do look simmilar:

```
# sudo systemctl status fwupd
○ fwupd.service - Firmware update daemon
     Loaded: loaded (/usr/lib/systemd/system/fwupd.service; static)
     Active: inactive (dead)
       Docs: https://fwupd.org/

Sep 20 13:19:09 laptwo systemd[1]: Starting Firmware update daemon...
Sep 20 13:19:10 laptwo fwupd[13550]: 10:19:10.264 FuEngine             failed to add device usb:03:00:03: failed to get version: error status: 0x315
Sep 20 13:19:10 laptwo fwupd[13550]: 10:19:10.273 FuEngine             failed to add device usb:03:00:05:04: failed to read SPI chip ID: failed to read chip ID: endpoint stalled or request not supported
Sep 20 13:19:10 laptwo fwupd[13550]: 10:19:10.567 FuEngine             failed to add device /sys/devices/pci0000:00/0000:00:1f.5: failed to mmap SPIBAR: Operation not permitted
Sep 20 13:19:35 laptwo fwupd[13550]: 10:19:35.638 FuMain               Daemon ready for requests (locale en_US.UTF-8)
Sep 20 13:19:35 laptwo systemd[1]: Started Firmware update daemon.
Sep 20 13:22:03 laptwo fwupd[13550]: 10:22:03.328 FuMain               another service has claimed the dbus name org.freedesktop.fwupd
Sep 20 13:22:03 laptwo systemd[1]: fwupd.service: Deactivated successfully.
```

Also attaching the result of `/usr/lib/fwupd/fwupd -vv &> /tmp/fwupd_verbose.log`:
[fwupd_verbose.log](https://github.com/fwupd/fwupd/files/12672041/fwupd_verbose.log)

Not sure what's considered fishy here, mayhaps the following:

```
10:26:37.553 FuEngine             failed to load Passim: Error calling StartServiceByName for org.freedesktop.Passim: Failed to activate service 'org.freedesktop.Passim': timed out (service_start_timeout=25000ms)
10:26:37.553 FuMain               Emitting PropertyChanged('Status'='idle')
10:26:37.553 FuEngine             resetting update motd timeout

../fwupd-1.9.5/src/fu-daemon.c:2169 [25836.21ms]:
    ../fwupd-1.9.5/src/fu-engine.c:8151:load-engine [25833.72ms]:
```

hughsie, 2023-09-21:
> Failed to activate service 'org.freedesktop.Passim': timed out (service_start_timeout=25000ms)

I don't know why this is taking 25s to timeout, but that's the problem.

michaellass, 2023-09-22:
> > Failed to activate service 'org.freedesktop.Passim': timed out (service_start_timeout=25000ms)
> 
> I don't know why this is taking 25s to timeout, but that's the problem.

That's https://github.com/hughsie/passim/issues/15.

hughsie, 2023-09-22:
Want a new passim tarball?

michaellass, 2023-09-22:
That should stop people running into this issue, so in my view that would be nice :slightly_smiling_face:. I'll make sure to let the Arch Linux packager know that avahi is a dependency for passim, just like [you introduced in the RPM spec](https://github.com/hughsie/passim/commit/85785c54db19c474be41680d664406db5b3b92a9).

hughsie, 2023-09-25:
> so in my view that would be nice

Done! https://github.com/hughsie/passim/releases/tag/0.1.2

superm1, 2023-09-25:
I've just bumped the min passim version for fwupd to use to match that. I think we should be good to close this now.

superm1, 2023-10-06:
You can install avahi, recompile fwupd without passim, or update passim.

renozion, 2023-10-06:
> You can install avahi, recompile fwupd without passim, or update passim.

mine here says avahi already installed still have the error, cant fix , been here for weeks now... 
:(

superm1, 2023-10-06:
> > You can install avahi, recompile fwupd without passim, or update passim.
> 
> mine here says avahi already installed still have the error, cant fix , been here for weeks now... 
> :( 

Can you please upgrade to 1.9.6? If you can reproduce still please provide the journal output for fwupd.service.

renozion, 2023-10-06:
ok mr mario bro, here it is, is no arch here is kali, debian stuff (sorry bad english mario bro): 
fwupd.service - Firmware update daemon
     Loaded: loaded (/lib/systemd/system/fwupd.service; static)
     Active: failed (Result: exit-code) since Fri 2023-10-06 09:03:06 -03; 10min ago
       Docs: https://fwupd.org/
    Process: 15881 ExecStart=/usr/libexec/fwupd/fwupd (code=exited, status=1/FAILURE)
   Main PID: 15881 (code=exited, status=1/FAILURE)
        CPU: 262ms

out 06 09:02:41 kali systemd[1]: Starting fwupd.service - Firmware update daemon...
out 06 09:02:41 kali fwupd[15881]: 12:02:41.415 FuPluginIntelMe      failed to get public key using /fpf/OemCred: generic failure [0x9e]
out 06 09:03:06 kali fwupd[15881]: Failed to load daemon: failed to load authority: Error initializing authority: Erro ao chamar StartServiceByName para org.freedeskt>
out 06 09:03:06 kali fwupd[15881]: 12:03:06.494 GLib-GObject         g_object_unref: assertion 'G_IS_OBJECT (object)' failed
out 06 09:03:06 kali systemd[1]: fwupd.service: Main process exited, code=exited, status=1/FAILURE
out 06 09:03:06 kali systemd[1]: fwupd.service: Failed with result 'exit-code'.
out 06 09:03:06 kali systemd[1]: Failed to start fwupd.service - Firmware update daemon.

also run sudo fwupdtool security --force
Carregando…              [******************                     ]12:17:20.199 FuPluginIntelMe      failed to get public key using /fpf/OemCred: generic failure [0x9e]
Carregando…              [************************************** ]
AVISO: Firmware UEFI não pode ser atualizado no modo BIOS legado
Veja https://github.com/fwupd/fwupd/wiki/PluginFlag:legacy-bios para mais informações.
12:17:20.336 FuPluginPciMei       ME family not supported for 0:7.0.4.1197
12:17:20.337 FuEngine             HSI attribute org.fwupd.hsi.Kernel.Lockdown (from linux_lockdown) had unknown result
12:17:20.337 FuEngine             HSI attribute org.fwupd.hsi.Mei.Version (from pci_mei) had unknown result

also my friend, must say ive done some /gshadow fuzzin around maybe thats the issue, ive seen some usermod commands but not quite sure how to get back to default on the files :(

jvitous81, 2023-10-19:
Hi renozion,
If this can any help you, in my case the fix was to use kernel 6.5.5-1, and after an update, the problem disappeared.

renozion, 2023-10-19:
thanks bro will do it.

renozion, 2023-10-19:
> Hi renozion, If this can any help you, in my case the fix was to use kernel 6.5.5-1, and after an update, the problem disappeared.

thx my dude,fixed no sytemctl fails. all done!

Germano0, 2024-04-05:
Can we please reopen this?
I experienced that on
- fwupd-1.9.15-1.fc39.x86_64
- 6.7.11-200.fc39.x86_64
```
# fwupdmgr get-updates
Connessione al demone non riuscita: Errore nel chiamare StartServiceByName per org.freedesktop.fwupd: È stato raggiunto il timeout
```
journalctl
```
apr 04 22:00:00 T16-user systemd[1]: Starting fwupd.service - Firmware update daemon...
apr 04 22:03:00 T16-user systemd[1]: fwupd.service: start operation timed out. Terminating.
apr 04 22:03:00 T16-user systemd[1]: fwupd.service: Failed with result 'timeout'.
apr 04 22:03:00 T16-user systemd[1]: Failed to start fwupd.service - Firmware update daemon.
apr 04 22:03:00 T16-user audit[1]: SERVICE_START pid=1 uid=0 auid=4294967295 ses=4294967295 subj=system_u:system_r:init_t:s0 msg='unit=fwupd comm="systemd" exe="/usr/lib/systemd/systemd" hostname=? addr=? >
```

superm1, 2024-04-05:
> Can we please reopen this?
> I experienced that on
> - fwupd-1.9.15-1.fc39.x86_64
> - 6.7.11-200.fc39.x86_64
> ```
> # fwupdmgr get-updates
> Connessione al demone non riuscita: Errore nel chiamare [...]

Open a new issue with all your details. It's totally possible it's a kernel bug, we just saw that very recently.