fwupd/fwupd #6023: UEFI update failed (Thinkpad t14s) -- Failed to find 'HD(1,GPT...)...cap' DevicePath

gdonval, 2023-07-22:
**Describe the bug**

Fwupd 1.4 fails to update:

```
fwupd-efi version 1.4
WARNING: Failed to find 'HD(1,GPT,0C8FC2F7-EF45-472E-BCAA-D5691BD44E83)/\EFI\arch\fw\fwupd-66d6a3ef-a771-4302-9cd0-d062c79c5ef2.cap' DevicePath
WARNING: Could not locate device handle: Unsupported
```

**Steps to Reproduce**

* `fwupdmgr refresh`
* `fwupdmgr get-updates`
* `fwupdmgr update`
* Reboot when prompted.

**Expected behavior**

The cap files should be found and the system should be updated.

**fwupd version information**

```shell
compile   org.freedesktop.fwupd         1.9.3
compile   com.hughsie.libxmlb           0.3.11
compile   com.hughsie.libjcat           0.1.14
runtime   org.freedesktop.fwupd-efi     1.4
compile   org.freedesktop.gusb          0.4.6
runtime   com.hughsie.libjcat           0.1.14
runtime   com.dell.libsmbios            2.4
runtime   org.freedesktop.gusb          0.4.6
runtime   org.kernel                    6.4.4-arch1-1
runtime   org.freedesktop.fwupd         1.9.3
```

Please note how you installed it (`apt`, `dnf`, `pacman`, source, etc): `pacman`.

<details>

<summary>**fwupd device information**</summary>

Please provide the output of the fwupd devices recognized in your system.

```shell
LENOVO 21CRS1N300
│
├─ELAN0678:00 04F3:3195:
│     Device ID:          f8bebdac7d9634f3890f6f6cf3eec0908c2c700b
│     Summary:            Touchpad
│     Current version:    0x0003
│     Bootloader Version: 0x0002
│     Vendor:             ELAN Microelectronics (HIDRAW:0x04F3)
│     GUIDs:              4e8cda57-37ba-5dc6-b114-7610e38714e2 ← HIDRAW\VEN_04F3&DEV_3195
│                         cc456594-6f7e-5d1b-82a6-b4558a813152 ← HIDRAW\VEN_04F3&DEV_3195&MOD_0006
│                         4b808d27-c833-50cc-8177-9a401f5c08fe ← ELANTP\ICTYPE_14
│                         d15e804c-570f-511b-803a-b38a4eddae29 ← ELANTP\ICTYPE_14&MOD_0006
│                         296e64f7-9e08-5717-bff2-651e41b0477e ← ELANTP\ICTYPE_14&MOD_0006&DRIVER_HID
│     Device Flags:       • Internal device
│                         • Updatable
│
├─AMD Ryzen 7 PRO 6850U with Radeon Graphics:
│ │   Device ID:          4bde70ba4e39b28f9eab1628f9dd6e6244c03027
│ │   Current version:    0x0a404102
│ │   Vendor:             Advanced Micro Devices, Inc.
│ │   GUIDs:              b9a2dd81-159e-5537-a7db-e7101d164d3f ← cpu
│ │                       22f9ecf4-588d-5c0a-8326-6ebff3655c6d ← CPUID\PRO_0&FAM_19
│ │                       52f8f9af-1ca9-5352-bef4-ceb232c888a5 ← CPUID\PRO_0&FAM_19&MOD_44
│ │                       e94372a3-3ffb-5d1c-a579-c415b7313e52 ← CPUID\PRO_0&FAM_19&MOD_44&STP_1
│ │   Device Flags:       • Internal device
[... 324 lines cut ...]
│   │   │   └── fwupd-6ab943b7-f4d4-aaa1-2f40-cb03a0c8cf3c.cap
│   │   └── fwupdx64.efi
│   ├── BOOT
│   │   └── BOOTX64.EFI
│   ├── Linux
│   │   ├── archlinux.efi
│   │   └── archlinux-fallback.efi
│   └── systemd
│       └── systemd-bootx64.efi
└── loader
    ├── entries
    │   └── fwupd.conf
    ├── entries.srel
    ├── loader.conf
    └── random-seed
```

**Additional questions**

- Operating system and version: Archlinux (up-to-date)
- Have you tried rebooting? Yes.
- Is this a regression? Yes.
- Are you using an NVMe disk? Yes.
- Is secure boot enabled? Yes (tried without too! fwupdx64.efi **is** signed)
- Is this a Lenovo system with 'Boot Order Lock' turned on in the BIOS? No.

gdonval, 2023-07-22:
I have little idea of what might be going on so sorry if my suggestion is stupid but is the path `fwupd-efi` supposed to do correct, sc.

```HD(1,GPT,0C8FC2F7-EF45-472E-BCAA-D5691BD44E83)/\EFI\arch\fw\fwupd-66d6a3ef-a771-4302-9cd0-d062c79c5ef2.cap```

whereas the path reported by `efibootmgr` has an extra `/File` (here altered by me to include the `.cap` file):

```HD(1,GPT,0c8fc2f7-ef45-472e-bcaa-d5691bd44e83,0x800,0x200000)/File(\EFI\arch\fw\fwupd-66d6a3ef-a771-4302-9cd0-d062c79c5ef2.cap)```

And just to make things evidently clear:

```script
$ ls /efi/EFI/arch/fw
fwupd-66d6a3ef-a771-4302-9cd0-d062c79c5ef2.cap
fwupd-6ab943b7-f4d4-aaa1-2f40-cb03a0c8cf3c.cap
```

```script
$ fwupdmgr get-bios-setting
...
BootOrderLock:
  Setting type:         Enumeration
  Current Value:        Disable
  Description:          BootOrderLock
  Read Only:            False
  Possible Values:
    0:                  Disable
    1:                  Enable
...
```

gdonval, 2023-07-24:
```
$ fwupdmgr --version
compile   org.freedesktop.fwupd         1.9.2
compile   com.hughsie.libxmlb           0.3.11
compile   com.hughsie.libjcat           0.1.13
runtime   org.freedesktop.fwupd-efi     1.4
compile   org.freedesktop.gusb          0.4.6
runtime   com.hughsie.libjcat           0.1.14
runtime   com.dell.libsmbios            2.4
runtime   org.freedesktop.gusb          0.4.6
runtime   org.kernel                    6.4.4-arch1-1
runtime   org.freedesktop.fwupd         1.9.2
```

```
$ efibootmgr  -v
BootNext: 0000
BootCurrent: 0004
Timeout: 0 seconds
BootOrder: 0004,001C,001A,001B,001D,001F,0020,001E,0021,0000
Boot0000* Linux-Firmware-Updater	HD(1,GPT,0c8fc2f7-ef45-472e-bcaa-d5691bd44e83,0x100,0x40000)/File(\EFI\arch\fwupdx64.efi)
      dp: 04 01 2a 00 01 00 00 00 00 01 00 00 00 00 00 00 00 00 04 00 00 00 00 00 f7 c2 8f 0c 45 ef 2e 47 bc aa d5 69 1b d4 4e 83 02 02 / 04 04 32 00 5c 00 45 00 46 00 49 00 5c 00 61 00 72 00 63 00 68 00 5c 00 66 00 77 00 75 00 70 00 64 00 78 00 36 00 34 00 2e 00 65 00 66 00 69 00 00 00 / 7f ff 04 00
Boot0004* Linux Boot Manager	HD(1,GPT,0c8fc2f7-ef45-472e-bcaa-d5691bd44e83,0x800,0x200000)/File(\EFI\systemd\systemd-bootx64.efi)
      dp: 04 01 2a 00 01 00 00 00 00 08 00 00 00 00 00 00 00 00 20 00 00 00 00 00 f7 c2 8f 0c 45 ef 2e 47 bc aa d5 69 1b d4 4e 83 02 02 / 04 04 46 00 5c 00 45 00 46 00 49 00 5c 00 73 00 79 00 73 00 74 00 65 00 6d 00 64 00 5c 00 73 00 79 00 73 00 74 00 65 00 6d 00 64 00 2d 00 62 00 6f 00 6f 00 74 00 78 00 36 00 34 00 2e 00 65 00 66 00 69 00 00 00 / 7f ff 04 00
```

(elided to what I think is relevant, please ask if you want the full output again).

hughsie, 2023-07-25:
So, it would appear that the update to 1.9.3 changed the DevicePath from `HD(1,GPT,0c8fc2f7-ef45-472e-bcaa-d5691bd44e83,0x100,0x40000)` to `HD(1,GPT,0c8fc2f7-ef45-472e-bcaa-d5691bd44e83,0x800,0x200000)` -- i.e. the offset and size are both x8 bigger than they should be. Can you attach the output of `udisksctl dump` please.

Also, is it possible your block size might not be 512 bytes? e.g. the output of `blockdev --getsize64 /dev/YOUR_ESP_HERE`

gdonval, 2023-07-25:
<details>
 <summary>udiskctl dump</summary>

Same as before, I tried to be relevant, just ask and I'll provide the full dump.

```
$ udisksctl dump

/org/freedesktop/UDisks2/Manager:
  org.freedesktop.UDisks2.Manager:
    DefaultEncryptionType:      luks1
    SupportedEncryptionTypes:   luks1
                                luks2
    SupportedFilesystems:       ext2
                                ext3
                                ext4
                                vfat
                                ntfs
                                exfat
                                xfs
                                reiserfs
                                nilfs2
                                btrfs
                                minix
                                udf
                                f2fs
                                swap
    Version:                    2.9.4

...

/org/freedesktop/UDisks2/block_devices/nvme0n1:
  org.freedesktop.UDisks2.Block:
    Configuration:              []
    CryptoBackingDevice:        '/'
    Device:                     /dev/nvme0n1
    DeviceNumber:               66304
    Drive:                      '/org/freedesktop/UDisks2/drives/SHGP31_2000GM_AJB8N422910108664'
    HintAuto:                   false
    HintIconName:
    HintIgnore:                 false
    HintName:
    HintPartitionable:          true
    HintSymbolicIconName:
    HintSystem:                 true
    Id:                         by-id-nvme-SHGP31-2000GM_AJB8N422910108664
    IdLabel:
    IdType:
    IdUUID:
    IdUsage:
    IdVersion:
    MDRaid:                     '/'
    MDRaidMember:               '/'
    PreferredDevice:            /dev/nvme0n1
    ReadOnly:                   false
    Size:                       2000398934016
    Symlinks:                   /dev/disk/by-diskseq/1
                                /dev/disk/by-id/nvme-SHGP31-2000GM_AJB8N422910108664
                                /dev/disk/by-id/nvme-SHGP31-2000GM_AJB8N422910108664_1
                                /dev/disk/by-id/nvme-nvme.1c5c-414a42384e343232393130313038363634-5348475033312d32303030474d-00000001
                                /dev/disk/by-path/pci-0000:02:00.0-nvme-1
    UserspaceMountOptions:
  org.freedesktop.UDisks2.PartitionTable:
    Partitions:         /org/freedesktop/UDisks2/block_devices/nvme0n1p2
                        /org/freedesktop/UDisks2/block_devices/nvme0n1p3
                        /org/freedesktop/UDisks2/block_devices/nvme0n1p1
    Type:               gpt

...

/org/freedesktop/UDisks2/block_devices/nvme0n1p1:
  org.freedesktop.UDisks2.Block:
    Configuration:              [('fstab', {'fsname': <b'UUID=9275-DF17'>, 'dir': <b'/efi'>, 'type': <b'vfat'>, 'opts': <b'rw,noauto,noatime,x-systemd.automoun
t,x-systemd.idle-timeout=120,fmask=0077,dmask=0077,codepage=437,iocharset=ascii,shortname=mixed,utf8,errors=remount-ro'>, 'freq': <0>, 'passno': <2>})]
    CryptoBackingDevice:        '/'
    Device:                     /dev/nvme0n1p1
    DeviceNumber:               66305
    Drive:                      '/org/freedesktop/UDisks2/drives/SHGP31_2000GM_AJB8N422910108664'
    HintAuto:                   false
    HintIconName:
    HintIgnore:                 true
    HintName:
    HintPartitionable:          true
    HintSymbolicIconName:
    HintSystem:                 true
    Id:                         by-id-nvme-SHGP31-2000GM_AJB8N422910108664-part1
    IdLabel:
    IdType:                     vfat
    IdUUID:                     9275-DF17
    IdUsage:                    filesystem
    IdVersion:                  FAT32
    MDRaid:                     '/'
    MDRaidMember:               '/'
    PreferredDevice:            /dev/nvme0n1p1
    ReadOnly:                   false
    Size:                       1073741824
    Symlinks:                   /dev/disk/by-diskseq/1-part1
                                /dev/disk/by-id/nvme-SHGP31-2000GM_AJB8N422910108664-part1
                                /dev/disk/by-id/nvme-SHGP31-2000GM_AJB8N422910108664_1-part1
                                /dev/disk/by-id/nvme-nvme.1c5c-414a42384e343232393130313038363634-5348475033312d32303030474d-00000001-part1
                                /dev/disk/by-partlabel/EFI\x20system\x20partition
                                /dev/disk/by-partuuid/0c8fc2f7-ef45-472e-bcaa-d5691bd44e83
                                /dev/disk/by-path/pci-0000:02:00.0-nvme-1-part1
                                /dev/disk/by-uuid/9275-DF17
    UserspaceMountOptions:      x-systemd.automount
                                x-systemd.idle-timeout=120
  org.freedesktop.UDisks2.Filesystem:
    MountPoints:        /efi
    Size:               0
  org.freedesktop.UDisks2.Partition:
    Flags:              0
    IsContained:        false
    IsContainer:        false
    Name:               EFI system partition
    Number:             1
    Offset:             1048576
    Size:               1073741824
    Table:              '/org/freedesktop/UDisks2/block_devices/nvme0n1'
    Type:               c12a7328-f81f-11d2-ba4b-00a0c93ec93b
    UUID:               0c8fc2f7-ef45-472e-bcaa-d5691bd44e83
```

</details>

```
# blockdev --getsize64 /dev/nvme0n1p1
1073741824
```

> Also, is it possible your block size might not be 512 bytes?

Yep, my disk uses 4k block sizes so `mkfs.vfat` used that:

```
$ stat /efi
  File: /efi
  Size: 4096      	Blocks: 8          IO Block: 4096   directory
Device: 259,1	Inode: 1           Links: 4
Access: (0700/drwx------)  Uid: (    0/    root)   Gid: (    0/    root)
Access: 1970-01-01 01:00:00.000000000 +0100
Modify: 1970-01-01 01:00:00.000000000 +0100
Change: 1970-01-01 01:00:00.000000000 +0100
 Birth: 1970-01-01 01:00:00.000000000 +0100
```

hughsie, 2023-07-25:
Hmm, so the big fat comment of:

    /* Linux always considers sectors to be 512 bytes */
    #define BLOCK_SIZE 0x200

looks like it's probably not so true :)

gdonval, 2023-07-25:
It is likely wrong indeed. I was under the impression that this is the difference between logical and physical block size as reported by the kernel in `/sys/block/*/queue/{logical,physical}_block_size` and both can be 4096 bytes.

hughsie, 2023-07-25:
Sooo.... what does `cat /sys/block/nvme0n1/queue/{logical,physical}_block_size` say for you?

hughsie, 2023-07-25:
i.e. what *should* I be using -- the logical or physical block size?

hughsie, 2023-07-25:
Maybe `sudo blockdev --getbsz /dev/nvme1n1p1` knows

gdonval, 2023-07-25:
Both are giving `4096`. `physical` should be at disk firmware level. `logical` is at Linux kernel level and should match the filesystem's block size. 

I really don't know enough about this (nor UEFI) to answer with confidence.

Would there be anyone we could ask?

---

```
$ sudo blockdev --getbsz /dev/nvme0n1p1
4096
```

But I have no idea whether this is the "correct" 4096 or if any other variant should be used. Is the offset provided at boot FAT32 offsets or a real sector offset?

hughsie, 2023-07-25:
> Is the offset provided at boot FAT32 offsets or a real sector offset

It's specified "in blocks" -- give me a few mins I've got a patch in progress.

closed 2023-07-26 by hughsie