fwupd/fwupd #6024: Update Error:       partition UUID required

leeli0, 2023-07-24:
**Describe the bug**
Update Error:       partition UUID required

**Steps to Reproduce**
`fwupdmgr update`

**Expected behavior**
Update successfully

**fwupd version information**
```
compile   org.freedesktop.fwupd         1.9.3
compile   com.hughsie.libxmlb           0.3.11
compile   com.hughsie.libjcat           0.1.14
runtime   org.freedesktop.fwupd-efi     1.4
compile   org.freedesktop.gusb          0.4.6
runtime   com.hughsie.libjcat           0.1.14
runtime   com.dell.libsmbios            2.4
runtime   org.freedesktop.gusb          0.4.6
runtime   org.kernel                    6.4.4-arch1-1
runtime   org.freedesktop.fwupd         1.9.3
```

Please note how you installed it (`apt`, `dnf`, `pacman`, source, etc):
`pacman`

<details>

<summary>**fwupd device information**</summary>

Please provide the output of the fwupd devices recognized in your system.

```
Dell Inc. XPS 17 9700
│
├─DELL098F:00 04F3:311C:
│     Device ID:          56635aa15eda663e5a7b77d3447191853ccea322
│     Summary:            Touchpad
│     Current version:    0x000a
│     Bootloader Version: 0x0001
│     Vendor:             ELAN Microelectronics (HIDRAW:0x04F3)
│     GUIDs:              46598f8d-5777-50c4-8da9-cda0cc8bbc26 ← HIDRAW\VEN_04F3&DEV_311C
│                         d00322f0-ff1f-5458-9c64-8a771fb0b72e ← HIDRAW\VEN_04F3&DEV_311C&MOD_0000
│                         4f9fe0ae-f26e-5c1d-8a4e-a7506fb02f96 ← ELANTP\ICTYPE_13
│                         6cf7ae6a-8b98-5b63-9af8-b23b932f65de ← ELANTP\ICTYPE_13&MOD_0000
│                         ac3627a9-8463-5e63-b6b1-be9e0bcf27d2 ← ELANTP\ICTYPE_13&MOD_0000&DRIVER_HID
│     Device Flags:       • Internal device
│                         • Updatable
│   
├─Unknown Device:
│     Device ID:          4a4907dbb1b96c6a1177dfd1b95eb41c781d1265
│     GUID:               1199a818-4c52-5137-b536-d59e2e2cada9 ← GPIO\ID_INT3450:00
│   
├─CometLake-H GT2 [UHD Graphics]:
│     Device ID:          5792b48846ce271fab11c4a545f7a3df0d36e00a
│     Current version:    05
│     Vendor:             Intel Corporation (PCI:0x8086)
│     GUIDs:              3777783a-3f83-56a5-95f4-533eb6a2bd19 ← PCI\VEN_8086&DEV_9BC4
│                         3749193a-6bc6-56c2-bf80-a28b4f77f5fc ← PCI\VEN_8086&DEV_9BC4&SUBSYS_1028098F
│     Device Flags:       • Internal device
│                         • Cryptographic hash verification is available
│   
├─Core™ i7-10875H CPU @ 2.30GHz:
│     Device ID:          4bde70ba4e39b28f9eab1628f9dd6e6244c03027
│     Current version:    0x000000f6
│     Vendor:             Intel
│     GUIDs:              b9a2dd81-159e-5537-a7db-e7101d164d3f ← cpu
│                         30249f37-d140-5d3e-9319-186b1bd5cac3 ← CPUID\PRO_0&FAM_06
│                         a45b0522-5722-54bd-b802-86cd044262df ← CPUID\PRO_0&FAM_06&MOD_A5
│                         7b9b6e8c-226c-5db6-86cb-ea3187578013 ← CPUID\PRO_0&FAM_06&MOD_A5&STP_2
[... 769 lines cut ...]
    MediaAvailable:             true
    MediaChangeDetected:        true
    MediaCompatibility:         
    MediaRemovable:             false
    Model:                      PM981a NVMe Samsung 1024GB
    Optical:                    false
    OpticalBlank:               false
    OpticalNumAudioTracks:      0
    OpticalNumDataTracks:       0
    OpticalNumSessions:         0
    OpticalNumTracks:           0
    Removable:                  false
    Revision:                   15305129
    RotationRate:               0
    Seat:                       seat0
    Serial:                           S4GXNX0NA03718
    SiblingId:                  
    Size:                       1024209543168
    SortKey:                    00coldplug/00fixed/nvme0n1
    TimeDetected:               1690192006264542
    TimeMediaDetected:          1690192006264542
    Vendor:                     
    WWN:                        eui.344758304ea037180025385800000001

```

hughsie, 2023-07-24:
So you have /boot and / -- but where's your ESP? You need a volume with type `c12a7328-f81f-11d2-ba4b-00a0c93ec93b`

leeli0, 2023-07-24:
The problem is that I did not change the partition type, thus the default type **Linux filesystem** is used.
I changed the type by using the `fdisk -t` option.

[How to change partition type id without formatting](https://unix.stackexchange.com/questions/508890/how-to-change-partition-type-id-without-formatting)

I think maybe the error output should be more informative, then people know what's the problem.

closed 2023-07-24 by leeli0

hughsie, 2023-07-25:
> I think maybe the error output should be more informative, then people know what's the problem.

Hmm, you should have got a warning when running `fwupdmgr get-devices` -- what does `fwupdmgr get-plugins` and `sudo fwupdtool get-devices --plugins uefi-capsule -vv` say please?

hughsie, 2023-07-27:
@leeli0 can you test that https://github.com/fwupd/fwupd/pull/6033 gives you the expected warning with `Linux filesystem` please.

leeli0, 2023-07-27:
```
➜  Downloads fwupdtool --version
Loading…                 [************************************** ]
WARNING: This package has not been validated, it may not work properly.
runtime   org.freedesktop.fwupd         1.9.4
compile   com.hughsie.libxmlb           0.3.11
runtime   com.dell.libsmbios            2.4
runtime   com.hughsie.libjcat           0.1.14
runtime   org.kernel                    6.4.6-arch1-1
compile   org.freedesktop.gusb          0.4.6
compile   com.hughsie.libjcat           0.1.14
compile   org.freedesktop.fwupd         1.9.4
runtime   org.freedesktop.gusb          0.4.6
```

```
➜  Downloads fwupdmgr --version
compile   org.freedesktop.fwupd         1.9.4
compile   com.hughsie.libxmlb           0.3.11
compile   com.hughsie.libjcat           0.1.14
compile   org.freedesktop.gusb          0.4.6
runtime   com.hughsie.libjcat           0.1.14
runtime   com.dell.libsmbios            2.4
runtime   org.freedesktop.gusb          0.4.6
runtime   org.kernel                    6.4.6-arch1-1
runtime   org.freedesktop.fwupd         1.9.4

```

```
➜  Downloads fwupdmgr refresh
WARNING: UEFI ESP partition not detected or configured
See https://github.com/fwupd/fwupd/wiki/PluginFlag:esp-not-found for more information.
WARNING: This package has not been validated, it may not work properly.
No remotes are currently enabled so no metadata is available.
Metadata can be obtained from the Linux Vendor Firmware Service.

```

```
➜  Downloads fwupdmgr update
WARNING: UEFI ESP partition not detected or configured
See https://github.com/fwupd/fwupd/wiki/PluginFlag:esp-not-found for more information.
WARNING: This package has not been validated, it may not work properly.
Devices with no available firmware updates: 
 • UEFI Device Firmware
 • UEFI Device Firmware
 • DELL098F:00 04F3:311C
 • Internal SPI Controller
 • PM981a NVMe Samsung 1024GB
Devices with the latest available firmware version:
 • System Firmware
 • Thunderbolt host controller
 • Thunderbolt host controller
 • UEFI dbx

```

<details>
<summary>fwupdmgr get-devices</summary>

```
[31m[1mWARNING[0m: UEFI ESP partition not detected or configured
See https://github.com/fwupd/fwupd/wiki/PluginFlag:esp-not-found for more information.
[31m[1mWARNING[0m: This package has not been validated, it may not work properly.
Dell Inc. XPS 17 9700
│
├─DELL098F:00 04F3:311C:
│     Device ID:          56635aa15eda663e5a7b77d3447191853ccea322
│     Summary:            Touchpad
[... 1656 lines cut ...]
21:30:59.027 FuMain               FuUefiNvramDevice:
  DeviceId:             2292ae5236790b47884e37cf162dcf23bfcd1c60
  Name:                 UEFI Device Firmware
  Guid:                 3850c9f7-c6f9-ae4a-833b-673332ea8d7e
  Summary:              UEFI ESRT device
  Plugin:               uefi_capsule
  Protocol:             org.uefi.capsule
  Flags:                internal|require-ac|registered|needs-reboot|usable-during-update|updatable-hidden
  VendorId:             DMI:Dell Inc.
  Version:              355488041
  VersionLowest:        355488041
  VersionFormat:        number
  VersionRaw:           0x15305129
  VersionLowestRaw:     0x15305129
  Created:              2023-07-27
  UpdateState:          success
  UpdateError:          Not updatable as UEFI ESP partition not detected

21:30:59.028 FuPluginTest         destroy
21:30:59.029 FuProgress           raw timing data was { 0.016, 0.001, 0.003, 0.000, 0.009, 0.001 } -- steps were set as [ 63 1 8 1 26 1 ] but should have been [ 53 3 10 1 29 4 ] at ../plugins/uefi-capsule/fu-uefi-capsule-plugin.c:1059

```
</details>

I think it's good now. Thank you for your excellent work.