fwupd/fwupd #6031: Aerox 3 mouse doesn't update version immidiatelly after update/downgrade

d4s, 2023-07-26:
**Describe the bug**
After update or downgrade it takes some time for `fwupd` daemon to re-read a correct version of the mouse after successful update or downgrade. It takes more than 30 sec to re-read version.

**Steps to Reproduce**
```
$ fwupdmgr update 221e7a4c65d6649a574dd6ead4fe436541b20a47 --json && fwupdmgr get-devices
```

**Expected behavior**
Should read proper version just after successful update.

**fwupd version information**

```shell
$ fwupdmgr --version
compile   org.freedesktop.fwupd         1.9.4
compile   com.hughsie.libxmlb           0.3.8
compile   com.hughsie.libjcat           0.1.14
compile   org.freedesktop.gusb          0.4.6
runtime   com.hughsie.libjcat           0.1.14
runtime   com.dell.libsmbios            2.4
runtime   org.freedesktop.gusb          0.4.6
runtime   org.kernel                    6.1.38-un-def-alt1
runtime   org.freedesktop.fwupd         1.9.4
```

Installed from `main` branch.

<details>

<summary>**fwupd device information**</summary>

This output shows correct version of the mouse due some time passed after the downgrade.

```shell
# fwupdmgr get-devices --show-all-devices

Intel(R) Client Systems NUC8i7HVK
│
├─Unknown Device:
│     Device ID:          3f7509968cd38e34ac767b883f6a25edcfa0b6e4
│     GUID:               3159d3db-61b8-52be-a3ec-e9b8eb730e3c ← GPIO\ID_INT345D:00
│   
├─Aerox 3 Wireless USB Receiver:
│ │   Device ID:          5a787a05e004e8597402288d253b959d1cab16a9
│ │   Current version:    1.4.2
│ │   Vendor:             SteelSeries (USB:0x1038)
│ │   Install Duration:   13 seconds
│ │   GUIDs:              df0cb7f0-be88-532f-87f9-65662da376b9
│ │                       291b0582-8d08-5714-88db-986911c744d9 ← USB\VID_1038&PID_1838
│ │   Device Flags:       • Updatable
│ │                       • Supported on remote server
│ │                       • Cryptographic hash verification is available
│ │ 
│ └─Aerox 3 Wireless Mouse via USB Receiver:
│       Device ID:        2213896d4ce7cdfddbed1530a42137173b111360
│       Current version:  1.10.9
│       Vendor:           SteelSeries (USB:0x1038)
│       Install Duration: 38 seconds
│       Update State:     Success
│       Battery:          80% (threshold 20%)
│       Last modified:    2023-07-26 22:58
│       GUID:             541b2713-d367-5240-a443-bed07f09ffbf ← STEELSERIES\VID_1038&PID_1838&PROTOCOL_FIZZ_TUNNEL
│       Device Flags:     • Updatable
│                         • Supported on remote server
│                         • Cryptographic hash verification is available
│     
├─Core™ i7-8809G CPU @ 3.10GHz:
│     Device ID:          4bde70ba4e39b28f9eab1628f9dd6e6244c03027
│     Current version:    0x000000f0
[... 150 lines cut ...]
      GUIDs:              69749ddc-cd7a-542e-981f-620656c59918 ← NVME\VEN_15B7&DEV_5030
                          53d3c0f5-42b2-5cc9-8a52-fa350528dc02 ← NVME\VEN_15B7&DEV_5030&SUBSYS_15B75030
                          c6686f27-9161-5f60-a26e-12268be7d913 ← WD_BLACK SN850X 2000GB
      Device Flags:       • Internal device
                          • Updatable
                          • System requires external power source
                          • Needs a reboot after installation
                          • Device is usable for the duration of the update
    
────────────────────────────────────────────────
Devices that have been updated successfully:
 • EPOS ADAPT 1x5 (8.41.34.0 → 8.41.38.0)
 • Aerox 3 Wireless Mouse via USB Receiver (1.11.4 → 1.10.9)
```

</details>

**Additional questions**

- Operating system and version:
  - Chrome OS several versions, ALT Linux
- Have you tried rebooting?
  -  Yes
- Is this a regression?
  -  No

hughsie, 2023-07-27:
When fwupd does ->reload() to get the new version does the mouse return with the old data? Is it the mouse that takes some time to realize the new version, or are we not waiting for the replug correctly?

A `fwupdtool -vv install-blob` output would be super-useful too.

d4s, 2023-07-27:
@hughsie I'll check the core reason shortly.
That's issue is mostly for me to not forget about the issue and because replug issue is possible as well.