fwupd/fwupd #6032: Failed to load signature: Need more input

shenki, 2023-07-27:
**Describe the bug**
`fwupdmgr refresh` downloads, but then reports `Failed to load signature: Need more input`.

**Steps to Reproduce**
`fwupdmgr refresh`

**fwupd version information**

```shell
compile   org.freedesktop.fwupd         1.9.3
compile   com.hughsie.libxmlb           0.3.10
compile   com.hughsie.libjcat           0.1.9
runtime   org.freedesktop.fwupd-efi     1.4
compile   org.freedesktop.gusb          0.4.5
runtime   com.dell.libsmbios            2.4
runtime   org.freedesktop.gusb          0.4.5
runtime   org.kernel                    6.3.0-1-amd64
runtime   org.freedesktop.fwupd         1.9.3
```

Debian testing, installed from apt. Fresh install, first boot. Had to disable secure boot to install.

<details>

<summary>**fwupd device information**</summary>

Please provide the output of the fwupd devices recognized in your system.

```shell
LENOVO 21CCS29201
│
├─Unknown Device:
│     Device ID:          6b5a06645e27c4e94b6e67d5185c46d11ec9ad67
│     GUID:               c96751f4-16e8-59ea-975d-69d3dac88d7d ← GPIO\ID_INTC1055:00
│   
├─12th Gen Intel Core™ i7-1270P:
│ │   Device ID:          4bde70ba4e39b28f9eab1628f9dd6e6244c03027
│ │   Current version:    0x0000042a
│ │   Vendor:             Intel
│ │   GUIDs:              b9a2dd81-159e-5537-a7db-e7101d164d3f ← cpu
│ │                       30249f37-d140-5d3e-9319-186b1bd5cac3 ← CPUID\PRO_0&FAM_06
│ │                       ab855c04-4ff6-54af-8a8a-d8193daa0cd8 ← CPUID\PRO_0&FAM_06&MOD_9A
│ │                       3ebbde86-d03e-549a-a8fd-02ebf9aa537a ← CPUID\PRO_0&FAM_06&MOD_9A&STP_3
│ │   Device Flags:       • Internal device
│ │ 
│ ├─USB4 host controller:
│ │     Device ID:        40a03cdbee3f958a36cd1f3b6fd9469b52aee4b8
│ │     Summary:          Unmatched performance for high-speed I/O
│ │     Vendor:           INTEL (THUNDERBOLT:0x8087)
│ │     GUIDs:            5041b15d-6e1c-5681-86b3-bb1d56993357 ← THUNDERBOLT\VEN_8087&DEV_463E
│ │                       e72e778e-94f7-5ed2-b560-1c1262ee217c ← TBT-fixed
│ │     Device Flags:     • Internal device
│ │                       • System requires external power source
│ │                       • Signed Payload
│ │   
│ └─USB4 host controller:
│       Device ID:        94aa454d747cacb16437c732155d773e40539e13
│       Summary:          Unmatched performance for high-speed I/O
│       Vendor:           INTEL (THUNDERBOLT:0x8087)
│       GUIDs:            5041b15d-6e1c-5681-86b3-bb1d56993357 ← THUNDERBOLT\VEN_8087&DEV_463E
│                         e72e778e-94f7-5ed2-b560-1c1262ee217c ← TBT-fixed
│       Device Flags:     • Internal device
│                         • System requires external power source
│                         • Signed Payload
│     
├─Alder Lake-P Integrated Graphics Controller:
│     Device ID:          5792b48846ce271fab11c4a545f7a3df0d36e00a
│     Current version:    0c
│     Vendor:             Intel Corporation (PCI:0x8086)
│     GUIDs:              eaad9970-8e4d-56da-88ab-41a8c1e2811f ← PCI\VEN_8086&DEV_46A6
[... 276 lines cut ...]
│     Vendor:             DMI:LENOVO
│     Update State:       Success
│     GUID:               88523ddf-3aac-41f6-b15f-dc7bea434b39
│     Device Flags:       • Internal device
│                         • Updatable
│                         • System requires external power source
│                         • Needs a reboot after installation
│                         • Device is usable for the duration of the update
│   
└─UEFI Device Firmware:
      Device ID:          a083ebc5138e5e071ef7270cc9a8280722cc7adf
      Summary:            UEFI ESRT device
      Current version:    3549562
      Vendor:             DMI:LENOVO
      Update State:       Success
      GUID:               8bb5952d-450d-440b-8e40-026edcb5e638
      Device Flags:       • Internal device
                          • Updatable
                          • System requires external power source
                          • Needs a reboot after installation
                          • Device is usable for the duration of the update

```

</details>

shenki, 2023-07-27:
I was working out why I had to disable secure boot, and it turns out Lenovo disables the Microsoft 3rd Party UEFI CA by default. Enabling that allowed the `refresh` operation to complete.

If that was really the cause, perhaps this issue could be about modifying the error message to indicate the signature failed due to UEFI CA issues?

hughsie, 2023-07-27:
I don't *think* disabling secure boot should affect anything like this -- can you try turning off secure boot again and doing `fwupdmgr refresh --force` again? It might have just been a transient network issue.

cwrau, 2023-08-02:
I have the same problem, my secure boot is disabled and neither `sudo fwupdmgr refresh` nor `sudo fwupdmgr refresh --force` are working and both are failing with the following error;

```
Updating lvfs
Downloading…             [  -                                    ]
Failed to load signature: Need more input
```

jknaus, 2023-08-02:
Same here:
```sh
sudo fwupdmgr refresh --force
```
```
Updating lvfs
Downloading…             [        |                              ]
Failed to load signature: Need more input
```

ousatov-ua, 2023-08-02:
Same here, Fedora 38. Appeared today.

`fwupdmgr refresh --force 
Updating lvfs
Downloading…             [     \                                 ]
Failed to load signature: Need more input
`

ousatov-ua, 2023-08-02:
Basically, this happens on enabling remote too:

`fwupdmgr enable-remote lvfs`
...

`Agree and enable the remote? [Y|n]: Y
Authenticating…          [                                    \  ]
Do you want to refresh this remote now? (Requires internet connection) [Y|n]: y
Downloading…             [                                   -   ]
Failed to load signature: Need more input`

artemkovalyov, 2023-08-02:
I've the same issue on Arch

GodAndEmperor, 2023-08-02:
Lenovo P50, Fedora 38:
same issue as described by others above. After enabling CMS, the Download progress indicator looks a bit different, but issue remains:

fwupdmgr refresh --force
Updating lvfs
Downloading…             [*******************                    ]
Failed to load signature: Need more input

hughsie, 2023-08-02:
This error is when we try to parse the signature in libjcat from the thing the CDN delivered. I'm on holiday on a little island with not much internet, so this will need someone else to debug this. I just refreshed just fine on my f38 machine I updated a few days ago, do it's not *everyone* affected. Did any recent update start causing this perhaps? Is it a transient issue?

hughsie, 2023-08-02:
For instance, the -vv output is going to be super useful, as is downgrading fwupd, libjcat, glib2, libcurl etc

ousatov-ua, 2023-08-02:
Hi @hughsie !

> fwupdmgr refresh --force -vv 
> (fwupdmgr:15194): GLib-DEBUG: 18:44:15.298: setenv()/putenv() are not thread-safe and should not be used after threads are created
> (pkttyagent:15196): GLib-GIO-DEBUG: 18:44:15.300: Using cross-namespace EXTERNAL authentication (this will deadlock if server is GDBus < 2.73.3)
> (fwupdmgr:15194): GLib-GIO-DEBUG: 18:44:15.305: _g_io_module_get_default: Found default implementation dconf (DConfSettingsBackend) for ‘gsettings-backend’
> (fwupdmgr:15194): dconf-DEBUG: 18:44:15.305: watch_fast: "/system/proxy/" (establishing: 0, active: 0)
> (fwupdmgr:15194): dconf-DEBUG: 18:44:15.305: watch_fast: "/system/proxy/http/" (establishing: 0, active: 0)
> (fwupdmgr:15194): dconf-DEBUG: 18:44:15.305: watch_fast: "/system/proxy/https/" (establishing: 0, active: 0)
> (fwupdmgr:15194): dconf-DEBUG: 18:44:15.305: watch_fast: "/system/proxy/ftp/" (establishing: 0, active: 0)
> (fwupdmgr:15194): dconf-DEBUG: 18:44:15.305: watch_fast: "/system/proxy/socks/" (establishing: 0, active: 0)
> (fwupdmgr:15194): dconf-DEBUG: 18:44:15.305: unwatch_fast: "/system/proxy/" (active: 0, establishing: 1)
> (fwupdmgr:15194): dconf-DEBUG: 18:44:15.305: unwatch_fast: "/system/proxy/http/" (active: 0, establishing: 1)
> (fwupdmgr:15194): dconf-DEBUG: 18:44:15.305: unwatch_fast: "/system/proxy/https/" (active: 0, establishing: 1)
> (fwupdmgr:15194): dconf-DEBUG: 18:44:15.305: unwatch_fast: "/system/proxy/ftp/" (active: 0, establishing: 1)
> (fwupdmgr:15194): dconf-DEBUG: 18:44:15.305: unwatch_fast: "/system/proxy/socks/" (active: 0, establishing: 1)
> (fwupdmgr:15194): GLib-GIO-DEBUG: 18:44:15.305: Using cross-namespace EXTERNAL authentication (this will deadlock if server is GDBus < 2.73.3)
> (fwupdmgr:15194): GLib-GIO-DEBUG: 18:44:15.306: Using cross-namespace EXTERNAL authentication (this will deadlock if server is GDBus < 2.73.3)
> (fwupdmgr:15194): GLib-GIO-DEBUG: 18:44:15.306: _g_io_module_get_default: Found default implementation libproxy (GLibproxyResolver) for ‘gio-proxy-resolver’
> (fwupdmgr:15194): dconf-DEBUG: 18:44:15.307: watch_established: "/system/proxy/" (establishing: 0)
> (fwupdmgr:15194): dconf-DEBUG: 18:44:15.307: watch_established: "/system/proxy/http/" (establishing: 0)
> (fwupdmgr:15194): dconf-DEBUG: 18:44:15.307: watch_established: "/system/proxy/https/" (establishing: 0)
> (fwupdmgr:15194): dconf-DEBUG: 18:44:15.307: watch_established: "/system/proxy/ftp/" (establishing: 0)
> (fwupdmgr:15194): dconf-DEBUG: 18:44:15.307: watch_established: "/system/proxy/socks/" (establishing: 0)
> (fwupdmgr:15194): Fwupd-DEBUG: 18:44:15.307: Emitting ::status-changed() [idle]
> Updating lvfs
> Fwupd-INFO: 18:44:15.310: downloading https://cdn.fwupd.org/downloads/firmware.xml.xz.jcat
> (fwupdmgr:15194): Fwupd-DEBUG: 18:44:15.310: Emitting ::status-changed() [downloading]
> Downloading…             [      \                                ]Fwupd-INFO: 18:44:15.512: download progress: 100%
> (fwupdmgr:15194): Fwupd-DEBUG: 18:44:15.512: Emitting ::status-changed() [idle]
> Fwupd-INFO: 18:44:15.512: status-code was 200
> 
> Failed to load signature: Need more input
>

hughsie, 2023-08-02:
And does the jcat cli load the file?

ousatov-ua, 2023-08-02:
@hughsie I removed cdn. from config and it worked!

ousatov-ua, 2023-08-02:
@hughsie Cannot attach that .jcat - github does not support it
[archive.tar.gz](https://github.com/fwupd/fwupd/files/12242239/archive.tar.gz)

hughsie, 2023-08-02:
Can you compare what the CDN sent and what the main server sent please?

ousatov-ua, 2023-08-02:
@hughsie 
It is strange, but their content is identical

ousatov-ua, 2023-08-02:
So maybe the problem with archive itself

ousatov-ua, 2023-08-02:
[Archive.zip](https://github.com/fwupd/fwupd/files/12242731/Archive.zip)

ousatov-ua, 2023-08-02:
@hughsie Seems like the issue is resolved.  I can run 
`fwupdmgr refresh --force`

for cdn

GodAndEmperor, 2023-08-02:
I also don't get an error anymore:
fwupdmgr refresh --force
Updating lvfs
Downloading…             [   -                                   ]
Successfully downloaded new metadata: 2 local devices supported

cwrau, 2023-08-02:
I also don't get the error anymore when executing it by hand, but the systemd service still fails, but without any error message, just the following;

```log
2023-08-02T21:44:16.879379+0200 steve systemd[1]: Starting Refresh fwupd metadata and update motd...
2023-08-02T21:44:16.902677+0200 steve systemd[1]: fwupd-refresh.service: Main process exited, code=exited, status=1/FAILURE
2023-08-02T21:44:16.902802+0200 steve systemd[1]: fwupd-refresh.service: Failed with result 'exit-code'.
2023-08-02T21:44:16.916016+0200 steve systemd[1]: Failed to start Refresh fwupd metadata and update motd.
``'

cwrau, 2023-08-03:
> So maybe the CDN went crazy? For the systemd service, can you add -v and see any more debug lines?

Sure;

```log
2023-08-03T13:28:13.325515+0200 steve systemd[1]: Starting Refresh fwupd metadata and update motd...
2023-08-03T13:28:13.341499+0200 steve fwupdmgr[2999241]: (/usr/bin/fwupdmgr:2999241): GLib-DEBUG: 13:28:13.341: setenv()/putenv() are not thread-safe and should not be used after threads are created
2023-08-03T13:28:13.343537+0200 steve fwupdmgr[2999241]: (/usr/bin/fwupdmgr:2999241): GLib-GIO-DEBUG: 13:28:13.343: _g_io_module_get_default: Found default implementation dconf (DConfSettingsBackend) for ‘gsettings-backend’
2023-08-03T13:28:13.343567+0200 steve fwupdmgr[2999241]: (/usr/bin/fwupdmgr:2999241): dconf-DEBUG: 13:28:13.343: watch_fast: "/system/proxy/" (establishing: 0, active: 0)
2023-08-03T13:28:13.343711+0200 steve fwupdmgr[2999241]: (/usr/bin/fwupdmgr:2999241): dconf-DEBUG: 13:28:13.343: watch_fast: "/system/proxy/http/" (establishing: 0, active: 0)
2023-08-03T13:28:13.343756+0200 steve fwupdmgr[2999241]: (/usr/bin/fwupdmgr:2999241): dconf-DEBUG: 13:28:13.343: watch_fast: "/system/proxy/https/" (establishing: 0, active: 0)
2023-08-03T13:28:13.343786+0200 steve fwupdmgr[2999241]: (/usr/bin/fwupdmgr:2999241): dconf-DEBUG: 13:28:13.343: watch_fast: "/system/proxy/ftp/" (establishing: 0, active: 0)
2023-08-03T13:28:13.343808+0200 steve fwupdmgr[2999241]: (/usr/bin/fwupdmgr:2999241): dconf-DEBUG: 13:28:13.343: watch_fast: "/system/proxy/socks/" (establishing: 0, active: 0)
2023-08-03T13:28:13.343840+0200 steve fwupdmgr[2999241]: (/usr/bin/fwupdmgr:2999241): dconf-DEBUG: 13:28:13.343: unwatch_fast: "/system/proxy/" (active: 0, establishing: 1)
2023-08-03T13:28:13.343840+0200 steve fwupdmgr[2999241]: (/usr/bin/fwupdmgr:2999241): dconf-DEBUG: 13:28:13.343: unwatch_fast: "/system/proxy/http/" (active: 0, establishing: 1)
2023-08-03T13:28:13.343921+0200 steve fwupdmgr[2999241]: (/usr/bin/fwupdmgr:2999241): dconf-DEBUG: 13:28:13.343: unwatch_fast: "/system/proxy/https/" (active: 0, establishing: 1)
2023-08-03T13:28:13.343921+0200 steve fwupdmgr[2999241]: (/usr/bin/fwupdmgr:2999241): dconf-DEBUG: 13:28:13.343: unwatch_fast: "/system/proxy/ftp/" (active: 0, establishing: 1)
2023-08-03T13:28:13.343921+0200 steve fwupdmgr[2999241]: (/usr/bin/fwupdmgr:2999241): dconf-DEBUG: 13:28:13.343: unwatch_fast: "/system/proxy/socks/" (active: 0, establishing: 1)
2023-08-03T13:28:13.344117+0200 steve fwupdmgr[2999241]: (/usr/bin/fwupdmgr:2999241): dconf-DEBUG: 13:28:13.344: watch_established: "/system/proxy/" (establishing: 0)
2023-08-03T13:28:13.344152+0200 steve fwupdmgr[2999241]: (/usr/bin/fwupdmgr:2999241): dconf-DEBUG: 13:28:13.344: watch_established: "/system/proxy/http/" (establishing: 0)
2023-08-03T13:28:13.344192+0200 steve fwupdmgr[2999241]: (/usr/bin/fwupdmgr:2999241): dconf-DEBUG: 13:28:13.344: watch_established: "/system/proxy/https/" (establishing: 0)
2023-08-03T13:28:13.344223+0200 steve fwupdmgr[2999241]: (/usr/bin/fwupdmgr:2999241): dconf-DEBUG: 13:28:13.344: watch_established: "/system/proxy/ftp/" (establishing: 0)
2023-08-03T13:28:13.344257+0200 steve fwupdmgr[2999241]: (/usr/bin/fwupdmgr:2999241): dconf-DEBUG: 13:28:13.344: watch_established: "/system/proxy/socks/" (establishing: 0)
2023-08-03T13:28:13.344993+0200 steve fwupdmgr[2999241]: (/usr/bin/fwupdmgr:2999241): GLib-GIO-DEBUG: 13:28:13.344: _g_io_module_get_default: Found default implementation local (GLocalVfs) for ‘gio-vfs’
2023-08-03T13:28:13.345027+0200 steve fwupdmgr[2999241]: (/usr/bin/fwupdmgr:2999241): pxbackend-DEBUG: 13:28:13.345: px_config_sysconfig_set_config_file: Could not read file /etc/sysconfig/proxy
2023-08-03T13:28:13.345027+0200 steve fwupdmgr[2999241]: (/usr/bin/fwupdmgr:2999241): pxbackend-DEBUG: 13:28:13.345: Active config plugins:
2023-08-03T13:28:13.345027+0200 steve fwupdmgr[2999241]: (/usr/bin/fwupdmgr:2999241): pxbackend-DEBUG: 13:28:13.345:  - config-env
2023-08-03T13:28:13.345027+0200 steve fwupdmgr[2999241]: (/usr/bin/fwupdmgr:2999241): pxbackend-DEBUG: 13:28:13.345:  - config-kde
2023-08-03T13:28:13.345027+0200 steve fwupdmgr[2999241]: (/usr/bin/fwupdmgr:2999241): pxbackend-DEBUG: 13:28:13.345:  - config-gnome
2023-08-03T13:28:13.345091+0200 steve fwupdmgr[2999241]: (/usr/bin/fwupdmgr:2999241): pxbackend-DEBUG: 13:28:13.345:  - config-sysconfig
2023-08-03T13:28:13.346094+0200 steve fwupdmgr[2999241]: (/usr/bin/fwupdmgr:2999241): GLib-GIO-DEBUG: 13:28:13.346: Failed to initialize portal (GNetworkMonitorPortal) for gio-network-monitor: Not using portals
2023-08-03T13:28:13.346448+0200 steve fwupdmgr[2999241]: (/usr/bin/fwupdmgr:2999241): GLib-GIO-DEBUG: 13:28:13.346: Using cross-namespace EXTERNAL authentication (this will deadlock if server is GDBus < 2.73.3)
2023-08-03T13:28:13.346818+0200 steve fwupdmgr[2999241]: (/usr/bin/fwupdmgr:2999241): GLib-GIO-DEBUG: 13:28:13.346: Failed to initialize networkmanager (GNetworkMonitorNM) for gio-network-monitor: The connection is closed
2023-08-03T13:28:13.346870+0200 steve fwupdmgr[2999241]: (/usr/bin/fwupdmgr:2999241): GLib-GIO-DEBUG: 13:28:13.346: _g_io_module_get_default: Found default implementation netlink (GNetworkMonitorNetlink) for ‘gio-network-monitor’
2023-08-03T13:28:13.346915+0200 steve fwupdmgr[2999241]: (/usr/bin/fwupdmgr:2999241): pxbackend-DEBUG: 13:28:13.346: px_manager_on_network_changed: Network connection changed, clearing pac data
2023-08-03T13:28:13.346915+0200 steve fwupdmgr[2999241]: (/usr/bin/fwupdmgr:2999241): pxbackend-DEBUG: 13:28:13.346: px_manager_constructed: Up and running
2023-08-03T13:28:13.346915+0200 steve fwupdmgr[2999241]: (/usr/bin/fwupdmgr:2999241): GLib-GIO-DEBUG: 13:28:13.346: _g_io_module_get_default: Found default implementation libproxy (GLibproxyResolver) for ‘gio-proxy-resolver’
2023-08-03T13:28:13.348511+0200 steve systemd[1]: fwupd-refresh.service: Main process exited, code=exited, status=1/FAILURE
```

operutka, 2023-08-08:
Same problem here. Downloading `firmware.xml.xz.jcat` from the main server instead of CDN also works for me.

I compared `firmware.xml.xz.jcat` from the main server with the one I get from CDN. Both of them are gzipped JSONs. After decompressing the one from the main server, I get the following JSON:
```json
{"JcatVersionMajor": 0, "JcatVersionMinor": 1, "Items": [{"Id": "firmware-06672-stable.xml.xz", "AliasIds": ["firmware.xml.xz"], "Blobs": [{"Kind": 4, "Flags": 1, "Timestamp": 1691483156, "Data": "eacfbe188f1e337c983ed9c262e9b15a24182aac"}, {"Kind": 1, "Flags": 1, "Timestamp": 1691483156, "Data": "29ec1995117637f7c336064170dc911c22312b83554b13e1c694de0581320e14"}, {"Kind": 2, "Flags": 1, "Timestamp": 1691483162, "Data": "-----BEGIN PGP SIGNATURE-----\n\niQEzBAABCAAdFiEEP8a4BEEO0IQNjy+XSKbYDkU4usIFAmTR/BoACgkQSKbYDkU4\nusIqFgf/cm4MpOvBq3nOHQIfjJ8/SEQzf1onvVWAD231m4oVE3xgWMBWzc4uQRvt\nCJajR1j6OoHPezb3aZiXOUeTAv7adQZO7Hg4OK6kGIZl1hgL2kYnwbDldccVwUPo\n1I0pq3naoGWxgN+Z0i5IkdC2ul4KR9fnar/vzzh4vRimV9WAJJtM9vZdUM0/+miT\nVubLNbEWrIIJcCyIbdVpMWkyGkSRllxjmPGZpwd/73vn6AKI+oumlXsOQzA1Uf8W\nJCKeOCjC7/B53bX8dNCf9ThgWa3wfEVvVJSCxkq5gkhSPTP85QUA09xPcOZXao2A\n9nA7gf8jTFNEqhTde+VLrQq463TkcQ==\n=BnoP\n-----END PGP SIGNATURE-----\n"}, {"Kind": 3, "Flags": 1, "Timestamp": 1691483162, "Data": "-----BEGIN PKCS7-----\nMIIGUgYJKoZIhvcNAQcCoIIGQzCCBj8CAQExDTALBglghkgBZQMEAgEwCwYJKoZI\nhvcNAQcBoIIEOjCCBDYwggKeoAMCAQICDFprhisibP88kP07YjANBgkqhkiG9w0B\nAQsFADA6MRAwDgYDVQQDEwdMVkZTIENBMSYwJAYDVQQKEx1MaW51eCBWZW5kb3Ig\nRmlybXdhcmUgUHJvamVjdDAeFw0xODAxMTYwMDAwMDBaFw0yODAxMTYwMDAwMDBa\nMBkxFzAVBgNVBAMTDlJpY2hhcmQgSHVnaGVzMIIBIjANBgkqhkiG9w0BAQEFAAOC\nAQ8AMIIBCgKCAQEArR5nseG3+Zs+o41P9LTspiVSGVcp6ifNHhbKKxBvZZXsy0gX\n+P/VRlsHLiKQrFulQ8GbPODytFv0o+y/0MkiJxv/fY3yEZ2bwNpsSeXFQSGHI6yz\njaVNeNCu8lnnDtD7kiC8UNUNHcnA4+2h/Yv4k+KPqYF+Qb6nWAEIID1ObMnjeJUb\ndbPPvy12aasV3gZcZ+goYNFc0ua6OU/CNEuzAVVCTAJ/EpCdGpll1+6BGU5ImIUG\nTlMTWq2xmCfCPugakHrmWA66yHWwE6LmC/U7qQDWFemsSNnmzyBB9HPkqsW1DjHr\n+ZmNUPj3+q2UGnNwP/Cne462XbsZB569w7pnzQIDAQABo4HcMIHZMAwGA1UdEwEB\n/wQCMAAwNwYDVR0RBDAwLoYXaHR0cHM6Ly9md3VwZC5vcmcvbHZmcy+BE3JpY2hh\ncmRAaHVnaHNpZS5jb20wEwYDVR0lBAwwCgYIKwYBBQUHAwMwDwYDVR0PAQH/BAUD\nAweAADAdBgNVHQ4EFgQUMcrnDWk3DM1GJCqJK7EbAycgB+UwHwYDVR0jBBgwFoAU\nsY3q5COnfgmOte4x4GrdnjQ3ZawwKgYDVR0fBCMwITAfoB2gG4YZaHR0cDovL3d3\ndy5md3VwZC5vcmcvcGtpLzANBgkqhkiG9w0BAQsFAAOCAYEAjoXBOtVb1qPLuE7I\nShdSkk66JMNmzZnODbun3BaViUF3PPhuRiJ2y7Bu7loCnxHwzKpq5Hn4Untg1WhQ\nSsPEv6H2oc2E4HU1Gds2HEE1UL0VzulPEwaOFX0OZ37LmJ2VFvfgwfNmjAWZ+ngN\nsjuff3PWeHNsOwbS3To0CBlHqz9LlcA7Gxpxbz4DbUEvgTTKZFbasgUS6/QmbpPy\n880ThgD93yCg6Q002AooPMw8zW6gGUP7/2D4aKBjgM/IJDrJEicTdKDFtUacsN+d\nCfIEcn6F1rzPgwo0B/yLYVMpbL8WzxSS7cX5dBlkreH6Nf7g+A2NTryU/6FxoFxo\nnWJmLssNQCgx42Ywc8CgJRt1yWXxoZXOPap8zPvFAAmrfUlqfZlr2Wzcf2nxpSZJ\noQRzGDXDRiYNt28xjX408cJ+Wy+0oWwgnXCkfzMDnJQB8MX6+5Ah4v4mS/XjmOIi\nQ+hKCT99Yf0Eq3TWzFhyYducqLU904FbrrUSoi07DfxXa/8RMYIB3jCCAdoCAQEw\nSjA6MRAwDgYDVQQDEwdMVkZTIENBMSYwJAYDVQQKEx1MaW51eCBWZW5kb3IgRmly\nbXdhcmUgUHJvamVjdAIMWmuGKyJs/zyQ/TtiMAsGCWCGSAFlAwQCAaBpMBgGCSqG\nSIb3DQEJAzELBgkqhkiG9w0BBwEwHAYJKoZIhvcNAQkFMQ8XDTIzMDgwODA4MjYw\nMlowLwYJKoZIhvcNAQkEMSIEICnsGZURdjf3wzYGQXDckRwiMSuDVUsT4caU3gWB\nMg4UMA0GCSqGSIb3DQEBAQUABIIBABng4wodCLYNGreeKti2m4cG2JahoNDDrrv2\nLRMFPpkHvOmlPdO2GW00+xsD+EYbv1Xu3X6OD/AOpQwkupugA4kaLh4bE3iSvR3R\ncLi6bpMcA3icQpQXcfNtgQMUMkbzx+EkWmKGqRKlVSOn/rGd/031lSriev0P7DUQ\n4UEwoXai8iC/KDQsd2i4/GEBeUVtOzba1SH2q5uyBduSE0e5N6IxNISL7EywAb0K\n2UCS6E92xd0WyDQgV9QMzxfaqcqYKdnWtHJUgpHsbX1jXzmpmxER8WZrnHWmsyn2\nLU6BBtTOEzbP4gqUkCxqXh5Z0mfWyEDKYGdzMNi0fk25dFvvwVk=\n-----END PKCS7-----\n"}]}]}
```

However, decompressing the one I get from CDN fails:
```text
# gzip -d firmware.xml.xz.jcat.gz 

gzip: firmware.xml.xz.jcat.gz: unexpected end of file
```

I also compared the gzipped binaries right after the download (i.e. without decompressing them) and it appears that the CDN truncates the original file. **The file downloaded from CDN is four bytes shorter and it is a prefix of the original file** from the main server (i.e. if I remove the last four bytes from the file downloaded from the main server, it matches exactly to the file downloaded from CDN).

hughsie, 2023-08-08:
What's the file size in bytes of the truncated version out of interest?

vajdao, 2023-08-08:
I was facing this issue on Debian 12.1 Bookworm aswell, when I opened up "Discover" software centre. An error message popped up: "Failed to load signature: need more input". 

Now, what I tried based on the previous comments are and every single attemts are failure: 

Updating my BIOS/UEFI to the latest version: **negative**
Enabling secure boot: **negative**
Disabling secure boot: **negative**
sudo fwupdmgr refresh and sudo fwupdmgr refresh --force: **negative** & **negative**
Adding "invalid;bios" to the /etc/fwupd/daemon.conf's 'DisabledPlugins' line then doing a restart: **negative**

My UEFI menu doesn't have anything such as enable/disable capsule UEFI update or such, BUT all I know that my machine was perfectly working as intended, everything was normal, until my distro received a big update, and after that, I'm receiving these error messages are bombarding me, and it seems everyone else. I'm on Debian stable for a reason, for not to receive such bugs, but it seems, not even that is a warranty...

I can't seem to find any fix on this, and based on the comments, for now we are groping in the darkness.

Lavode, 2023-08-08:
> What's the file size in bytes of the truncated version out of interest?

```
› du -b *.jcat
2223	cdn.xml.xz.jcat
2227	main.xml.xz.jcat
```
I.e. they differ by four bytes. Specifically the last four bytes are missing on the CDN:
```
› diff <(xxd cdn.xml.xz.jcat) <(xxd main.xml.xz.jcat)
139c139,140
< 000008a0: d20f 2f33 5fff fedb fefd 07cd ccb9 da    ../3_..........
---
> 000008a0: d20f 2f33 5fff fedb fefd 07cd ccb9 dac4  ../3_...........
> 000008b0: 0c00 00                                  ...
```

- `cdn.xml.xz.jcat` is the one at https://cdn.fwupd.org/downloads/firmware.xml.xz.jcat
- `main.xml.xz.jcat` is the one at https://fwupd.org/downloads/firmware.xml.xz.jcat

PocketSam, 2023-08-17:
Problem still occurs from time to time.

hughsie, 2023-08-17:
@PocketSam if you're comfortable, can you share where in the world you are? I'm trying to reproduce with a VPN. I think the whole CDN hostname might be helpful too.

cwrau, 2023-08-17:
Still happens to me as well;

```log
2023-08-17T15:29:19.197599+0200 steve systemd[1]: Starting Refresh fwupd metadata and update motd...
2023-08-17T15:29:19.254006+0200 steve fwupdmgr[3383222]: (/usr/bin/fwupdmgr:3383222): GLib-DEBUG: 15:29:19.253: setenv()/putenv() are not thread-safe and should not be used after threads are created
2023-08-17T15:29:19.262094+0200 steve fwupdmgr[3383222]: (/usr/bin/fwupdmgr:3383222): GLib-GIO-DEBUG: 15:29:19.262: _g_io_module_get_default: Found default implementation dconf (DConfSettingsBackend) for ‘gsettings-backend’
2023-08-17T15:29:19.262094+0200 steve fwupdmgr[3383222]: (/usr/bin/fwupdmgr:3383222): dconf-DEBUG: 15:29:19.262: watch_fast: "/system/proxy/" (establishing: 0, active: 0)
2023-08-17T15:29:19.262315+0200 steve fwupdmgr[3383222]: (/usr/bin/fwupdmgr:3383222): dconf-DEBUG: 15:29:19.262: watch_fast: "/system/proxy/http/" (establishing: 0, active: 0)
2023-08-17T15:29:19.264903+0200 steve fwupdmgr[3383222]: (/usr/bin/fwupdmgr:3383222): dconf-DEBUG: 15:29:19.262: watch_fast: "/system/proxy/https/" (establishing: 0, active: 0)
2023-08-17T15:29:19.264903+0200 steve fwupdmgr[3383222]: (/usr/bin/fwupdmgr:3383222): dconf-DEBUG: 15:29:19.262: watch_fast: "/system/proxy/ftp/" (establishing: 0, active: 0)
2023-08-17T15:29:19.264903+0200 steve fwupdmgr[3383222]: (/usr/bin/fwupdmgr:3383222): dconf-DEBUG: 15:29:19.262: watch_fast: "/system/proxy/socks/" (establishing: 0, active: 0)
2023-08-17T15:29:19.264903+0200 steve fwupdmgr[3383222]: (/usr/bin/fwupdmgr:3383222): dconf-DEBUG: 15:29:19.262: unwatch_fast: "/system/proxy/" (active: 0, establishing: 1)
2023-08-17T15:29:19.264903+0200 steve fwupdmgr[3383222]: (/usr/bin/fwupdmgr:3383222): dconf-DEBUG: 15:29:19.262: unwatch_fast: "/system/proxy/http/" (active: 0, establishing: 1)
2023-08-17T15:29:19.264903+0200 steve fwupdmgr[3383222]: (/usr/bin/fwupdmgr:3383222): dconf-DEBUG: 15:29:19.262: unwatch_fast: "/system/proxy/https/" (active: 0, establishing: 1)
2023-08-17T15:29:19.264903+0200 steve fwupdmgr[3383222]: (/usr/bin/fwupdmgr:3383222): dconf-DEBUG: 15:29:19.262: watch_established: "/system/proxy/" (establishing: 0)
2023-08-17T15:29:19.264903+0200 steve fwupdmgr[3383222]: (/usr/bin/fwupdmgr:3383222): dconf-DEBUG: 15:29:19.262: unwatch_fast: "/system/proxy/ftp/" (active: 0, establishing: 1)
2023-08-17T15:29:19.264903+0200 steve fwupdmgr[3383222]: (/usr/bin/fwupdmgr:3383222): dconf-DEBUG: 15:29:19.262: unwatch_fast: "/system/proxy/socks/" (active: 0, establishing: 1)
2023-08-17T15:29:19.264903+0200 steve fwupdmgr[3383222]: (/usr/bin/fwupdmgr:3383222): dconf-DEBUG: 15:29:19.262: watch_established: "/system/proxy/http/" (establishing: 0)
2023-08-17T15:29:19.264903+0200 steve fwupdmgr[3383222]: (/usr/bin/fwupdmgr:3383222): dconf-DEBUG: 15:29:19.262: watch_established: "/system/proxy/https/" (establishing: 0)
2023-08-17T15:29:19.264903+0200 steve fwupdmgr[3383222]: (/usr/bin/fwupdmgr:3383222): dconf-DEBUG: 15:29:19.262: watch_established: "/system/proxy/ftp/" (establishing: 0)
2023-08-17T15:29:19.264903+0200 steve fwupdmgr[3383222]: (/usr/bin/fwupdmgr:3383222): dconf-DEBUG: 15:29:19.262: watch_established: "/system/proxy/socks/" (establishing: 0)
2023-08-17T15:29:19.266875+0200 steve fwupdmgr[3383222]: (/usr/bin/fwupdmgr:3383222): GLib-GIO-DEBUG: 15:29:19.266: _g_io_module_get_default: Found default implementation local (GLocalVfs) for ‘gio-vfs’
2023-08-17T15:29:19.266914+0200 steve fwupdmgr[3383222]: (/usr/bin/fwupdmgr:3383222): pxbackend-DEBUG: 15:29:19.266: px_config_sysconfig_set_config_file: Could not read file /etc/sysconfig/proxy
2023-08-17T15:29:19.266914+0200 steve fwupdmgr[3383222]: (/usr/bin/fwupdmgr:3383222): pxbackend-DEBUG: 15:29:19.266: Active config plugins:
2023-08-17T15:29:19.266914+0200 steve fwupdmgr[3383222]: (/usr/bin/fwupdmgr:3383222): pxbackend-DEBUG: 15:29:19.266:  - config-env
2023-08-17T15:29:19.266980+0200 steve fwupdmgr[3383222]: (/usr/bin/fwupdmgr:3383222): pxbackend-DEBUG: 15:29:19.266:  - config-kde
2023-08-17T15:29:19.266980+0200 steve fwupdmgr[3383222]: (/usr/bin/fwupdmgr:3383222): pxbackend-DEBUG: 15:29:19.266:  - config-gnome
2023-08-17T15:29:19.266980+0200 steve fwupdmgr[3383222]: (/usr/bin/fwupdmgr:3383222): pxbackend-DEBUG: 15:29:19.266:  - config-sysconfig
2023-08-17T15:29:19.267756+0200 steve fwupdmgr[3383222]: (/usr/bin/fwupdmgr:3383222): GLib-GIO-DEBUG: 15:29:19.267: Failed to initialize portal (GNetworkMonitorPortal) for gio-network-monitor: Not using portals
2023-08-17T15:29:19.268057+0200 steve fwupdmgr[3383222]: (/usr/bin/fwupdmgr:3383222): GLib-GIO-DEBUG: 15:29:19.268: Using cross-namespace EXTERNAL authentication (this will deadlock if server is GDBus < 2.73.3)
2023-08-17T15:29:19.268515+0200 steve fwupdmgr[3383222]: (/usr/bin/fwupdmgr:3383222): GLib-GIO-DEBUG: 15:29:19.268: Failed to initialize networkmanager (GNetworkMonitorNM) for gio-network-monitor: The connection is closed
2023-08-17T15:29:19.268566+0200 steve fwupdmgr[3383222]: (/usr/bin/fwupdmgr:3383222): GLib-GIO-DEBUG: 15:29:19.268: _g_io_module_get_default: Found default implementation netlink (GNetworkMonitorNetlink) for ‘gio-network-monitor’
2023-08-17T15:29:19.268583+0200 steve fwupdmgr[3383222]: (/usr/bin/fwupdmgr:3383222): pxbackend-DEBUG: 15:29:19.268: px_manager_on_network_changed: Network connection changed, clearing pac data
2023-08-17T15:29:19.268583+0200 steve fwupdmgr[3383222]: (/usr/bin/fwupdmgr:3383222): pxbackend-DEBUG: 15:29:19.268: px_manager_constructed: Up and running
[... cut ...] und default implementation libproxy (GLibproxyResolver) for ‘gio-proxy-resolver’
2023-08-17T15:29:19.272375+0200 steve systemd[1]: fwupd-refresh.service: Main process exited, code=exited, status=1/FAILURE
2023-08-17T15:29:19.272629+0200 steve systemd[1]: fwupd-refresh.service: Failed with result 'exit-code'.
2023-08-17T15:29:19.284053+0200 steve systemd[1]: Failed to start Refresh fwupd metadata and update motd.
```

@hughsie I'm in Bielefeld, Germany. How do I find out what the CDN hostname is?

hughsie, 2023-08-17:
`curl -v https://cdn.fwupd.org/downloads/firmware.xml.xz.jcat` -- then look at x-served-by:

cwrau, 2023-08-17:
> `curl -v https://cdn.fwupd.org/downloads/firmware.xml.xz.jcat` -- then look at x-served-by:

```
server: gunicorn
x-served-by: cache-fra-eddf8230121-FRA
```

PocketSam, 2023-08-17:
> @PocketSam if you're comfortable, can you share where in the world you are? I'm trying to reproduce with a VPN. I think the whole CDN hostname might be helpful too.

Hi, @hughsie. I'm in Germany, Berlin using o2 mobile connection. External Operator IP 89.204.137.118

PocketSam, 2023-08-17:
> `curl -v https://cdn.fwupd.org/downloads/firmware.xml.xz.jcat` -- then look at x-served-by:

server: gunicorn
x-served-by: cache-fra-etou8220050-FRA

senden9, 2023-08-18:
Same for me. I am in Austria. Debug output:
```
➜  ~ fwupdmgr refresh -vv --force
(fwupdmgr:23155): GLib-DEBUG: 15:37:59.268: setenv()/putenv() are not thread-safe and should not be used after threads are created
(process:23157): GLib-GIO-DEBUG: 15:37:59.271: Using cross-namespace EXTERNAL authentication (this will deadlock if server is GDBus < 2.73.3)
(fwupdmgr:23155): GLib-GIO-DEBUG: 15:37:59.274: _g_io_module_get_default: Found default implementation dconf (DConfSettingsBackend) for ‘gsettings-backend’
(fwupdmgr:23155): dconf-DEBUG: 15:37:59.274: watch_fast: "/system/proxy/" (establishing: 0, active: 0)
(fwupdmgr:23155): dconf-DEBUG: 15:37:59.274: watch_fast: "/system/proxy/http/" (establishing: 0, active: 0)
(fwupdmgr:23155): dconf-DEBUG: 15:37:59.274: watch_fast: "/system/proxy/https/" (establishing: 0, active: 0)
(fwupdmgr:23155): dconf-DEBUG: 15:37:59.274: watch_fast: "/system/proxy/ftp/" (establishing: 0, active: 0)
(fwupdmgr:23155): dconf-DEBUG: 15:37:59.274: watch_fast: "/system/proxy/socks/" (establishing: 0, active: 0)
(fwupdmgr:23155): GLib-GIO-DEBUG: 15:37:59.274: _g_io_module_get_default: Found default implementation gnome (GProxyResolverGnome) for ‘gio-proxy-resolver’
(fwupdmgr:23155): GLib-GIO-DEBUG: 15:37:59.275: Using cross-namespace EXTERNAL authentication (this will deadlock if server is GDBus < 2.73.3)
(fwupdmgr:23155): GLib-GIO-DEBUG: 15:37:59.275: Using cross-namespace EXTERNAL authentication (this will deadlock if server is GDBus < 2.73.3)
(fwupdmgr:23155): dconf-DEBUG: 15:37:59.276: watch_established: "/system/proxy/" (establishing: 1)
(fwupdmgr:23155): dconf-DEBUG: 15:37:59.276: watch_established: "/system/proxy/http/" (establishing: 1)
(fwupdmgr:23155): dconf-DEBUG: 15:37:59.276: watch_established: "/system/proxy/https/" (establishing: 1)
(fwupdmgr:23155): dconf-DEBUG: 15:37:59.276: watch_established: "/system/proxy/ftp/" (establishing: 1)
(fwupdmgr:23155): dconf-DEBUG: 15:37:59.276: watch_established: "/system/proxy/socks/" (establishing: 1)
(fwupdmgr:23155): Fwupd-DEBUG: 15:37:59.276: Emitting ::status-changed() [idle]
Updating lvfs
(fwupdmgr:23155): Fwupd-DEBUG: 15:37:59.279: downloading https://cdn.fwupd.org/downloads/firmware.xml.xz.jcat
(fwupdmgr:23155): Fwupd-DEBUG: 15:37:59.279: Emitting ::status-changed() [downloading]
Downloading…             [    \                                  ](fwupdmgr:23155): Fwupd-DEBUG: 15:37:59.411: download progress: 0%
(fwupdmgr:23155): Fwupd-DEBUG: 15:37:59.411: download progress: 0%
(fwupdmgr:23155): Fwupd-DEBUG: 15:37:59.411: download progress: 0%
(fwupdmgr:23155): Fwupd-DEBUG: 15:37:59.411: download progress: 0%
(fwupdmgr:23155): Fwupd-DEBUG: 15:37:59.412: download progress: 0%
(fwupdmgr:23155): Fwupd-DEBUG: 15:37:59.412: download progress: 0%
(fwupdmgr:23155): Fwupd-DEBUG: 15:37:59.412: download progress: 100%
(fwupdmgr:23155): Fwupd-DEBUG: 15:37:59.412: download progress: 100%
(fwupdmgr:23155): Fwupd-DEBUG: 15:37:59.412: download progress: 100%
(fwupdmgr:23155): Fwupd-DEBUG: 15:37:59.412: download progress: 100%
(fwupdmgr:23155): Fwupd-DEBUG: 15:37:59.412: Emitting ::status-changed() [idle]
(fwupdmgr:23155): Fwupd-DEBUG: 15:37:59.412: status-code was 200

Failed to load signature: Need more input
```
My `firmware.xml.xz.jcat` in a zip -> [firmware_file.zip](https://github.com/fwupd/fwupd/files/12379890/firmware_file.zip).

CDN server IP resolved:
```
➜  Downloads dig A cdn.fwupd.org         

; <<>> DiG 9.18.12-1ubuntu1.1-Ubuntu <<>> A cdn.fwupd.org
;; global options: +cmd
;; Got answer:
;; ->>HEADER<<- opcode: QUERY, status: NOERROR, id: 20248
;; flags: qr rd ra; QUERY: 1, ANSWER: 5, AUTHORITY: 0, ADDITIONAL: 1

;; OPT PSEUDOSECTION:
; EDNS: version: 0, flags:; udp: 65494
;; QUESTION SECTION:
;cdn.fwupd.org.			IN	A

;; ANSWER SECTION:
cdn.fwupd.org.		60	IN	CNAME	dualstack.p2.shared.global.fastly.net.
dualstack.p2.shared.global.fastly.net. 30 IN A	151.101.66.49
dualstack.p2.shared.global.fastly.net. 30 IN A	151.101.130.49
dualstack.p2.shared.global.fastly.net. 30 IN A	151.101.194.49
dualstack.p2.shared.global.fastly.net. 30 IN A	151.101.2.49

;; Query time: 340 msec
;; SERVER: 127.0.0.53#53(127.0.0.53) (UDP)
;; WHEN: Fri Aug 18 15:40:46 CEST 2023
;; MSG SIZE  rcvd: 157

➜  Downloads dig AAAA cdn.fwupd.org

; <<>> DiG 9.18.12-1ubuntu1.1-Ubuntu <<>> AAAA cdn.fwupd.org
[... 65 lines cut ...]
> accept: */*
> 
< HTTP/2 200 
< content-type: application/octet-stream
< server: gunicorn
< content-disposition: inline; filename=firmware.xml.xz.jcat
< last-modified: Fri, 18 Aug 2023 08:26:26 GMT
< cache-control: public, max-age=14400
< expires: Fri, 18 Aug 2023 16:26:58 GMT
< etag: "1692347186.242-2226-4227534271"
< accept-ranges: bytes
< date: Fri, 18 Aug 2023 13:42:00 GMT
< via: 1.1 varnish
< age: 4502
< x-served-by: cache-vie6325-VIE
< x-cache: HIT
< x-cache-hits: 48
< content-length: 2226
< 
Warning: Binary output can mess up your terminal. Use "--output -" to tell 
Warning: curl to output it to your terminal anyway, or consider "--output 
Warning: <FILE>" to save to a file.
* Failure writing output to destination
* Connection #0 to host cdn.fwupd.org left intact
```

senden9, 2023-08-18:
Additional info: If I download the file via a Server in the Switzerland I get a file from the CDN that is two bytes larger! So it seams to be a CDN-Sync issue?

Austria Server (with a `x-served-by: cache-vie6325-VIE`):
```
ls -l Downloads/firmware.xml.xz.jcat
-rw-rw-r-- 1 stefano stefano 2226 Aug 18 10:26 Downloads/firmware.xml.xz.jcat
```
Switzerland Server (with a `x-served-by: cache-mxp6952-MXP`):
```
ls -l firmware.xml.xz.jcat 
-rw-rw-r-- 1 ubuntu ubuntu 2228 Aug 18 13:53 firmware.xml.xz.jcat
 ```

hughsie, 2023-08-21:
I've created a ticket with Linux Foundation IT and they're going to proxy it to Fastly -- thanks. I'll keep this bug updated with progress.

hughsie, 2023-09-06:
Does anybody still get this, or was it a [transient] issue with the CDN? Thanks.

PocketSam, 2023-09-06:
Seems like I had no issues for the last two weeks.

hughsie, 2023-09-07:
Does anyone else have issues still? I'm being asked to close the downstream ticket.

cwrau, 2023-09-07:
This still happens for me;

```
2023-09-07T13:45:17.646720+0200 steve systemd[1]: Starting Refresh fwupd metadata and update motd...
2023-09-07T13:45:17.674549+0200 steve fwupdmgr[3841321]: (/usr/bin/fwupdmgr:3841321): GLib-DEBUG: 13:45:17.674: setenv()/putenv() are not thread-safe and should not be used after threads are created
2023-09-07T13:45:17.681468+0200 steve fwupdmgr[3841321]: (/usr/bin/fwupdmgr:3841321): GLib-GIO-DEBUG: 13:45:17.681: _g_io_module_get_default: Found default implementation dconf (DConfSettingsBackend) for ‘gsettings-backend’
2023-09-07T13:45:17.681468+0200 steve fwupdmgr[3841321]: (/usr/bin/fwupdmgr:3841321): dconf-DEBUG: 13:45:17.681: watch_fast: "/system/proxy/" (establishing: 0, active: 0)
2023-09-07T13:45:17.681991+0200 steve fwupdmgr[3841321]: (/usr/bin/fwupdmgr:3841321): dconf-DEBUG: 13:45:17.681: watch_established: "/system/proxy/" (establishing: 1)
2023-09-07T13:45:17.682099+0200 steve fwupdmgr[3841321]: (/usr/bin/fwupdmgr:3841321): dconf-DEBUG: 13:45:17.682: watch_fast: "/system/proxy/http/" (establishing: 0, active: 0)
2023-09-07T13:45:17.682131+0200 steve fwupdmgr[3841321]: (/usr/bin/fwupdmgr:3841321): dconf-DEBUG: 13:45:17.682: watch_fast: "/system/proxy/https/" (establishing: 0, active: 0)
2023-09-07T13:45:17.682149+0200 steve fwupdmgr[3841321]: (/usr/bin/fwupdmgr:3841321): dconf-DEBUG: 13:45:17.682: watch_established: "/system/proxy/http/" (establishing: 1)
2023-09-07T13:45:17.682165+0200 steve fwupdmgr[3841321]: (/usr/bin/fwupdmgr:3841321): dconf-DEBUG: 13:45:17.682: watch_fast: "/system/proxy/ftp/" (establishing: 0, active: 0)
2023-09-07T13:45:17.682180+0200 steve fwupdmgr[3841321]: (/usr/bin/fwupdmgr:3841321): dconf-DEBUG: 13:45:17.682: watch_established: "/system/proxy/https/" (establishing: 1)
2023-09-07T13:45:17.682210+0200 steve fwupdmgr[3841321]: (/usr/bin/fwupdmgr:3841321): dconf-DEBUG: 13:45:17.682: watch_fast: "/system/proxy/socks/" (establishing: 0, active: 0)
2023-09-07T13:45:17.682229+0200 steve fwupdmgr[3841321]: (/usr/bin/fwupdmgr:3841321): dconf-DEBUG: 13:45:17.682: watch_established: "/system/proxy/ftp/" (establishing: 1)
2023-09-07T13:45:17.682244+0200 steve fwupdmgr[3841321]: (/usr/bin/fwupdmgr:3841321): dconf-DEBUG: 13:45:17.682: unwatch_fast: "/system/proxy/" (active: 1, establishing: 0)
2023-09-07T13:45:17.682266+0200 steve fwupdmgr[3841321]: (/usr/bin/fwupdmgr:3841321): dconf-DEBUG: 13:45:17.682: unwatch_fast: "/system/proxy/http/" (active: 1, establishing: 0)
2023-09-07T13:45:17.682266+0200 steve fwupdmgr[3841321]: (/usr/bin/fwupdmgr:3841321): dconf-DEBUG: 13:45:17.682: unwatch_fast: "/system/proxy/https/" (active: 1, establishing: 0)
2023-09-07T13:45:17.682266+0200 steve fwupdmgr[3841321]: (/usr/bin/fwupdmgr:3841321): dconf-DEBUG: 13:45:17.682: unwatch_fast: "/system/proxy/ftp/" (active: 1, establishing: 0)
2023-09-07T13:45:17.682311+0200 steve fwupdmgr[3841321]: (/usr/bin/fwupdmgr:3841321): dconf-DEBUG: 13:45:17.682: watch_established: "/system/proxy/socks/" (establishing: 1)
2023-09-07T13:45:17.682311+0200 steve fwupdmgr[3841321]: (/usr/bin/fwupdmgr:3841321): dconf-DEBUG: 13:45:17.682: unwatch_fast: "/system/proxy/socks/" (active: 1, establishing: 0)
2023-09-07T13:45:17.683916+0200 steve fwupdmgr[3841321]: (/usr/bin/fwupdmgr:3841321): GLib-GIO-DEBUG: 13:45:17.683: _g_io_module_get_default: Found default implementation local (GLocalVfs) for ‘gio-vfs’
2023-09-07T13:45:17.683939+0200 steve fwupdmgr[3841321]: (/usr/bin/fwupdmgr:3841321): pxbackend-DEBUG: 13:45:17.683: px_config_sysconfig_set_config_file: Could not read file /etc/sysconfig/proxy
2023-09-07T13:45:17.683957+0200 steve fwupdmgr[3841321]: (/usr/bin/fwupdmgr:3841321): pxbackend-DEBUG: 13:45:17.683: Active config plugins:
2023-09-07T13:45:17.683957+0200 steve fwupdmgr[3841321]: (/usr/bin/fwupdmgr:3841321): pxbackend-DEBUG: 13:45:17.683:  - config-env
2023-09-07T13:45:17.683957+0200 steve fwupdmgr[3841321]: (/usr/bin/fwupdmgr:3841321): pxbackend-DEBUG: 13:45:17.683:  - config-kde
2023-09-07T13:45:17.683957+0200 steve fwupdmgr[3841321]: (/usr/bin/fwupdmgr:3841321): pxbackend-DEBUG: 13:45:17.683:  - config-gnome
2023-09-07T13:45:17.684014+0200 steve fwupdmgr[3841321]: (/usr/bin/fwupdmgr:3841321): pxbackend-DEBUG: 13:45:17.683:  - config-sysconfig
2023-09-07T13:45:17.684781+0200 steve fwupdmgr[3841321]: (/usr/bin/fwupdmgr:3841321): GLib-GIO-DEBUG: 13:45:17.684: Failed to initialize portal (GNetworkMonitorPortal) for gio-network-monitor: Not using portals
2023-09-07T13:45:17.685074+0200 steve fwupdmgr[3841321]: (/usr/bin/fwupdmgr:3841321): GLib-GIO-DEBUG: 13:45:17.685: Using cross-namespace EXTERNAL authentication (this will deadlock if server is GDBus < 2.73.3)
2023-09-07T13:45:17.685387+0200 steve fwupdmgr[3841321]: (/usr/bin/fwupdmgr:3841321): GLib-GIO-DEBUG: 13:45:17.685: Failed to initialize networkmanager (GNetworkMonitorNM) for gio-network-monitor: The connection is closed
2023-09-07T13:45:17.685442+0200 steve fwupdmgr[3841321]: (/usr/bin/fwupdmgr:3841321): GLib-GIO-DEBUG: 13:45:17.685: _g_io_module_get_default: Found default implementation netlink (GNetworkMonitorNetlink) for ‘gio-network-monitor’
2023-09-07T13:45:17.685442+0200 steve fwupdmgr[3841321]: (/usr/bin/fwupdmgr:3841321): pxbackend-DEBUG: 13:45:17.685: px_manager_on_network_changed: Network connection changed, clearing pac data
2023-09-07T13:45:17.685442+0200 steve fwupdmgr[3841321]: (/usr/bin/fwupdmgr:3841321): pxbackend-DEBUG: 13:45:17.685: px_manager_constructed: Up and running
[... cut ...] bin/fwupdmgr:3841321): GLib-GIO-DEBUG: 13:45:17.685: _g_io_module_get_default: Found default implementation libproxy (GLibproxyResolver) for ‘gio-proxy-resolver’
2023-09-07T13:45:17.688315+0200 steve systemd[1]: fwupd-refresh.service: Main process exited, code=exited, status=1/FAILURE
2023-09-07T13:45:17.688449+0200 steve systemd[1]: fwupd-refresh.service: Failed with result 'exit-code'.
2023-09-07T13:45:17.706454+0200 steve systemd[1]: Failed to start Refresh fwupd metadata and update motd.
```

hughsie, 2023-09-07:
@cwrau that's a different issue, no? what does "fwupdmgr refresh --force" say for you on the interactive CLI?

cwrau, 2023-09-07:
> @cwrau that's a different issue, no? what does "fwupdmgr refresh --force" say for you on the interactive CLI?

Ah, ok, so I should open a new issue?

Running interactively works, as it has before, just the systemd refresh service fails

hughsie, 2023-09-07:
Yup, new issue please.

Pentasis, 2023-09-15:
I'm new at linux so forgive me if I sound like a nood. I'm on Kubuntu 23.04 for e few days now. Today (after updating OpenSSH via Discovery) I am getting this error whenever I open Discovery.

This is the output using -vv
```
(fwupdmgr:4100): GLib-DEBUG: 11:33:53.489: setenv()/putenv() are not thread-safe and should not be used after threads are created
(process:4102): GLib-GIO-DEBUG: 11:33:53.493: Using cross-namespace EXTERNAL authentication (this will deadlock if server is GDBus < 2.73.3)
(fwupdmgr:4100): GLib-GIO-DEBUG: 11:33:53.498: _g_io_module_get_default: Found default implementation dconf (DConfSettingsBackend) for ?gsettings-backend?
(fwupdmgr:4100): dconf-DEBUG: 11:33:53.498: watch_fast: "/system/proxy/" (establishing: 0, active: 0)
(fwupdmgr:4100): dconf-DEBUG: 11:33:53.498: watch_fast: "/system/proxy/http/" (establishing: 0, active: 0)
(fwupdmgr:4100): dconf-DEBUG: 11:33:53.498: watch_fast: "/system/proxy/https/" (establishing: 0, active: 0)
(fwupdmgr:4100): dconf-DEBUG: 11:33:53.498: watch_fast: "/system/proxy/ftp/" (establishing: 0, active: 0)
(fwupdmgr:4100): dconf-DEBUG: 11:33:53.498: watch_fast: "/system/proxy/socks/" (establishing: 0, active: 0)
(fwupdmgr:4100): dconf-DEBUG: 11:33:53.498: unwatch_fast: "/system/proxy/" (active: 0, establishing: 1)
(fwupdmgr:4100): dconf-DEBUG: 11:33:53.498: unwatch_fast: "/system/proxy/http/" (active: 0, establishing: 1)
(fwupdmgr:4100): dconf-DEBUG: 11:33:53.498: unwatch_fast: "/system/proxy/https/" (active: 0, establishing: 1)
(fwupdmgr:4100): dconf-DEBUG: 11:33:53.498: unwatch_fast: "/system/proxy/ftp/" (active: 0, establishing: 1)
(fwupdmgr:4100): dconf-DEBUG: 11:33:53.498: unwatch_fast: "/system/proxy/socks/" (active: 0, establishing: 1)
(fwupdmgr:4100): GLib-GIO-DEBUG: 11:33:53.499: _g_io_module_get_default: Found default implementation libproxy (GLibproxyResolver) for ?gio-proxy-resolver?
(fwupdmgr:4100): GLib-GIO-DEBUG: 11:33:53.499: Using cross-namespace EXTERNAL authentication (this will deadlock if server is GDBus < 2.73.3)
(fwupdmgr:4100): GLib-GIO-DEBUG: 11:33:53.499: Using cross-namespace EXTERNAL authentication (this will deadlock if server is GDBus < 2.73.3)
(fwupdmgr:4100): dconf-DEBUG: 11:33:53.500: watch_established: "/system/proxy/" (establishing: 0)
(fwupdmgr:4100): dconf-DEBUG: 11:33:53.500: watch_established: "/system/proxy/http/" (establishing: 0)
(fwupdmgr:4100): dconf-DEBUG: 11:33:53.501: watch_established: "/system/proxy/https/" (establishing: 0)
(fwupdmgr:4100): dconf-DEBUG: 11:33:53.501: watch_established: "/system/proxy/ftp/" (establishing: 0)
(fwupdmgr:4100): dconf-DEBUG: 11:33:53.501: watch_established: "/system/proxy/socks/" (establishing: 0)
(fwupdmgr:4100): Fwupd-DEBUG: 11:33:53.502: Emitting ::status-changed() [idle]
Updating lvfs
(fwupdmgr:4100): Fwupd-DEBUG: 11:33:53.505: downloading https://cdn.fwupd.org/downloads/firmware.xml.xz.jcat
(fwupdmgr:4100): Fwupd-DEBUG: 11:33:53.505: Emitting ::status-changed() [downloading]
Downloading?             [     \                                 ](fwupdmgr:4100): Fwupd-DEBUG: 11:33:53.678: download progress: 100%
(fwupdmgr:4100): Fwupd-DEBUG: 11:33:53.678: download progress: 100%
(fwupdmgr:4100): Fwupd-DEBUG: 11:33:53.678: download progress: 100%
(fwupdmgr:4100): Fwupd-DEBUG: 11:33:53.678: download progress: 100%
(fwupdmgr:4100): Fwupd-DEBUG: 11:33:53.678: Emitting ::status-changed() [idle]
Downloading?             [***************************************](fwupdmgr:4100): Fwupd-DEBUG: 11:33:53.679: status-code was 200

Failed to load signature: Need more input
```

travier, 2023-09-25:
I have this issue since today, on an updated Fedora 39 (Kinoite but should not matter).

I've noticed that if I move the URL back to `gz`, then it works again, suggesting that there is an issue with the server or the xz metadata:

```
$ cat /etc/fwupd/remotes.d/lvfs.conf
[fwupd Remote]

# this remote provides metadata and firmware marked as 'stable' from the LVFS
Enabled=true
Title=Linux Vendor Firmware Service
MetadataURI=https://cdn.fwupd.org/downloads/firmware.xml.gz
ReportURI=https://fwupd.org/lvfs/firmware/report
SecurityReportURI=https://fwupd.org/lvfs/hsireports/upload
AutomaticReports=true
AutomaticSecurityReports=true
ApprovalRequired=false
$ sudo fwupdmgr refresh --force
Updating lvfs
Downloading…             [  -                                    ]
Successfully downloaded new metadata: 6 local devices supported

$ cat /etc/fwupd/remotes.d/lvfs.conf
[fwupd Remote]

# this remote provides metadata and firmware marked as 'stable' from the LVFS
Enabled=true
Title=Linux Vendor Firmware Service
MetadataURI=https://cdn.fwupd.org/downloads/firmware.xml.xz
ReportURI=https://fwupd.org/lvfs/firmware/report
SecurityReportURI=https://fwupd.org/lvfs/hsireports/upload
AutomaticReports=true
AutomaticSecurityReports=true
ApprovalRequired=false
$ sudo fwupdmgr refresh --force
Updating lvfs
Downloading…             [  -                                    ]
Failed to load signature: Need more input
```

travier, 2023-09-26:
And it works again today.

rynofinn, 2023-10-04:
Working with our CDN provider, we've verified that there is no difference between the files hosted in each region and the origin file.  Does anyone still see the symptom at this time?

cwrau, 2023-10-05:
> Working with our CDN provider, we've verified that there is no difference between the files hosted in each region and the origin file. Does anyone still see the symptom at this time?

~Currently `/usr/bin/fwupdmgr refresh -v` via systemd service results in exit code `2` with the following in the journal;~

<details>
<summary>Details</summary>
```log
2023-10-05T10:05:23.992632+0200 steve systemd[1]: Starting Refresh fwupd metadata and update motd...
2023-10-05T10:05:24.016430+0200 steve fwupdmgr[2566873]: (/usr/bin/fwupdmgr:2566873): GLib-DEBUG: 10:05:24.016: setenv()/putenv() are not thread-safe and should not be used after threads are created
2023-10-05T10:05:24.019615+0200 steve fwupdmgr[2566873]: (/usr/bin/fwupdmgr:2566873): GLib-GIO-DEBUG: 10:05:24.019: _g_io_module_get_default: Found default implementation dconf (DConfSettingsBackend) for ‘gsettings-backend’
2023-10-05T10:05:24.019615+0200 steve fwupdmgr[2566873]: (/usr/bin/fwupdmgr:2566873): dconf-DEBUG: 10:05:24.019: watch_fast: "/system/proxy/" (establishing: 0, active: 0)
2023-10-05T10:05:24.019824+0200 steve fwupdmgr[2566873]: (/usr/bin/fwupdmgr:2566873): dconf-DEBUG: 10:05:24.019: watch_fast: "/system/proxy/http/" (establishing: 0, active: 0)
2023-10-05T10:05:24.019856+0200 steve fwupdmgr[2566873]: (/usr/bin/fwupdmgr:2566873): dconf-DEBUG: 10:05:24.019: watch_fast: "/system/proxy/https/" (establishing: 0, active: 0)
2023-10-05T10:05:24.019887+0200 steve fwupdmgr[2566873]: (/usr/bin/fwupdmgr:2566873): dconf-DEBUG: 10:05:24.019: watch_fast: "/system/proxy/ftp/" (establishing: 0, active: 0)
2023-10-05T10:05:24.019913+0200 steve fwupdmgr[2566873]: (/usr/bin/fwupdmgr:2566873): dconf-DEBUG: 10:05:24.019: watch_fast: "/system/proxy/socks/" (establishing: 0, active: 0)
2023-10-05T10:05:24.019941+0200 steve fwupdmgr[2566873]: (/usr/bin/fwupdmgr:2566873): dconf-DEBUG: 10:05:24.019: unwatch_fast: "/system/proxy/" (active: 0, establishing: 1)
2023-10-05T10:05:24.019959+0200 steve fwupdmgr[2566873]: (/usr/bin/fwupdmgr:2566873): dconf-DEBUG: 10:05:24.019: unwatch_fast: "/system/proxy/http/" (active: 0, establishing: 1)
2023-10-05T10:05:24.019959+0200 steve fwupdmgr[2566873]: (/usr/bin/fwupdmgr:2566873): dconf-DEBUG: 10:05:24.019: unwatch_fast: "/system/proxy/https/" (active: 0, establishing: 1)
2023-10-05T10:05:24.019959+0200 steve fwupdmgr[2566873]: (/usr/bin/fwupdmgr:2566873): dconf-DEBUG: 10:05:24.019: unwatch_fast: "/system/proxy/ftp/" (active: 0, establishing: 1)
2023-10-05T10:05:24.019959+0200 steve fwupdmgr[2566873]: (/usr/bin/fwupdmgr:2566873): dconf-DEBUG: 10:05:24.019: unwatch_fast: "/system/proxy/socks/" (active: 0, establishing: 1)
2023-10-05T10:05:24.020061+0200 steve fwupdmgr[2566873]: (/usr/bin/fwupdmgr:2566873): dconf-DEBUG: 10:05:24.020: watch_established: "/system/proxy/" (establishing: 0)
2023-10-05T10:05:24.020097+0200 steve fwupdmgr[2566873]: (/usr/bin/fwupdmgr:2566873): dconf-DEBUG: 10:05:24.020: watch_established: "/system/proxy/http/" (establishing: 0)
2023-10-05T10:05:24.020125+0200 steve fwupdmgr[2566873]: (/usr/bin/fwupdmgr:2566873): dconf-DEBUG: 10:05:24.020: watch_established: "/system/proxy/https/" (establishing: 0)
2023-10-05T10:05:24.020155+0200 steve fwupdmgr[2566873]: (/usr/bin/fwupdmgr:2566873): dconf-DEBUG: 10:05:24.020: watch_established: "/system/proxy/ftp/" (establishing: 0)
2023-10-05T10:05:24.020197+0200 steve fwupdmgr[2566873]: (/usr/bin/fwupdmgr:2566873): dconf-DEBUG: 10:05:24.020: watch_established: "/system/proxy/socks/" (establishing: 0)
2023-10-05T10:05:24.020843+0200 steve fwupdmgr[2566873]: (/usr/bin/fwupdmgr:2566873): GLib-GIO-DEBUG: 10:05:24.020: _g_io_module_get_default: Found default implementation local (GLocalVfs) for ‘gio-vfs’
2023-10-05T10:05:24.020875+0200 steve fwupdmgr[2566873]: (/usr/bin/fwupdmgr:2566873): pxbackend-DEBUG: 10:05:24.020: px_config_sysconfig_set_config_file: Could not read file /etc/sysconfig/proxy
2023-10-05T10:05:24.020914+0200 steve fwupdmgr[2566873]: (/usr/bin/fwupdmgr:2566873): pxbackend-DEBUG: 10:05:24.020: Active config plugins:
2023-10-05T10:05:24.020914+0200 steve fwupdmgr[2566873]: (/usr/bin/fwupdmgr:2566873): pxbackend-DEBUG: 10:05:24.020:  - config-env
2023-10-05T10:05:24.020914+0200 steve fwupdmgr[2566873]: (/usr/bin/fwupdmgr:2566873): pxbackend-DEBUG: 10:05:24.020:  - config-kde
2023-10-05T10:05:24.020914+0200 steve fwupdmgr[2566873]: (/usr/bin/fwupdmgr:2566873): pxbackend-DEBUG: 10:05:24.020:  - config-gnome
2023-10-05T10:05:24.020914+0200 steve fwupdmgr[2566873]: (/usr/bin/fwupdmgr:2566873): pxbackend-DEBUG: 10:05:24.020:  - config-sysconfig
2023-10-05T10:05:24.021717+0200 steve fwupdmgr[2566873]: (/usr/bin/fwupdmgr:2566873): GLib-GIO-DEBUG: 10:05:24.021: Failed to initialize portal (GNetworkMonitorPortal) for gio-network-monitor: Not using portals
2023-10-05T10:05:24.022054+0200 steve fwupdmgr[2566873]: (/usr/bin/fwupdmgr:2566873): GLib-GIO-DEBUG: 10:05:24.022: Using cross-namespace EXTERNAL authentication (this will deadlock if server is GDBus < 2.73.3)
2023-10-05T10:05:24.023299+0200 steve fwupdmgr[2566873]: (/usr/bin/fwupdmgr:2566873): GLib-GIO-DEBUG: 10:05:24.023: _g_io_module_get_default: Found default implementation networkmanager (GNetworkMonitorNM) for ‘gio-network-monitor’
2023-10-05T10:05:24.023299+0200 steve fwupdmgr[2566873]: (/usr/bin/fwupdmgr:2566873): pxbackend-DEBUG: 10:05:24.023: px_manager_on_network_changed: Network connection changed, clearing pac data
[... cut ...]
2023-10-05T10:05:24.023346+0200 steve fwupdmgr[2566873]: (/usr/bin/fwupdmgr:2566873): GLib-GIO-DEBUG: 10:05:24.023: _g_io_module_get_default: Found default implementation libproxy (GLibproxyResolver) for ‘gio-proxy-resolver’
2023-10-05T10:05:24.026569+0200 steve fwupdmgr[2566873]: (/usr/bin/fwupdmgr:2566873): Fwupd-DEBUG: 10:05:24.026: Emitting ::status-changed() [idle]
2023-10-05T10:05:24.034378+0200 steve systemd[1]: fwupd-refresh.service: Deactivated successfully.
2023-10-05T10:05:24.043320+0200 steve systemd[1]: Finished Refresh fwupd metadata and update motd.
```
</details>

---

Nevermind, exit code `2` is success 😁

Then this looks like it's working now 👍

Amr-Ibra, 2023-10-06:
I'm affected too. From Germany.

leture, 2023-10-10:
Hi,

also experience this issue under Manjaro.

I switched in `/etc/fwupd/remotes.d/lvfs.conf` from `MetadataURI=https://cdn.fwupd.org/downloads/firmware.xml.xz` to `MetadataURI=https://fwupd.org/downloads/firmware.xml.xz` and this solved the problem.

davidlt, 2023-10-10:
Happened to me on Fedora 38 too. Removing cdn solved the problem as suggested in a number of replies. It would be great if it could re-try without CDN by default.

hughsie, 2023-10-10:
> It would be great if it could re-try without CDN by default.

That would destroy the main server any time the CDN went wrong. Outgoing bandwidth in AWS is expensive. :/

hughsie, 2023-10-10:
I guess the hacky workaround is to just to append 6 bytes of "XXXXXX" to the JCAT file -- libjcat seems to parse it just fine with the extra data appended and then if the CDN did truncate the file then ¯\_(ツ)_/¯

hughsie, 2023-10-10:
Can anyone affected go back to the CDN and try again please? There's 6 bytes of junk on the end of https://cdn.fwupd.org/downloads/firmware.xml.xz.jcat now

Lavode, 2023-10-10:
> I guess the hacky workaround is to just to append 6 bytes of "XXXXXX" to the JCAT file -- libjcat seems to parse it just fine with the extra data appended and then if the CDN did truncate the file then ¯_(ツ)_/¯

I doubt that's sustainable. Last time the file was missing four bytes, this time it's six. Next time it might be eight, and you're back to square one. And that's not to mention just /how awful/ of a hack that is to begin with. ;)

That aside - works for me now. Main website & CDN deliver the same file for now - including the `IH8CDN` postfix. :grin:

hughsie, 2023-10-10:
> including the IH8CDN postfix

That was supposed to be an easter egg! :)

rynofinn, 2023-10-10:
Thanks for helping us to track this down.  If you are affected still, can you post the output of these two commands:
curl -v https://cdn.fwupd.org/downloads/firmware.xml.xz.jcat
curl -v https://fwupd.org/downloads/firmware.xml.xz.jcat

hughsie, 2023-10-10:
If this hack, cough, workaround, fixed the issue for you please say. If it's still broken, please shout louder!

lordtaytertot, 2023-10-30:
@hughsie, I had to use @leture's fix: https://github.com/fwupd/fwupd/issues/6032#issuecomment-1755085972.

DomiStyle, 2023-11-14:
My Discover on Fedora 39 KDE just started throwing this error today, same from the command line.

Checksums are different between CDN and direct download:
```
f7eeea093ef8edee01413a38ee3d9507  cdn.xml.xz.jcat
9e1820f33cda3403fdcde480bafe828c  direct.xml.xz.jcat
```

curl output to cdn.fwupd.org: https://gist.github.com/DomiStyle/5814ef8ffe082ee56098138437622274
curl output to fwupd.org: https://gist.github.com/DomiStyle/acf60e69d14b10c570b4f4e8ce5046ea

[5 more replies]