fwupd/fwupd #6034: --force option for "switch-branch" doesn't work when fwupd falsely thinks I'm not on AC

ell1e, 2023-07-27:
**Describe the bug**
Using the `--force` option for "switch-branch" doesn't work as advertised in the error message when fwupd falsely thinks I'm not on AC. I just keep getting `QUECTEL Mobile Broadband Module is not currently updatable: Cannot install update when not on AC power unless forced`. This is kind of frustrating given fwupd seems to quite often think I'm not on AC when it's just completely wrong (and I'm at 100% battery anyway, maybe it shouldn't stop with that error if above a specific threshold?).

**Steps to Reproduce**
1. Make fwupd falsely think you're not on AC
2. Try `sudo fwupdmgr switch-branch <some device of yours where a branch can be switched>` with `--force`

**Expected behavior**
Works, instead I keep getting `QUECTEL Mobile Broadband Module is not currently updatable: Cannot install update when not on AC power unless forced` as if I hadn't used 
`--force`

**fwupd version information**
Please provide the version of the daemon and client.

```shell
$ fwupdmgr --version
runtime   org.freedesktop.fwupd         1.9.3
compile   com.hughsie.libxmlb           0.3.11
compile   org.freedesktop.gusb          0.4.6
runtime   com.hughsie.libjcat           0.1.13
compile   com.hughsie.libjcat           0.1.13
runtime   org.kernel                    6.3.5
compile   org.freedesktop.fwupd         1.9.3
runtime   org.freedesktop.gusb          0.4.6
```

Please note how you installed it (`apt`, `dnf`, `pacman`, source, etc):

<details>

<summary>**fwupd device information**</summary>

Please provide the output of the fwupd devices recognized in your system.

```shell
$ fwupdmgr get-devices --show-all-devices
WARNING: UEFI capsule updates not available or enabled in firmware setup
See https://github.com/fwupd/fwupd/wiki/PluginFlag:capsules-unsupported for more information.
WARNING: This package has not been validated, it may not work properly.
pine64 Pine64 PinePhone (1.2)
│
├─Unknown Device:
│     Device ID:          <removed>
│     GUID:               <removed> ← GPIO\ID_1c20800.pinctrl
│   
├─Unknown Device:
│     Device ID:          <removed>
│     GUID:               <removed> ← GPIO\ID_1f02c00.pinctrl
│   
├─DA4032:
│     Device ID:          <removed>
│     Current version:    0x3034313430363139
│     Vendor:             EMMC:0x000045
│     Serial Number:      0xc3612415
│     GUIDs:              <removed> ← EMMC\NAME_DA4032
│                         <removed> ← EMMC\MAN_0045&OEM_0100
│                         <removed> ← EMMC\MAN_0045&OEM_0100&NAME_DA4032
│                         <removed> ← EMMC\MAN_0045&NAME_DA4032&REV_0x3034313430363139
│                         <removed> ← EMMC\MAN_0045&OEM_0100&NAME_DA4032&REV_0x3034313430363139
│                         <removed> ← EMMC\0069&0256&DA4032
│     Device Flags:       • Internal device
│                         • Updatable
│   
└─QUECTEL Mobile Broadband Module:
      Device ID:          <removed>
      Summary:            Quectel EG25-G modem
      Current version:    0.7.2
      Vendor:             Qualcomm (USB:0x2C7C)
      Release Branch:     community
      GUIDs:              <removed>
                          <removed> ← USB\VID_2C7C&PID_0125&REV_0318&NAME_EG25GGB
                          <removed> ← USB\VID_2C7C&PID_0125&REV_0318
                          <removed> ← USB\VID_2C7C&PID_0125
                          <removed> ← USB\VID_2C7C
      Device Flags:       • Updatable
                          • System requires external power source
                          • Supported on remote server
                          • Unsigned Payload

```

</details>

**Additional questions**

- Operating system and version: postmarketOS 23.06
- Have you tried rebooting? yes but then the AC situation was different then
- Is this a regression? I've never used fwupd before so I don't know

hughsie, 2023-07-27:
Hey. We need to know the AC state correctly as some vendors insist on specific battery thresholds or AC power states. What does `upower --dump` say about your device?

ell1e, 2023-07-30:
Wouldn't the `--force` switch not working be kind of independent of whatever upower does? (Or shouldn't it be, if it's not?)

hughsie, 2023-07-30:
I don't think so, using force like that is a way to shoot yourself in the foot. For instance, using a capsule update the payload is going to check for battery level itself, and there's no way to "force" that one away.

ell1e, 2023-07-30:
In that case I think the error message should be reworded. Also, I think that's actually a problem. Because upower has bugs, and the distribution I use already confirmed this seems to be a upower bug, but now if you don't allow any switch to override that people just can't update their device properly anymore whenever there's a upower bug like this. That seems bad to me, what's the point of `--force` then?

ell1e, 2023-07-30:
Is there a command with which I can check if fwupd currently thinks I'm in a state where I can install or upgrade firmware without actually installing or upgrading anything?

I think I'm in the state right now where it broke (which is on AC but battery is so full that it discharges anyway) although I could be wrong. Here's the upower dump:
```
$ upower --dump
Device: /org/freedesktop/UPower/devices/battery_axp20x_battery
  native-path:          axp20x-battery
  power supply:         yes
  updated:              Sun Jul 30 09:21:29 2023 (25 seconds ago)
  has history:          yes
  has statistics:       yes
  battery
    present:             yes
    rechargeable:        yes
    state:               discharging
    warning-level:       none
    energy:              0 Wh
    energy-empty:        0 Wh
    energy-full:         0 Wh
    energy-full-design:  0 Wh
    energy-rate:         1.2874 W
    voltage:             4.065 V
    charge-cycles:       N/A
    percentage:          100%
    icon-name:          'battery-full-symbolic'
  History (rate):
    1690701689	1.287	discharging
    1690701659	0.672	discharging
    1690701634	0.500	charging
    1690701604	0.529	charging

Device: /org/freedesktop/UPower/devices/line_power_axp20x_usb
  native-path:          axp20x-usb
  power supply:         yes
  updated:              Sun Jul 30 09:12:23 2023 (571 seconds ago)
  has history:          no
  has statistics:       no
  line-power
    warning-level:       none
    online:              yes
    icon-name:          'ac-adapter-symbolic'

Device: /org/freedesktop/UPower/devices/DisplayDevice
  power supply:         yes
  updated:              Sun Jul 30 09:21:29 2023 (25 seconds ago)
  has history:          no
  has statistics:       no
  battery
    present:             yes
    state:               discharging
    warning-level:       none
    energy:              0 Wh
    energy-full:         0 Wh
    energy-rate:         1.2874 W
    charge-cycles:       N/A
    percentage:          100%
    icon-name:          'battery-full-symbolic'

Daemon:
  daemon-version:  1.90.0
  on-battery:      no
  lid-is-closed:   no
  lid-is-present:  no
  critical-action: PowerOff
```
It actually seems to me like the upower output is correct so if I'm remembering right this is how it looked like when it didn't work (I'm guessing based on the notifications bar icon which no longer shows it charging) then maybe fwupd is actually reading the upower data wrong rather than upower being bugged, but this is just a very uneducated guess from my side I wouldn't know.

hughsie, 2023-07-30:
I think the UPower data is wrong:

    battery
      present:             yes
      state:               discharging

If the battery is not discharging but on AC the state should be fully-charged, no? e.g. see https://github.com/fwupd/fwupd/blob/main/plugins/upower/fu-upower-plugin.c#L54

hughsie, 2023-07-31:
> I think --force should override this situation

Force really shouldn't be used to work around bugs. I think the right thing to do here is to copy the design used by the powerd plugin, i.e. use the line-power:online property as the AC state, and the DisplayDevice for the battery state.

hughsie, 2024-11-11:
I'm pretty sure we don't have that error now. Please re-open if this still happens in 2.0.x

closed 2024-11-11 by hughsie