fwupd/fwupd #6035: fwupdmgr: Dead URL in WARNING: UEFI ESP partition may not be set up correctly

vmatare, 2023-07-27:
**Describe the bug**
Whenever I execute `fwupdmgr`, I get:
```
WARNING: UEFI ESP partition may not be set up correctly
See https://github.com/fwupd/fwupd/wiki/PluginFlag:esp-not-valid for more information.
```

**Steps to Reproduce**
Don't know what's causing it. That is precisely the problem. `fwupdtool esp-list --verbose` seems to be working just fine:
```
10:34:09.115 FuDebug              verbose to info (on console 1)
10:34:09.115 FuEngine             starting fwupd 1.9.3…
Selected volume: /org/freedesktop/UDisks2/block_devices/nvme0n1p1
/boot/efi/EFI/BOOT/BOOTIA32.EFI
/boot/efi/EFI/BOOT/BOOTX64.EFI
/boot/efi/EFI/BOOT/fbia32.efi
/boot/efi/EFI/BOOT/fbx64.efi
/boot/efi/EFI/fedora/fw/fwupd-a28cc671-fabc-4eb1-9a7b-a10a5223c64e.cap
/boot/efi/EFI/fedora/fwupdx64.efi
/boot/efi/EFI/fedora/BOOTIA32.CSV
/boot/efi/EFI/fedora/BOOTX64.CSV
/boot/efi/EFI/fedora/gcdia32.efi
/boot/efi/EFI/fedora/gcdx64.efi
/boot/efi/EFI/fedora/grubia32.efi
/boot/efi/EFI/fedora/grubx64.efi
/boot/efi/EFI/fedora/mmia32.efi
/boot/efi/EFI/fedora/mmx64.efi
/boot/efi/EFI/fedora/shim.efi
/boot/efi/EFI/fedora/grub.cfg.rpmsave
/boot/efi/EFI/fedora/shimia32.efi
/boot/efi/EFI/fedora/grubenv.rpmsave
/boot/efi/EFI/fedora/shimx64.efi
/boot/efi/EFI/fedora/grub.cfg
/boot/efi/System/Library/CoreServices/SystemVersion.plist
/boot/efi/mach_kernel
/boot/efi/System Volume Information/IndexerVolumeGuid
/boot/efi/System Volume Information/WPSettings.dat
/boot/efi/$RECYCLE.BIN/desktop.ini
```

**Expected behavior**
I expect some info on why `fwupdmgr` thinks something is wrong with my ESP.

**fwupd version information**

```shell
fwupdmgr --version
```
```
compile   org.freedesktop.fwupd         1.9.3
compile   com.hughsie.libxmlb           0.3.11
compile   com.hughsie.libjcat           0.1.13
runtime   org.freedesktop.fwupd-efi     1.4
compile   org.freedesktop.gusb          0.4.5
runtime   com.hughsie.libjcat           0.1.13
runtime   org.freedesktop.gusb          0.4.5
runtime   org.freedesktop.fwupd         1.9.3
runtime   org.kernel                    6.4.4-200.fc38.x86_64
```

Please note how you installed it (`apt`, `dnf`, `pacman`, source, etc): `dnf`

<details>

<summary>**fwupd device information**</summary>

Please provide the output of the fwupd devices recognized in your system.

```shell
fwupdmgr get-devices --show-all-devices
[... 143 lines cut ...]
│                         • Signed Payload
│     
└─TPM:
      Device ID:          c6a80ac3a22083423992a3cb15018989f37834d6
      Current version:    7.61.10.57600
      Vendor:             Infineon (TPM:IFX)
      GUIDs:              ff71992e-52f7-5eea-94ef-883e56e034c6 ← system-tpm
                          5eebb112-75ad-5536-b173-a11eb3399402 ← TPM\VEN_IFX&DEV_0000
                          ddf995da-1b32-5a8a-bc1b-8d5af4b38b51 ← TPM\VEN_IFX&MOD_SLB9670
                          6d81ab63-db2e-50ac-934f-6be9accf5e02 ← TPM\VEN_IFX&DEV_0000&VER_2.0
                          301555de-680d-5ddc-b995-7553fc9138f1 ← TPM\VEN_IFX&MOD_SLB9670&VER_2.0
      Device Flags:       • Internal device
                          • System requires external power source
                          • Needs a reboot after installation
                          • Device can recover flash failures
                          • Full disk encryption secrets may be invalidated when updating
                          • Signed Payload
```
</details>

**Additional questions**

- Operating system and version: Fedora 38
- Have you tried rebooting? sure
- Is this a regression? don't know

hughsie, 2023-07-27:
Hmm, that page sure used to exist. Perhaps someone deleted it?

hughsie, 2023-07-27:
I've added a link to https://github.com/fwupd/fwupd/wiki/LVFS-Triaged-Issue:-Invalid-ESP-Partition as that's the one I think should be used to describe this.  fwupdtool esp-list --verbose will work fine -- but the firmware might not. Newer machines are much more strict about partitions being tagged with the correct GUID.

hughsie, 2023-07-27:
@vmatare did you create the ESP partition, or did Anaconda do it for you?

vmatare, 2023-08-01:
> @vmatare did you create the ESP partition, or did Anaconda do it for you?

I'm pretty sure Anaconda did it. After I got this message from fwupd, I noticed that it had a weird type flag (I think it was `0x04`, which fdisk says is `FAT16 <32M`). I then tried both `0xef` and `0x0b`, but none of that made any difference. Currently I have the following:

```
# fdisk -l /dev/nvme0n1
Disk /dev/nvme0n1: 238.47 GiB, 256060514304 bytes, 500118192 sectors
Disk model: ADATA SX8000NP                          
Units: sectors of 1 * 512 = 512 bytes
Sector size (logical/physical): 512 bytes / 512 bytes
I/O size (minimum/optimal): 512 bytes / 512 bytes
Disklabel type: dos
Disk identifier: 0x772dea53

Device         Boot   Start       End   Sectors   Size Id Type
/dev/nvme0n1p1 *       2048   1230847   1228800   600M ef EFI (FAT-12/16/32)
/dev/nvme0n1p2      1230848   3327999   2097152     1G 83 Linux
/dev/nvme0n1p3      3328000 500117503 496789504 236.9G 8e Linux LVM
```
So apparently it's not even a GPT partitioning scheme. Anaconda might have just left it the way it was before. But normally GPT shouldn't be strictly necessary, right?

hughsie, 2023-08-02:
The firmware used to not care, but it seems more and more firmware does nowadays. It's probably more because it's not being tested on MBR partitions anymore. If firmware updates do work for you it's safe to ignore.

XenonSup, 2023-09-29:
I'm having the same problem.

**Describe the bug**
Same as above

**Steps to Reproduce**
Same as above

**Expected behavior**
Same as above. Or display no warnings.

**fwupd version information**

```shell
$ fwupdmgr --version
compile   org.freedesktop.fwupd         1.9.5
compile   com.hughsie.libxmlb           0.3.12
compile   com.hughsie.libjcat           0.1.13
runtime   org.freedesktop.fwupd-efi     1.4
compile   org.freedesktop.gusb          0.4.5
runtime   com.hughsie.libjcat           0.1.13
runtime   org.freedesktop.gusb          0.4.5
runtime   org.freedesktop.fwupd         1.9.5
runtime   org.kernel                    6.5.5-200.fc38.x86_64
```

Please note how you installed it (`apt`, `dnf`, `pacman`, source, etc): `dnf`

<details>

**<summary>fwupd device information</summary>**

```shell
$ fwupdmgr get-devices --show-all-devices
WARNING: UEFI ESP partition may not be set up correctly
See https://github.com/fwupd/fwupd/wiki/PluginFlag:esp-not-valid for more information.
Dell Inc. XPS 15 9560
│
├─Core™ i7-7700HQ CPU @ 2.80GHz:
│     Device ID:          4bde70ba4e39b28f9eab1628f9dd6e6244c03027
│     Current version:    0x000000f4
│     Vendor:             Intel
│     GUIDs:              b9a2dd81-159e-5537-a7db-e7101d164d3f ← cpu
│                         30249f37-d140-5d3e-9319-186b1bd5cac3 ← CPUID\PRO_0&FAM_06
│                         809a0b93-8a12-5338-a571-ad5583acf896 ← CPUID\PRO_0&FAM_06&MOD_9E
│                         d0f754d5-1395-5573-bc83-85ba955da70a ← CPUID\PRO_0&FAM_06&MOD_9E&STP_9
│     Device Flags:       • Internal device
│   
├─GP107M [GeForce GTX 1050 Mobile]:
│     Device ID:          ce4c74a5188d5b9cdb1e72ed32dad2d313c1c999
│     Current version:    a1
│     Vendor:             NVIDIA Corporation (PCI:0x10DE, PCI:0x8086)
│     GUIDs:              ef41e989-4e6e-560d-a860-ea71aa8c21a3 ← PCI\VEN_10DE&DEV_1C8D
│                         5fdc8f12-8985-5dcb-95ea-5761f769cb4d ← PCI\VEN_10DE&DEV_1C8D&SUBSYS_102807BE
│                         d29269b6-e458-5f45-975c-9d52bb38e35f ← PCI\VEN_8086&DEV_1901
│                         6949aa50-f4e1-578b-81c8-d8c61915cfe2 ← PCI\VEN_8086&DEV_1901&SUBSYS_102807BE
│     Device Flags:       • Internal device
│                         • Cryptographic hash verification is available
│   
├─HD Graphics 630:
│     Device ID:          5792b48846ce271fab11c4a545f7a3df0d36e00a
│     Current version:    04
│     Vendor:             Intel Corporation (PCI:0x8086)
│     GUIDs:              f744dc01-a247-5543-b9db-2adba3091233 ← PCI\VEN_8086&DEV_591B
│                         089d6890-1705-50c9-92bd-55a1bd648ed9 ← PCI\VEN_8086&DEV_591B&SUBSYS_102807BE
│     Device Flags:       • Internal device
│                         • Cryptographic hash verification is available
│   
├─SHGP31-1000GM:
│     Device ID:          04e17fcf7d3de91da49a163ffe4907855c3648be
[... 269 lines cut ...]
/dev/nvme0n1p2     206848     239615      32768    16M Microsoft reserved
/dev/nvme0n1p3     239616  522866687  522627072 249.2G Microsoft basic data
/dev/nvme0n1p4  522866688  524523519    1656832   809M Windows recovery environment
/dev/nvme0n1p5  524525568  526622719    2097152     1G Linux filesystem
/dev/nvme0n1p6  526622720 1758902271 1232279552 587.6G Linux filesystem
/dev/nvme0n1p7 1758902272 1953523711  194621440  92.8G Microsoft basic data

# parted --list
Model: SHGP31-1000GM (nvme)
Disk /dev/nvme0n1: 1000GB
Sector size (logical/physical): 512B/512B
Partition Table: gpt
Disk Flags: 

Number  Start   End     Size    File system  Name                          Flags
 1      1049kB  106MB   105MB   fat32        EFI System Partition          boot, esp, no_automount
 2      106MB   123MB   16.8MB               Microsoft reserved partition  msftres, no_automount
 3      123MB   268GB   268GB   ntfs         Basic data partition          msftdata, no_automount
 4      268GB   269GB   848MB   ntfs                                       hidden, diag, no_automount
 5      269GB   270GB   1074MB  ext4
 6      270GB   901GB   631GB
 7      901GB   1000GB  99.6GB  fat32                                      msftdata
...
```
</details>

rehrauer, 2023-10-31:
I have the same problem.  I believe that  fwupd thinks that a partition with the msftdata flag set is an EFI partition.  I do not run windows only linux on my laptop but have a recovery partition with the msftdata flag set.  fwupdtool esp-list --verbose also lists my recovery partition along with my 2 EFI partitions.  This only happens with fedora.  I have debian running on another partition and do not get any error when running debian.  It also does not happen with ubuntu.  I can do firmware updates from ubuntu or debian but the only one I attempted from fedora 38 hung my system.

T0MuX, 2023-12-05:
[ EDIT : SOLVED TO ME ]
[ I just added the "esp" flag on my EFI partition ... lol ]

Hi, I have exactly the same issue, even a bit worse.

- Archlinux freshly updated, linux 6.1.64-1-lts
- Udisks is installed.
- EFI partition is mounted on boot via fstab.
- Already edited /etc/fwupd/uefi_capsule.conf to add `OverrideESPMountPoint=/boot/efi` (which is the mount point of my EFI partition), but when I restart the fwupd service, the config file gets renamed into /etc/fwupd/uefi_capsule.conf.old with no change at all (still not working properly).

```
[t0mux@BCD-GAP-L004 ~]$ cat /etc/fstab | grep vfat
UUID=0FF4-FC82      	/boot/efi 	vfat      	rw,relatime,fmask=0022,dmask=0022,codepage=437,iocharset=ascii,shortname=mixed,utf8,errors=remount-ro	0 2

[t0mux@BCD-GAP-L004 ~]$ sudo fwupdtool esp-list --verbose
05:08:29.197 FuDebug              verbose to info (on console 1)
05:08:29.198 FuEngine             starting fwupd 1.9.9…
No ESP or BDP found

[t0mux@BCD-GAP-L004 ~]$ sudo fdisk -l
Disque /dev/nvme0n1 : 476,94 GiB, 512110190592 octets, 1000215216 secteurs
Modèle de disque : KBG50ZNT512G LS KIOXIA                  
Unités : secteur de 1 × 512 = 512 octets
Taille de secteur (logique / physique) : 512 octets / 512 octets
taille d'E/S (minimale / optimale) : 512 octets / 512 octets
Type d'étiquette de disque : gpt
Identifiant de disque : 650E1D25-1BA0-4B12-82EA-9F4A255C2E11

Périphérique     Début        Fin  Secteurs Taille Type
/dev/nvme0n1p1      34     585937    585904 286,1M Système de fichiers Linux
/dev/nvme0n1p2 1048576 1000214527 999165952 476,4G Système de fichiers Linux

[t0mux@BCD-GAP-L004 ~]$ ls /boot/efi
EFI

[t0mux@BCD-GAP-L004 ~]$ ls /boot/efi/EFI/
BOOT  crystal

[t0mux@BCD-GAP-L004 ~]$ ls /boot/efi/EFI/BOOT/
BOOTX64.EFI

[t0mux@BCD-GAP-L004 ~]$ ls /etc/fwupd/
bios-settings.d  fwupd.conf  remotes.d  uefi_capsule.conf.old

-------------- EDIT (after added "esp" flag on EFI partition) --------------------

[t0mux@BCD-GAP-L004 ~]$ sudo fwupdtool esp-list --verbose
05:10:15.045 FuDebug              verbose to info (on console 1)
05:10:15.045 FuEngine             starting fwupd 1.9.9…
Volume sélectionné: /org/freedesktop/UDisks2/block_devices/nvme0n1p1
/boot/efi/EFI/BOOT/BOOTX64.EFI
/boot/efi/EFI/crystal/grubx64.efi
```

tomhughes, 2023-12-06:
Also broken for me, and I do have the `esp` flag:

```
arden [~] % sudo fwupdmgr          
WARNING: UEFI ESP partition may not be set up correctly
See https://github.com/fwupd/fwupd/wiki/PluginFlag:esp-not-valid for more information.
Command not found

Use fwupdmgr --help for help

arden [~] % sudo parted -l /dev/nvme0n1
Model: THNSN5512GPUK NVMe TOSHIBA 512GB (nvme)
Disk /dev/nvme0n1: 512GB
Sector size (logical/physical): 512B/512B
Partition Table: gpt
Disk Flags: 

Number  Start   End     Size    File system     Name                  Flags
 1      1049kB  630MB   629MB   fat32           EFI System Partition  boot, esp
 2      630MB   3851MB  3221MB  fat32           Basic data partition  msftdata
 3      3851MB  478GB   474GB   ext4
 4      478GB   512GB   34.1GB  linux-swap(v1)                        swap

arden [~] % sudo fwupdtool esp-list
0.	Cancel
1.	/org/freedesktop/UDisks2/block_devices/nvme0n1p1
2.	/org/freedesktop/UDisks2/block_devices/nvme0n1p2
Choose volume [0-2]: 1
/boot/efi/EFI/BOOT/BOOTX64.EFI
/boot/efi/EFI/BOOT/fbx64.efi
/boot/efi/EFI/fedora/fw/fwupd-3b8c8162-188c-46a4-aec9-be43f1d65697.cap
/boot/efi/EFI/fedora/fw/fwupd-5ffdbc0d-f340-441c-a803-8439c8c0ae10.cap
/boot/efi/EFI/fedora/fwupdx64.efi
/boot/efi/EFI/fedora/shim.efi
/boot/efi/EFI/fedora/BOOTX64.CSV
/boot/efi/EFI/fedora/mmx64.efi
/boot/efi/EFI/fedora/grubx64.efi
/boot/efi/EFI/fedora/grub.cfg
/boot/efi/EFI/fedora/shimx64.efi
/boot/efi/EFI/Dell/Bios/Recovery/BIOS_PRE.rcv
/boot/efi/EFI/Dell/Bios/Recovery/BIOS_CUR.RCV
/boot/efi/EFI/Dell/logs/diags_previous.xml
/boot/efi/EFI/Dell/logs/diags_current.xml
```

closed 2023-12-07 by hughsie

tomhughes, 2023-12-07:
Yes that patch fixes it - this is the output:

```
11:55:49.507 FuPluginUefiCapsule  found /boot/efi/EFI/fedora/grubx64.efi which indicates a Linux ESP, using /boot/efi
11:55:49.507 FuVolume             mounting /org/freedesktop/UDisks2/block_devices/nvme0n1p2
11:55:49.681 FuPluginUefiCapsule  found /run/media/root/OS/efi.factory/boot/grubx64.efi which indicates a Linux ESP, using /run/media/root/OS
11:55:49.681 FuVolume             unmounting /org/freedesktop/UDisks2/block_devices/nvme0n1p2
11:55:49.718 FuPluginUefiCapsule  more than one ESP possible:
 - 0x30258:	/org/freedesktop/UDisks2/block_devices/nvme0n1p1
 - 0x10c00:	/org/freedesktop/UDisks2/block_devices/nvme0n1p2
```

hughsie, 2023-12-07:
> OS/efi.factory/boot

Is that a recovery partition?

hughsie, 2023-12-07:
> Basic data partition

Bleh, that's the fallback IIRC for when the name isn't set. Maybe the UUID is something we could key-from? i.e. if you google your `UUID` from `org.freedesktop.UDisks2.Partition` is it obviously just dd'd in the factory?

tomhughes, 2023-12-07:
The UUID is `f618a4bd-8b4f-4c55-a88f-c88a8365c58d` which doesn't seem to get any hits at all...

diegorondini, 2023-12-07:
Hi,

same situation, Dell XPS 13 9370 preloaded with Ubuntu reinstalled with Fedora.
The **GPT partition label** is indeed `Basic data partition`, but **FAT filesystem label** is `OS`:
```
[root@localhost ~]# ll /dev/disk/by-partlabel/Basic\\x20data\\x20partition
lrwxrwxrwx. 1 root root 15 Dec  6 08:55 '/dev/disk/by-partlabel/Basic\x20data\x20partition' -> ../../nvme0n1p2
[root@localhost ~]# ll /dev/disk/by-label/OS
lrwxrwxrwx. 1 root root 15 Dec  6 08:55 /dev/disk/by-label/OS -> ../../nvme0n1p2
[root@localhost ~]# dosfslabel /dev/nvme0n1p2 
OS
```

Regards

superm1, 2023-12-07:
> @diegorondini is the `Name` in `org.freedesktop.UDisks2.Partition` when running `udisksctl dump`? @superm1 do you think it makes sense to add to the ignorelist?

I could do one better.  Here's a relatively complete ignore list:
https://github.com/storaged-project/udisks/blob/master/data/80-udisks2.rules#L158

superm1, 2023-12-07:
I almost wonder if `UDISKS_IGNORE` would be worth using but I'm pretty sure it will catch real ESP's too then.

hughsie, 2023-12-07:
`UDISKS_IGNORE` doesn't work indeed as https://github.com/storaged-project/udisks/blob/master/data/80-udisks2.rules#L136 but `Recovery|RECOVERY|Lenovo_Recovery|HP_RECOVERY|Recovery_Partition|DellUtility|DellRestore|IBM_SERVICE|SERVICEV001|SERVICEV002|SYSTEM_RESERVED|System_Reserved|WINRE_DRV|DIAGS|IntelRST` makes sense to me. @superm1 shall I do a PR with all those?

hughsie, 2023-12-07:
e.g. like https://github.com/fwupd/fwupd/pull/6512 perhaps.

Forge64, 2025-03-11:
Could the amended/extended ignore list get considered/implemented please? Both new and existing installs of PopOS are triggering this message, as they create a "Recovery" partition which is *not* an actual ESP, but it's being IDed as such by fwupdmgr and triggering the complaint. I do not want the ESP flag set on that partition, or it treated as one, as it doesn't have a complete and usable ESP layout.
``16:16:31.237 FuVolume             mounting /org/freedesktop/UDisks2/block_devices/nvme2n1p1
16:16:31.343 FuPluginUefiCapsule  not a Linux ESP: did not find grubx64.efi|shimx64.efi|systemd-bootx64.efi in /boot/efi
16:16:31.343 FuVolume             unmounting /org/freedesktop/UDisks2/block_devices/nvme2n1p1
16:16:31.378 FuPluginUefiCapsule  not a Linux ESP: did not find grubx64.efi|shimx64.efi|systemd-bootx64.efi in /recovery
16:16:31.378 FuPluginUefiCapsule  more than one ESP possible:
 - 0xfff:	/org/freedesktop/UDisks2/block_devices/nvme2n1p4
 - 0x800:	/org/freedesktop/UDisks2/block_devices/nvme2n1p1
16:16:31.389 FuDevice             using a45df35ac0e948ee180fe216a5f703f32dda163f for /sys/firmware/efi/esrt/entries/entry0
16:16:31.389 FuPlugin             emit added from uefi_capsule: a45df35ac0e948ee180fe216a5f703f32dda163f``

hughsie, 2025-03-11:
@Forge64 what's your output of `udisksctl dump` please?

Forge64, 2025-03-13:
> [@Forge64](https://github.com/Forge64) what's your output of `udisksctl dump` please?

Attached, since it's quite a wall of text.

[diskdump.txt](https://github.com/user-attachments/files/19237454/diskdump.txt)

superm1, 2025-03-24:
I've moved our conversation to https://github.com/fwupd/fwupd/issues/8612, let's follow up there.
/lock