fwupd/fwupd #6055: Error while starting upgrade : text must be UTF-8

ELDiablO59152, 2023-08-04:
**Describe the bug**
I can't upgrade packages anymore, tells me that text must be UTF-8.

**Steps to Reproduce**
```shell
$ sudo fwupdmgr upgrade
Devices with no available firmware updates: 
 • UEFI Device Firmware
 • UEFI Device Firmware
Devices with the latest available firmware version:
 • KXG60ZNV1T02 NVMe KIOXIA 1024GB
╔══════════════════════════════════════════════════════════════════════════════╗
║ Upgrade System Firmware from 1.28.1 to 1.29.0?                               ║
╠══════════════════════════════════════════════════════════════════════════════╣
║ This stable release fixes the following issues:                              ║
║                                                                              ║
║ • This release contains security updates as disclosed in the Dell            ║
║ Security Advisories.                                                         ║
║ • Fixed the issue where the error 2000-0620 error message is displayed in    ║
║ SupportAssist.                                                               ║
║ • Fixed the issue where the system cannot power on when you plug in the      ║
║ AC adapter to the system.                                                    ║
║ • Fixed the issue where the system boots automatically when connected to     ║
║ a Dell Power Button Sync (DPBS) monitor.This issue occurs when the system    ║
║ enters Sleep mode.                                                           ║
║ • Fixed the issue where the system does not turn on after you update the     ║
║ BIOS.                                                                        ║
║                                                                              ║
║ Latitude 5420 must remain plugged into a power source for the duration of    ║
║ the update to avoid damage.                                                  ║
╚══════════════════════════════════════════════════════════════════════════════╝
Perform operation? [Y|n]: 
Decompressing…           [                                       ]
text must be UTF-8

$ echo $LANG
fr_FR.UTF-8
```

**Expected behavior**
Succeed to upgrade.

**fwupd version information**
Please provide the version of the daemon and client.

```shell
compile   org.freedesktop.fwupd         1.9.3
compile   com.hughsie.libxmlb           0.3.6
compile   com.hughsie.libjcat           0.1.9
runtime   org.freedesktop.fwupd-efi     1.4
compile   org.freedesktop.gusb          0.3.10
runtime   com.dell.libsmbios            2.4
runtime   org.freedesktop.gusb          0.3.10
runtime   org.kernel                    5.15.0-76-generic
runtime   org.freedesktop.fwupd         1.9.3
```

Please note how you installed it (`apt`, `dnf`, `pacman`, source, etc): apt

<details>

<summary>**fwupd device information**</summary>

Please provide the output of the fwupd devices recognized in your system.

```shell
Dell Inc. Latitude 5420
│
├─Unknown Device:
│     Device ID:          63223cc2d1b0b8d4c45ef8cedcb1f45315d3ff20
[... 126 lines cut ...]
│   
└─UEFI Device Firmware:
      Device ID:          2292ae5236790b47884e37cf162dcf23bfcd1c60
      Summary:            UEFI ESRT device
      Current version:    274743559
      Minimum Version:    274743559
      Vendor:             DMI:Dell Inc.
      Update State:       Success
      GUID:               344c2804-4948-e811-842f-0ed5f89f718b
      Device Flags:       • Internal device
                          • Updatable
                          • System requires external power source
                          • Needs a reboot after installation
                          • Device is usable for the duration of the update
    
────────────────────────────────────────────────
```

</details>

**Additional questions**

- Operating system and version: Ubuntu 22.04
- Have you tried rebooting? Of course in 2 months
- Is this a regression? I think so because it was working before

hughsie, 2023-08-04:
Hey. Can you attach the log from "sudo fwupdtool update -vv" to this issue please. I've got an idea where it's coming from but I'm a bit confused why.

ELDiablO59152, 2023-08-04:
I don't have fwupdtool but here is the output of fwupdmgr update -vv

```shell
$ sudo fwupdmgr update -vv
(/snap/fwupd/5039/bin/fwupdmgr:69634): GLib-DEBUG: 19:35:47.313: setenv()/putenv() are not thread-safe and should not be used after threads are created
(process:69662): GLib-DEBUG: 19:35:47.315: Failed to set thread scheduler attributes: Operation not permitted
(/snap/fwupd/5039/bin/fwupdmgr:69634): GLib-GIO-DEBUG: 19:35:47.330: _g_io_module_get_default: Found default implementation local (GLocalVfs) for ‘gio-vfs’
(/snap/fwupd/5039/bin/fwupdmgr:69634): GLib-GIO-DEBUG: 19:35:47.331: _g_io_module_get_default: Found default implementation keyfile (GKeyfileSettingsBackend) for ‘gsettings-backend’
(/snap/fwupd/5039/bin/fwupdmgr:69634): GLib-GIO-DEBUG: 19:35:47.331: _g_io_module_get_default: Found default implementation gnome (GProxyResolverGnome) for ‘gio-proxy-resolver’
(/snap/fwupd/5039/bin/fwupdmgr:69634): GLib-DEBUG: 19:35:47.331: Failed to set thread scheduler attributes: Operation not permitted
(/snap/fwupd/5039/bin/fwupdmgr:69634): Fwupd-DEBUG: 19:35:47.368: Emitting ::status-changed() [idle]
FuMain-INFO: 19:35:47.370: systemd is not accessible: GDBus.Error:org.freedesktop.DBus.Error.AccessDenied: An AppArmor policy prevents this sender from sending this message to this recipient; type="method_call", sender=":1.748" (uid=0 pid=69634 comm="/snap/fwupd/5039/bin/fwupdmgr update -vv " label="snap.fwupd.fwupdmgr (enforce)") interface="org.freedesktop.systemd1.Manager" member="GetDefaultTarget" error name="(unset)" requested_reply="0" destination=":1.5" (uid=0 pid=1 comm="/sbin/init " label="unconfined")
Devices with no available firmware updates: 
 • UEFI Device Firmware
 • UEFI Device Firmware
Devices with the latest available firmware version:
 • KXG60ZNV1T02 NVMe KIOXIA 1024GB
(/snap/fwupd/5039/bin/fwupdmgr:69634): FuMain-DEBUG: 19:35:47.377: current version is 10604107: 10604107=same, 10604106=older
╔══════════════════════════════════════════════════════════════════════════════╗
║ Upgrade System Firmware from 1.28.1 to 1.29.0?                               ║
╠══════════════════════════════════════════════════════════════════════════════╣
║ This stable release fixes the following issues:                              ║
║                                                                              ║
║ • This release contains security updates as disclosed in the Dell            ║
║ Security Advisories.                                                         ║
║ • Fixed the issue where the error 2000-0620 error message is displayed in    ║
║ SupportAssist.                                                               ║
║ • Fixed the issue where the system cannot power on when you plug in the      ║
║ AC adapter to the system.                                                    ║
║ • Fixed the issue where the system boots automatically when connected to     ║
║ a Dell Power Button Sync (DPBS) monitor.This issue occurs when the system    ║
║ enters Sleep mode.                                                           ║
║ • Fixed the issue where the system does not turn on after you update the     ║
║ BIOS.                                                                        ║
║                                                                              ║
║ Latitude 5420 must remain plugged into a power source for the duration of    ║
║ the update to avoid damage.                                                  ║
╚══════════════════════════════════════════════════════════════════════════════╝
Perform operation? [Y|n]: 
Fwupd-INFO: 19:35:52.623: downloading https://fwupd.org/downloads/269d2226a11d7680b89c946837afb969870e06fe5924870b2f3fc3be97585b64-Latitude_5420_TGL_1.29.0.cab
(/snap/fwupd/5039/bin/fwupdmgr:69634): Fwupd-DEBUG: 19:35:52.623: Emitting ::status-changed() [downloading]
Downloading…             [                            /          ]Fwupd-INFO: 19:35:53.738: download progress: 100%
Downloading…             [***************************************]Fwupd-INFO: 19:35:56.887: download progress: 0%
Downloading…             [  -                                    ]Fwupd-INFO: 19:35:56.942: download progress: 1%
Downloading…             [                                       ]Fwupd-INFO: 19:35:56.971: download progress: 2%
Downloading…             [                                       ]Fwupd-INFO: 19:35:56.994: download progress: 3%
Downloading…             [*                                      ]Fwupd-INFO: 19:35:57.011: download progress: 4%
Downloading…             [*                                      ]Fwupd-INFO: 19:35:57.032: download progress: 5%
Downloading…             [*                                      ]Fwupd-INFO: 19:35:57.063: download progress: 6%
Downloading…             [**                                     ]Fwupd-INFO: 19:35:57.093: download progress: 7%
Downloading…             [**                                     ]Fwupd-INFO: 19:35:57.115: download progress: 8%
Downloading…             [***                                    ]Fwupd-INFO: 19:35:57.136: download progress: 9%
Downloading…             [***                                    ]Fwupd-INFO: 19:35:57.155: download progress: 10%
Downloading…             [***                                    ]Fwupd-INFO: 19:35:57.179: download progress: 11%
Downloading…             [****                                   ]Fwupd-INFO: 19:35:57.191: download progress: 12%
[... cut ...]
(/snap/fwupd/5039/bin/fwupdmgr:69634): Fwupd-DEBUG: 19:35:59.791: Emitting ::device-changed(a45df35ac0e948ee180fe216a5f703f32dda163f)
(/snap/fwupd/5039/bin/fwupdmgr:69634): Fwupd-DEBUG: 19:35:59.794: Emitting ::device-changed(a45df35ac0e948ee180fe216a5f703f32dda163f)
(/snap/fwupd/5039/bin/fwupdmgr:69634): Fwupd-DEBUG: 19:35:59.794: Emitting ::device-changed(a45df35ac0e948ee180fe216a5f703f32dda163f)
(/snap/fwupd/5039/bin/fwupdmgr:69634): Fwupd-DEBUG: 19:35:59.797: Emitting ::device-changed(a45df35ac0e948ee180fe216a5f703f32dda163f)

text must be UTF-8
```

ELDiablO59152, 2023-08-10:
> Can you attach the contents of /sys/firmware/efi/efivars/SbatLevelRT-605dab50-e046-4300-abb6-3dd810dd8b23 to this issue please.

Sure, here it is:

```shell
$ cat /sys/firmware/efi/efivars/SbatLevelRT-605dab50-e046-4300-abb6-3dd810dd8b23
sbat,1,2022052400
grub,2
```

> @ELDiablO59152 if you can use a snap package of the latest main branch (e.g. with ^^) the error message should be a lot more helpful.

What do you me by "latest main branch" ?
```shell
$ snap info fwupd
name:      fwupd
summary:   Firmware updates for Linux
publisher: Richard Hughes
store-url: https://snapcraft.io/fwupd
contact:   https://github.com/fwupd/fwupd
license:   unset
description: |
  The full fwupd firmware updating stack for Linux.  Can be used to perform updates on all supported
  devices via https://fwupd.org
commands:
  - fwupd.dbxtool
  - fwupd.fwupdmgr
  - fwupd.fwupdtool
services:
  fwupd: dbus, enabled, active
snap-id:      HpOj37PuyuaMUZY0NQhtwnp7oS5P8u5R
tracking:     latest/stable
refresh-date: Il y a 28 jours, à 08 h 59 HNR
channels:
  latest/stable:    1.9.3               2023-07-12 (5039) 24MB -
  latest/candidate: 1.9.3               2023-07-11 (5039) 24MB -
  latest/beta:      ↑                                          
  latest/edge:      1.9.3-72-gf11590bd3 2023-08-10 (5085) 24MB -
installed:          1.9.3                          (5039) 24MB -
```
Are you talking about the edge one ?

ELDiablO59152, 2023-08-10:
@hughsie that's interesting

```shell
$ sudo fwupdmgr update -vv
(/snap/fwupd/5086/bin/fwupdmgr:15758): GLib-DEBUG: 09:49:14.625: setenv()/putenv() are not thread-safe and should not be used after threads are created
(process:15790): GLib-DEBUG: 09:49:14.628: Failed to set thread scheduler attributes: Operation not permitted
(/snap/fwupd/5086/bin/fwupdmgr:15758): GLib-GIO-DEBUG: 09:49:14.643: _g_io_module_get_default: Found default implementation local (GLocalVfs) for ‘gio-vfs’
(/snap/fwupd/5086/bin/fwupdmgr:15758): GLib-GIO-DEBUG: 09:49:14.643: _g_io_module_get_default: Found default implementation keyfile (GKeyfileSettingsBackend) for ‘gsettings-backend’
(/snap/fwupd/5086/bin/fwupdmgr:15758): GLib-GIO-DEBUG: 09:49:14.644: _g_io_module_get_default: Found default implementation gnome (GProxyResolverGnome) for ‘gio-proxy-resolver’
(/snap/fwupd/5086/bin/fwupdmgr:15758): GLib-DEBUG: 09:49:14.644: Failed to set thread scheduler attributes: Operation not permitted
(/snap/fwupd/5086/bin/fwupdmgr:15758): Fwupd-DEBUG: 09:49:14.647: Emitting ::status-changed() [idle]
WARNING: This package has not been validated, it may not work properly.
FuMain-INFO: 09:49:14.650: systemd is not accessible: GDBus.Error:org.freedesktop.DBus.Error.AccessDenied: An AppArmor policy prevents this sender from sending this message to this recipient; type="method_call", sender=":1.215" (uid=0 pid=15758 comm="/snap/fwupd/5086/bin/fwupdmgr update -vv " label="snap.fwupd.fwupdmgr (enforce)") interface="org.freedesktop.systemd1.Manager" member="GetDefaultTarget" error name="(unset)" requested_reply="0" destination=":1.6" (uid=0 pid=1 comm="/sbin/init " label="unconfined")
Devices with no available firmware updates: 
 • UEFI Device Firmware
 • UEFI Device Firmware
Devices with the latest available firmware version:
 • KXG60ZNV1T02 NVMe KIOXIA 1024GB
(/snap/fwupd/5086/bin/fwupdmgr:15758): FuMain-DEBUG: 09:49:14.656: current version is 10604107: 10604107=same, 10604106=older
╔══════════════════════════════════════════════════════════════════════════════╗
║ Upgrade System Firmware from 1.28.1 to 1.29.0?                               ║
╠══════════════════════════════════════════════════════════════════════════════╣
║ This stable release fixes the following issues:                              ║
║                                                                              ║
║ • This release contains security updates as disclosed in the Dell            ║
║ Security Advisories.                                                         ║
║ • Fixed the issue where the error 2000-0620 error message is displayed in    ║
║ SupportAssist.                                                               ║
║ • Fixed the issue where the system cannot power on when you plug in the      ║
║ AC adapter to the system.                                                    ║
║ • Fixed the issue where the system boots automatically when connected to     ║
║ a Dell Power Button Sync (DPBS) monitor.This issue occurs when the system    ║
║ enters Sleep mode.                                                           ║
║ • Fixed the issue where the system does not turn on after you update the     ║
║ BIOS.                                                                        ║
║                                                                              ║
║ Latitude 5420 must remain plugged into a power source for the duration of    ║
║ the update to avoid damage.                                                  ║
╚══════════════════════════════════════════════════════════════════════════════╝
Perform operation? [Y|n]: 
Fwupd-INFO: 09:49:31.335: downloading https://fwupd.org/downloads/269d2226a11d7680b89c946837afb969870e06fe5924870b2f3fc3be97585b64-Latitude_5420_TGL_1.29.0.cab
(/snap/fwupd/5086/bin/fwupdmgr:15758): Fwupd-DEBUG: 09:49:31.335: Emitting ::status-changed() [downloading]
Downloading…             [                          |            ]Fwupd-INFO: 09:49:32.379: download progress: 100%
Downloading…             [***************************************]Fwupd-INFO: 09:49:32.512: download progress: 0%
Downloading…             [  -                                    ]Fwupd-INFO: 09:49:32.552: download progress: 1%
Downloading…             [                                       ]Fwupd-INFO: 09:49:32.567: download progress: 2%
Downloading…             [                                       ]Fwupd-INFO: 09:49:32.581: download progress: 3%
Downloading…             [*                                      ]Fwupd-INFO: 09:49:32.592: download progress: 4%
Downloading…             [*                                      ]Fwupd-INFO: 09:49:32.613: download progress: 5%
Downloading…             [*                                      ]Fwupd-INFO: 09:49:32.628: download progress: 6%
Downloading…             [**                                     ]Fwupd-INFO: 09:49:32.643: download progress: 7%
Downloading…             [**                                     ]Fwupd-INFO: 09:49:32.657: download progress: 8%
Downloading…             [***                                    ]Fwupd-INFO: 09:49:32.672: download progress: 9%
Downloading…             [***                                    ]Fwupd-INFO: 09:49:32.686: download progress: 10%
Downloading…             [***                                    ]Fwupd-INFO: 09:49:32.696: download progress: 11%
Downloading…             [****                                   ]Fwupd-INFO: 09:49:32.713: download progress: 12%
[... cut ...]
(/snap/fwupd/5086/bin/fwupdmgr:15758): Fwupd-DEBUG: 09:49:34.896: Emitting ::device-changed(a45df35ac0e948ee180fe216a5f703f32dda163f)
(/snap/fwupd/5086/bin/fwupdmgr:15758): Fwupd-DEBUG: 09:49:34.900: Emitting ::device-changed(a45df35ac0e948ee180fe216a5f703f32dda163f)
(/snap/fwupd/5086/bin/fwupdmgr:15758): Fwupd-DEBUG: 09:49:34.904: Emitting ::device-changed(a45df35ac0e948ee180fe216a5f703f32dda163f)
(/snap/fwupd/5086/bin/fwupdmgr:15758): Fwupd-DEBUG: 09:49:34.904: Emitting ::status-changed() [idle]

failed to load /snap/fwupd/5086/libexec/fwupd/efi/shimx64.efi.signed: failed to parse .sbat: text must be UTF-8
```

hughsie, 2023-08-10:
> /snap/fwupd/5086/libexec/fwupd/efi/shimx64.efi.signed

Can you attach that to this issue please. You might need to add it to a password-protected zip file so that github will let you attach it.

ELDiablO59152, 2023-08-16:
@hughsie Apparently, Github seams to accept it as is.
Thanks for your help, here it is : [shimx64.efi.signed.tar.gz](https://github.com/fwupd/fwupd/files/12357119/shimx64.efi.signed.tar.gz)

PR #6091 (2023-08-16): fu-pefile-firmware: Load effective size of section instead of raw size
Type of pull request:

- [ ] New plugin (Please include [new plugin checklist](https://github.com/fwupd/fwupd/wiki/New-plugin-checklist))
- [x] Code fix
- [ ] Feature
- [ ] Documentation

Fixes #6055
closed 2023-08-16 by hughsie