fwupd/fwupd #6056: fwupd >1.9.1 is broken on my system

Rustmilian, 2023-08-05:
**Describe the bug**
fwupd versions above 1.9.1 no longer show relevant HSI info. 

**Steps to Reproduce**
install fwupd >= 1.9.2

**Expected behavior**
show relevant HSI info

Please note how you installed it (`apt`, `dnf`, `pacman`, source, etc):
pacman

<details>

<summary>**fwupd device information**</summary>

```
HP HP Laptop 14-fq0xxx
│
├─AMD Athlon Silver 3050U with Radeon Graphics:
│ │   Device ID:          4bde70ba4e39b28f9eab1628f9dd6e6244c03027
│ │   Current version:    0x08108109
│ │   Vendor:             Advanced Micro Devices, Inc.
│ │   Problems:           • System power is too low to perform the update
│ │   GUIDs:              b9a2dd81-159e-5537-a7db-e7101d164d3f ← cpu
│ │                       997104da-ec11-5e04-9d94-ac5501fbf609 ← CPUID\PRO_0&FAM_17
│ │                       38cd08d3-c885-5730-bff8-fdf71774a4a2 ← CPUID\PRO_0&FAM_17&MOD_18
│ │                       cbe5ceff-2b79-5935-9e79-ad728587b62f ← CPUID\PRO_0&FAM_17&MOD_18&STP_1
│ │   Device Flags:       • Internal device
│ │
│ ├─GPIO controller:
│ │     Device ID:        f685512aa07369c9e77742acef941d779d31e766
│ │     Problems:         • System power is too low to perform the update
│ │     GUID:             37b440a9-2473-5087-a39b-db84f32a8ed8 ← GPIO\ID_AMDI0030:00
│ │
│ ├─Graphics Processing Unit (GPU):
│ │     Device ID:        3743975ad7f64f8d6575a9ae49fb3a8856fe186f
│ │     Current version:  113-RAVEN2-117
│ │     Vendor:           Advanced Micro Devices, Inc. [AMD/ATI] (PCI:0x1002)
│ │     Problems:         • System power is too low to perform the update
│ │     GUIDs:            765e82f6-d884-59a2-b73c-e3d4960090c4 ← PCI\VEN_1002&DEV_15D8
│ │                       2c2506d9-f37a-566d-972e-4f15807ea02a ← PCI\VEN_1002&DEV_15D8&REV_CD
│ │                       5ef8e6cc-b4e3-59d1-8534-60aea4d6dc45 ← PCI\VEN_1002&DEV_15D8&SUBSYS_103C87B7
│ │                       1c19d82b-6d63-52dd-9178-9ca84e9b2f61 ← PCI\VEN_1002&DEV_15D8&SUBSYS_103C87B7&REV_CD
│ │     Device Flags:     • Internal device
│ │
│ └─Secure Processor:
│       Device ID:        c54ab0237d7a8db8c717b68e0be78e4374a2a079
│       Vendor:           Advanced Micro Devices, Inc. (PCI:0x1022)
│       Problems:         • System power is too low to perform the update
│       GUIDs:            0e8dc554-a0a2-51fb-b439-1eb72b14ec38 ← PCI\VEN_1022&DEV_15DF
│                         1adad2a8-21d1-5354-ae84-b68876203f50 ← PCI\VEN_1022&DEV_15DF&REV_00
│                         b87ead33-97cb-5aa1-9d0a-99853a7df8fd ← PCI\VEN_1022&DEV_15DF&SUBSYS_103C87B7
│                         d0d5d79a-983e-528a-982e-4ae527ec1ddb ← PCI\VEN_1022&DEV_15DF&SUBSYS_103C87B7&REV_00
│       Device Flags:     • Internal device
│
├─MZNLH128HBHQ-000H1:
│     Device ID:          ff5bf9f6641211b248157b73007e1b638b3f0a05
│     Summary:            ATA drive
│     Current version:    RVT22H0Q
│     Vendor:             Samsung (ATA:0x144D, OUI:002538)
│     Serial Number:      S4HENX0N783407
│     Problems:           • System power is too low to perform the update
│     GUIDs:              1eb31b07-56c1-5716-9bda-1777ba8adf9f ← IDE\SAMSUNG_MZNLH128HBHQ-000H1______________RVT22H0Q
│                         206bee03-dac0-5aa6-b638-3f1a84d6253d ← IDE\0SAMSUNG_MZNLH128HBHQ-000H1______________
│                         2e2b6b01-2023-519c-8063-c92cade85e74 ← SAMSUNG MZNLH128HBHQ-000H1
│     Device Flags:       • Internal device
│                         • System requires external power source
│                         • Needs a reboot after installation
│                         • Device is usable for the duration of the update
│                         • Updatable
│
├─System Firmware:
│ │   Device ID:          a45df35ac0e948ee180fe216a5f703f32dda163f
│ │   Summary:            UEFI ESRT device
│ │   Current version:    258342912
│ │   Minimum Version:    258342912
│ │   Vendor:             HP (DMI:AMI)
│ │   Update State:       Success
│ │   Problems:           • System power is too low to perform the update
│ │   GUIDs:              0a77b8ba-3848-4478-9923-f6029b7dbcc1
│ │                       230c8b18-8d9b-53ec-838b-6cfc0383493a ← main-system-firmware
│ │   Device Flags:       • Internal device
│ │                       • System requires external power source
│ │                       • Needs a reboot after installation
│ │                       • Cryptographic hash verification is available
│ │                       • Device is usable for the duration of the update
│ │                       • Updatable
│ │
│ ├─UEFI Platform Key:
│ │     Device ID:        6924110cde4fa051bfdc600a60620dc7aa9d3c6a
│ │     Summary:          HP Inc. PK 2016 CA
│ │     Vendor:           HP Inc.
│ │     Problems:         • System power is too low to perform the update
│ │     GUID:             77c6cc94-a10e-5208-916d-2a805c661547 ← UEFI\CRT_9FD5D4B39F4374A3C6CAA212D4007AB905C58591
│ │
│ └─UEFI dbx:
│       Device ID:        362301da643102b9f38477387e2193e57abaa590
│       Summary:          UEFI revocation database
│       Current version:  371
│       Minimum Version:  371
│       Vendor:           UEFI:Linux Foundation
│       Install Duration: 1 second
│       Problems:         • System power is too low to perform the update
│       GUIDs:            95641a6b-bc03-5b8b-bdfc-c92dab523a06 ← UEFI\CRT_C8B4C6D9755968B6DEB48D4768DD5E9020D3FF4ADC9A62521F54252DD4241678
│                         7689caf4-c147-5c67-bff9-5dbe59a441bd ← UEFI\CRT_C8B4C6D9755968B6DEB48D4768DD5E9020D3FF4ADC9A62521F54252DD4241678&ARCH_X64
│                         c6682ade-b5ec-57c4-b687-676351208742 ← UEFI\CRT_A1117F516A32CEFCBA3F2D1ACE10A87972FD6BBE8FE0D0B996E09E65D802A503
│                         f8ba2887-9411-5c36-9cee-88995bb39731 ← UEFI\CRT_A1117F516A32CEFCBA3F2D1ACE10A87972FD6BBE8FE0D0B996E09E65D802A503&ARCH_X64
│       Device Flags:     • Internal device
│                         • Supported on remote server
│                         • Needs a reboot after installation
│                         • Device is usable for the duration of the update
│                         • Updatable
│                         • Only version upgrades are allowed
│                         • Signed Payload
│
└─TPM:
      Device ID:          c6a80ac3a22083423992a3cb15018989f37834d6
      Summary:            TPM 2.0 Device
      Current version:    3.78.0.4
      Vendor:             Advanced Micro Devices, Inc. (TPM:AMD)
      Problems:           • System power is too low to perform the update
      GUIDs:              ff71992e-52f7-5eea-94ef-883e56e034c6 ← system-tpm
                          9305de1c-1e12-5665-81c4-37f8e51219b8 ← TPM\VEN_AMD&DEV_0001
                          78a291ae-b499-5b0f-8f1d-74e1fefd0b1c ← TPM\VEN_AMD&MOD_AMD
                          65a3fced-b423-563f-8098-bf5c329fc063 ← TPM\VEN_AMD&DEV_0001&VER_2.0
                          5e704f0d-83cb-5364-8384-f46d725a23b8 ← TPM\VEN_AMD&MOD_AMD&VER_2.0
      Device Flags:       • Internal device
                          • System requires external power source
                          • Needs a reboot after installation
                          • Device can recover flash failures
                          • Full disk encryption secrets may be invalidated when updating
                          • Signed Payload
```

</details>

**Additional questions**

- Operating system and version:
Arch, Rolling
- Have you tried rebooting?
Yes
- Is this a regression?
Yes

**Additional info**
fwupd v1.9.2 gives no HSI info.
fwupd v1.9.3 gives no HSI info & the following error : 
```
HSI attribute org.fwupd.hsi.SupportedCpu (from pci_psp) had unknown result
```

superm1, 2023-08-05:
Your system doesn't provide enough data to make a calculation. This is controlled by the platform firmware.

If it was working previously I suspect you needed to use --force.

Rustmilian, 2023-08-05:
> Your system doesn't provide enough data to make a calculation. This is controlled by the platform firmware.
> 
> If it was working previously I suspect you needed to use --force.

It was working in <=1.9.1, `--force` doesn't work.

superm1, 2023-08-05:
Can you please provide the output for 1.9.1?

Rustmilian, 2023-08-05:
gladly, v1.9.1 output below.
```
HSI-1
✔ TPM empty PCRs:                Valid
✔ TPM v2.0:                      Found
✔ UEFI platform key:             Valid
✘ Supported CPU:                 Failed

HSI-2
✔ IOMMU:                         Enabled
✔ TPM PCR0 reconstruction:       Valid

HSI-3
✘ Pre-boot DMA protection:       Disabled
✘ Suspend-to-idle:               Disabled
✘ Suspend-to-ram:                Enabled

HSI-4
✘ Encrypted RAM:                 Not supported

Runtime Suffix -!
✔ Linux kernel:                  Untainted
✔ Linux kernel lockdown:         Enabled
✔ Linux swap:                    Encrypted
✔ fwupd plugins:                 Untainted
✘ UEFI secure boot:              Disabled

This system has a low HSI security level.
 » https://fwupd.github.io/hsi.html#low-security-level

This system has HSI runtime issues.
 » https://fwupd.github.io/hsi.html#hsi-runtime-suffix
```

superm1, 2023-08-05:
Okay, so it's an unsupported CPU in 1.9.1, this is just confirming it. And as a result you can't get a score.

closed 2023-08-05 by superm1

Rustmilian, 2023-08-05:
> Okay, so it's an unsupported CPU in 1.9.1, this is just confirming it. And as a result you can't get a score.

Why change it?

superm1, 2023-08-05:
We removed force support in newer releases, because it was sending the wrong message to people using it.

You can't get a calculation unless that plugin reports data.

Rustmilian, 2023-08-05:
well then, I guess I'll uninstall it and remove it from kinfocenter.