fwupd/fwupd #6057: Regression: `fwupdtool firmware-patch` requires root permission to edit any dfu file

mobilinkd, 2023-08-05:
**Describe the bug**
`fwupdtool firmware-patch` now requires root permission to edit dfu files.

from #4142, this should not occur and appears to be a regression.

**Steps to Reproduce**

 1. Install fwupd-1.9.3-2.fc38.x86_64 RPM package.
 1. Create a dfu file.
 1. Attempt to use `fwupdtool firmware-patch` as a non-root user.

    $ fwupdtool firmware-patch ${filename} 293 ${next_serial_number} dfu
    Loading…                 [ -                                     ]
    Failed to load config: Failed to open file “/etc/fwupd/fwupd.conf”: Permission denied
    WARNING: This program may only work correctly as root

**Expected behavior**
`fwupdtool firmware-patch` runs successfully and the file is updated as specified when the file being updated is owned and writeable by the user running fwupdtool.

There does not appear to be any need to access fwupd.conf when modifying dfu files.

**fwupd version information**

```shell
$ fwupdmgr --version
compile   org.freedesktop.fwupd         1.9.3
compile   com.hughsie.libxmlb           0.3.11
compile   com.hughsie.libjcat           0.1.13
runtime   org.freedesktop.fwupd-efi     1.4
compile   org.freedesktop.gusb          0.4.5
runtime   com.hughsie.libjcat           0.1.13
runtime   org.freedesktop.gusb          0.4.5
runtime   org.freedesktop.fwupd         1.9.3
runtime   org.kernel                    6.4.6-200.fc38.x86_64
```

Please note how you installed it (`apt`, `dnf`, `pacman`, source, etc):

`dnf install fwupd-1.9.3-2.fc38.x86_64`

<summary>**fwupd device information**</summary>

Please provide the output of the fwupd devices recognized in your system.

Not applicable as I am not using fwupdtool to modify any device in this case.

**Additional questions**

- Operating system and version: Fedora 38
- Have you tried rebooting? This is not a problem that rebooting will address.
- Is this a regression? Yes, see #4142; note the comment about the warning issued.

superm1, 2023-08-05:
> 2. Create a dfu file.
> 3. Attempt to use fwupdtool firmware-patch as a non-root user.

I believe this should be fixed by https://github.com/fwupd/fwupd/commit/6f50e5814983f5fbad867b73a8d0e2a251416dd8 <https://github.com/fwupd/fwupd/commit/6f50e5814983f5fbad867b73a8d0e2a251416dd8> in the next release.

Sent with Shortwave <https://www.shortwave.com?utm_medium=email&utm_content=signature&utm_source=c3VwZXJtMUBnbWFpbC5jb20=>

hughsie, 2023-08-22:
Agreed, thanks.

closed 2023-08-22 by hughsie