fwupd/fwupd #6061: vli: ninja build error

memily, 2023-08-09:
**Describe the question**
Salutations, its us.

While inconspicuously adding three new firmware types (817S, 822T, 122), we ran into ninja build errors for each new type. The errors are similarly:

` 'FU_VLI_DEVICE_KIND_VL817S' undeclared (first use in this function); did you mean 'FU_VLI_DEVICE_KIND_VL817' `

We’ve attached the meson-log.txt and output of the command “ninja -v -C builddir” below. Perhaps you’ve encountered this before? Nevertheless it’s not an urgent issue for someone who’s just back from vacation. 

[meson-log.txt](https://github.com/fwupd/fwupd/files/12299267/meson-log.txt)

[ninja-log.txt](https://github.com/fwupd/fwupd/files/12299270/ninja-log.txt)

On a different note, Sherlock is wondering if fwupd has support for hid devices.
Thanks!

**fwupd version information**
Most recent pull from the codebase.

hughsie, 2023-08-09:
> Sherlock is wondering if fwupd has support for hid devices

Yup, about half the plugins in fwupd derive from FuHidDevice.

memily, 2023-08-09:
> Hey Emily. Did you add them to `plugins/vli/fu-vli.rs` ?

Good catch, this is probably the issue. I'll make the necessary fixes and push. 

> Yup, about half the plugins in fwupd derive from FuHidDevice.

We've had OEMs making inquiries so Sherlock was worried.

hughsie, 2023-08-09:
Sherlock is always worried :) Let me know the protocol like last time and I can give you some advice.

memily, 2023-08-09:
> Sherlock is always worried :) Let me know the protocol like last time and I can give you some advice.

That and coffee is what is holding him together... We'll keep you posted, but no specific device so far!

memily, 2023-08-09:
Failed on git push, attached test-log below. Running ninja build passed, meson test passed, and meson --reconfigure passed before git push.

[testlog.txt](https://github.com/fwupd/fwupd/files/12299971/testlog.txt)

memily, 2023-08-09:
Not the self tests but yes sudo on meson and ninja, I tried your suggestion, meson --wipe and the tests are still failing. 

`Bail out: ERROR: ../libfwupdplugin/fu-self-test.c:1232:fu_common_bytes_get_data_func: assertion failed (error == NULL): Failed to rename file ?/tmp/fwupdzero.0JVH91? to ?/tmp/fwupdzero?: g_rename() failed: Operation not permitted (g-file-error-quark, 22)`

New testlog attached. I must theatrically remind you that we are clowns, and thank you for your patience. 

[testlog.txt](https://github.com/fwupd/fwupd/files/12300168/testlog.txt)

memily, 2023-08-09:
Another round, "failed to push some refs to 'https:githube.com/fwupd/fwupd.git", new testlog after reboot.

[testlog.txt](https://github.com/fwupd/fwupd/files/12300304/testlog.txt)

hughsie, 2023-08-09:
Okay, tests look good now. What command are you using to push?

memily, 2023-08-09:
Worst case I save the files on the branch, git clone the repo, and try again from scratch.

 `git push -u origin wip/emily/vli-fw-cases`

hughsie, 2023-08-09:
Is it possible you committed with sudo? e.g. maybe a `sudo chown emily:emily .git -R` would fix things?

memily, 2023-08-09:
Don't think so. I tried variations of `sudo chown root:username .git -R` and looked at `ls -l`. I have read, write, and execute permissions but git push still fails. I'm going to put a pin in it for today and try from scratch tomorrow!

hughsie, 2023-08-09:
If it helps, sometimes I screw up git so completely I resort to a fresh clone.

closed 2023-08-10 by memily