fwupd/fwupd #6077: fwupdmgr stops if wrong input given for prompt

nullr0ute, 2023-08-11:
**Describe the bug**
When checking for updates if the wrong input is given for the upload of the firmware reports fwupdmgr doesn't error or re-prompt for valid input but rather basically just sits there for ever.

**Steps to Reproduce**
```
# fwupdmgr get-updates
<snip a bunch of output>
Uploading firmware reports helps hardware vendors to quickly identify failing and successful updates on real devices.
Upload report now? (Requires internet connection) [Y|n]: m

```

**Expected behavior**
I would expect it to print an error and re-prompt for valid input

**fwupd version information**
```
# fwupdmgr --version
compile   org.freedesktop.fwupd         1.9.3
compile   com.hughsie.libxmlb           0.3.11
compile   com.hughsie.libjcat           0.1.13
runtime   org.freedesktop.fwupd-efi     1.4
compile   org.freedesktop.gusb          0.4.5
runtime   com.hughsie.libjcat           0.1.13
runtime   org.freedesktop.gusb          0.4.5
runtime   org.freedesktop.fwupd         1.9.3
runtime   org.kernel                    6.4.7-200.fc38.x86_64
```

Please note how you installed it (`apt`, `dnf`, `pacman`, source, etc):

dnf, standard Fedora 38 install.

Please provide the output of the fwupd devices recognized in your system.

The devices are irrelevant, could recreate this across a number of both x86_64 and aarch64 devices.

**Additional questions**

- Operating system and version:
Fedora 38
- Have you tried rebooting?
Yes
- Is this a regression?
No

hughsie, 2023-08-11:
Weird! Did you do "m<enter>"?

nullr0ute, 2023-08-11:
> Weird! Did you do "m[enter]"?

Yep, and then I tried p<enter>. I did m originally because I wasn't paying attention.

nullr0ute, 2023-08-11:
I have to Ctrl+C a few times to get it to eventually quit.

closed 2023-08-11 by superm1