fwupd/fwupd #6101: GNOME Software shows failed to build result for 362301da643102b9f38477387e2193e57abaa590

bbhtt, 2023-08-18:
**Describe the bug**

After upgrading from Fedora 38 to Fedora 39 beta, GNOME software on each login shows the following warning:

[image]

Searching for the id and lokking at https://gitlab.gnome.org/GNOME/gnome-software/-/issues/1163 suggests it is coming from fwupd, but none of the fwupdmgr cli commands shows anything actionable. I've tried reinstalling it with `sudo fwupdmgr reinstall`, but the error comes up again on next restart.

What's the cause and what would be the way to resolve it?

Thanks!

**Steps to Reproduce**

Upgrade from Fedora 38 to 39 beta

**Expected behavior**

Shouldn't show any warning

**fwupd version information**
Please provide the version of the daemon and client.

```shell
fwupdmgr --version
compile   org.freedesktop.fwupd         1.9.3
compile   com.hughsie.libxmlb           0.3.11
compile   com.hughsie.libjcat           0.1.14
runtime   org.freedesktop.fwupd-efi     1.4
compile   org.freedesktop.gusb          0.4.6
runtime   com.hughsie.libjcat           0.1.14
runtime   org.freedesktop.gusb          0.4.6
runtime   org.freedesktop.fwupd         1.9.3
runtime   org.kernel                    6.5.0-0.rc6.43.fc39.x86_64
```

Please note how you installed it (`apt`, `dnf`, `pacman`, source, etc): dnf

<details>

<summary>**fwupd device information**</summary>

Please provide the output of the fwupd devices recognized in your system.

```shell
fwupdmgr get-devices --show-all-devices
Gigabyte Technology Co., Ltd. B450M DS3H WIFI
│
├─Unknown Device:
│     Device ID:          a29de8124928f7dc31db026069ac1d601a56056f
│     GUID:               2d9ac22e-0ffe-5691-9e21-f9fecce8761c ← GPIO\ID_AMDIF030:00
│   
├─400 Series Chipset SATA Controller:
│     Device ID:          7d29f2075dcafb4488b40c73f199cf46bb76bddb
│     Current version:    01
│     Vendor:             Advanced Micro Devices, Inc. [AMD] (PCI:0x1022)
│     GUIDs:              40dd51ce-4d2b-5863-a55b-e5e05b828e18 ← PCI\VEN_1022&DEV_43C8
│                         b315bc76-8da0-5bc6-82e8-296673694802 ← PCI\VEN_1022&DEV_43C8&SUBSYS_1B211062
│                         abcaf57e-ca4a-5f11-8dde-8b75cb993f28 ← PCI\VEN_1022&DEV_1634
│                         467ef116-7254-5747-9b3a-7e92b68def61 ← PCI\VEN_1022&DEV_1634&SUBSYS_10221453
│     Device Flags:       • Internal device
│                         • Cryptographic hash verification is available
│   
├─AMD Ryzen 5 5600G with Radeon Graphics:
│ │   Device ID:          4bde70ba4e39b28f9eab1628f9dd6e6244c03027
│ │   Current version:    0x0a50000d
│ │   Vendor:             Advanced Micro Devices, Inc.
│ │   GUIDs:              b9a2dd81-159e-5537-a7db-e7101d164d3f ← cpu
│ │                       22f9ecf4-588d-5c0a-8326-6ebff3655c6d ← CPUID\PRO_0&FAM_19
│ │                       79759cdc-94db-5098-be7b-eb02521fbbec ← CPUID\PRO_0&FAM_19&MOD_50
[... 145 lines cut ...]
|-- loader
|   `-- entries
|       |-- ce40a64afb9c4524bd6a0f238683ea9a-0-rescue.conf
|       |-- ce40a64afb9c4524bd6a0f238683ea9a-6.4.10-200.fc38.x86_64.conf
|       |-- ce40a64afb9c4524bd6a0f238683ea9a-6.4.9-200.fc38.x86_64.conf
|       `-- ce40a64afb9c4524bd6a0f238683ea9a-6.5.0-0.rc6.43.fc39.x86_64.conf
|-- lost+found
|-- symvers-6.4.10-200.fc38.x86_64.xz -> /lib/modules/6.4.10-200.fc38.x86_64/symvers.xz
|-- symvers-6.4.9-200.fc38.x86_64.xz -> /lib/modules/6.4.9-200.fc38.x86_64/symvers.xz
|-- symvers-6.5.0-0.rc6.43.fc39.x86_64.xz -> /lib/modules/6.5.0-0.rc6.43.fc39.x86_64/symvers.xz
|-- vmlinuz-0-rescue-ce40a64afb9c4524bd6a0f238683ea9a
|-- vmlinuz-6.4.10-200.fc38.x86_64
|-- vmlinuz-6.4.9-200.fc38.x86_64
`-- vmlinuz-6.5.0-0.rc6.43.fc39.x86_64

```

**Additional questions**

- Operating system and version: Fedora 39 beta
- Have you tried rebooting? Yes
- Is this a regression? No
- Are you using an NVMe disk? Yes
- Is secure boot enabled? Yes
- Is this a Lenovo system with 'Boot Order Lock' turned on in the BIOS? No

CleoMenezesJr, 2023-08-21:
Facing the same here.

hughsie, 2023-08-21:
Hmm, device 362301da643102b9f38477387e2193e57abaa590 clearly exists in the fwupdmgr output. The `Failed to build result` string comes from `plugins/fwupd/gs-plugin-fwupd.c` in gnome-software. The latter does `fwupd_client_get_results()` and gets a valid `FwupdDevice` and then calls `gs_plugin_fwupd_new_app_from_device()`. I think the only way this can fail is if somehow the AppStream ID is NULL -- I wonder if you're hitting https://github.com/fwupd/fwupd/commit/3974ea526fbfa93375fdce06719c961ff5b4025e too.

Two ways to test -- either downgrade libxmlb, or better, can you test the package in https://copr.fedorainfracloud.org/coprs/rhughes/fwupd/ please?

bbhtt, 2023-08-21:
> Two ways to test -- either downgrade libxmlb, or better, can you test the package in https://copr.fedorainfracloud.org/coprs/rhughes/fwupd/ please?

I downgraded linxmlb to `0.3.11-3.fc39` manually from koji and also tried `fwupd-1.9.4-0.1034.20230814git.fc39.x86_64` from your COPR. Neither seems to work. On restart or doing `gnome-software --quit`, `gnome-software` I get the popup again.

Thanks!

hughsie, 2023-08-21:
Can you attach the output of `fwupdmgr get-history --json` please?

bbhtt, 2023-08-21:
> Can you attach the output of `fwupdmgr get-history --json` please?

<details>
<summary> Output </summary>

```json
{
  "Devices" : [
    {
      "Name" : "UEFI dbx",
      "DeviceId" : "362301da643102b9f38477387e2193e57abaa590",
      "Guid" : [
        "c6682ade-b5ec-57c4-b687-676351208742"
      ],
      "Plugin" : "uefi_dbx",
      "Flags" : [
        "internal",
        "updatable",
        "supported",
        "needs-reboot",
        "historical",
        "usable-during-update",
        "only-version-upgrade",
        "signed-payload"
      ],
      "Version" : "371",
      "Created" : 1692361249,
      "Modified" : 1692362745,
      "UpdateState" : 2,
      "Releases" : [
        {
          "RemoteId" : "lvfs",
          "Version" : "371",
          "Filename" : "de143fc38b339d982079517b6f01bcec5246cf5e",
          "Protocol" : "org.uefi.dbx",
          "Checksum" : [
            "bc6a604b29918d67d5fef745ad4375ca3d43d05e"
          ],
          "HostEnclosureKind" : "3",
          "HostSku" : "Default string",
          "DistroVersion" : "39",
          "RuntimeVersion(org.freedesktop.fwupd)" : "1.9.3",
          "RuntimeVersion(org.kernel)" : "6.5.0-0.rc6.43.fc39.x86_64",
          "CpuArchitecture" : "x86_64",
          "UEFIUXCapsule" : "Enabled",
          "SecureBoot" : "Enabled",
          "HostFamily" : "B450 MB",
          "HostBiosVersion" : "F65a",
          "HostVendor" : "Gigabyte Technology Co., Ltd.",
          "RuntimeVersion(org.freedesktop.gusb)" : "0.4.6",
          "KernelName" : "Linux",
          "DistroPrettyName" : "Fedora Linux 39 (Workstation Edition Prerelease)",
          "FwupdSupported" : "True",
          "DistroName" : "Fedora Linux",
          "RuntimeVersion(com.hughsie.libjcat)" : "0.1.14",
          "DistroVariant" : "workstation",
          "KernelVersion" : "6.5.0-0.rc6.43.fc39.x86_64",
          "CompileVersion(org.freedesktop.gusb)" : "0.4.6",
          "HostFirmwareMajorRelease" : "ff",
          "HostProduct" : "B450M DS3H WIFI",
          "BootTime" : "1692361205",
          "RuntimeVersion(org.freedesktop.fwupd-efi)" : "1.4",
          "CpuModel" : "Advanced Micro Devices, Inc. AMD Ryzen 5 5600G with Radeon Graphics",
          "HostBaseboardManufacturer" : "Gigabyte Technology Co., Ltd.",
          "KernelCmdline" : "quiet\n",
          "HostBiosMajorRelease" : "05",
          "HostBiosVendor" : "American Megatrends International, LLC.",
          "CompileVersion(com.hughsie.libxmlb)" : "0.3.11",
          "CompileVersion(com.hughsie.libjcat)" : "0.1.14",
          "HostBaseboardProduct" : "B450M DS3H WIFI-CF",
          "EfivarNvramUsed" : "55293",
          "DistroId" : "fedora",
          "CompileVersion(org.freedesktop.fwupd)" : "1.9.3",
          "HostFirmwareMinorRelease" : "ff",
          "HostBiosMinorRelease" : "11"
        },
        {
          "AppstreamId" : "org.linuxfoundation.dbx.x64.firmware",
          "ReleaseId" : "35287",
          "RemoteId" : "lvfs",
          "Name" : "Secure Boot dbx",
          "NameVariantSuffix" : "x64",
          "Summary" : "UEFI Secure Boot Forbidden Signature Database",
          "Description" : "<p>Insecure versions of the Microsoft Windows boot manager affected by Black Lotus were added to the list of forbidden signatures due to a discovered security problem.This updates the dbx to the latest release from Microsoft.</p><p>Before installing the update, fwupd will check for any affected executables in the ESP and will refuse to update if it finds any boot binaries signed with any of the forbidden signatures.Applying this update may also cause some Windows install media to not start correctly.</p>",
          "Version" : "371",
          "Filename" : "DBXUpdate-20230509-x64.cab",
          "Protocol" : "org.uefi.dbx",
          "Categories" : [
            "X-Configuration",
            "X-System"
          ],
          "Issues" : [
            "CVE-2022-21894"
          ],
          "Checksum" : [
            "bc6a604b29918d67d5fef745ad4375ca3d43d05e",
            "fc3feb015df2710fcfa07583d31b5975ee398357016699cfff067f422ab91e13"
          ],
          "License" : "LicenseRef-proprietary",
          "Size" : 21170,
          "Created" : 1683590400,
          "Locations" : [
            "https://fwupd.org/downloads/fc3feb015df2710fcfa07583d31b5975ee398357016699cfff067f422ab91e13-DBXUpdate-20230509-x64.cab"
          ],
          "Uri" : "https://fwupd.org/downloads/fc3feb015df2710fcfa07583d31b5975ee398357016699cfff067f422ab91e13-DBXUpdate-20230509-x64.cab",
          "Homepage" : "https://uefi.org/revocationlistfile",
          "Vendor" : "Linux Foundation",
          "Flags" : [
            "trusted-metadata",
            "trusted-report"
          ],
          "InstallDuration" : 1,
          "Reports" : [
            {
              "DeviceName" : "FUJITSU CLIENT COMPUTING LIMITED",
              "DistroId" : "fedora",
              "DistroVariant" : "workstation",
              "DistroVersion" : "38",
              "VersionOld" : "211",
              "Vendor" : "DMC Group",
              "RemoteId" : "lvfs",
              "VendorId" : 1758,
              "RuntimeVersion(org.freedesktop.fwupd)" : "1.9.2"
            },
            {
              "DeviceName" : "LENOVO ThinkPad E14 Gen 2",
              "DistroId" : "ubuntu",
              "DistroVersion" : "22.04",
              "VersionOld" : "220",
              "Vendor" : "Jabra",
              "RemoteId" : "lvfs",
              "VendorId" : 21,
              "RuntimeVersion(org.freedesktop.fwupd)" : "1.9.3"
            }
          ]
        }
      ]
    }
  ]
}
```
</details>

hughsie, 2023-08-21:
Weird! There are two 371 releases on the device -- the first has no AppstreamId and the second does. I have no idea how that's happened. Can you attach your /var/lib/fwupd/pending.db please?

bbhtt, 2023-08-21:
> Weird! There are two 371 releases on the device -- the first has no AppstreamId and the second does. I have no idea how that's happened. Can you attach your /var/lib/fwupd/pending.db please?

Here (GitHub doesn't support .db so I changed it to .txt)

[pending.txt](https://github.com/fwupd/fwupd/files/12394697/pending.txt)

hughsie, 2023-08-21:
Okay, I can reproduce here with that pending database, debugging now --thanks.

hughsie, 2023-08-21:
Bleh, that's just https://github.com/fwupd/fwupd/pull/6103 which is harmless and not the fix for this. It seems that the AppStream ID isn't saved in the database for some reason and somehow we're not fixing it up before returning results to the client. I'm pondering the fix.

hughsie, 2023-08-21:
@bbhtt ^^ fixes it for me. I can do a Fedora COPR package if that helps?

bbhtt, 2023-08-21:
> @bbhtt ^^ fixes it for me. I can do a Fedora COPR package if that helps?

Thanks, that'd be great

hughsie, 2023-08-21:
@bbhtt COPR updated: https://copr.fedorainfracloud.org/coprs/rhughes/fwupd/build/6324661/

hughsie, 2023-08-21:
@bbhtt what's `fwupdmgr get-history --json` with that version installed?

bbhtt, 2023-08-21:
> @bbhtt what's `fwupdmgr get-history --json` with that version installed?

<details>
<summary> Output </summary>

```json
{
  "Devices" : [
    {
      "Name" : "UEFI dbx",
      "DeviceId" : "362301da643102b9f38477387e2193e57abaa590",
      "Guid" : [
        "c6682ade-b5ec-57c4-b687-676351208742"
      ],
      "Plugin" : "uefi_dbx",
      "Flags" : [
        "internal",
        "updatable",
        "supported",
        "needs-reboot",
        "historical",
        "usable-during-update",
        "only-version-upgrade",
        "signed-payload"
      ],
      "Version" : "371",
      "Created" : 1692361249,
      "Modified" : 1692362745,
      "UpdateState" : 2,
      "Releases" : [
        {
          "AppstreamId" : "org.linuxfoundation.dbx.x64.firmware",
          "RemoteId" : "lvfs",
          "Version" : "371",
          "Filename" : "de143fc38b339d982079517b6f01bcec5246cf5e",
          "Protocol" : "org.uefi.dbx",
          "Checksum" : [
            "bc6a604b29918d67d5fef745ad4375ca3d43d05e"
          ],
          "HostEnclosureKind" : "3",
          "HostSku" : "Default string",
          "DistroVersion" : "39",
          "RuntimeVersion(org.freedesktop.fwupd)" : "1.9.3",
          "RuntimeVersion(org.kernel)" : "6.5.0-0.rc6.43.fc39.x86_64",
          "CpuArchitecture" : "x86_64",
          "UEFIUXCapsule" : "Enabled",
          "SecureBoot" : "Enabled",
          "HostFamily" : "B450 MB",
          "HostBiosVersion" : "F65a",
          "HostVendor" : "Gigabyte Technology Co., Ltd.",
          "RuntimeVersion(org.freedesktop.gusb)" : "0.4.6",
          "KernelName" : "Linux",
          "DistroPrettyName" : "Fedora Linux 39 (Workstation Edition Prerelease)",
          "FwupdSupported" : "True",
          "DistroName" : "Fedora Linux",
          "RuntimeVersion(com.hughsie.libjcat)" : "0.1.14",
          "DistroVariant" : "workstation",
          "KernelVersion" : "6.5.0-0.rc6.43.fc39.x86_64",
          "CompileVersion(org.freedesktop.gusb)" : "0.4.6",
          "HostFirmwareMajorRelease" : "ff",
          "HostProduct" : "B450M DS3H WIFI",
          "BootTime" : "1692361205",
          "RuntimeVersion(org.freedesktop.fwupd-efi)" : "1.4",
          "CpuModel" : "Advanced Micro Devices, Inc. AMD Ryzen 5 5600G with Radeon Graphics",
          "HostBaseboardManufacturer" : "Gigabyte Technology Co., Ltd.",
          "KernelCmdline" : "quiet\n",
          "HostBiosMajorRelease" : "05",
          "HostBiosVendor" : "American Megatrends International, LLC.",
          "CompileVersion(com.hughsie.libxmlb)" : "0.3.11",
          "CompileVersion(com.hughsie.libjcat)" : "0.1.14",
          "HostBaseboardProduct" : "B450M DS3H WIFI-CF",
          "EfivarNvramUsed" : "55293",
          "DistroId" : "fedora",
          "CompileVersion(org.freedesktop.fwupd)" : "1.9.3",
          "HostFirmwareMinorRelease" : "ff",
          "HostBiosMinorRelease" : "11"
        }
      ]
    }
  ]
}

```
</details>

hughsie, 2023-08-21:
okay, that's setting `AppstreamId" : "org.linuxfoundation.dbx.x64.firmware` like I expected now -- are you sure you restarted gnome-software? What gnome-software version are you using?

hughsie, 2023-08-21:
> okay, that's setting AppstreamId" : "org.linuxfoundation.dbx.x64.firmware like I expected now

Aha, but gnome-software isn't using get-history, it's using get-results! I've extended the fix to fix up both cases. There's a new rpm building in the COPR now, you need fwupd release fwupd-1.9.4-0.**1037**.20230821git.fc38

bbhtt, 2023-08-21:
> Aha, but gnome-software isn't using get-history, it's using get-results! I've extended the fix to fix up both cases. There's a new rpm building in the COPR now, you need fwupd release [...]

Thanks a lot! That fixes it for me.

pwithnall, 2023-08-21:
Thanks @hughsie! That was a fast turnaround

romulasry, 2023-08-23:
I am getting this on Wed Aug 23 at at approximately 11 AM Pacific Daylight Savings time. Fedora rawhide currently- all up to date.

AdamWill, 2023-08-28:
@romulasry what does `rpm -q fwupd` say?

romulasry, 2023-08-28:
@AdamWill
fwupd-1.9.3-1.fc39.x86_64

AdamWill, 2023-08-28:
The fix is in 1.9.4-1 , you'll need to update to that. It was built August 22 and would have been in the August 23 nightly compose, your update was probably just too soon.

francesco-ballarin, 2023-08-30:
Hi @hughsie,
I am getting a similar problem with debian trixie.

```
Detailed errors from the package manager follow:

failed to build result for c085a3cbf4dc1d60d110b76168
```

<details>
<summary><code>$ lsb_release -a</code></summary>

```bash
No LSB modules are available.
Distributor ID:	Debian
Description:	Debian GNU/Linux trixie/sid
Release:	n/a
Codename:	trixie
```
</details>

<details>
<summary><code>$ fwupdmgr --version</code></summary>

```
compile   org.freedesktop.fwupd         1.9.4
compile   com.hughsie.libxmlb           0.3.10
compile   com.hughsie.libjcat           0.1.9
runtime   org.freedesktop.fwupd-efi     1.4
compile   org.freedesktop.gusb          0.4.5
runtime   com.dell.libsmbios            2.4
runtime   org.freedesktop.gusb          0.4.5
runtime   org.kernel                    6.4.0-3-amd64
runtime   org.freedesktop.fwupd         1.9.4
```
</details>

<details>
<summary><code>$ fwupdmgr get-history --json</code></summary>

```
{
  "Devices" : [
    {
      "Name" : "Embedded Controller",
      "DeviceId" : "c085a3cbf4dc1d60d110b7616816e2ffd279b64e",
      "Guid" : [
        "f72e048b-65bd-4e71-9071-1ac7045223e5"
      ],
      "Plugin" : "uefi_capsule",
      "Flags" : [
        "internal",
        "updatable",
        "require-ac",
        "needs-reboot",
        "reported",
        "historical",
        "usable-during-update"
      ],
      "Version" : "0.1.2",
      "Created" : 1637485252,
      "Modified" : 1637485281,
      "UpdateState" : 2,
      "Releases" : [
        {
          "Version" : "0.1.25",
          "Checksum" : [
            "ee91745253df4d875399c03fb793966d83a30432"
          ],
          "RuntimeVersion(org.freedesktop.gusb)" : "0.3.8",
          "HostFamily" : "ThinkPad X1 Carbon 7th",
          "BootMgrDesc" : "legacy",
          "CompileVersion(com.redhat.fwupdate)" : "12",
          "LastAttemptStatus" : "0x0",
          "SecureBoot" : "Disabled",
          "LinuxLockdown" : "none",
          "EspPath" : "/boot/efi",
          "UEFIUXCapsule" : "Enabled",
          "RuntimeVersion(com.redhat.fwupdate)" : "12",
          "MissingCapsuleHeader" : "False",
          "RuntimeVersion(org.kernel)" : "5.14.0-4-amd64",
          "CompileVersion(org.freedesktop.gusb)" : "0.3.5",
          "RuntimeVersion(com.dell.libsmbios)" : "2.4",
          "DistroId" : "debian",
          "CpuArchitecture" : "x86_64",
          "BootTime" : "1637485242",
          "HostVendor" : "LENOVO",
          "EfivarNvramUsed" : "43516",
          "RuntimeVersion(org.freedesktop.appstream-glib)" : "0.7.14",
          "KernelVersion" : "5.14.0-4-amd64",
          "HostProduct" : "20QD00LKIX",
          "LastAttemptVersion" : "0x0",
          "TpmFamily" : "2.0",
          "HostSku" : "LENOVO_MT_20QD_BU_Think_FM_ThinkPad X1 Carbon 7th",
          "RuntimeVersion(org.freedesktop.fwupd)" : "1.5.7",
          "CompileVersion(org.freedesktop.fwupd)" : "1.5.7"
        }
      ]
    },
    {
      "Name" : "UEFI dbx",
      "DeviceId" : "362301da643102b9f38477387e2193e57abaa590",
      "Guid" : [
        "14503b3d-73ce-5d06-8137-77c68972a341"
      ],
      "Plugin" : "uefi_dbx",
      "Flags" : [
        "internal",
        "updatable",
        "supported",
        "needs-reboot",
        "historical",
        "usable-during-update",
        "only-version-upgrade",
        "signed-payload"
      ],
      "Version" : "217",
      "Created" : 1693395599,
      "Modified" : 1693395949,
      "UpdateState" : 2,
      "Releases" : [
        {
          "AppstreamId" : "org.linuxfoundation.dbx.x64.firmware",
          "RemoteId" : "lvfs",
          "Version" : "371",
          "Filename" : "de143fc38b339d982079517b6f01bcec5246cf5e",
          "Protocol" : "org.uefi.dbx",
          "Checksum" : [
            "bc6a604b29918d67d5fef745ad4375ca3d43d05e"
          ],
          "HostEnclosureKind" : "a",
          "HostSku" : "LENOVO_MT_20QD_BU_Think_FM_ThinkPad X1 Carbon 7th",
          "RuntimeVersion(org.freedesktop.fwupd)" : "1.9.4",
          "RuntimeVersion(org.kernel)" : "6.4.0-3-amd64",
          "BootMgrDesc" : "legacy",
          "UEFIUXCapsule" : "Enabled",
          "CpuArchitecture" : "x86_64",
          "SecureBoot" : "Disabled",
          "HostFamily" : "ThinkPad X1 Carbon 7th",
          "HostBiosVersion" : "N2HET30W (1.13 )",
          "HostVendor" : "LENOVO",
          "RuntimeVersion(org.freedesktop.gusb)" : "0.4.5",
          "KernelName" : "Linux",
          "DistroPrettyName" : "Debian GNU/Linux trixie/sid",
          "RuntimeVersion(com.dell.libsmbios)" : "2.4",
          "DistroName" : "Debian GNU/Linux",
          "FwupdSupported" : "True",
          "KernelVersion" : "6.4.0-3-amd64",
          "CompileVersion(org.freedesktop.gusb)" : "0.4.5",
          "HostFirmwareMajorRelease" : "01",
          "HostProduct" : "20QD00LKIX",
          "BootTime" : "1693395574",
          "RuntimeVersion(org.freedesktop.fwupd-efi)" : "1.4",
          "CpuModel" : "Intel Core™ i7-8565U CPU @ 1.80GHz",
          "HostBaseboardManufacturer" : "LENOVO",
          "HostBiosMajorRelease" : "01",
          "HostBiosVendor" : "LENOVO",
          "CompileVersion(com.hughsie.libxmlb)" : "0.3.10",
          "CompileVersion(com.hughsie.libjcat)" : "0.1.9",
          "EfivarNvramUsed" : "52331",
          "HostBaseboardProduct" : "20QD00LKIX",
          "DistroId" : "debian",
          "HostFirmwareMinorRelease" : "19",
          "CompileVersion(org.freedesktop.fwupd)" : "1.9.4",
          "HostBiosMinorRelease" : "0d"
        }
      ]
    }
  ]
}
```
</details>

Thanks.

francesco-ballarin, 2023-08-30:
Apologies, I must have truncated it in the copy and paste

```
Detailed errors from the package manager follow:

failed to build result for c085a3cbf4dc1d60d110b7616816e2ffd279b64e
```

hughsie, 2023-08-30:
@francesco-ballarin what does `fwupdmgr get-results c085a3cbf4dc1d60d110b7616816e2ffd279b64e --json` say for you?

hughsie, 2023-08-30:
And `fwupdmgr get-devices` please?

francesco-ballarin, 2023-08-30:
<details><summary><code>$ fwupdmgr get-devices</code></summary>

```
LENOVO 20QD00LKIX
│
├─Core™ i7-8565U CPU @ 1.80GHz:
│     Device ID:          4bde70ba4e39b28f9eab1628f9dd6e6244c03027
│     Current version:    0x000000b8
│     Vendor:             Intel
│     GUIDs:              b9a2dd81-159e-5537-a7db-e7101d164d3f ← cpu
│                         30249f37-d140-5d3e-9319-186b1bd5cac3 ← CPUID\PRO_0&FAM_06
│                         561403e8-143a-5071-ab09-bf5e4c146983 ← CPUID\PRO_0&FAM_06&MOD_8E
│                         d5092916-1868-5182-8c70-12f83dab6372 ← CPUID\PRO_0&FAM_06&MOD_8E&STP_C
│     Device Flags:       • Internal device
│   
├─Embedded Controller:
│     Device ID:          2292ae5236790b47884e37cf162dcf23bfcd1c60
│     Summary:            UEFI ESRT device
│     Current version:    0.1.25
│     Minimum Version:    0.1.25
│     Vendor:             Lenovo (DMI:LENOVO)
│     Update State:       Success
│     Problems:           • Device requires AC power to be connected
│     GUID:               f72e048b-65bd-4e71-9071-1ac7045223e5
│     Device Flags:       • Internal device
│                         • System requires external power source
│                         • Supported on remote server
│                         • Needs a reboot after installation
│                         • Device is usable for the duration of the update
│                         • Updatable
│   
├─Intel Management Engine:
│     Device ID:          349bb341230b1a86e5effe7dfe4337e1590227bd
│     Summary:            UEFI ESRT device
│     Current version:    192.47.1524
│     Minimum Version:    0.0.1
│     Vendor:             Lenovo (DMI:LENOVO)
│     Update State:       Success
│     Problems:           • Device requires AC power to be connected
│     GUID:               c3e4be53-e714-4ea1-bb9c-7fe13a98b556
│     Device Flags:       • Internal device
│                         • System requires external power source
│                         • Supported on remote server
│                         • Needs a reboot after installation
│                         • Device is usable for the duration of the update
│                         • Updatable
│   
├─System Firmware:
│ │   Device ID:          a45df35ac0e948ee180fe216a5f703f32dda163f
│ │   Summary:            UEFI ESRT device
│ │   Current version:    0.1.13
│ │   Minimum Version:    0.0.1
│ │   Vendor:             Lenovo (DMI:LENOVO)
│ │   Update State:       Success
│ │   Problems:           • Device requires AC power to be connected
│ │   GUIDs:              4bbc40fa-f81e-4206-bc70-a1f7b744d964
│ │                       230c8b18-8d9b-53ec-838b-6cfc0383493a ← main-system-firmware
│ │   Device Flags:       • Internal device
│ │                       • System requires external power source
│ │                       • Needs a reboot after installation
│ │                       • Cryptographic hash verification is available
│ │                       • Device is usable for the duration of the update
│ │                       • Updatable
│ │ 
│ └─UEFI dbx:
│       Device ID:        362301da643102b9f38477387e2193e57abaa590
│       Summary:          UEFI revocation database
│       Current version:  371
│       Minimum Version:  371
│       Vendor:           UEFI:Linux Foundation
[... 84 lines cut ...]
│     GUIDs:              ff2112dc-038c-596d-90ca-d43c5077c6ec ← NVME\VEN_15B7&DEV_5002
│                         c528df4b-7972-5880-8cb1-330415e2dc6a ← NVME\VEN_15B7&DEV_5002&SUBSYS_15B75002
│                         63af1441-fb24-5368-b450-886fbc98610b ← WDC PC SN720 SDAQNTW-512G-1001
│     Device Flags:       • Internal device
│                         • System requires external power source
│                         • Supported on remote server
│                         • Needs a reboot after installation
│                         • Device is usable for the duration of the update
│                         • Updatable
│   
└─WhiskeyLake-U GT2 [UHD Graphics 620]:
      Device ID:          5792b48846ce271fab11c4a545f7a3df0d36e00a
      Current version:    02
      Vendor:             Intel Corporation (PCI:0x8086)
      GUIDs:              52aa8ac7-c837-561f-af32-b7979cd2f6ed ← PCI\VEN_8086&DEV_3EA0
                          4cf1c502-32ec-5639-85c3-5343edba4731 ← PCI\VEN_8086&DEV_3EA0&SUBSYS_17AA2292
      Device Flags:       • Internal device
                          • Cryptographic hash verification is available
    
────────────────────────────────────────────────
Devices that have been updated successfully:
 • UEFI dbx (217 → 371)
```

</details>

hughsie, 2023-08-30:
Ahh, get-history has:  `"RuntimeVersion(org.freedesktop.fwupd)" : "1.5.7",`  -- that's much older than 1.9.3. I think fwupd should probably filter out the devices that no longer exist when doing `get_results` with the `FWUPD_DEVICE_ID_ANY` mask. Can you open a new issue please and I'll fix that one this week.

francesco-ballarin, 2023-08-30:
> gnome-software version pls?

`gnome-software 45.beta`

> Can you open a new issue please and I'll fix that one this week.

Thanks! Issue is https://github.com/fwupd/fwupd/issues/6128