fwupd/fwupd #6108: 1.9.4: parallel build fails on missing `fu-efi-struct.h` header file

kloczek, 2023-08-22:
As long as build is executed with `-j1` everything is fine so looks like some build dependencies are not properly defined
```console
+ /usr/bin/meson compile -C x86_64-redhat-linux-gnu -j 48 --verbose

[..]

[72/1067] /usr/bin/gcc -Iplugins/acpi-dmar/libfu_plugin_acpi_dmar.a.p -Iplugins/acpi-dmar -I../plugins/acpi-dmar -I. -I.. -Ilibfwupd -I../libfwupd -Ilibfwupdplugin -I../libfwupdplugin -I/usr/include/glib-2.0 -I/usr/lib64/glib-2.0/include -I/usr/include/libmount -I/usr/include/gio-unix-2.0 -I/usr/include/gudev-1.0 -I/usr/include/gusb-1 -I/usr/include/libusb-1.0 -I/usr/include/json-glib-1.0 -I/usr/include/libxmlb-2 -fdiagnostics-color=always -D_FILE_OFFSET_BITS=64 -Wall -Winvalid-pch -Wextra -std=c11 -DFWUPD_COMPILATION -D_DEFAULT_SOURCE -D_BSD_SOURCE -D__BSD_VISIBLE -D_XOPEN_SOURCE=700 -D_GNU_SOURCE -D_DARWIN_C_SOURCE=900000 -O2 -g -grecord-gcc-switches -pipe -Wall -Werror=format-security -Wp,-U_FORTIFY_SOURCE,-D_FORTIFY_SOURCE=3 -Wp,-D_GLIBCXX_ASSERTIONS -specs=/usr/lib/rpm/redhat/redhat-hardened-cc1 -fstack-protector-strong -specs=/usr/lib/rpm/redhat/redhat-annobin-cc1 -fno-omit-frame-pointer -mno-omit-leaf-frame-pointer -m64 -mtune=generic -fasynchronous-unwind-tables -fstack-clash-protection -fcf-protection -fdata-sections -ffunction-sections -flto=auto -flto-partition=none -Os -fPIC -pthread '-DG_LOG_DOMAIN="FuPluginAcpiDmar"' -MD -MQ plugins/acpi-dmar/libfu_plugin_acpi_dmar.a.p/fu-acpi-dmar.c.o -MF plugins/acpi-dmar/libfu_plugin_acpi_dmar.a.p/fu-acpi-dmar.c.o.d -o plugins/acpi-dmar/libfu_plugin_acpi_dmar.a.p/fu-acpi-dmar.c.o -c ../plugins/acpi-dmar/fu-acpi-dmar.c
FAILED: plugins/acpi-dmar/libfu_plugin_acpi_dmar.a.p/fu-acpi-dmar.c.o
/usr/bin/gcc -Iplugins/acpi-dmar/libfu_plugin_acpi_dmar.a.p -Iplugins/acpi-dmar -I../plugins/acpi-dmar -I. -I.. -Ilibfwupd -I../libfwupd -Ilibfwupdplugin -I../libfwupdplugin -I/usr/include/glib-2.0 -I/usr/lib64/glib-2.0/include -I/usr/include/libmount -I/usr/include/gio-unix-2.0 -I/usr/include/gudev-1.0 -I/usr/include/gusb-1 -I/usr/include/libusb-1.0 -I/usr/include/json-glib-1.0 -I/usr/include/libxmlb-2 -fdiagnostics-color=always -D_FILE_OFFSET_BITS=64 -Wall -Winvalid-pch -Wextra -std=c11 -DFWUPD_COMPILATION -D_DEFAULT_SOURCE -D_BSD_SOURCE -D__BSD_VISIBLE -D_XOPEN_SOURCE=700 -D_GNU_SOURCE -D_DARWIN_C_SOURCE=900000 -O2 -g -grecord-gcc-switches -pipe -Wall -Werror=format-security -Wp,-U_FORTIFY_SOURCE,-D_FORTIFY_SOURCE=3 -Wp,-D_GLIBCXX_ASSERTIONS -specs=/usr/lib/rpm/redhat/redhat-hardened-cc1 -fstack-protector-strong -specs=/usr/lib/rpm/redhat/redhat-annobin-cc1 -fno-omit-frame-pointer -mno-omit-leaf-frame-pointer -m64 -mtune=generic -fasynchronous-unwind-tables -fstack-clash-protection -fcf-protection -fdata-sections -ffunction-sections -flto=auto -flto-partition=none -Os -fPIC -pthread '-DG_LOG_DOMAIN="FuPluginAcpiDmar"' -MD -MQ plugins/acpi-dmar/libfu_plugin_acpi_dmar.a.p/fu-acpi-dmar.c.o -MF plugins/acpi-dmar/libfu_plugin_acpi_dmar.a.p/fu-acpi-dmar.c.o.d -o plugins/acpi-dmar/libfu_plugin_acpi_dmar.a.p/fu-acpi-dmar.c.o -c ../plugins/acpi-dmar/fu-acpi-dmar.c
In file included from ../libfwupdplugin/fwupdplugin.h:47,
                 from ../plugins/acpi-dmar/fu-acpi-dmar.h:9,
                 from ../plugins/acpi-dmar/fu-acpi-dmar.c:9:
../libfwupdplugin/fu-efi-signature.h:11:10: fatal error: fu-efi-struct.h: No such file or directory
   11 | #include "fu-efi-struct.h"
      |          ^~~~~~~~~~~~~~~~~
compilation terminated.
[73/1067] /usr/bin/gcc -Iplugins/acpi-dmar/acpi-dmar-self-test.p -Iplugins/acpi-dmar -I../plugins/acpi-dmar -I. -I.. -Ilibfwupd -I../libfwupd -Ilibfwupdplugin -I../libfwupdplugin -I/usr/include/glib-2.0 -I/usr/lib64/glib-2.0/include -I/usr/include/libmount -I/usr/include/gio-unix-2.0 -I/usr/include/gudev-1.0 -I/usr/include/gusb-1 -I/usr/include/libusb-1.0 -I/usr/include/json-glib-1.0 -I/usr/include/libxmlb-2 -fdiagnostics-color=always -D_FILE_OFFSET_BITS=64 -Wall -Winvalid-pch -Wextra -std=c11 -DFWUPD_COMPILATION -D_DEFAULT_SOURCE -D_BSD_SOURCE -D__BSD_VISIBLE -D_XOPEN_SOURCE=700 -D_GNU_SOURCE -D_DARWIN_C_SOURCE=900000 -O2 -g -grecord-gcc-switches -pipe -Wall -Werror=format-security -Wp,-U_FORTIFY_SOURCE,-D_FORTIFY_SOURCE=3 -Wp,-D_GLIBCXX_ASSERTIONS -specs=/usr/lib/rpm/redhat/redhat-hardened-cc1 -fstack-protector-strong -specs=/usr/lib/rpm/redhat/redhat-annobin-cc1 -fno-omit-frame-pointer -mno-omit-leaf-frame-pointer -m64 -mtune=generic -fasynchronous-unwind-tables -fstack-clash-protection -fcf-protection -fdata-sections -ffunction-sections -flto=auto -flto-partition=none -Os -pthread '-DSRCDIR="/home/tkloczko/rpmbuild/BUILD/fwupd-1.9.4/plugins/acpi-dmar"' -MD -MQ plugins/acpi-dmar/acpi-dmar-self-test.p/fu-self-test.c.o -MF plugins/acpi-dmar/acpi-dmar-self-test.p/fu-self-test.c.o.d -o plugins/acpi-dmar/acpi-dmar-self-test.p/fu-self-test.c.o -c ../plugins/acpi-dmar/fu-self-test.c
FAILED: plugins/acpi-dmar/acpi-dmar-self-test.p/fu-self-test.c.o
/usr/bin/gcc -Iplugins/acpi-dmar/acpi-dmar-self-test.p -Iplugins/acpi-dmar -I../plugins/acpi-dmar -I. -I.. -Ilibfwupd -I../libfwupd -Ilibfwupdplugin -I../libfwupdplugin -I/usr/include/glib-2.0 -I/usr/lib64/glib-2.0/include -I/usr/include/libmount -I/usr/include/gio-unix-2.0 -I/usr/include/gudev-1.0 -I/usr/include/gusb-1 -I/usr/include/libusb-1.0 -I/usr/include/json-glib-1.0 -I/usr/include/libxmlb-2 -fdiagnostics-color=always -D_FILE_OFFSET_BITS=64 -Wall -Winvalid-pch -Wextra -std=c11 -DFWUPD_COMPILATION -D_DEFAULT_SOURCE -D_BSD_SOURCE -D__BSD_VISIBLE -D_XOPEN_SOURCE=700 -D_GNU_SOURCE -D_DARWIN_C_SOURCE=900000 -O2 -g -grecord-gcc-switches -pipe -Wall -Werror=format-security -Wp,-U_FORTIFY_SOURCE,-D_FORTIFY_SOURCE=3 -Wp,-D_GLIBCXX_ASSERTIONS -specs=/usr/lib/rpm/redhat/redhat-hardened-cc1 -fstack-protector-strong -specs=/usr/lib/rpm/redhat/redhat-annobin-cc1 -fno-omit-frame-pointer -mno-omit-leaf-frame-pointer -m64 -mtune=generic -fasynchronous-unwind-tables -fstack-clash-protection -fcf-protection -fdata-sections -ffunction-sections -flto=auto -flto-partition=none -Os -pthread '-DSRCDIR="/home/tkloczko/rpmbuild/BUILD/fwupd-1.9.4/plugins/acpi-dmar"' -MD -MQ plugins/acpi-dmar/acpi-dmar-self-test.p/fu-self-test.c.o -MF plugins/acpi-dmar/acpi-dmar-self-test.p/fu-self-test.c.o.d -o plugins/acpi-dmar/acpi-dmar-self-test.p/fu-self-test.c.o -c ../plugins/acpi-dmar/fu-self-test.c
In file included from ../libfwupdplugin/fwupdplugin.h:47,
                 from ../plugins/acpi-dmar/fu-acpi-dmar.h:9,
                 from ../plugins/acpi-dmar/fu-self-test.c:9:
../libfwupdplugin/fu-efi-signature.h:11:10: fatal error: fu-efi-struct.h: No such file or directory
   11 | #include "fu-efi-struct.h"
      |          ^~~~~~~~~~~~~~~~~
compilation terminated.
[74/1067] /usr/bin/python3 /home/tkloczko/rpmbuild/BUILD/fwupd-1.9.4/libfwupdplugin/rustgen.py ../libfwupdplugin/fu-elf.rs libfwupdplugin/fu-elf-struct.c libfwupdplugin/fu-elf-struct.h
[75/1067] /usr/bin/gcc -Iplugins/acpi-dmar/libfu_plugin_acpi_dmar.a.p -Iplugins/acpi-dmar -I../plugins/acpi-dmar -I. -I.. -Ilibfwupd -I../libfwupd -Ilibfwupdplugin -I../libfwupdplugin -I/usr/include/glib-2.0 -I/usr/lib64/glib-2.0/include -I/usr/include/libmount -I/usr/include/gio-unix-2.0 -I/usr/include/gudev-1.0 -I/usr/include/gusb-1 -I/usr/include/libusb-1.0 -I/usr/include/json-glib-1.0 -I/usr/include/libxmlb-2 -fdiagnostics-color=always -D_FILE_OFFSET_BITS=64 -Wall -Winvalid-pch -Wextra -std=c11 -DFWUPD_COMPILATION -D_DEFAULT_SOURCE -D_BSD_SOURCE -D__BSD_VISIBLE -D_XOPEN_SOURCE=700 -D_GNU_SOURCE -D_DARWIN_C_SOURCE=900000 -O2 -g -grecord-gcc-switches -pipe -Wall -Werror=format-security -Wp,-U_FORTIFY_SOURCE,-D_FORTIFY_SOURCE=3 -Wp,-D_GLIBCXX_ASSERTIONS -specs=/usr/lib/rpm/redhat/redhat-hardened-cc1 -fstack-protector-strong -specs=/usr/lib/rpm/redhat/redhat-annobin-cc1 -fno-omit-frame-pointer -mno-omit-leaf-frame-pointer -m64 -mtune=generic -fasynchronous-unwind-tables -fstack-clash-protection -fcf-protection -fdata-sections -ffunction-sections -flto=auto -flto-partition=none -Os -fPIC -pthread '-DG_LOG_DOMAIN="FuPluginAcpiDmar"' -MD -MQ plugins/acpi-dmar/libfu_plugin_acpi_dmar.a.p/fu-acpi-dmar-plugin.c.o -MF plugins/acpi-dmar/libfu_plugin_acpi_dmar.a.p/fu-acpi-dmar-plugin.c.o.d -o plugins/acpi-dmar/libfu_plugin_acpi_dmar.a.p/fu-acpi-dmar-plugin.c.o -c ../plugins/acpi-dmar/fu-acpi-dmar-plugin.c
FAILED: plugins/acpi-dmar/libfu_plugin_acpi_dmar.a.p/fu-acpi-dmar-plugin.c.o
[... cut ...] s/acpi-dmar/libfu_plugin_acpi_dmar.a.p/fu-acpi-dmar-plugin.c.o -c ../plugins/acpi-dmar/fu-acpi-dmar-plugin.c
In file included from ../libfwupdplugin/fwupdplugin.h:47,
                 from ../plugins/acpi-dmar/fu-acpi-dmar-plugin.h:9,
                 from ../plugins/acpi-dmar/fu-acpi-dmar-plugin.c:9:
../libfwupdplugin/fu-efi-signature.h:11:10: fatal error: fu-efi-struct.h: No such file or directory
   11 | #include "fu-efi-struct.h"
      |          ^~~~~~~~~~~~~~~~~
compilation terminated.
```

hughsie, 2023-08-22:
Hmm, I thought I was doing the right thing here -- the `fu-efi.rs` (all of them are in `fwupdplugin_structs`) gets build into `fu-efi-struct.h` using custom_target -- and all the targets are added to `fwupdplugin_struct_targets` which is a dep of `fwupdplugin` -- which in turn is added to `plugin_libs` -- which all plugins use with `link_with`. Maybe a meson issue?

I can't reproduce on my system with meson-1.0.1-1.fc38.noarch and even `meson compile -j 200`

kloczek, 2023-08-22:
I'm using meson 1.2.1.

hughsie, 2023-08-22:
A meson regression perhaps?

hughsie, 2023-08-22:
Does it build correctly with meson 1.0.1?

hughsie, 2023-08-22:
I've just tried with meson-1.2.1-1.fc38.noarch and `meson compile -j 200 --verbose` -- with no non-default options set and it worked. i.e. `meson setup build-mesontest && meson compile -j 200 --verbose -C build-mesontest` and it worked.

kloczek, 2023-08-22:
Hmm .. do you run that on actual box with 100/200 cores/HT threads to have really started 200 parallel builds physically building at the same time? 🤔 
In my case I have 24/48 physical cores/HT threads system.
In cases like this one all it is about timing and physically running parallel compile processes ..

hughsie, 2023-08-22:
>  In cases like this one all it is about timing and physically running parallel compile processes ..

Ohh, I have just 12 cores to play with. Does it build correctly with meson 1.0.1 on your machine?

kloczek, 2023-08-22:
In my case build fails on 72 meson target so it is close to run 3rd batch of compiles.

hughsie, 2023-08-22:
> In my case build fails on 72 meson target so it is close to run 3rd batch of compiles.

Sorry, I don't understand.

kloczek, 2023-08-22:
> Ohh, I have just 12 cores to play with. Does it build correctly with meson 1.0.1 on your machine?

My build infrastructure is a bit busy right now.
Will try to reproduce that in next 4-5 hours.

kloczek, 2023-08-22:
> Sorry, I don't understand.

With assumption that each compile takes aprox the same time first 48 meson tasks will be running in real parallel.
If build in my case it failed on 72 task it means that none of th first 48 depends on that header files.
Before start second batch of task generating that header file should already be finished if meson is not queuing those tasks depend on availability of generated fu-efi-struct.h file.

hughsie, 2023-08-22:
I think you're going to have to debug this then; I have no way to debug or test any potential fix.

kloczek, 2023-08-22:
Still my request is in the queue however in the error output are detail which one .c files do not depends on fu-efi-struct.h.

mkszuba, 2023-08-25:
We have got the same problem in Gentoo too, now that I have pushed a 1.9.4 ebuild: https://bugs.gentoo.org/913003. Our CI system uses _-j 64_, no clue how many actual cores it has got though.

hughsie, 2023-08-29:
@mkszuba I can't test this unless someone sends me a new CPU :) Any debugging most appreciated.

jackpot51, 2023-08-30:
I've seen this issue also on fu-cfu-struct.h. We have a build server with 128 cores that has this issue. I should be able to fix up the meson files to prevent it.

jackpot51, 2023-08-30:
A minimal reproduction can be done on any CPU with the following, making sure `build` is not already created:

```
meson build
ninja plugins/elanfp/libfu_plugin_elanfp.a
```

hughsie, 2023-08-30:
@jackpot51 i can reproduce with that, thanks. I can try to fix tomorrow unless you get there first.

jackpot51, 2023-08-30:
@kloczek if you can, please test #6131

kloczek, 2023-08-30:
On sec .. 😋

kloczek, 2023-08-30:
Nope .. It DoesNotWork ™️  😞 
<details>

```console
+ /usr/bin/meson compile -C x86_64-redhat-linux-gnu -j 48 --verbose
ninja: Entering directory `/home/tkloczko/rpmbuild/BUILD/fwupd-1.9.4/x86_64-redhat-linux-gnu'
[1/2923] /usr/bin/msgfmt -o po/af/LC_MESSAGES/fwupd.mo ../po/af.po
[2/2923] /usr/bin/msgfmt -o po/ast/LC_MESSAGES/fwupd.mo ../po/ast.po
[3/2923] /usr/bin/msgfmt -o po/eo/LC_MESSAGES/fwupd.mo ../po/eo.po
[4/2923] /usr/bin/gcab --create --nopath data/installed-tests/fakedevice123.cab ../data/installed-tests/fakedevice123.bin ../data/installed-tests/fakedevice123.bin.asc ../data/installed-tests/fakedevice123.metainfo.xml
[5/2923] /usr/bin/gcab --create --nopath data/installed-tests/fakedevice124.cab ../data/installed-tests/fakedevice124.bin ../data/installed-tests/fakedevice124.bin.asc ../data/installed-tests/fakedevice124.metainfo.xml
[6/2923] /usr/bin/msgfmt -o po/eu/LC_MESSAGES/fwupd.mo ../po/eu.po
[7/2923] /usr/bin/msgfmt -o po/hi/LC_MESSAGES/fwupd.mo ../po/hi.po
[8/2923] /usr/bin/msgfmt -o po/kk/LC_MESSAGES/fwupd.mo ../po/kk.po
[9/2923] /usr/bin/msgfmt -o po/fr/LC_MESSAGES/fwupd.mo ../po/fr.po
[10/2923] /usr/bin/msgfmt -o po/ja/LC_MESSAGES/fwupd.mo ../po/ja.po
[11/2923] /usr/bin/msgfmt -o po/ky/LC_MESSAGES/fwupd.mo ../po/ky.po
[12/2923] /usr/bin/jcat-tool --basename --appstream-id localhost self-sign libfwupd/tests/auth/metadata.xml.gz.jcat ../libfwupd/tests/auth/metadata.xml.gz
[13/2923] /usr/bin/msgfmt -o po/gl/LC_MESSAGES/fwupd.mo ../po/gl.po
[14/2923] /usr/bin/msgfmt -o po/oc/LC_MESSAGES/fwupd.mo ../po/oc.po
[15/2923] /usr/bin/msgfmt -o po/pa/LC_MESSAGES/fwupd.mo ../po/pa.po
[16/2923] /usr/bin/msgfmt -o po/he/LC_MESSAGES/fwupd.mo ../po/he.po
[17/2923] /usr/bin/msgfmt -o po/lt/LC_MESSAGES/fwupd.mo ../po/lt.po
[18/2923] /usr/bin/msgfmt -o po/da/LC_MESSAGES/fwupd.mo ../po/da.po
[19/2923] /usr/bin/msgfmt -o po/fur/LC_MESSAGES/fwupd.mo ../po/fur.po
[20/2923] /usr/bin/msgfmt -o po/ka/LC_MESSAGES/fwupd.mo ../po/ka.po
[21/2923] /usr/bin/msgfmt -o po/sk/LC_MESSAGES/fwupd.mo ../po/sk.po
[22/2923] /usr/bin/msgfmt -o po/sr/LC_MESSAGES/fwupd.mo ../po/sr.po
[23/2923] /usr/bin/msgfmt -o po/nl/LC_MESSAGES/fwupd.mo ../po/nl.po
[24/2923] /usr/bin/msgfmt -o po/de/LC_MESSAGES/fwupd.mo ../po/de.po
[25/2923] /usr/bin/msgfmt -o po/id/LC_MESSAGES/fwupd.mo ../po/id.po
[26/2923] /usr/bin/msgfmt -o po/it/LC_MESSAGES/fwupd.mo ../po/it.po
[27/2923] /usr/bin/msgfmt -o po/zh_TW/LC_MESSAGES/fwupd.mo ../po/zh_TW.po
[28/2923] /usr/bin/msgfmt -o po/ca/LC_MESSAGES/fwupd.mo ../po/ca.po
[29/2923] /usr/bin/msgfmt -o po/en_GB/LC_MESSAGES/fwupd.mo ../po/en_GB.po
[30/2923] /usr/bin/gcab --create --nopath libfwupdplugin/tests/colorhug/colorhug-als-3.0.2.cab ../libfwupdplugin/tests/colorhug/firmware.bin ../libfwupdplugin/tests/colorhug/firmware.bin.asc ../libfwupdplugin/tests/colorhug/firmware.metainfo.xml
[31/2923] /usr/bin/msgfmt -o po/cs/LC_MESSAGES/fwupd.mo ../po/cs.po
[32/2923] /usr/bin/msgfmt -o po/es/LC_MESSAGES/fwupd.mo ../po/es.po
[33/2923] /usr/bin/msgfmt -o po/fi/LC_MESSAGES/fwupd.mo ../po/fi.po
[34/2923] /usr/bin/msgfmt -o po/hr/LC_MESSAGES/fwupd.mo ../po/hr.po
[35/2923] /usr/bin/msgfmt -o po/hu/LC_MESSAGES/fwupd.mo ../po/hu.po
[36/2923] /usr/bin/msgfmt -o po/ru/LC_MESSAGES/fwupd.mo ../po/ru.po
[37/2923] /usr/bin/msgfmt -o po/tr/LC_MESSAGES/fwupd.mo ../po/tr.po
[38/2923] /usr/bin/msgfmt -o po/ko/LC_MESSAGES/fwupd.mo ../po/ko.po
[39/2923] /usr/bin/msgfmt -o po/pt/LC_MESSAGES/fwupd.mo ../po/pt.po
[40/2923] /usr/bin/msgfmt -o po/pt_BR/LC_MESSAGES/fwupd.mo ../po/pt_BR.po
[41/2923] /usr/bin/msgfmt -o po/zh_CN/LC_MESSAGES/fwupd.mo ../po/zh_CN.po
[42/2923] /usr/bin/msgfmt -o po/si/LC_MESSAGES/fwupd.mo ../po/si.po
[43/2923] /usr/bin/msgfmt -o po/pl/LC_MESSAGES/fwupd.mo ../po/pl.po
[44/2923] /usr/bin/msgfmt -o po/uk/LC_MESSAGES/fwupd.mo ../po/uk.po
[45/2923] /usr/bin/msgfmt -o po/sv/LC_MESSAGES/fwupd.mo ../po/sv.po
[46/2923] /usr/bin/python3 ../generate-build/generate-metainfo.py ../data/org.freedesktop.fwupd.metainfo.xml data/org.freedesktop.fwupd.metainfo.xml
[47/2923] /usr/bin/python3 -m zipfile -c libfwupdplugin/tests/firmware.zip ../libfwupdplugin/tests/colorhug/firmware.bin ../libfwupdplugin/tests/colorhug/firmware.bin.asc
[48/2923] /usr/bin/gcc -Ilibfwupd/libfwupd.so.2.0.0.p -Ilibfwupd -I../libfwupd -I. -I.. -I/usr/include/gio-unix-2.0 -I/usr/include/glib-2.0 -I/usr/lib64/glib-2.0/include -I/usr/include/libmount -I/usr/include/libjcat-1 -I/usr/include/json-glib-1.0 -fdiagnostics-color=always -D_FILE_OFFSET_BITS=64 -Wall -Winvalid-pch -Wextra -std=c11 -DFWUPD_COMPILATION -D_DEFAULT_SOURCE -D_BSD_SOURCE -D__BSD_VISIBLE -D_XOPEN_SOURCE=700 -D_GNU_SOURCE -D_DARWIN_C_SOURCE=900000 -O2 -g -grecord-gcc-switches -pipe -Wall -Werror=format-security -Wp,-U_FORTIFY_SOURCE,-D_FORTIFY_SOURCE=3 -Wp,-D_GLIBCXX_ASSERTIONS -specs=/usr/lib/rpm/redhat/redhat-hardened-cc1 -fstack-protector-strong -specs=/usr/lib/rpm/redhat/redhat-annobin-cc1 -fno-omit-frame-pointer -mno-omit-leaf-frame-pointer -m64 -mtune=generic -fasynchronous-unwind-tables -fstack-clash-protection -fcf-protection -fdata-sections -ffunction-sections -flto=auto -flto-partition=none -Os -fPIC -pthread '-DG_LOG_DOMAIN="Fwupd"' '-DLOCALSTATEDIR="/var"' -MD -MQ libfwupd/libfwupd.so.2.0.0.p/fwupd-version.c.o -MF libfwupd/libfwupd.so.2.0.0.p/fwupd-version.c.o.d -o libfwupd/libfwupd.so.2.0.0.p/fwupd-version.c.o -c ../libfwupd/fwupd-version.c
[... cut ...]
      |                                     ~~~~~~~~~~~~~~^~~~~~~~~~~
[123/2923] /usr/bin/python3 ../plugins/uefi-capsule/make-images.py --podir /home/tkloczko/rpmbuild/BUILD/fwupd-1.9.4/po --label 'Installing firmware update…' --out plugins/uefi-capsule/uefi-capsule-ux.tar.xz
ninja: build stopped: subcommand failed.
INFO: autodetecting backend as ninja
INFO: calculating backend command to run: /usr/bin/ninja -C /home/tkloczko/rpmbuild/BUILD/fwupd-1.9.4/x86_64-redhat-linux-gnu -j 48 -v
```
</details>

jackpot51, 2023-08-30:
There are issues with that change that we are going to resolve on the PR itself.

kloczek, 2023-08-30:
> @kloczek Can you show me `git status` and `git rev-parse HEAD` of your fwupd source?

I'm using autogenerated from git tag + PR. From my spec file
```spec
URL:            https://github.com/fwupd/fwupd/
VCS:            https://github.com/fwupd/fwupd/
Source:         %{VCS}/releases/download/%{version}//%{name}-%{version}.tar.xz
Patch:          %{VCS}/pull/6131.psatc#/%{name}-Add-libfwupdplugin-generated-structs-as-input-to-all-built-in.patch

[..]

%prep
%autosetup -p1
```
If you want me to test current master + PR please let me know.

hughsie, 2023-08-31:
@kloczek https://github.com/fwupd/fwupd/pull/6136 might be worth testing too

kloczek, 2023-08-31:
OK .. one sec 😋

kloczek, 2023-08-31:
Looks like new PR solves reported issue 😄 👍

hughsie, 2023-09-01:
> It seems fwupdplugin_rs_dep needs to also be added to client_dep in src/meson.build to fully solve this issue for us.

Good catch! I've pushed this directly to the main branch, thanks.

jackpot51, 2023-09-01:
@hughsie I really appreciate your speed in fixing this issue. Since this was affecting some Linux distro builds, would it be a good time for a tagged release?

hughsie, 2023-09-01:
@jackpot51 Monday? Today I've got a lot of stuff to finish up before EOD.

LinuxUserGD, 2023-09-09:
I can still reproduce the error with 1.9.5 on Gentoo Linux (amd64/llvm/musl):
<details><summary>Error:</summary>

```bash
samu: job failed: clang -Isrc/libfwupdengine.so.p -Isrc -I../fwupd-1.9.5/src -I. -I../fwupd-1.9.5 -Ilibfwupd -I../fwupd-1.9.5/libfwupd -Ilibfwupdplugin -I../fwupd-1.9.5/libfwupdplugin -I/usr/include/libjcat-1 -I/usr/include/glib-2.0 -I/usr/lib/glib-2.0/include -I/usr/include/json-glib-1.0 -I/usr/lib/libffi/include -I/usr/include/libmount -I/usr/include/blkid -I/usr/include/libxmlb-2 -I/usr/include/libgcab-1.0 -I/usr/include/gio-unix-2.0 -I/usr/include/gudev-1.0 -I/usr/include/gusb-1 -I/usr/include/libusb-1.0 -I/usr/include/polkit-1 -fcolor-diagnostics -D_FILE_OFFSET_BITS=64 -Wall -Winvalid-pch -Wextra -std=c11 -Waggregate-return -Wunused -Warray-bounds -Wcast-align -Wdeclaration-after-statement -Wempty-body -Wformat=2 -Wformat-nonliteral -Wformat-security -Wignored-qualifiers -Wimplicit-function-declaration -Wimplicit-int -Winit-self -Wint-conversion -Wmissing-declarations -Wmissing-format-attribute -Wmissing-include-dirs -Wmissing-noreturn -Wmissing-prototypes -Wnested-externs -Wno-cast-function-type -Wno-address-of-packed-member -Wno-unknown-pragmas -Wno-missing-field-initializers -Wno-strict-aliasing -Wno-typedef-redefinition -Wno-unknown-warning-option -Wno-unused-parameter -Wold-style-definition -Woverride-init -Wpointer-arith -Wredundant-decls -Wreturn-type -Wshadow -Wsign-compare -Wstrict-aliasing -Wstrict-prototypes -Wswitch-default -Wtype-limits -Wundef -Wuninitialized -Wunused-but-set-variable -Wunused-variable -Wvla -Wwrite-strings -fstack-protector-strong -DFWUPD_COMPILATION -D_DEFAULT_SOURCE -D_BSD_SOURCE -D__BSD_VISIBLE -D_XOPEN_SOURCE=700 -D_GNU_SOURCE -D_DARWIN_C_SOURCE=900000 -O3 -pipe -march=native -mtune=native -D_FORTIFY_SOURCE=3 -g0 -flto -Wno-incompatible-function-pointer-types -fPIC -pthread -MD -MQ src/libfwupdengine.so.p/fu-udev-backend.c.o -MF src/libfwupdengine.so.p/fu-udev-backend.c.o.d -o src/libfwupdengine.so.p/fu-udev-backend.c.o -c ../fwupd-1.9.5/src/fu-udev-backend.c
In file included from ../fwupd-1.9.5/src/fu-udev-backend.c:11:
In file included from ../fwupd-1.9.5/libfwupdplugin/fwupdplugin.h:47:
../fwupd-1.9.5/libfwupdplugin/fu-efi-signature.h:11:10: fatal error: 'fu-efi-struct.h' file not found
   11 | #include "fu-efi-struct.h"
      |          ^~~~~~~~~~~~~~~~~
1 error generated.
samu: job failed: clang -Isrc/libfwupdengine.so.p -Isrc -I../fwupd-1.9.5/src -I. -I../fwupd-1.9.5 -Ilibfwupd -I../fwupd-1.9.5/libfwupd -Ilibfwupdplugin -I../fwupd-1.9.5/libfwupdplugin -I/usr/include/libjcat-1 -I/usr/include/glib-2.0 -I/usr/lib/glib-2.0/include -I/usr/include/json-glib-1.0 -I/usr/lib/libffi/include -I/usr/include/libmount -I/usr/include/blkid -I/usr/include/libxmlb-2 -I/usr/include/libgcab-1.0 -I/usr/include/gio-unix-2.0 -I/usr/include/gudev-1.0 -I/usr/include/gusb-1 -I/usr/include/libusb-1.0 -I/usr/include/polkit-1 -fcolor-diagnostics -D_FILE_OFFSET_BITS=64 -Wall -Winvalid-pch -Wextra -std=c11 -Waggregate-return -Wunused -Warray-bounds -Wcast-align -Wdeclaration-after-statement -Wempty-body -Wformat=2 -Wformat-nonliteral -Wformat-security -Wignored-qualifiers -Wimplicit-function-declaration -Wimplicit-int -Winit-self -Wint-conversion -Wmissing-declarations -Wmissing-format-attribute -Wmissing-include-dirs -Wmissing-noreturn -Wmissing-prototypes -Wnested-externs -Wno-cast-function-type -Wno-address-of-packed-member -Wno-unknown-pragmas -Wno-missing-field-initializers -Wno-strict-aliasing -Wno-typedef-redefinition -Wno-unknown-warning-option -Wno-unused-parameter -Wold-style-definition -Woverride-init -Wpointer-arith -Wredundant-decls -Wreturn-type -Wshadow -Wsign-compare -Wstrict-aliasing -Wstrict-prototypes -Wswitch-default -Wtype-limits -Wundef -Wuninitialized -Wunused-but-set-variable -Wunused-variable -Wvla -Wwrite-strings -fstack-protector-strong -DFWUPD_COMPILATION -D_DEFAULT_SOURCE -D_BSD_SOURCE -D__BSD_VISIBLE -D_XOPEN_SOURCE=700 -D_GNU_SOURCE -D_DARWIN_C_SOURCE=900000 -O3 -pipe -march=native -mtune=native -D_FORTIFY_SOURCE=3 -g0 -flto -Wno-incompatible-function-pointer-types -fPIC -pthread -MD -MQ src/libfwupdengine.so.p/fu-security-attr-common.c.o -MF src/libfwupdengine.so.p/fu-security-attr-common.c.o.d -o src/libfwupdengine.so.p/fu-security-attr-common.c.o -c ../fwupd-1.9.5/src/fu-security-attr-common.c
In file included from ../fwupd-1.9.5/src/fu-security-attr-common.c:14:
In file included from ../fwupd-1.9.5/src/fu-security-attr-common.h:9:
In file included from ../fwupd-1.9.5/libfwupdplugin/fwupdplugin.h:47:
../fwupd-1.9.5/libfwupdplugin/fu-efi-signature.h:11:10: fatal error: 'fu-efi-struct.h' file not found
   11 | #include "fu-efi-struct.h"
      |          ^~~~~~~~~~~~~~~~~
1 error generated.
samu: job failed: clang -Isrc/libfwupdengine.so.p -Isrc -I../fwupd-1.9.5/src -I. -I../fwupd-1.9.5 -Ilibfwupd -I../fwupd-1.9.5/libfwupd -Ilibfwupdplugin -I../fwupd-1.9.5/libfwupdplugin -I/usr/include/libjcat-1 -I/usr/include/glib-2.0 -I/usr/lib/glib-2.0/include -I/usr/include/json-glib-1.0 -I/usr/lib/libffi/include -I/usr/include/libmount -I/usr/include/blkid -I/usr/include/libxmlb-2 -I/usr/include/libgcab-1.0 -I/usr/include/gio-unix-2.0 -I/usr/include/gudev-1.0 -I/usr/include/gusb-1 -I/usr/include/libusb-1.0 -I/usr/include/polkit-1 -fcolor-diagnostics -D_FILE_OFFSET_BITS=64 -Wall -Winvalid-pch -Wextra -std=c11 -Waggregate-return -Wunused -Warray-bounds -Wcast-align -Wdeclaration-after-statement -Wempty-body -Wformat=2 -Wformat-nonliteral -Wform [... cut ...] fu-daemon.c.o -MF src/fwupd.p/fu-daemon.c.o.d -o src/fwupd.p/fu-daemon.c.o -c ../fwupd-1.9.5/src/fu-daemon.c
In file included from ../fwupd-1.9.5/src/fu-daemon.c:11:
In file included from ../fwupd-1.9.5/libfwupdplugin/fwupdplugin.h:47:
../fwupd-1.9.5/libfwupdplugin/fu-efi-signature.h:11:10: fatal error: 'fu-efi-struct.h' file not found
   11 | #include "fu-efi-struct.h"
      |          ^~~~~~~~~~~~~~~~~
1 error generated.
samu: subcommands failed
```

</details>

Should I open a new bug report?

kloczek, 2023-11-01:
Just testes 1.9.7 and looks like this issue still is not resolved.
```console
[65/1107] /usr/bin/gcc -Iplugins/cpu/libfwupdcethelper.a.p -Iplugins/cpu -I../plugins/cpu -I. -I.. -fdiagnostics-color=always -D_FILE_OFFSET_BITS=64 -Wall -Winvalid-pch -Wextra -std=c11 -DFWUPD_COMPILATION -D_DEFAULT_SOURCE -D_BSD_SOURCE -D__BSD_VISIBLE -D_XOPEN_SOURCE=700 -D_GNU_SOURCE -D_DARWIN_C_SOURCE=900000 -O2 -g -grecord-gcc-switches -pipe -Wall -Werror=format-security -Wp,-U_FORTIFY_SOURCE,-D_FORTIFY_SOURCE=3 -Wp,-D_GLIBCXX_ASSERTIONS -specs=/usr/lib/rpm/redhat/redhat-hardened-cc1 -fstack-protector-strong -specs=/usr/lib/rpm/redhat/redhat-annobin-cc1 -fno-omit-frame-pointer -mno-omit-leaf-frame-pointer -m64 -mtune=generic -fasynchronous-unwind-tables -fstack-clash-protection -fcf-protection -fdata-sections -ffunction-sections -flto=auto -flto-partition=none -Os -fPIC -fcf-protection=none -MD -MQ plugins/cpu/libfwupdcethelper.a.p/fu-cpu-helper-cet-common.c.o -MF plugins/cpu/libfwupdcethelper.a.p/fu-cpu-helper-cet-common.c.o.d -o plugins/cpu/libfwupdcethelper.a.p/fu-cpu-helper-cet-common.c.o -c ../plugins/cpu/fu-cpu-helper-cet-common.c
[66/1107] /usr/bin/python3 /home/tkloczko/rpmbuild/BUILD/fwupd-1.9.7/libfwupdplugin/rustgen.py ../libfwupdplugin/fu-ifwi.rs libfwupdplugin/fu-ifwi-struct.c libfwupdplugin/fu-ifwi-struct.h
[67/1107] /usr/bin/gcc -Ilibfwupdplugin/gcab.p -Ilibfwupdplugin -I../libfwupdplugin -I. -I.. -Ilibfwupd -I../libfwupd -I/usr/include/libxmlb-2 -I/usr/include/glib-2.0 -I/usr/lib64/glib-2.0/include -I/usr/include/gio-unix-2.0 -I/usr/include/gusb-1 -I/usr/include/libusb-1.0 -I/usr/include/json-glib-1.0 -I/usr/include/gudev-1.0 -I/usr/include/valgrind -fdiagnostics-color=always -D_FILE_OFFSET_BITS=64 -Wall -Winvalid-pch -Wextra -std=c11 -DFWUPD_COMPILATION -D_DEFAULT_SOURCE -D_BSD_SOURCE -D__BSD_VISIBLE -D_XOPEN_SOURCE=700 -D_GNU_SOURCE -D_DARWIN_C_SOURCE=900000 -O2 -g -grecord-gcc-switches -pipe -Wall -Werror=format-security -Wp,-U_FORTIFY_SOURCE,-D_FORTIFY_SOURCE=3 -Wp,-D_GLIBCXX_ASSERTIONS -specs=/usr/lib/rpm/redhat/redhat-hardened-cc1 -fstack-protector-strong -specs=/usr/lib/rpm/redhat/redhat-annobin-cc1 -fno-omit-frame-pointer -mno-omit-leaf-frame-pointer -m64 -mtune=generic -fasynchronous-unwind-tables -fstack-clash-protection -fcf-protection -fdata-sections -ffunction-sections -flto=auto -flto-partition=none -Os -pthread -MD -MQ libfwupdplugin/gcab.p/fu-gcab.c.o -MF libfwupdplugin/gcab.p/fu-gcab.c.o.d -o libfwupdplugin/gcab.p/fu-gcab.c.o -c ../libfwupdplugin/fu-gcab.c
FAILED: libfwupdplugin/gcab.p/fu-gcab.c.o
/usr/bin/gcc -Ilibfwupdplugin/gcab.p -Ilibfwupdplugin -I../libfwupdplugin -I. -I.. -Ilibfwupd -I../libfwupd -I/usr/include/libxmlb-2 -I/usr/include/glib-2.0 -I/usr/lib64/glib-2.0/include -I/usr/include/gio-unix-2.0 -I/usr/include/gusb-1 -I/usr/include/libusb-1.0 -I/usr/include/json-glib-1.0 -I/usr/include/gudev-1.0 -I/usr/include/valgrind -fdiagnostics-color=always -D_FILE_OFFSET_BITS=64 -Wall -Winvalid-pch -Wextra -std=c11 -DFWUPD_COMPILATION -D_DEFAULT_SOURCE -D_BSD_SOURCE -D__BSD_VISIBLE -D_XOPEN_SOURCE=700 -D_GNU_SOURCE -D_DARWIN_C_SOURCE=900000 -O2 -g -grecord-gcc-switches -pipe -Wall -Werror=format-security -Wp,-U_FORTIFY_SOURCE,-D_FORTIFY_SOURCE=3 -Wp,-D_GLIBCXX_ASSERTIONS -specs=/usr/lib/rpm/redhat/redhat-hardened-cc1 -fstack-protector-strong -specs=/usr/lib/rpm/redhat/redhat-annobin-cc1 -fno-omit-frame-pointer -mno-omit-leaf-frame-pointer -m64 -mtune=generic -fasynchronous-unwind-tables -fstack-clash-protection -fcf-protection -fdata-sections -ffunction-sections -flto=auto -flto-partition=none -Os -pthread -MD -MQ libfwupdplugin/gcab.p/fu-gcab.c.o -MF libfwupdplugin/gcab.p/fu-gcab.c.o.d -o libfwupdplugin/gcab.p/fu-gcab.c.o -c ../libfwupdplugin/fu-gcab.c
In file included from ../libfwupdplugin/fwupdplugin.h:52,
                 from ../libfwupdplugin/fu-gcab.c:9:
../libfwupdplugin/fu-efi-signature.h:11:10: fatal error: fu-efi-struct.h: No such file or directory
   11 | #include "fu-efi-struct.h"
      |          ^~~~~~~~~~~~~~~~~
compilation terminated.
```

superm1, 2023-11-01:
I guess this regressed when the native CAB support was added (in lieu of `libgcab`)

superm1, 2023-11-01:
I CND, but maybe this helps?

```
diff --git a/libfwupdplugin/meson.build b/libfwupdplugin/meson.build
index 153f7c035..fd94096fd 100644
--- a/libfwupdplugin/meson.build
+++ b/libfwupdplugin/meson.build
@@ -386,7 +386,8 @@ gcab = executable(
     fwupd_incdir,
   ],
   dependencies: [
-    library_deps
+    library_deps,
+    fwupdplugin_rs_dep,
   ],
   link_with: [
     fwupd,
```

superm1, 2023-11-01:
I've merged that to main, see if it helps.

closed 2023-11-01 by kloczek