fwupd/fwupd #6109: Memory leak crashes system on attempt to update VMM5322

michaellass, 2023-08-22:
**Describe the bug**
fwupd shows the following available update for my system:
```
LENOVO 20UJS00K00
│
└─VMM5322:
  │   Device ID:          b9bf993c1d78d3022437375a91dd6f8181957dc6
  │   Summary:            Multi-stream transport device
  │   Current version:    5.04.01
  │   Vendor:             Synaptics (DRM_DP_AUX_DEV:0x06CB, DRM_DP_AUX_DEV:0x1002)
  │   GUIDs:              676a78a9-6e81-588f-bd38-1ea5d174801a ← MST-596
  │                       832f0673-f0f9-5bdb-b0d1-c1a8e7db76d3 ← MST-dock-596
  │                       b025a790-7adb-5f93-ba47-b5e1ce6e8588 ← MST-dock-vmm5322-596
  │                       f15aa55c-9cd5-5942-85ae-a6bf8740b96c ← MST-panamera
  │                       cf1bdd9f-29bc-5dab-9d14-dad7a4e06d16 ← MST-panamera-vmm5322-596
  │                       fa28f9c4-c5da-57d8-802b-c644e06159b4 ← MST-panamera-596
  │   Device Flags:       • Updatable
  │                       • Supported on remote server
  │                       • Device stages updates
  │                       • Unsigned Payload
  │ 
  └─ThinkPad Gen2 DP USB Dock Update:
        New version:      5.05.00
        Remote ID:        lvfs
        Release ID:       6036
        Summary:          Lenovo ThinkPad USB-C Dock Gen2 DP Firmware
        License:          MIT
        Size:             524.3 kB
        Created:          2020-08-06
        Urgency:          High
        Vendor:           Lenovo
        Release Flags:    • Trusted metadata
                          • Is upgrade
        Description:      
        Lenovo ThinkPad USB-C Dock Gen2 MSTHUB Firmware
        Checksum:         dbda394e4de684ff29674c93d59ac41bd6d09dc9fb92c9d9088a093ec1d00954
```

When trying to update via `sudo fwupdmgr update`, the process got stuck at `Writing ...`. A look at top showed that fwupdmgr was consuming Gigabytes of memory, increasing very quickly. When approaching the 32 GiB installed in my system, I attempted to cancel the update which made the system freeze.

**Steps to Reproduce**
I don't know if this problem is related to the specific hardware or firmware update. On a similar system, it should be reproducible by running `sudo fwupdmgr update`.

**Expected behavior**
A successful update. In case of errors, an error message.

**fwupd version information**
Please provide the version of the daemon and client.

```shell
compile   org.freedesktop.fwupd         1.9.4
compile   com.hughsie.libxmlb           0.3.11
compile   com.hughsie.libjcat           0.1.14
runtime   org.freedesktop.fwupd-efi     1.4
compile   org.freedesktop.gusb          0.4.6
runtime   com.hughsie.libjcat           0.1.14
runtime   com.dell.libsmbios            2.4
runtime   org.freedesktop.gusb          0.4.6
runtime   org.freedesktop.fwupd         1.9.4
runtime   org.kernel                    6.4.11-arch2-1
```

Please note how you installed it (`apt`, `dnf`, `pacman`, source, etc):

I'm running Arch Linux and fwupd is installed from Arch's repositories (package version 1.9.4-1). The build source for the current package can be found here:
https://gitlab.archlinux.org/archlinux/packaging/packages/fwupd/-/blob/43e8a866453c2363377242b7cca714572928a2d8/PKGBUILD

<details>

<summary>**fwupd device information**</summary>
[... 302 lines cut ...]
│   
└─WDC PC SN730 SDBQNTY-1T00-1001:
      Device ID:          71b677ca0f1bc2c5b804fa1d59e52064ce589293
      Summary:            NVM Express solid state drive
      Current version:    11170101
      Vendor:             Sandisk Corp (NVME:0x15B7)
      Serial Number:      2018JE443513
      GUIDs:              fccbb6ea-e20e-58ad-bf8a-7fb7d43ff4c2 ← NVME\VEN_15B7&DEV_5006
                          12c86995-0b90-5ec5-98f3-7a6ed4ca50e0 ← NVME\VEN_15B7&DEV_5006&SUBSYS_15B75006
                          ad5de143-d4f4-5a5f-8b26-b118b2ea81d4 ← WDC PC SN730 SDBQNTY-1T00-1001
      Device Flags:       • Internal device
                          • Updatable
                          • System requires external power source
                          • Supported on remote server
                          • Needs a reboot after installation
                          • Device is usable for the duration of the update
```

</details>

**Additional questions**

- Operating system and version: Arch Linux (rolling release), Linux kernel 6.4.11.
- Have you tried rebooting? I have not dared to reattempt the update. I am worried of bricking my USB-C dock.
- Is this a regression? This might have been the first update for that specific device for me. So if it is a device-specific issue, I can't tell.

hughsie, 2023-08-22:
I see it too! Spicy!

[image]

hughsie, 2023-08-22:
This is what I see:

	#1  0x00007ffff7504b4e in __libc_calloc (n=n@entry=1, elem_size=elem_size@entry=72) at malloc.c:3664
	#2  0x00007ffff76d9092 in g_malloc0 (n_bytes=n_bytes@entry=72) at ../glib/gmem.c:163
	#3  0x00007ffff77d22e1 in g_closure_new_simple (sizeof_closure=sizeof_closure@entry=40, data=data@entry=0x6aa640) at ../gobject/gclosure.c:222
	#4  0x00007ffff77d2987 in g_cclosure_new (callback_func=0x7ffff7b94f0e <fu_progress_child_percentage_changed_cb>, user_data=0x6aa640, destroy_data=0x0)
	    at ../gobject/gclosure.c:978
	#5  0x00007ffff77ee1e6 in g_signal_connect_data
	    (instance=0x7e14bef0, detailed_signal=<optimized out>, c_handler=0x7ffff7b94f0e <fu_progress_child_percentage_changed_cb>, data=0x6aa640, destroy_data=0x0, connect_flags=G_CONNECT_DEFAULT) at ../gobject/gsignal.c:2613
	#6  0x00007ffff7b95369 in fu_progress_add_step (self=0x6aa640, status=FWUPD_STATUS_DEVICE_WRITE, value=0, name=0x0) at ../libfwupdplugin/fu-progress.c:684
	#7  0x00007ffff7b94c45 in fu_progress_set_steps (self=0x6aa640, step_max=64617493) at ../libfwupdplugin/fu-progress.c:518
	#8  0x00007ffff7de5293 in fu_synaptics_mst_device_update_panamera_firmware
	    (self=0x7342c0, payload_len=524288, payload_data=0xa832a0 "", progress=0x6aa640, error=0x7fffffffd8a0) at ../plugins/synaptics-mst/fu-synaptics-mst-device.c:615

...which I think is 64617493×72 bytes, probably a few more for the progress accounting. Debugging now.

hughsie, 2023-08-22:
> 15:13:16.268 FuPluginSynapticsMST calculated fw size as 4135519504

That's 3.85GB as a firmware which seems *implausible*.

michaellass, 2023-08-22:
Looks like you are much quicker debugging than I am in responding. Still, on your questions:

> > sudo fwupdmgr update
> 
> Are you sure it was the `fwupdmgr` process that was consuming huge amounts of ram, or the `fwupd` process?

No. It's very well possible that it was `fwupd`.

> > ThinkPad Gen2
> 
> Can you link to the dock product page on Lenovo please? I've got a lot of docks and also a lot of "Gen2" ThinkPad ones :)

It should be this one: https://support.lenovo.com/de/en/accessories/acc500106-thinkpad-usb-c-dock-gen-2-overview-and-service-parts

The bottom of the case shows:
* Model NO: LDC-G2
* ASM PN: SD20S97543
* FRU PN: 03X7609
* TYPE: 40AS

In particular, it should be the 40AS0090EU variant, i.e., the one sold in the EU.

> Probably not related to this bug, but why the sudo?

I always assumed, `fwupdmgr` would require root permissions but never tested running it without.

hughsie, 2023-08-22:
I've demoted this firmware on the lvfs until we've worked out what's going on.

closed 2023-08-24 by hughsie

superm1, 2023-08-24:
> I've demoted this firmware on the lvfs until we've worked out what's going on.

Given it's been broken a while I guess any of this firmware needs a match added to only install with fwupd 1.9.5 or later.

hughsie, 2023-08-24:
> Given it's been broken a while I guess any of this firmware needs a match added to only install with fwupd 1.9.5 or later.

Done.

michaellass, 2023-09-04:
Thanks for fixing this so quickly! I just updated fwupd to 1.9.5 but the update is not shown again. Is it still blocked entirely instead of being limited to >1.9.5?

hughsie, 2023-09-05:
> Is it still blocked entirely instead of being limited to >1.9.5?

Ahh yes, good catch. I've returned the firmware to the stable remote now it has the check for fwupd >= 1.9.5 -- it'll take at least 4 hours to show up.

michaellass, 2023-09-05:
Great! The update now worked flawlessly :slightly_smiling_face: