fwupd/fwupd #6118: Acpi bug

QueDemais, 2023-08-27:
**Describe the bug**
A clear and concise description of what the bug is. Hello, throughout the boot, acpi errors appear, acpi bios bug, x509, etc.

**Steps to Play**
Steps to reproduce the behavior. Just starting the computer, even with everything updated, these errors appear. These errors come from many years ago, since I started using Ubuntu, ubuntu 14 lts. And it is also in any kernel version, even the most current one I used.

**Expected behavior**
A clear and concise description of what you expected to happen. That after the most recent update, the bug was resolved and corrected, without affecting anything, but rather gaining performance without showing the errors.

**fwupd version information**
Please provide the version of the daemon and client.
fwupdmgr --version
runtime org.freedesktop.fwupd 1.7.9
runtime com.dell.libsmbios 2.4
compile org.freedesktop.gusb 0.3.10
runtime org.kernel 6.2.0-26-generic
compile com.hughsie.libjcat 0.1.9
compile org.freedesktop.fwupd 1.7.9
runtime org.freedesktop.gusb 0.3.10

Please note how you installed it (`apt`, `dnf`, `pacman`, source, etc): apenas utlizei o comando fwupdmgr --force refresh && fwupdmgr update

<details>

<summary>**fwupd device information**</summary>

Please provide the output of the fwupd devices recognized in your system.

``` shell
fwupdmgr get-devices --show-all-devices
```
82MD
│
├─0000:00:1f.5:
│ Device ID:
│ Summary: Memory Technology Device
│ Vendor: DMI:LENOVO
│ GUID:
│ Device Flags: • Internal device
│ • Updatable
│ • Needs a reboot after installation
│
├─11th Gen Intel Core™ i3-1115G4 @ 3.00GHz:
│ Device ID:
│ Current version: 0x0000002c
│ Vendor: Intel
│ GUID:
│ Device Flags: • Internal device
│
├─Display controller:
│ Device ID:
│ Current version: 03
│ Vendor: Intel Corporation (PCI:0x8086)
│ GUID:
│ Device Flags: • Internal device
│ • Cryptographic hash verification is available
│
├─NVMe  256GB:
│ Device ID:
│ Summary: NVM Express solid state drive
│ Current version: 1B114215
│ Vendor: ADATA Technology Co., Ltd. (NVME:0x1CC1)
│ GUID:
│ Device Flags: • Updateable
│ • System requires external power source
│ • Needs a reboot after installation
│
├─System Firmware:
│ │ Device ID:
│ │ Summary: UEFI ESRT device
│ │ Current version: 1497841749
│ │ Minimum Version: 1380122624
│ │ Vendor: LENOVO (DMI:LENOVO)
│ │ Update State: Success
│ │ GUID:
│ │ Device Flags: • Internal device
│ │ • Updatable
│ │ • System requires external power source
│ │ • Needs a reboot after installation
│ │ • Cryptographic hash verification is available
│ │ • Device is usable for the duration of the update
│ │
│ └─UEFI dbx:
│ Device ID:
│ Summary: UEFI revocation database
│ Current version: 421
│ Minimum Version: 421
│ Vendor: UEFI:Linux Foundation
│ Install Duration: 1 second
│ GUID:
│ Device Flags: • Internal device
│ • Updatable
│ • Needs a reboot after installation
│ • Only version upgrades are allowed
│ • Signed Payload
│
├─TPM:
│ Device ID:
│ Summary: TPM 2.0 Device
│ Current version: 600.7.0.0
│ Vendor: Intel (TPM:INTC)
│ GUID:
│ Device Flags: • Internal device
│
├─UEFI Device Firmware:
│ Device ID:
│ Summary: UEFI ESRT device
│ Current version: 1951
│ Minimum Version: 1000
│ Vendor: DMI:LENOVO
│ Update State: Success
│ GUID:
│ Device Flags: • Internal device
│ • Updatable
│ • System requires external power source
│ • Needs a reboot after installation
│ • Device is usable for the duration of the update
│
├─UEFI Device Firmware:
│ Device ID:

</details>

**System UEFI configuration**
Please provide the output of the following commands:

```shell
efibootmgr -v
```
BootCurrent: 0002
Timeout: 0 seconds
BootOrder: 0002,2001,2002,2003
Boot0002* ubuntu        HD(1,GPT,)/File(\EFI\ubuntu\shimx64.efi)

**Additional questions**

- Operating system and version: Kubuntu Lts 22.04.3
- Have you tried rebooting? yes
- Is this a regression? I don't know
- Are you using an NVMe disk? Yes
- Is secure boot enabled? Yes
- Is this a Lenovo system with 'Boot Order Lock' turned on in the BIOS? and laptop lenovo ideapad 3 15itl6, with device guard enabled. I don't know about the boot order lock.

hughsie, 2023-08-27:
Why do you think this is a fwupd bug?

QueDemais, 2023-08-27:
> Por que você acha que isso é um bug do fwupd?

Ola, acredito que seja no nivel de firmware.
Dai queria deixar o feedback 👍😄

hughsie, 2023-08-28:
Right, but fwupd doesn't solve firmware bugs. I think you need to contact your laptop OEM, sorry.

closed 2023-08-28 by hughsie

QueDemais, 2023-08-28:
👍