fwupd/fwupd #6129: Possible incorrect archive UEFI 1.26.0 update for dell latitude 5401 for fwupd.

stargladius, 2023-08-30:
When I try to update there is an error.  I am trying to find out the reason why there is an error with update for dell latitude 5401.
Please tell me if I'm right and if I understand correctly the cause of the error.

Error with _fwupdmgr update_
```
Perform operation? [Y|n]: y
Downloading…             [***************************************]
Downloading…             [***************************************]
cannot find Latitude_5X01_Precision_3541_1.26.0.cab in archive
```

If there is such a lines in the file firmware.metainfo.xml , then the archive must also contain such a file Latitude_5X01_Precision_3541_1.26.0.cab ?

```
<artifact type="binary">
           <filename>Latitude_5X01_Precision_3541_1.26.0.cab</filename>
</artifact>
```

But it isn't there.

_fwupdmgr get-updates --json_
```
 "Locations" : [
            "https://fwupd.org/downloads/75716e9132fdbe86671f4853e08e6d947c19b04f771071b56b8978c0f133d2cd-Latitude_5X01_Precision_3541_1.26.0.cab"
          ],
          "Uri" : "https://fwupd.org/downloads/75716e9132fdbe86671f4853e08e6d947c19b04f771071b56b8978c0f133d2cd-Latitude_5X01_Precision_3541_1.26.0.cab",
```

```
cabextract -t 75716e9132fdbe86671f4853e08e6d947c19b04f771071b56b8978c0f133d2cd-Latitude_5X01_Precision_3541_1.26.0.cab 
Testing cabinet: 75716e9132fdbe86671f4853e08e6d947c19b04f771071b56b8978c0f133d2cd-Latitude_5X01_Precision_3541_1.26.0.cab
  README.txt  OK                               47dec8984e330f7e490f614f9231d571
  catalog.cat  OK                              eaa6cd7d1aac6a9485a5c0ddf15c59db
  firmware.bin  OK                             66e17ddb2fb9f65ab2c1abf9e6ee4eec
  firmware.bin.asc  OK                         727c51bb652c4a7f39ea1b4f3c779a1e
  firmware.inf  OK                             817f4bdbab1ecaad6bab0d2b593a843b
  firmware.jcat  OK                            be01d4258e5a544ba479cfd27c4055dc
  firmware.metainfo.xml  OK                    9c7b14fc1c07f88c40f7d5821da12e58
```

**fwupd version information**
runtime   org.freedesktop.fwupd         1.7.9
runtime   com.dell.libsmbios            2.4
compile   org.freedesktop.gusb          0.3.10
compile   com.hughsie.libjcat           0.1.9
runtime   org.kernel                    6.2.0-31-generic
compile   org.freedesktop.fwupd         1.7.9
runtime   org.freedesktop.gusb          0.3.10

hughsie, 2023-08-30:
this was a regression on the LVFS. I've got a cron job running that's 88% complete that's resigning all affected firmware, and that should be complete in a few hours. Can you try again in ~4 hours please.

hughsie, 2023-08-30:
Nah, it's not Dells fault, it's mine :) https://gitlab.com/fwupd/lvfs-website/-/commit/fd42f1bf0a472e50f54488870cf3d6f1e2c10f63 for the curious.

stargladius, 2023-08-31:
Hello. I have checked it today in morning as you asked to do it later. Still incorrect archive.

hughsie, 2023-08-31:
Did you `fwupdmgr refresh --force` before?

closed 2023-08-31 by hughsie