fwupd/fwupd #6141: some USB devices seem to not work in snap

superm1, 2023-09-02:
A [downstream bug report](https://bugs.launchpad.net/ubuntu/+source/fwupd/+bug/2033573) was opened about how FPC fingerprint readers aren't working in the snap.  I was intrigued and found I can reproduce it with DFU devices on my machine too.

```
00:54:34.246 FuPluginDfu          DFU v1.1
00:54:34.246 FuEngine             failed to add device usb:01:00:03: USB error on device 04d8:0039 : Input/Output Error [-1]
00:54:34.256 FuPluginDfu          DFU v1.1
00:54:34.256 FuEngine             failed to add device usb:01:00:04: USB error on device 04f2:b768 : Input/Output Error [-1]
```

I rebuilt the exact same version of fwupd outside the snap and can't reproduce it.

[fwupdtoll_get_devices_dfu_strace.txt](https://github.com/fwupd/fwupd/files/12502159/fwupdtoll_get_devices_dfu_strace.txt)
[no_snap.txt](https://github.com/fwupd/fwupd/files/12502160/no_snap.txt)

@valentindavid any ideas?

valentindavid, 2023-09-06:
Do you have any log from apparmor or audit logs? I have tried lots of usb devices, but I am not sure I have anything that can be updated.

superm1, 2023-09-07:
This is a different machine but that shows same error.

```
[  277.277158] audit: type=1400 audit(1694059561.720:131): apparmor="DENIED" operation="capable" class="cap" profile="/snap/snapd/19993/usr/lib/snapd/snap-confine" pid=4693 comm="snap-confine" capability=12  capname="net_admin"
[  277.277169] audit: type=1400 audit(1694059561.720:132): apparmor="DENIED" operation="capable" class="cap" profile="/snap/snapd/19993/usr/lib/snapd/snap-confine" pid=4693 comm="snap-confine" capability=38  capname="perfmon"
[  277.326107] audit: type=1326 audit(1694059561.768:133): auid=1000 uid=0 gid=0 ses=3 subj=snap.fwupd.fwupdtool pid=4693 comm="fwupdtool" exe="/snap/fwupd/5125/bin/fwupdtool" sig=0 arch=c000003e syscall=314 compat=0 ip=0x7ff0d7bb9a3d code=0x50000
```

binli, 2023-09-07:
I've a ThinkPad E14 with  FPC (10a5:9800) USB fingerprint device. Here are the log when I run 'sudo /snap/bin/fwupd.fwupdtool get-devices'.

Sep 07 02:43:27 u-ThinkPad-E14-Gen-5 sudo[46294]: pam_unix(sudo:session): session opened for user root(uid=0) by (uid=1001)                       
Sep 07 02:43:27 u-ThinkPad-E14-Gen-5 systemd[1]: Started snap.fwupd.fwupdtool.a5363c7a-fe65-4509-8aab-c7e3590cb817.scope.                         
Sep 07 02:43:27 u-ThinkPad-E14-Gen-5 audit[46296]: SECCOMP auid=1001 uid=0 gid=0 ses=3 subj=snap.fwupd.fwupdtool pid=46296 comm="fwupdtool" exe="/
snap/fwupd/5139/bin/fwupdtool" sig=0 arch=c000003e syscall=314 compat=0 ip=0x7f9df3945a3d code=0x50000                                            
Sep 07 02:43:27 u-ThinkPad-E14-Gen-5 kernel: audit: type=1326 audit(1694069007.944:80): auid=1001 uid=0 gid=0 ses=3 subj=snap.fwupd.fwupdtool pid=
46296 comm="fwupdtool" exe="/snap/fwupd/5139/bin/fwupdtool" sig=0 arch=c000003e syscall=314 compat=0 ip=0x7f9df3945a3d code=0x50000               
Sep 07 02:43:28 u-ThinkPad-E14-Gen-5 udisksd[837]: Mounted /dev/nvme0n1p2 at /media/root/PQSERVICE on behalf of uid 0                             
Sep 07 02:43:28 u-ThinkPad-E14-Gen-5 audit[46296]: AVC apparmor="DENIED" operation="open" profile="snap.fwupd.fwupdtool" name="/media/root/PQSERVI
CE/" pid=46296 comm="fwupdtool" requested_mask="r" denied_mask="r" fsuid=0 ouid=0                                                                 
Sep 07 02:43:28 u-ThinkPad-E14-Gen-5 kernel: audit: type=1400 audit(1694069008.088:81): apparmor="DENIED" operation="open" profile="snap.fwupd.fwu
pdtool" name="/media/root/PQSERVICE/" pid=46296 comm="fwupdtool" requested_mask="r" denied_mask="r" fsuid=0 ouid=0                                
Sep 07 02:43:28 u-ThinkPad-E14-Gen-5 systemd[1]: media-root-PQSERVICE.mount: Deactivated successfully.                                            
Sep 07 02:43:28 u-ThinkPad-E14-Gen-5 udisksd[837]: Cleaning up mount point /media/root/PQSERVICE (device 259:2 is not mounted)                    
Sep 07 02:43:28 u-ThinkPad-E14-Gen-5 udisksd[837]: Unmounted /dev/nvme0n1p2 on behalf of uid 0                                                    
Sep 07 02:43:28 u-ThinkPad-E14-Gen-5 systemd[1]: snap.fwupd.fwupdtool.a5363c7a-fe65-4509-8aab-c7e3590cb817.scope: Deactivated successfully.

binli, 2023-09-07:
I also attched the log of 'sudo /snap/bin/fwupd.fwupdtool get-devices -v'
[b.log](https://github.com/fwupd/fwupd/files/12545623/b.log)

Loading…                 [**********                             ]04:55:57.957 FuEngine             failed to add device usb:03:00:06: USB error o
n device 10a5:9800 : Input/Output Error [-1]                                                                                                      
Loading…                 [************                           ]04:55:57.967 FuPluginDfu          DFU v1.1                                      
04:55:57.967 FuEngine             failed to add device usb:03:00:07: USB error on device 04f2:b7b4 : Input/Output Error [-1]                      
Loading…                 [****************                       ]04:55:57.992 FuDeviceList         ::added 5792b48846ce271fab11c4a545f7a3df0d36e0
0a [Display controller]                                                                                                                           
Loading…                 [*****************                      ]04:55:58.109 FuDeviceList         ::added 71b677ca0f1bc2c5b804fa1d59e52064ce5892
93 [KBG5AZNT512G LA KIOXIA]                                                                                                                       
Loading…                 [******************                     ]04:55:58.122 FuDevice             setting parent of USB4 host controller [c4a1b8
a457fcd7f2cdedf0f243d7a466a75ad0ba] to be 13th Gen Intel CoreT i5-1340P [4bde70ba4e39b28f9eab1628f9dd6e6244c03027]                                
04:55:58.122 FuDeviceList         ::added c4a1b8a457fcd7f2cdedf0f243d7a466a75ad0ba [USB4 host controller]

valentindavid, 2023-09-08:
The problem does not look like from apparmor. But the strace does not show EIO. I suppose you did not use `-f`. I see some `clone3` syscalls, but now new processes. You can maybe try to strace with `snap run --strace fwupd.fwupdtool [...]`. That should follow forks.

I have tried to fine some hardware at home that can be updated, but failed to do so. Do you know any cheap USB device and available worldwide that can be updated with fwupd? That would be helpful in the future to do manual testing when I need to.

binli, 2023-09-08:
[snap.strace.log](https://github.com/fwupd/fwupd/files/12558367/snap.strace.log)
Hi Valentin,
 Here is the strace log, but I didn't find any USB errors. And I've no idea for any cheap USB devices, the fingerprint is integrated into my laptop directlly, :)

hughsie, 2023-09-08:
@valentindavid I can send you a colorhug PCB if that helps. If so, email me your address.

valentindavid, 2023-09-08:
@binli It printed out "Command not found". I think you missed an "s" for `get-devices`. You wrote `get-device`.

binli, 2023-09-14:
@valentindavid Oh, sorry, my bad. I could find the 'EPERM' errors now.

[strace.log](https://github.com/fwupd/fwupd/files/12605344/strace.log)

[pid  4511] openat(AT_FDCWD, "/dev/bus/usb/003/002", O_RDWR|O_CLOEXEC) = -1 EPERM (Operation not permitted)                                       
[pid  4511] write(2, "06:00:45.997 ", 1306:00:45.997 ) = 13                                                                                       
[pid  4511] write(2, "FuEngine             ", 21FuEngine             ) = 21                                                                       
[pid  4511] write(2, "failed to add device usb:03:00:0"..., 91failed to add device usb:03:00:06: USB error on device 10a5:9800 : Input/Output Erro
r [-1]

binli, 2023-09-14:
I also tried the snapd from edge channel, 2.60.3+git1321.gd0a3a9a and it still didn't work.

https://github.com/snapcore/snapd/commit/31df9fa676dc00a34f040c52c449f861b89dc095

I found you had added USB permission in snapd. Should we change the permission from 'r' to 'rw'? Thanks!

  # Plugin for Logitech Whiteboard camera and fingerprint devices
  /dev/bus/usb/[0-9][0-9][0-9]/[0-9][0-9][0-9] rw,
  /dev/video[0-9]* r,

valentindavid, 2023-09-21:
Thank you very much @binli

binli, 2023-09-22:
@valentindavid My pleasure, thanks for the commit.  I could help verify it when the new version snapd is released.

closed 2023-09-22 by superm1

superm1, 2023-09-22:
@binli the raw-usb plug is added to the edge snap.  Can you upgrade to that and manually connect it?  If it works, then I think we should open a store request to auto-connect it.

binli, 2023-09-28:
@superm1 hi, it doesn't work in my side, did I make something wrong? Thanks!

$ sudo snap refresh snapd --channel latest/edge

$ snap list | grep snapd
snapd                      2.60.4+git1361.g2725b1f  20390  latest/edge      canonical✓      snapd

valentindavid, 2023-09-28:
Try `snap connect fwupd:usb-raw`

binli, 2023-09-29:
> Try `snap connect fwupd:usb-raw`

@valentindavid oh, thanks, now with the edge snapd, it works fine now.

From fwupdmgr get-devices I could find the USB FPC device now.

├─Sensor Controller:
│     Device ID:          65a54fb6ce182f0e75edf0e43047d547a0d61f0e
│     Summary:            FPC fingerprint sensor
│     Current version:    27.26.23.5
│     Vendor:             FPC (USB:0x10A5)
│     Install Duration:   15 seconds
│     GUID:               bcb666d2-d13b-5295-b6ef-32a3f4ba8737 ← USB\VID_10A5&PID_9800
│     Device Flags:       • Updatable
│                         • Supported on remote server
│                         • Signed Payload

superm1, 2023-09-29:
Ok, I think we can close this after next fwupd release. 

We also need a request to store.

binli, 2023-09-29:
okok, please.

superm1, 2023-09-29:
> okok, please.

Btw I don't think you need edge snaps with this solution, just edge fwupd.

PR #6201 (2023-09-22): snap: Use raw-usb plug
This will need a request from the store.

Fixes #6141

Type of pull request:

- [ ] New plugin (Please include [new plugin checklist](https://github.com/fwupd/fwupd/wiki/New-plugin-checklist))
- [x] Code fix
- [ ] Feature
- [ ] Documentation
commit b6e426df (2023-09-22): snap: Use raw-usb plug

This will need a request from the store.

Fixes #6141