fwupd/fwupd #6144: How to clear host security events log

julianfairfax, 2023-09-04:
**Describe the question**
I would like to clear the (Host) Security Events log from fwupd (that also shows in GNOME Settings). I've tried deleting /var/log, rotating and vacuuming journald logs, clearing dmesg logs. None of those methods worked. I believe it is possible to do this in some way, as when I reinstalled my OS (switched from Arch to Debian), the previously collected log was now non-existent.

Here is a visual example of what I'm talking about:
```
$ fwupdmgr security --force
Host security events
  2023-09-03 13:52:49   Linux Kernel Lockdown        Pass (Not Enabled → Enabled)
  2023-09-03 13:52:49   UEFI Secure Boot             Pass (Not Enabled → Enabled)
  2023-09-03 11:52:39   Linux Kernel Lockdown      ! Fail (Enabled → Not Enabled)
  2023-09-03 11:52:39   UEFI Secure Boot           ! Fail (Enabled → Not Enabled)
```

[image]

For information on the contents of this report, see https://fwupd.github.io/hsi.html

**fwupd version information**
Please provide the version of the daemon and client if applicable to your current installation of `fwupd`.

```shell
$ fwupdmgr --version
compile   org.freedesktop.fwupd         1.9.4
compile   com.hughsie.libxmlb           0.3.10
compile   com.hughsie.libjcat           0.1.9
runtime   org.freedesktop.fwupd-efi     1.4
compile   org.freedesktop.gusb          0.4.5
runtime   com.dell.libsmbios            2.4
runtime   org.freedesktop.gusb          0.4.5
runtime   org.kernel                    6.4.0-3-amd64
runtime   org.freedesktop.fwupd         1.9.4
```

hughsie, 2023-09-04:
Does `sudo rm /var/lib/fwupd/pending.db` and then restarting fwupd help?

julianfairfax, 2023-09-04:
> Does `sudo rm /var/lib/fwupd/pending.db` and then restarting fwupd help?

That worked! Thank you!

Sidenote (and maybe this needs to be another issue): but, I lied about being able to use that terminal command to get the info. What it really returns is this:
```
Not enough data was provided by the platform to make an HSI calculation.
 » https://fwupd.github.io/hsi.html#not-enough-info
```

I didn't have that on Arch. What's more odd, is that GNOME Settings also says not enough information, but let's me copy the "technical report", which is just `fwupdmgr security --force`'s would-be output:
```
Device Security Report
======================

Report details
  Date generated:                                  2023-09-04 13:58:06
  fwupd version:                                   1.9.4

System details
  Hardware model:                                  TUXEDO TUXEDO Pulse 15 Gen1
  Processor:                                       AMD Ryzen 7 4800H with Radeon Graphics
  OS:                                              Debian GNU/Linux trixie/sid
  Security level:                                  HSI:INVALID:missing-data

HSI-1 Tests
  UEFI Platform Key:                               Pass (Valid)
  UEFI Bootservice Variables:                      Pass (Locked)
  TPM v2.0:                                        Pass (Found)
  UEFI Secure Boot:                                Pass (Enabled)
  TPM Platform Configuration:                      Pass (Valid)

HSI-2 Tests
  TPM Reconstruction:                              Pass (Valid)
  IOMMU Protection:                                Pass (Enabled)

HSI-3 Tests
  Pre-boot DMA Protection:                       ! Fail (Not Enabled)
  Suspend To RAM:                                ! Fail (Enabled)
  Suspend To Idle:                               ! Fail (Not Enabled)

HSI-4 Tests
  Encrypted RAM:                                 ! Fail (Not Supported)

Runtime Tests
  Firmware Updater Verification:                   Pass (Not Tainted)
  Linux Swap:                                    ! Fail (Not Encrypted)
  Linux Kernel Lockdown:                           Pass (Enabled)
  Linux Kernel Verification:                       Pass (Not Tainted)

Host security events

For information on the contents of this report, see https://fwupd.github.io/hsi.html
```

Maybe I'm missing a package or something, since this is a minimal Debian install of mine. But, this is odd to me.

hughsie, 2023-09-04:
I suspect this is because you don't have the AMD psp reporting stuff in your firmware. Is there a warning in dmesg about this?

julianfairfax, 2023-09-04:
> Try grepping with `ccp`

```
$ sudo dmesg | grep ccp
[   23.344793] ccp 0000:03:00.2: enabling device (0000 -> 0002)
[   23.345275] ccp 0000:03:00.2: ccp: unable to access the device: you might be running a broken BIOS.
[   23.355476] ccp 0000:03:00.2: tee enabled
[   23.355485] ccp 0000:03:00.2: psp enabled
```

julianfairfax, 2023-09-04:
> Is there a BIOS update available for your device (which may be direct from the vendor, rather than on the LVFS)?

It's possible. They are unfortunately not on LVFS. I will get back to you once I have the answer to that.

julianfairfax, 2023-09-04:
> Is there a BIOS update available for your device (which may be direct from the vendor, rather than on the LVFS)?

There was one, and I installed it. However, this did not change anything:
```
Device Security Report
======================

Report details
  Date generated:                                  2023-09-04 15:44:09
  fwupd version:                                   1.9.4

System details
  Hardware model:                                  TUXEDO TUXEDO Pulse 15 Gen1
  Processor:                                       AMD Ryzen 7 4800H with Radeon Graphics
  OS:                                              Debian GNU/Linux trixie/sid
  Security level:                                  HSI:INVALID:missing-data

HSI-1 Tests
  UEFI Platform Key:                               Pass (Valid)
  UEFI Bootservice Variables:                      Pass (Locked)
  TPM v2.0:                                        Pass (Found)
  UEFI Secure Boot:                                Pass (Enabled)
  TPM Platform Configuration:                      Pass (Valid)

HSI-2 Tests
  TPM Reconstruction:                              Pass (Valid)
  IOMMU Protection:                                Pass (Enabled)

HSI-3 Tests
  Pre-boot DMA Protection:                       ! Fail (Not Enabled)
  Suspend To RAM:                                ! Fail (Enabled)
  Suspend To Idle:                               ! Fail (Not Enabled)

HSI-4 Tests
  Encrypted RAM:                                 ! Fail (Not Supported)

Runtime Tests
  Firmware Updater Verification:                   Pass (Not Tainted)
  Linux Swap:                                    ! Fail (Not Encrypted)
  Linux Kernel Lockdown:                           Pass (Enabled)
  Linux Kernel Verification:                       Pass (Not Tainted)

Host security events

For information on the contents of this report, see https://fwupd.github.io/hsi.html
```

```
sudo dmesg | grep psp
[sudo] password for user: 
[    4.365443] [drm] add ip block number 3 <psp>
[    5.053372] [drm] psp gfx command LOAD_TA(0x1) failed and response status is (0x7)
[    5.053497] [drm] psp gfx command INVOKE_CMD(0x3) failed and response status is (0x4)
[   14.360717] ccp 0000:03:00.2: psp enabled
```

```
$ sudo dmesg | grep ccp
[   14.350399] ccp 0000:03:00.2: enabling device (0000 -> 0002)
[   14.350547] ccp 0000:03:00.2: ccp: unable to access the device: you might be running a broken BIOS.
[   14.360705] ccp 0000:03:00.2: tee enabled
[   14.360717] ccp 0000:03:00.2: psp enabled
```

To recap, I didn't have this issue on Arch, and I have a minimal Debian install, but fwupd and its dependencies are present, so it's odd.

superm1, 2023-09-04:
If the kernel module doesn't export the data there is nothing fwupd can do.

The kernel passes the information from the platform.

closed 2023-09-04 by superm1