fwupd/fwupd #6148: How to install the self created CA certificate to the system-wide system store?

chinyakao, 2023-09-06:
I followed the doc [1], created a self-signed cab file, and pasted the `ACME-CA.pem` to the directory `/etc/pki/fwupd/`.
However, when I tried to install the self-signed firmware, it still showed the signature not trusted.
I also restarted the fwupd service and did `sudo update-ca-certificates` to update the CA, but it still failed.
How could I get my system to recognize the self-created CA and install self-signed firmware?

[1] https://github.com/fwupd/fwupd/blob/main/docs/only-trusted.md

```
u@u-system:~/work/selfsigned_firmware$ fwupdmgr install selfsigned__firmware.cab
Decompressing…          [***************************************]
firmware signature missing or not trusted; set OnlyTrusted=false in /etc/fwupd/daemon.conf ONLY if you are a firmware developer
```

**fwupd version information**
```
u@u-system:~/work$ fwupdmgr --version
runtime   org.freedesktop.fwupd         1.7.9
runtime   com.dell.libsmbios            2.4
compile   org.freedesktop.gusb          0.3.10
runtime   org.kernel                    6.1.0-1020-oem
compile   com.hughsie.libjcat           0.1.9
compile   org.freedesktop.fwupd         1.7.9
runtime   org.freedesktop.gusb          0.3.10
```

hughsie, 2023-09-06:
What does `sudo fwupdtool get-details selfsigned__firmware.cab -v` say?

> org.freedesktop.fwupd         1.7.9

This is very old -- does a newer version work as you expect?

chinyakao, 2023-09-06:
Hi Hughsie,

Thank you for the reply!
The command `sudo fwupdtool get-details selfsigned__firmware.cab -v` shows as following.

```
05:52:14:0697 FuCabinet            failed to verify firmware.metainfo.xml: checksums were required, but none supplied
05:52:14:0697 FuCabinet            processing file: firmware.metainfo.xml
05:52:14:0697 FuCabinet            processing release: 0x10600
05:52:14:0697 FuCabinet            failed to verify payload firmware.bin: checksums were required, but none supplied
```

The `.jact` file didn't include the SHA-256 checksum. After running the command `jcat-tool self-sign firmware.jcat firmware.bin --kind sha256` and re-cab the file, it works!! Thanks you!

BTW, the version `org.freedesktop.fwupd 1.7.9` also works. :)

closed 2023-09-06 by chinyakao

hughsie, 2023-09-06:
How did you create the jcat file? maybe we need to modify the readme to make it clearer.

chinyakao, 2023-09-06:
I originally followed this readme[1] 
```
$ ../contrib/build-certs.py
$ jcat-tool sign firmware.jcat firmware.bin rileyk_signed.pem rileyk.key
$ jcat-tool verify firmware.jcat --public-key ACME-CA.pem
```
It all worked as expect. When I tried to `fwupdmgr install firmware.cab`, then started to face the problem.
I didn't realize it is required to add a SHA256 checksum in `.jcat`, even I read the another readme[2]
Maybe somewhere could have a note that the SHA256 checksum is required in `.jcat` to install trusted firmware.

[1] https://github.com/hughsie/libjcat#public-key-signing
[2] https://github.com/fwupd/fwupd/blob/main/docs/only-trusted.md

hughsie, 2023-09-06:
@chinyakao would https://github.com/fwupd/fwupd/pull/6150 help?

chinyakao, 2023-09-06:
@hughsie Yes, it's more straightforward. Knowing the debugging method and taking the common mistake as an example is great. Thank you