fwupd/fwupd #6153: systemd service `fwupd-refresh.service` fails to start

cwrau, 2023-09-07:
```
2023-09-07T13:45:17.646720+0200 steve systemd[1]: Starting Refresh fwupd metadata and update motd...
2023-09-07T13:45:17.674549+0200 steve fwupdmgr[3841321]: (/usr/bin/fwupdmgr:3841321): GLib-DEBUG: 13:45:17.674: setenv()/putenv() are not thread-safe and should not be used after threads are created
2023-09-07T13:45:17.681468+0200 steve fwupdmgr[3841321]: (/usr/bin/fwupdmgr:3841321): GLib-GIO-DEBUG: 13:45:17.681: _g_io_module_get_default: Found default implementation dconf (DConfSettingsBackend) for ‘gsettings-backend’
2023-09-07T13:45:17.681468+0200 steve fwupdmgr[3841321]: (/usr/bin/fwupdmgr:3841321): dconf-DEBUG: 13:45:17.681: watch_fast: "/system/proxy/" (establishing: 0, active: 0)
2023-09-07T13:45:17.681991+0200 steve fwupdmgr[3841321]: (/usr/bin/fwupdmgr:3841321): dconf-DEBUG: 13:45:17.681: watch_established: "/system/proxy/" (establishing: 1)
2023-09-07T13:45:17.682099+0200 steve fwupdmgr[3841321]: (/usr/bin/fwupdmgr:3841321): dconf-DEBUG: 13:45:17.682: watch_fast: "/system/proxy/http/" (establishing: 0, active: 0)
2023-09-07T13:45:17.682131+0200 steve fwupdmgr[3841321]: (/usr/bin/fwupdmgr:3841321): dconf-DEBUG: 13:45:17.682: watch_fast: "/system/proxy/https/" (establishing: 0, active: 0)
2023-09-07T13:45:17.682149+0200 steve fwupdmgr[3841321]: (/usr/bin/fwupdmgr:3841321): dconf-DEBUG: 13:45:17.682: watch_established: "/system/proxy/http/" (establishing: 1)
2023-09-07T13:45:17.682165+0200 steve fwupdmgr[3841321]: (/usr/bin/fwupdmgr:3841321): dconf-DEBUG: 13:45:17.682: watch_fast: "/system/proxy/ftp/" (establishing: 0, active: 0)
2023-09-07T13:45:17.682180+0200 steve fwupdmgr[3841321]: (/usr/bin/fwupdmgr:3841321): dconf-DEBUG: 13:45:17.682: watch_established: "/system/proxy/https/" (establishing: 1)
2023-09-07T13:45:17.682210+0200 steve fwupdmgr[3841321]: (/usr/bin/fwupdmgr:3841321): dconf-DEBUG: 13:45:17.682: watch_fast: "/system/proxy/socks/" (establishing: 0, active: 0)
2023-09-07T13:45:17.682229+0200 steve fwupdmgr[3841321]: (/usr/bin/fwupdmgr:3841321): dconf-DEBUG: 13:45:17.682: watch_established: "/system/proxy/ftp/" (establishing: 1)
2023-09-07T13:45:17.682244+0200 steve fwupdmgr[3841321]: (/usr/bin/fwupdmgr:3841321): dconf-DEBUG: 13:45:17.682: unwatch_fast: "/system/proxy/" (active: 1, establishing: 0)
2023-09-07T13:45:17.682266+0200 steve fwupdmgr[3841321]: (/usr/bin/fwupdmgr:3841321): dconf-DEBUG: 13:45:17.682: unwatch_fast: "/system/proxy/http/" (active: 1, establishing: 0)
2023-09-07T13:45:17.682266+0200 steve fwupdmgr[3841321]: (/usr/bin/fwupdmgr:3841321): dconf-DEBUG: 13:45:17.682: unwatch_fast: "/system/proxy/https/" (active: 1, establishing: 0)
2023-09-07T13:45:17.682266+0200 steve fwupdmgr[3841321]: (/usr/bin/fwupdmgr:3841321): dconf-DEBUG: 13:45:17.682: unwatch_fast: "/system/proxy/ftp/" (active: 1, establishing: 0)
2023-09-07T13:45:17.682311+0200 steve fwupdmgr[3841321]: (/usr/bin/fwupdmgr:3841321): dconf-DEBUG: 13:45:17.682: watch_established: "/system/proxy/socks/" (establishing: 1)
2023-09-07T13:45:17.682311+0200 steve fwupdmgr[3841321]: (/usr/bin/fwupdmgr:3841321): dconf-DEBUG: 13:45:17.682: unwatch_fast: "/system/proxy/socks/" (active: 1, establishing: 0)
2023-09-07T13:45:17.683916+0200 steve fwupdmgr[3841321]: (/usr/bin/fwupdmgr:3841321): GLib-GIO-DEBUG: 13:45:17.683: _g_io_module_get_default: Found default implementation local (GLocalVfs) for ‘gio-vfs’
2023-09-07T13:45:17.683939+0200 steve fwupdmgr[3841321]: (/usr/bin/fwupdmgr:3841321): pxbackend-DEBUG: 13:45:17.683: px_config_sysconfig_set_config_file: Could not read file /etc/sysconfig/proxy
2023-09-07T13:45:17.683957+0200 steve fwupdmgr[3841321]: (/usr/bin/fwupdmgr:3841321): pxbackend-DEBUG: 13:45:17.683: Active config plugins:
2023-09-07T13:45:17.683957+0200 steve fwupdmgr[3841321]: (/usr/bin/fwupdmgr:3841321): pxbackend-DEBUG: 13:45:17.683:  - config-env
2023-09-07T13:45:17.683957+0200 steve fwupdmgr[3841321]: (/usr/bin/fwupdmgr:3841321): pxbackend-DEBUG: 13:45:17.683:  - config-kde
2023-09-07T13:45:17.683957+0200 steve fwupdmgr[3841321]: (/usr/bin/fwupdmgr:3841321): pxbackend-DEBUG: 13:45:17.683:  - config-gnome
2023-09-07T13:45:17.684014+0200 steve fwupdmgr[3841321]: (/usr/bin/fwupdmgr:3841321): pxbackend-DEBUG: 13:45:17.683:  - config-sysconfig
2023-09-07T13:45:17.684781+0200 steve fwupdmgr[3841321]: (/usr/bin/fwupdmgr:3841321): GLib-GIO-DEBUG: 13:45:17.684: Failed to initialize portal (GNetworkMonitorPortal) for gio-network-monitor: Not using portals
2023-09-07T13:45:17.685074+0200 steve fwupdmgr[3841321]: (/usr/bin/fwupdmgr:3841321): GLib-GIO-DEBUG: 13:45:17.685: Using cross-namespace EXTERNAL authentication (this will deadlock if server is GDBus < 2.73.3)
2023-09-07T13:45:17.685387+0200 steve fwupdmgr[3841321]: (/usr/bin/fwupdmgr:3841321): GLib-GIO-DEBUG: 13:45:17.685: Failed to initialize networkmanager (GNetworkMonitorNM) for gio-network-monitor: The connection is closed
2023-09-07T13:45:17.685442+0200 steve fwupdmgr[3841321]: (/usr/bin/fwupdmgr:3841321): GLib-GIO-DEBUG: 13:45:17.685: _g_io_module_get_default: Found default implementation netlink (GNetworkMonitorNetlink) for ‘gio-network-monitor’
2023-09-07T13:45:17.685442+0200 steve fwupdmgr[3841321]: (/usr/bin/fwupdmgr:3841321): pxbackend-DEBUG: 13:45:17.685: px_manager_on_network_changed: Network connection changed, clearing pac data
2023-09-07T13:45:17.685442+0200 steve fwupdmgr[3841321]: (/usr/bin/fwupdmgr:3841321): pxbackend-DEBUG: 13:45:17.685: px_manager_constructed: Up and running
2023-09-07T13:45:17.685523+0200 steve fwupdmgr[3841321]: (/usr/bin/fwupdmgr:3841321): GLib-GIO-DEBUG: 13:45:17.685: _g_io_module_get_default: Found default implementation libproxy (GLibproxyResolver) for ‘gio-proxy-resolver’
2023-09-07T13:45:17.688315+0200 steve systemd[1]: fwupd-refresh.service: Main process exited, code=exited, status=1/FAILURE
2023-09-07T13:45:17.688449+0200 steve systemd[1]: fwupd-refresh.service: Failed with result 'exit-code'.
2023-09-07T13:45:17.706454+0200 steve systemd[1]: Failed to start Refresh fwupd metadata and update motd.
```

_Originally posted by @cwrau in https://github.com/fwupd/fwupd/issues/6032#issuecomment-1710009629_

hughsie, 2023-09-07:
>2023-09-07T13:45:17.683939+0200 steve fwupdmgr[3841321]: (/usr/bin/fwupdmgr:3841321): pxbackend-DEBUG: >13:45:17.683: px_config_sysconfig_set_config_file: Could not read file /etc/sysconfig/proxy

What's px_config_sysconfig_set_config_file() doing? Do you have something to manage a corporate proxy installed perhaps?

cwrau, 2023-09-08:
> What's px_config_sysconfig_set_config_file() doing? Do you have something to manage a corporate proxy installed perhaps?

I don't think so? At least I've never done something like that.

That file, and it's parent directory, don't exist, I thought that could be it, but creating them didn't help, just this error message vanished

hughsie, 2023-09-08:
What distro please? How did you install fwupd?

cwrau, 2023-09-08:
> What distro please? How did you install fwupd?

Arch Linux, I installed it via the package manager. I have version 1.9.5 installed.

hughsie, 2023-09-08:
and to confirm, `fwupdmgr refresh` works okay, just not in the fwupd-refresh.service?

cwrau, 2023-09-08:
> and to confirm, `fwupdmgr refresh` works okay, just not in the fwupd-refresh.service?

Yessss

hughsie, 2023-09-08:
@superm1 any ideas on how to debug this? I'm no systemd expert.

iyanmv, 2023-09-09:
I think there are two issues here:
1. The first time that `fwupd-refresh.service` is triggered by `fwupd-refresh.timer` will fail because it takes too long for `fwupd.service` to start, so `fwupdmgr refresh` will timeout. This I can reproduce by running `fwupdmgr refresh` without the service running. And I think this is something I didn't observe before version 1.9.5.
```bash
$ sudo systemctl stop fwupd.service
$ fwupdmgr refresh
Failed to connect to daemon: Error calling StartServiceByName for org.freedesktop.fwupd: Failed to activate service 'org.freedesktop.fwupd': timed out (service_start_timeout=25000ms)
$ echo $?
1
```
2. The second issue maybe is a systemd bug. Arch recently updated systemd as well as fwupd. I see that `SuccessExitStatus=2` is added to the service file, which is the status `fwupdmgr refresh` returns without `--force` when metadata is up to date, but maybe it's not working.

iyanmv, 2023-09-09:
> 2\. The second issue maybe is a systemd bug. Arch recently updated systemd as well as fwupd. I see that `SuccessExitStatus=2` is added to the service file, which is the status `fwupdmgr refresh` returns without `--force` when metadata is up to date, but maybe it's not working.

After checking a bit, I think this is not the case. Even after modifying the service file to pass `--force` to the refresh command, it still fails. However removing the line `DynamicUser=yes` solves the issue for me. So I still think it may be a systemd bug.

hughsie, 2023-09-09:
With DynamicUser=true, if you add `--verbose` to the fwupdmgr call in fwupd-refresh.service -- do you get any more debug output?

iyanmv, 2023-09-09:
This is the output of `journalctl -xeu fwupd-refresh.service` after running the modified service with `--verbose`:

```
Sep 09 22:24:05 ilum systemd[1]: Starting Refresh fwupd metadata and update motd...
░░ Subject: A start job for unit fwupd-refresh.service has begun execution
░░ Defined-By: systemd
░░ Support: https://lists.freedesktop.org/mailman/listinfo/systemd-devel
░░ 
░░ A start job for unit fwupd-refresh.service has begun execution.
░░ 
░░ The job identifier is 1396.
Sep 09 22:24:05 ilum fwupdmgr[81295]: (/usr/bin/fwupdmgr:81295): GLib-DEBUG: 22:24:05.401: setenv()/putenv() are not thread-safe and should not be used after threads are created
Sep 09 22:24:05 ilum fwupdmgr[81295]: (/usr/bin/fwupdmgr:81295): GLib-GIO-DEBUG: 22:24:05.402: _g_io_module_get_default: Found default implementation dconf (DConfSettingsBackend) for ‘gsettings-backend’
Sep 09 22:24:05 ilum fwupdmgr[81295]: (/usr/bin/fwupdmgr:81295): dconf-DEBUG: 22:24:05.402: watch_fast: "/system/proxy/" (establishing: 0, active: 0)
Sep 09 22:24:05 ilum fwupdmgr[81295]: (/usr/bin/fwupdmgr:81295): dconf-DEBUG: 22:24:05.402: watch_fast: "/system/proxy/http/" (establishing: 0, active: 0)
Sep 09 22:24:05 ilum fwupdmgr[81295]: (/usr/bin/fwupdmgr:81295): dconf-DEBUG: 22:24:05.402: watch_fast: "/system/proxy/https/" (establishing: 0, active: 0)
Sep 09 22:24:05 ilum fwupdmgr[81295]: (/usr/bin/fwupdmgr:81295): dconf-DEBUG: 22:24:05.402: watch_fast: "/system/proxy/ftp/" (establishing: 0, active: 0)
Sep 09 22:24:05 ilum fwupdmgr[81295]: (/usr/bin/fwupdmgr:81295): dconf-DEBUG: 22:24:05.402: watch_fast: "/system/proxy/socks/" (establishing: 0, active: 0)
Sep 09 22:24:05 ilum fwupdmgr[81295]: (/usr/bin/fwupdmgr:81295): dconf-DEBUG: 22:24:05.402: unwatch_fast: "/system/proxy/" (active: 0, establishing: 1)
Sep 09 22:24:05 ilum fwupdmgr[81295]: (/usr/bin/fwupdmgr:81295): dconf-DEBUG: 22:24:05.402: unwatch_fast: "/system/proxy/http/" (active: 0, establishing: 1)
Sep 09 22:24:05 ilum fwupdmgr[81295]: (/usr/bin/fwupdmgr:81295): dconf-DEBUG: 22:24:05.402: unwatch_fast: "/system/proxy/https/" (active: 0, establishing: 1)
Sep 09 22:24:05 ilum fwupdmgr[81295]: (/usr/bin/fwupdmgr:81295): dconf-DEBUG: 22:24:05.402: unwatch_fast: "/system/proxy/ftp/" (active: 0, establishing: 1)
Sep 09 22:24:05 ilum fwupdmgr[81295]: (/usr/bin/fwupdmgr:81295): dconf-DEBUG: 22:24:05.402: unwatch_fast: "/system/proxy/socks/" (active: 0, establishing: 1)
Sep 09 22:24:05 ilum fwupdmgr[81295]: (/usr/bin/fwupdmgr:81295): dconf-DEBUG: 22:24:05.402: watch_established: "/system/proxy/" (establishing: 0)
Sep 09 22:24:05 ilum fwupdmgr[81295]: (/usr/bin/fwupdmgr:81295): dconf-DEBUG: 22:24:05.402: watch_established: "/system/proxy/http/" (establishing: 0)
Sep 09 22:24:05 ilum fwupdmgr[81295]: (/usr/bin/fwupdmgr:81295): dconf-DEBUG: 22:24:05.402: watch_established: "/system/proxy/https/" (establishing: 0)
Sep 09 22:24:05 ilum fwupdmgr[81295]: (/usr/bin/fwupdmgr:81295): dconf-DEBUG: 22:24:05.402: watch_established: "/system/proxy/ftp/" (establishing: 0)
Sep 09 22:24:05 ilum fwupdmgr[81295]: (/usr/bin/fwupdmgr:81295): dconf-DEBUG: 22:24:05.402: watch_established: "/system/proxy/socks/" (establishing: 0)
Sep 09 22:24:05 ilum fwupdmgr[81295]: (/usr/bin/fwupdmgr:81295): GLib-GIO-DEBUG: 22:24:05.403: _g_io_module_get_default: Found default implementation local (GLocalVfs) for ‘gio-vfs’
Sep 09 22:24:05 ilum fwupdmgr[81295]: (/usr/bin/fwupdmgr:81295): pxbackend-DEBUG: 22:24:05.403: px_config_sysconfig_set_config_file: Could not read file /etc/sysconfig/proxy
Sep 09 22:24:05 ilum fwupdmgr[81295]: (/usr/bin/fwupdmgr:81295): pxbackend-DEBUG: 22:24:05.403: Active config plugins:
Sep 09 22:24:05 ilum fwupdmgr[81295]: (/usr/bin/fwupdmgr:81295): pxbackend-DEBUG: 22:24:05.403:  - config-env
Sep 09 22:24:05 ilum fwupdmgr[81295]: (/usr/bin/fwupdmgr:81295): pxbackend-DEBUG: 22:24:05.403:  - config-kde
Sep 09 22:24:05 ilum fwupdmgr[81295]: (/usr/bin/fwupdmgr:81295): pxbackend-DEBUG: 22:24:05.403:  - config-gnome
Sep 09 22:24:05 ilum fwupdmgr[81295]: (/usr/bin/fwupdmgr:81295): pxbackend-DEBUG: 22:24:05.403:  - config-sysconfig
Sep 09 22:24:05 ilum fwupdmgr[81295]: (/usr/bin/fwupdmgr:81295): GLib-GIO-DEBUG: 22:24:05.403: Failed to initialize portal (GNetworkMonitorPortal) for gio-network-monitor: Not using portals
Sep 09 22:24:05 ilum fwupdmgr[81295]: (/usr/bin/fwupdmgr:81295): GLib-GIO-DEBUG: 22:24:05.403: Using cross-namespace EXTERNAL authentication (this will deadlock if server is GDBus < 2.73.3)
Sep 09 22:24:05 ilum fwupdmgr[81295]: (/usr/bin/fwupdmgr:81295): GLib-GIO-DEBUG: 22:24:05.404: Failed to initialize networkmanager (GNetworkMonitorNM) for gio-network-monitor: The connection is closed
Sep 09 22:24:05 ilum fwupdmgr[81295]: (/usr/bin/fwupdmgr:81295): GLib-GIO-DEBUG: 22:24:05.404: _g_io_module_get_default: Found default implementation netlink (GNetworkMonitorNetlink) for ‘gio-network-monitor’
Sep 09 22:24:05 ilum fwupdmgr[81295]: (/usr/bin/fwupdmgr:81295): pxbackend-DEBUG: 22:24:05.404: px_manager_on_network_changed: Network connection changed, clearing pac data
Sep 09 22:24:05 ilum fwupdmgr[81295]: (/usr/bin/fwupdmgr:81295): pxbackend-DEBUG: 22:24:05.404: px_manager_constructed: Up and running
Sep 09 22:24:05 ilum fwupdmgr[81295]: (/usr/bin/fwupdmgr:81295): GLib-GIO-DEBUG: 22:24:05.404: _g_io_module_get_default: Found default implementation libproxy (GLibproxyResolver) for ‘gio-proxy-resolver’
Sep 09 22:24:05 ilum systemd[1]: fwupd-refresh.service: Main process exited, code=exited, status=1/FAILURE
[... cut ...]
░░ Support: https://lists.freedesktop.org/mailman/listinfo/systemd-devel
░░ 
░░ The unit fwupd-refresh.service has entered the 'failed' state with result 'exit-code'.
Sep 09 22:24:05 ilum systemd[1]: Failed to start Refresh fwupd metadata and update motd.
░░ Subject: A start job for unit fwupd-refresh.service has failed
░░ Defined-By: systemd
░░ Support: https://lists.freedesktop.org/mailman/listinfo/systemd-devel
░░ 
░░ A start job for unit fwupd-refresh.service has finished with a failure.
░░ 
░░ The job identifier is 1396 and the job result is failed.
```

cwrau, 2023-09-10:
> With DynamicUser=true, if you add `--verbose` to the fwupdmgr call in fwupd-refresh.service -- do you get any more debug output?

My log output is already with `-v`, and `DynamicUser=yes` is also set.

superm1, 2023-09-11:
This is solved at least in Debian and Ubuntu packaging by creating a dedicated user for the service.

Perhaps you can do that too?

iyanmv, 2023-09-11:
> > Perhaps you can do that too?
> 
> I've [opened a bug](https://bugs.archlinux.org/task/79653) in Arch Linux's Flyspray. Hopefully, they can fix it ;)

I think the permission issues were solved in the last version: [extra/fwupd 1.9.5-2](https://gitlab.archlinux.org/archlinux/packaging/packages/fwupd/-/commit/e64ca501f89a629380d80743853e931d8e1006b5).

hughsie, 2023-09-11:
@superm1 do you think we should ship sysusers.d/fwupd.conf upstream?

superm1, 2023-09-11:
> @superm1 do you think we should ship sysusers.d/fwupd.conf upstream?

I had no idea such a concept existed. Given the systemd bug still lingers yes totally!

michaellass, 2023-09-22:
> What about `fwupd.service` taking a long time to start? Can you replicate this issue on your end? In previous versions it was quite fast, now it takes more than 20 seconds.

I think this was not addressed in the later comments: This is probably caused by https://github.com/hughsie/passim/issues/15.