fwupd/fwupd #6156: What's the Windows/.msi build?

sorenkrabbe, 2023-09-08:
I notice the FWUPD releases on Github come with .msi assets. E.g. "fwupd-1.9.5-setup-x86_64.msi" for the [1.9.5](https://github.com/fwupd/fwupd/releases/tag/1.9.5) release.

When I try running it on a Windows 11 machine I am able to get some basics working, but also face issues with both missing dll dependencies and instability of the plugins I tried.

I was not able to find info on Windows support in docs, so, I'm curious what the intention - and maybe status - of this build for Windows is?

Thanks.

hughsie, 2023-09-08:
> face issues with both missing dll dependencies

Ohh? What did you see? I only test in a Win10 VM.

> and instability of the plugins I tried

Got any more details? I'm testing with the ColorHug device.

> I'm curious what the intention - and maybe status - of this build for Windows is?

I guess it's a technical preview that's only really used by a couple of companies. I'd certainly like it to work better.

sorenkrabbe, 2023-09-08:
> Ohh? What did you see? I only test in a Win10 VM.

I did get it to run, but it required a fair effort: When running `fwupdtool` from the bin\ folder after install, it's unable to find the dlls installed in the fwupd-"version" sub-folder. If I naively move the files into the bin\ folder, it's resolved. Maybe there's a better way.
Further libzstd.dll and libpcre2-8-0.dll was missing on a local Windows 11 machine. Extracting them from https://packages.msys2.org/ packages and putting them in the fwupd\bin\ folder made it work

> and instability of the plugins I tried

Never mind this. Turned out to be mostly due to an unstable USB hub in my setup. Getting that fixed made it behave much more reliably.

> I guess it's a technical preview that's only really used by a couple of companies. I'd certainly like it to work better.

This is the type of indication I was curious about. I read that as a request to have it reported if we run into issues on Windows?

Thanks and cool to see the push to have fwupd run on Win and Mac too.

hughsie, 2023-09-08:
@sorenkrabbe do the .msi builds created by https://github.com/fwupd/fwupd/pull/6165 work without the extra msys2 binaries?

hughsie, 2023-09-09:
@sorenkrabbe if it helps, the msi you want to test is https://output.circle-artifacts.com/output/job/43de5a74-2cd5-4696-9202-c7075fa846c9/artifacts/0/dist/setup/fwupd-1.9.6-setup-x86_64.msi

sorenkrabbe, 2023-09-11:
> @sorenkrabbe do the .msi builds created by https://github.com/fwupd/fwupd/pull/6165 work without the extra msys2 binaries?

Yes, thanks. It resolves the issue with missing DLLs after I've copied the fwupd-1.9.6\ files into the bin\ folder.

closed 2023-09-11 by sorenkrabbe