fwupd/fwupd #6179: Make passim optional

ckotte, 2023-09-13:
**Is your feature request related to a problem? Please describe.**
The passim.service gets started every time I use fwupdmgr. I can disable or even mask the service, I still have systemd errors.  

(At least) on Arch Linux, passim.service requires avahi-daemon.service to successfully start. To not have any errors, I need to install avahi and enable and start both services.

However, I only use fwupd on my laptop.  I don't need caching and I prefer to not have services running I don't really need.

**Describe the solution you'd like**
Make passim optional. As far as I understand there should be a build option. I don't know what concerns the Arch Linux packagers have: https://bugs.archlinux.org/task/79614. They don't want to create two versions.

So, maybe you can introduce a config option to disable passim. Then users can disable and stop the service without any issues.

hughsie, 2023-09-13:
> However, I only use fwupd on my laptop. I don't need caching and I prefer to not have services running I don't really need.

In fedora `passim-libs` is installed by default, but the `passim` package (the daemon) is only installed when wanted. Can arch split up the passim package like that too?

> mask the service, I still have systemd errors

Masking the passim service should not give systemd errors -- what do you see?

ckotte, 2023-09-13:
I masked the service, but it didn't work. I will try again and let you know.

ckotte, 2023-09-13:
Sorry. I had those issues directly after the update of fwupd a short while ago but didn't have time to observer more. Looks like now after a few reboots / updates, passim service doesn't get started automatically when using fwupdmgr. Will close the issue for now.

closed 2023-09-13 by ckotte

marek22k, 2024-05-21:
I would also appreciate a solution to the problem. I have fwupd on a laptop and think that many background services are not necessarily energy-saving.

superm1, 2024-05-21:
> I would also appreciate a solution to the problem. I have fwupd on a laptop and think that many background services are not necessarily energy-saving.

You can disable the passim service or use `P2pPolicy=none` as described by https://fwupd.github.io/libfwupdplugin/fwupd.conf.html