fwupd/fwupd #6222: fwupdmgr refresh times out

alexis-aquanty, 2023-09-27:
**Describe the bug**
`fwupd-refresh.service` keeps failing because it times out waiting for `fwupd.service` to start. This is the error from `fwupdmgr refresh`:

```
Failed to connect to daemon: Error calling StartServiceByName for org.freedesktop.fwupd: Failed to activate service 'org.freedesktop.fwupd': timed out (service_start_timeout=25000ms)
```

The verbose output from `fwupd` itself is too long to post here, so I'll post it in a comment instead.

**Steps to Reproduce**
1. Ensure `fwupd` is not running
2. Run `fwupdmgr refresh`

**Expected behavior**
`fwupdmgr refresh` should not time out waiting for `fwupd` to start.

**fwupd version information**
Please provide the version of the daemon and client.

```shell
$ fwupdmgr --version
compile   org.freedesktop.fwupd         1.8.12
compile   com.hughsie.libxmlb           0.3.10
compile   com.hughsie.libjcat           0.1.9
runtime   org.freedesktop.fwupd-efi     1.4
compile   org.freedesktop.gusb          0.3.10
runtime   com.dell.libsmbios            2.4
runtime   org.freedesktop.gusb          0.3.10
runtime   org.freedesktop.fwupd         1.8.12
runtime   org.kernel                    6.1.0-12-amd64
```

Please note how you installed it (`apt`, `dnf`, `pacman`, source, etc): `apt`

<details>

<summary>**fwupd device information**</summary>

Please provide the output of the fwupd devices recognized in your system.

```shell
$ fwupdmgr get-devices --show-all-devices
GIGABYTE R183-Z92-AAD2-000
│
├─7450 PRO NVMe SSD:
│     Device ID:          ef61bfab28730bc9e4f9bda4854d472e0cd08eb4
│     Current version:    01
│     Vendor:             Micron Technology Inc (PCI:0x1344, PCI:0x1022)
│     GUIDs:              e6caad9e-fbdd-580d-b960-2e26e816b0f3 ← PCI\VEN_1344&DEV_51C3
│                         c6aca21f-3074-5bd6-a699-067eb14cee54 ← PCI\VEN_1344&DEV_51C3&REV_01
│                         897ccc7a-d566-5be8-80e4-2b1353688d02 ← PCI\VEN_1344&DEV_51C3&SUBSYS_13442100
│                         c0eba68d-c9ca-567e-b98c-b8557b06e58a ← PCI\VEN_1344&DEV_51C3&SUBSYS_13442100&REV_01
│                         db2d6da2-967a-5f20-9484-1cd4f02ab67b ← PCI\VEN_1022&DEV_14AA
│                         f57b028c-4c83-5d21-a3e7-f3e063f85a1c ← PCI\VEN_1022&DEV_14AA&REV_01
│                         def1b36b-3e13-558c-a439-44dbdd67bc13 ← PCI\VEN_1022&DEV_14AA&SUBSYS_10221234
│                         f06bb296-4710-5baa-a29d-5e544722e1bb ← PCI\VEN_1022&DEV_14AA&SUBSYS_10221234&REV_01
│     Device Flags:       • Internal device
│                         • Cryptographic hash verification is available
│
├─AMD EPYC 9654 96-Core Processor:
│ │   Device ID:          4bde70ba4e39b28f9eab1628f9dd6e6244c03027
│ │   Current version:    0x0a101111
│ │   Vendor:             Advanced Micro Devices, Inc.
│ │   GUIDs:              b9a2dd81-159e-5537-a7db-e7101d164d3f ← cpu
│ │                       22f9ecf4-588d-5c0a-8326-6ebff3655c6d ← CPUID\PRO_0&FAM_19
│ │                       f9203adb-b862-5263-88e9-373204438174 ← CPUID\PRO_0&FAM_19&MOD_11
│ │                       3e9793d5-b325-517a-81ec-6ebc6f5a6c02 ← CPUID\PRO_0&FAM_19&MOD_11&STP_1
│ │   Device Flags:       • Internal device
│ │
│ ├─GPIO controller:
│ │     Device ID:        f685512aa07369c9e77742acef941d779d31e766
│ │     GUID:             37b440a9-2473-5087-a39b-db84f32a8ed8 ← GPIO\ID_AMDI0030:00
│ │
│ └─Secure Processor:
│       Device ID:        c54ab0237d7a8db8c717b68e0be78e4374a2a079
│       Vendor:           Advanced Micro Devices, Inc. (PCI:0x1022)
│       GUIDs:            48501ece-4ef3-5888-9a08-f8b04e929cb7 ← PCI\VEN_1022&DEV_14CA
│                         9ef19700-a3ae-5770-aba1-b557da7c96c7 ← PCI\VEN_1022&DEV_14CA&REV_00
│                         82959eaf-114c-52ac-9691-5fcc77b4164a ← PCI\VEN_1022&DEV_14CA&SUBSYS_14581000
│                         eb7e5b6b-aeb2-5356-b7e6-32f9e959f281 ← PCI\VEN_1022&DEV_14CA&SUBSYS_14581000&REV_00
│       Device Flags:     • Internal device
│
├─INTEL SSDPF2KX076T1:
│     Device ID:          83eba1fa14bdefee1f6b4bcf06069b298e1c338f
│     Summary:            NVM Express solid state drive
│     Current version:    9CV10410
│     Vendor:             Intel Corporation (NVME:0x8086)
│     Serial Number:      PHAX2512008T7P6DGN
│     GUIDs:              4a0dba5e-7801-5954-b7ca-e23bc44cb7fa ← NVME\VEN_8086&DEV_0B60
│                         400b0094-fe57-5a67-b6d8-dee4a7c44c50 ← NVME\VEN_8086&DEV_0B60&REV_00
│                         a0fdd4fe-46f0-5e60-9af4-63430c993ed2 ← NVME\VEN_8086&DEV_0B60&SUBSYS_80869008
│                         73d72062-84c0-508b-9898-5949abc153d1 ← NVME\VEN_8086&DEV_0B60&SUBSYS_80869008&REV_00
│                         18abb88c-a545-586c-b41c-9de2ebefb6b9 ← INTEL SSDPF2KX076T1
│     Device Flags:       • Internal device
│                         • Updatable
│                         • System requires external power source
│                         • Needs a reboot after installation
│                         • Device is usable for the duration of the update
│
├─Micron 7450 MTFDKBA960TFR:
│     Device ID:          d42679745d228f25980bc14b6eaa5625b63367c6
│     Summary:            NVM Express solid state drive
│     Current version:    E2MU200
│     Vendor:             Micron Technology Inc (NVME:0x1344)
│     Serial Number:      22313BA6B538
│     GUIDs:              c8e1822c-2149-5bcb-9a06-aa35e2582a0f ← NVME\VEN_1344&DEV_51C3
│                         80fbda61-fc1b-5c19-92f1-d6bf84b09ead ← NVME\VEN_1344&DEV_51C3&REV_01
│                         c1952255-7a1a-5e26-90ff-54fa69b20e6f ← NVME\VEN_1344&DEV_51C3&SUBSYS_13442100
│                         b80c9795-9df5-53a1-a51c-45e1c75c02c6 ← NVME\VEN_1344&DEV_51C3&SUBSYS_13442100&REV_01
│                         f55bb2b1-385d-54d7-a7d1-db33a96b4b08 ← Micron_7450_MTFDKBA960TFR
│     Device Flags:       • Internal device
│                         • Updatable
│                         • System requires external power source
│                         • Needs a reboot after installation
│                         • Device is usable for the duration of the update
│
└─NVMe DC SSD [3DNAND, Sentinel Rock Controller]:
      Device ID:          6f772083971f483baa1771cde3563f754579fdd6
      Current version:    01
      Vendor:             Intel Corporation (PCI:0x8086, PCI:0x1022)
      GUIDs:              2edbbdc0-e0d2-550f-b485-0f1f789dca7d ← PCI\VEN_8086&DEV_0B60
                          092b8207-cfb6-57d8-ad17-66da6852527e ← PCI\VEN_8086&DEV_0B60&REV_00
                          d602810d-c2a2-5093-a4b8-e3355ebbc9b5 ← PCI\VEN_8086&DEV_0B60&SUBSYS_80869008
                          5fa6d8a3-ee1e-56fe-b1cd-275c90bf4948 ← PCI\VEN_8086&DEV_0B60&SUBSYS_80869008&REV_00
                          a23514a0-1ce1-5a4a-994a-bd31d0b6b3a2 ← PCI\VEN_1022&DEV_14AB
                          9d8cca9f-3474-5c3c-8c8a-eb14b868edb7 ← PCI\VEN_1022&DEV_14AB&REV_01
                          19957f2a-ee73-5294-86ed-e059ae64dea9 ← PCI\VEN_1022&DEV_14AB&SUBSYS_14581000
                          f44a65c6-061c-5629-a6c7-f1509ba3aaac ← PCI\VEN_1022&DEV_14AB&SUBSYS_14581000&REV_01
      Device Flags:       • Internal device
                          • Cryptographic hash verification is available
```

</details>

**Additional questions**

- Operating system and version: Debian 12 Bookworm.
- Have you tried rebooting? Yes.
- Is this a regression? Unclear. This is a fresh install.

alexis-aquanty, 2023-09-27:
This is the output when running `fwupd` manually:

<details>

<summary>fwupd output</summary>

```shell
$ sudo /usr/libexec/fwupd/fwupd -vv
15:24:42.856 FuDebug              Verbose debugging enabled (on console 1)
15:24:42.860 FuConfig             trying to load config values from /etc/fwupd/daemon.conf
15:24:42.860 FuCommon             reading /etc/fwupd/daemon.conf with 2563 bytes
15:24:42.860 FuConfig             trying to load config values from /var/etc/fwupd/daemon.conf
15:24:42.860 FuConfig             using autodetected max archive size 4.3 GB
15:24:42.861 FuRemoteList         loading /usr/share/fwupd/metainfo
15:24:42.861 XbSilo               attempting to load /var/cache/fwupd/metainfo.xmlb
15:24:42.861 XbSilo               file: 270eb25b-8503-976d-827a-c1d5bd02113f, current:270eb25b-8503-976d-827a-c1d5bd02113f, cached: (null)
15:24:42.861 XbSilo               loading silo with file contents
15:24:42.861 FuRemoteList         loading remote from /etc/fwupd/remotes.d/dell-esrt.conf
15:24:42.861 FuRemoteList         loading remote from /etc/fwupd/remotes.d/vendor-directory.conf
15:24:42.861 FuRemoteList         loading remote from /etc/fwupd/remotes.d/lvfs-testing.conf
15:24:42.861 FuRemoteList         loading remote from /etc/fwupd/remotes.d/lvfs.conf
15:24:42.861 FuRemoteList         loading remote from /etc/fwupd/remotes.d/vendor.conf
15:24:42.861 FuRemoteList         ordering lvfs-testing=lvfs+1
15:24:42.861 FuRemoteList         ignoring unfound remote fwupd
15:24:42.861 Jcat                 ignoring GPG-KEY-Linux-Vendor-Firmware-Service as not PKCS-7 certificate
15:24:42.861 Jcat                 ignoring GPG-KEY-Linux-Foundation-Firmware as not PKCS-7 certificate
15:24:42.861 Jcat                 trying to load certificate from /etc/pki/fwupd/LVFS-CA.pem
15:24:42.861 Jcat                 reading /etc/pki/fwupd/LVFS-CA.pem with 1679 bytes
15:24:42.861 Jcat                 loaded 1 certificates
15:24:42.861 Jcat                 ignoring GPG-KEY-Linux-Vendor-Firmware-Service as not PKCS-7 certificate
15:24:42.861 Jcat                 trying to load certificate from /etc/pki/fwupd-metadata/LVFS-CA.pem
15:24:42.861 Jcat                 reading /etc/pki/fwupd-metadata/LVFS-CA.pem with 1679 bytes
15:24:42.861 Jcat                 loaded 1 certificates
15:24:42.861 Jcat                 ignoring GPG-KEY-Linux-Foundation-Metadata as not PKCS-7 certificate
15:24:42.861 Jcat                 reading /var/lib/fwupd/pki/secret.key with 2455 bytes
15:24:42.861 Jcat                 reading /var/lib/fwupd/pki/client.pem with 1383 bytes
15:24:42.861 FuEngine             client certificate now exists: nothing to do
15:24:42.861 FuHistory            trying to open database '/var/lib/fwupd/pending.db'
15:24:42.861 FuHistory            got schema version of 8
15:24:42.864 FuPlugin             load(/usr/lib/x86_64-linux-gnu/fwupd-1.8.12/libfu_plugin_modem_manager.so)
15:24:42.865 FuContext            added udev subsystem watch of platform
15:24:42.865 FuContext            added udev subsystem watch of pci
15:24:42.865 FuContext            added udev subsystem watch of block
15:24:42.866 GLib                 setenv()/putenv() are not thread-safe and should not be used after threads are created
15:24:42.866 FuPluginDell         Using libsmbios 2.4
15:24:42.866 FuContext            added udev subsystem watch of i2c
15:24:42.866 FuContext            added udev subsystem watch of i2c-dev
15:24:42.866 FuContext            added udev subsystem watch of hidraw
15:24:42.866 FuContext            added udev subsystem watch of gpio
15:24:42.866 FuContext            added udev subsystem watch of mei
15:24:42.866 FuContext            added udev subsystem watch of iommu
15:24:42.866 FuContext            added udev subsystem watch of video4linux
15:24:42.866 FuContext            added udev subsystem watch of msr
15:24:42.866 FuContext            added udev subsystem watch of mtd
15:24:42.866 FuContext            added udev subsystem watch of nvme
15:24:42.866 FuContext            added udev subsystem watch of drm
15:24:42.866 FuContext            added udev subsystem watch of drm_dp_aux_dev
15:24:42.866 FuContext            added udev subsystem watch of serio
15:24:42.866 FuPluginTest         init
15:24:42.866 FuContext            added udev subsystem watch of thunderbolt
15:24:42.866 FuContext            added udev subsystem watch of tpm
15:24:42.867 FuIdle               setting timeout to 7200s
15:24:42.867 FuQuirks             loading quirks from /usr/share/fwupd/quirks.d
15:24:42.867 FuQuirks             loading quirks from /var/lib/fwupd/quirks.d
15:24:42.867 XbSilo               attempting to load /var/cache/fwupd/quirks.xmlb
15:24:42.867 XbSilo               file: ac7e1fd3-457f-6d98-2b4c-179526944fc4, current:ac7e1fd3-457f-6d98-2b4c-179526944fc4, cached: (null)
15:24:42.867 XbSilo               loading silo with file contents
15:24:42.867 FuContext            loading HwId overrides from /etc/fwupd/daemon.conf
15:24:42.867 FuContext            not loading HwId overrides from /var/lib/fwupd/daemon.conf
15:24:42.867 FuContext            smbios property Manufacturer=GIGABYTE
[... 689 lines cut ...]
15:25:09.608 FuEngine             skipping writing HSI attrs to database as unchanged
15:25:14.435 FuEngine             4a0dba5e-7801-5954-b7ca-e23bc44cb7fa was not found: no results for XPath query 'components/component['type',attr(),'firmware',eq()]/provides/firmware['type',attr(),'flashed',eq()text(),?0,eq()]/../..'
[... cut ...]
15:25:14.463 FuEngine             b80c9795-9df5-53a1-a51c-45e1c75c02c6 was not found: no results for XPath query 'components/component['type',attr(),'firmware',eq()]/provides/firmware['type',attr(),'flashed',eq()text(),?0,eq()]/../..'
15:25:14.465 FuEngine             f55bb2b1-385d-54d7-a7d1-db33a96b4b08 was not found: no results for XPath query 'components/component['type',attr(),'firmware',eq()]/provides/firmware['type',attr(),'flashed',eq()text(),?0,eq()]/../..'
15:25:14.465 FuEngine             writing motd target /var/cache/fwupd/motd.d/85-fwupd
```

</details>

superm1, 2023-09-27:
Can you try installing avahi daemon?
If that fixes it then it's a duplicate of an issue fixed in main that will be part of next release

alexis-aquanty, 2023-09-27:
> Can you try installing avahi daemon? If that fixes it then it's a duplicate of an issue fixed in main that will be part of next release

avahi-daemon seems to already be installed and running.

hughsie, 2023-09-27:
> 15:24:48.829 FuPlugin             fu_plugin_device_added(gpio)
> 15:25:09.242 FuDevice             changing verfmt for (null): plain->triplet

There's a huge amount of time being blocked here -- can you run fwupd under gdb and ctrl+c when it blocks and then get us a backtrace?

alexis-aquanty, 2023-09-27:
I don't seem to have debugging symbols for fwupd, so I'm not sure how useful this will be, but here you go:

<details>

<summary>gdb backtrace</summary>

```
$ sudo gdb --args /usr/libexec/fwupd/fwupd -vv
GNU gdb (Debian 13.1-3) 13.1
Copyright (C) 2023 Free Software Foundation, Inc.
License GPLv3+: GNU GPL version 3 or later <http://gnu.org/licenses/gpl.html>
This is free software: you are free to change and redistribute it.
There is NO WARRANTY, to the extent permitted by law.
Type "show copying" and "show warranty" for details.
This GDB was configured as "x86_64-linux-gnu".
Type "show configuration" for configuration details.
For bug reporting instructions, please see:
<https://www.gnu.org/software/gdb/bugs/>.
Find the GDB manual and other documentation resources online at:
    <http://www.gnu.org/software/gdb/documentation/>.

For help, type "help".
Type "apropos word" to search for commands related to "word"...
Reading symbols from /usr/libexec/fwupd/fwupd...
(No debugging symbols found in /usr/libexec/fwupd/fwupd)
(gdb) r
Starting program: /usr/libexec/fwupd/fwupd -vv
[Thread debugging using libthread_db enabled]
Using host libthread_db library "/lib/x86_64-linux-gnu/libthread_db.so.1".
15:52:54.626 FuDebug              Verbose debugging enabled (on console 1)
15:52:54.633 FuConfig             trying to load config values from /etc/fwupd/daemon.conf
15:52:54.633 FuCommon             reading /etc/fwupd/daemon.conf with 2563 bytes
15:52:54.633 FuConfig             trying to load config values from /var/etc/fwupd/daemon.conf
15:52:54.633 FuConfig             using autodetected max archive size 4.3 GB
[New Thread 0x7ffff31ff6c0 (LWP 43985)]
15:52:54.633 FuRemoteList         loading /usr/share/fwupd/metainfo
15:52:54.634 XbSilo               attempting to load /var/cache/fwupd/metainfo.xmlb
15:52:54.634 XbSilo               file: 270eb25b-8503-976d-827a-c1d5bd02113f, current:270eb25b-8503-976d-827a-c1d5bd02113f, cached: (null)
15:52:54.634 XbSilo               loading silo with file contents
15:52:54.634 FuRemoteList         loading remote from /etc/fwupd/remotes.d/dell-esrt.conf
15:52:54.634 FuRemoteList         loading remote from /etc/fwupd/remotes.d/vendor-directory.conf
15:52:54.634 FuRemoteList         loading remote from /etc/fwupd/remotes.d/lvfs-testing.conf
15:52:54.634 FuRemoteList         loading remote from /etc/fwupd/remotes.d/lvfs.conf
15:52:54.634 FuRemoteList         loading remote from /etc/fwupd/remotes.d/vendor.conf
15:52:54.634 FuRemoteList         ordering lvfs-testing=lvfs+1
15:52:54.634 FuRemoteList         ignoring unfound remote fwupd
15:52:54.634 Jcat                 ignoring GPG-KEY-Linux-Vendor-Firmware-Service as not PKCS-7 certificate
15:52:54.634 Jcat                 ignoring GPG-KEY-Linux-Foundation-Firmware as not PKCS-7 certificate
15:52:54.634 Jcat                 trying to load certificate from /etc/pki/fwupd/LVFS-CA.pem
15:52:54.634 Jcat                 reading /etc/pki/fwupd/LVFS-CA.pem with 1679 bytes
15:52:54.634 Jcat                 loaded 1 certificates
15:52:54.634 Jcat                 ignoring GPG-KEY-Linux-Vendor-Firmware-Service as not PKCS-7 certificate
15:52:54.634 Jcat                 trying to load certificate from /etc/pki/fwupd-metadata/LVFS-CA.pem
15:52:54.634 Jcat                 reading /etc/pki/fwupd-metadata/LVFS-CA.pem with 1679 bytes
15:52:54.635 Jcat                 loaded 1 certificates
15:52:54.635 Jcat                 ignoring GPG-KEY-Linux-Foundation-Metadata as not PKCS-7 certificate
15:52:54.635 Jcat                 reading /var/lib/fwupd/pki/secret.key with 2455 bytes
15:52:54.635 Jcat                 reading /var/lib/fwupd/pki/client.pem with 1383 bytes
15:52:54.635 FuEngine             client certificate now exists: nothing to do
15:52:54.635 FuHistory            trying to open database '/var/lib/fwupd/pending.db'
15:52:54.635 FuHistory            got schema version of 8
15:52:54.683 FuPlugin             load(/usr/lib/x86_64-linux-gnu/fwupd-1.8.12/libfu_plugin_modem_manager.so)
15:52:54.691 FuContext            added udev subsystem watch of platform
15:52:54.692 FuContext            added udev subsystem watch of pci
15:52:54.692 FuContext            added udev subsystem watch of block
15:52:54.692 GLib                 setenv()/putenv() are not thread-safe and should not be used after threads are created
15:52:54.692 FuPluginDell         Using libsmbios 2.4
15:52:54.692 FuContext            added udev subsystem watch of i2c
15:52:54.692 FuContext            added udev subsystem watch of i2c-dev
15:52:54.692 FuContext            added udev subsystem watch of hidraw
[... 332 lines cut ...]
#0  0x00007ffff775b03f in __GI___poll (fds=0x5555555c7c00, nfds=2, timeout=3996) at ../sysdeps/unix/sysv/linux/poll.c:29
#1  0x00007ffff78be9ae in ?? () from /lib/x86_64-linux-gnu/libglib-2.0.so.0
#2  0x00007ffff78beacc in g_main_context_iteration () from /lib/x86_64-linux-gnu/libglib-2.0.so.0
#3  0x00007ffff78beb11 in ?? () from /lib/x86_64-linux-gnu/libglib-2.0.so.0
#4  0x00007ffff78e8cfd in ?? () from /lib/x86_64-linux-gnu/libglib-2.0.so.0
#5  0x00007ffff76e8044 in start_thread (arg=<optimized out>) at ./nptl/pthread_create.c:442
#6  0x00007ffff77685fc in clone3 () at ../sysdeps/unix/sysv/linux/x86_64/clone3.S:81

Thread 1 (Thread 0x7ffff599f000 (LWP 43982) "fwupd"):
#0  0x00007ffff785c00c in ?? () from /lib/x86_64-linux-gnu/libxmlb.so.2
#1  0x00007ffff785c1b5 in ?? () from /lib/x86_64-linux-gnu/libxmlb.so.2
[... cut ...]
#6  0x00007ffff7d22566 in ?? () from /usr/lib/x86_64-linux-gnu/fwupd-1.8.12/libfwupdplugin.so
#7  0x00007ffff7cf0e05 in fu_device_probe () from /usr/lib/x86_64-linux-gnu/fwupd-1.8.12/libfwupdplugin.so
#8  0x00007ffff7e2f166 in ?? () from /usr/lib/x86_64-linux-gnu/fwupd-1.8.12/libfwupdengine.so
#9  0x00007ffff7e3d675 in fu_engine_load () from /usr/lib/x86_64-linux-gnu/fwupd-1.8.12/libfwupdengine.so
#10 0x0000555555561779 in fu_daemon_setup ()
#11 0x000055555555c08e in main ()
(gdb)
```

</details>

hughsie, 2023-09-27:
> I don't seem to have debugging symbols for fwupd

I think you do -- can you install the symbols for libxmlb and then grab the bt again please. I'm quite surprised to see that calltrace.

alexis-aquanty, 2023-09-27:
Debian doesn't seem to have a package for symbols for libxmlb2, but debuginfod seems to work:

<details>

<summary>gdb backtrace with debuginfod</summary>

```
$ sudo DEBUGINFOD_URLS="https://debuginfod.debian.net" gdb --args /usr/libexec/fwupd/fwupd -vv
GNU gdb (Debian 13.1-3) 13.1
Copyright (C) 2023 Free Software Foundation, Inc.
License GPLv3+: GNU GPL version 3 or later <http://gnu.org/licenses/gpl.html>
This is free software: you are free to change and redistribute it.
There is NO WARRANTY, to the extent permitted by law.
Type "show copying" and "show warranty" for details.
This GDB was configured as "x86_64-linux-gnu".
Type "show configuration" for configuration details.
For bug reporting instructions, please see:
<https://www.gnu.org/software/gdb/bugs/>.
Find the GDB manual and other documentation resources online at:
    <http://www.gnu.org/software/gdb/documentation/>.

For help, type "help".
Type "apropos word" to search for commands related to "word"...
Reading symbols from /usr/libexec/fwupd/fwupd...

This GDB supports auto-downloading debuginfo from the following URLs:
  <https://debuginfod.debian.net>
Enable debuginfod for this session? (y or [n]) y
Debuginfod has been enabled.
To make this setting permanent, add 'set debuginfod enabled on' to .gdbinit.
Reading symbols from /root/.cache/debuginfod_client/fdc8b3d112711edf07b2afcd8c8023b4847d44fa/debuginfo...
(gdb) r
Starting program: /usr/libexec/fwupd/fwupd -vv
[Thread debugging using libthread_db enabled]
Using host libthread_db library "/lib/x86_64-linux-gnu/libthread_db.so.1".
16:19:22.170 FuDebug              Verbose debugging enabled (on console 1)
16:19:22.178 FuConfig             trying to load config values from /etc/fwupd/daemon.conf
16:19:22.178 FuCommon             reading /etc/fwupd/daemon.conf with 2563 bytes
16:19:22.178 FuConfig             trying to load config values from /var/etc/fwupd/daemon.conf
16:19:22.178 FuConfig             using autodetected max archive size 4.3 GB
[New Thread 0x7ffff31ff6c0 (LWP 46657)]
16:19:22.179 FuRemoteList         loading /usr/share/fwupd/metainfo
16:19:22.179 XbSilo               attempting to load /var/cache/fwupd/metainfo.xmlb
16:19:22.180 XbSilo               file: 270eb25b-8503-976d-827a-c1d5bd02113f, current:270eb25b-8503-976d-827a-c1d5bd02113f, cached: (null)
16:19:22.180 XbSilo               loading silo with file contents
16:19:22.180 FuRemoteList         loading remote from /etc/fwupd/remotes.d/dell-esrt.conf
16:19:22.180 FuRemoteList         loading remote from /etc/fwupd/remotes.d/vendor-directory.conf
16:19:22.180 FuRemoteList         loading remote from /etc/fwupd/remotes.d/lvfs-testing.conf
16:19:22.180 FuRemoteList         loading remote from /etc/fwupd/remotes.d/lvfs.conf
16:19:22.180 FuRemoteList         loading remote from /etc/fwupd/remotes.d/vendor.conf
16:19:22.180 FuRemoteList         ordering lvfs-testing=lvfs+1
16:19:22.180 FuRemoteList         ignoring unfound remote fwupd
16:19:22.180 Jcat                 ignoring GPG-KEY-Linux-Vendor-Firmware-Service as not PKCS-7 certificate
16:19:22.180 Jcat                 ignoring GPG-KEY-Linux-Foundation-Firmware as not PKCS-7 certificate
16:19:22.180 Jcat                 trying to load certificate from /etc/pki/fwupd/LVFS-CA.pem
16:19:22.180 Jcat                 reading /etc/pki/fwupd/LVFS-CA.pem with 1679 bytes
16:19:22.181 Jcat                 loaded 1 certificates
16:19:22.181 Jcat                 ignoring GPG-KEY-Linux-Vendor-Firmware-Service as not PKCS-7 certificate
16:19:22.181 Jcat                 trying to load certificate from /etc/pki/fwupd-metadata/LVFS-CA.pem
16:19:22.181 Jcat                 reading /etc/pki/fwupd-metadata/LVFS-CA.pem with 1679 bytes
16:19:22.181 Jcat                 loaded 1 certificates
16:19:22.181 Jcat                 ignoring GPG-KEY-Linux-Foundation-Metadata as not PKCS-7 certificate
16:19:22.181 Jcat                 reading /var/lib/fwupd/pki/secret.key with 2455 bytes
16:19:22.181 Jcat                 reading /var/lib/fwupd/pki/client.pem with 1383 bytes
16:19:22.181 FuEngine             client certificate now exists: nothing to do
16:19:22.181 FuHistory            trying to open database '/var/lib/fwupd/pending.db'
16:19:22.181 FuHistory            got schema version of 8
16:19:22.312 FuPlugin             load(/usr/lib/x86_64-linux-gnu/fwupd-1.8.12/libfu_plugin_modem_manager.so)
16:19:22.337 FuContext            added udev subsystem watch of platform
16:19:22.337 FuContext            added udev subsystem watch of pci
[... 299 lines cut ...]
#6  0x00007ffff785c1b5 in xb_silo_query_section_root (self=self@entry=0x5555555cbb20, sn=<optimized out>, sn@entry=0x0, i=i@entry=0, bindings_offset=bindings_offset@entry=0, helper=helper@entry=0x7fffffffdd60,
    error=error@entry=0x7fffffffdeb8) at ../src/xb-silo-query.c:221
#7  0x00007ffff785c2f5 in xb_silo_query_part (self=self@entry=0x5555555cbb20, sroot=0x0, results=results@entry=0x55556006f340, results_hash=results_hash@entry=0x55555d91dea0, query=query@entry=0x55555567b0a0,
    context=context@entry=0x7fffffffdec0, first_result_only=0, query_data=0x7fffffffde30, flags=XB_SILO_QUERY_HELPER_NONE, error=0x7fffffffdeb8) at ../src/xb-silo-query.c:271
#8  0x00007ffff785ca24 in xb_silo_query_with_root_full (self=0x5555555cbb20, n=n@entry=0x0, query=0x55555567b0a0, context=context@entry=0x7fffffffdec0, first_result_only=first_result_only@entry=0,
    error=error@entry=0x7fffffffdeb8) at ../src/xb-silo-query.c:500
[... cut ...]
#18 fu_engine_backends_coldplug (progress=0x5555555bd690, self=0x5555555b2000) at ../src/fu-engine.c:8353
#19 fu_engine_load (self=<optimized out>, flags=<optimized out>, flags@entry=78, progress=0x5555555a9910, error=error@entry=0x7fffffffe348) at ../src/fu-engine.c:8712
#20 0x0000555555561779 in fu_daemon_setup (self=self@entry=0x5555555a9810, socket_address=socket_address@entry=0x0, error=error@entry=0x7fffffffe348) at ../src/fu-daemon.c:2231
#21 0x000055555555c08e in main (argc=<optimized out>, argv=<optimized out>) at ../src/fu-main.c:172
(gdb)
```

</details>

hughsie, 2023-09-27:
Would it be possible for you to try libxmlb 0.3.14? There are a couple of [fixed] bugs you *might* somehow be hitting.

alexis-aquanty, 2023-09-27:
libxmlb 0.3.14 seems to more or less fix the issue. There's still a slight delay, but it's much smaller than before, and the timeout no longer occurs:

<details>

<summary>fwupd output with libxmlb 0.3.14</summary>

```
$ sudo /usr/libexec/fwupd/fwupd -vv
17:08:30.152 FuDebug              Verbose debugging enabled (on console 1)
17:08:30.156 FuConfig             trying to load config values from /etc/fwupd/daemon.conf
17:08:30.156 FuCommon             reading /etc/fwupd/daemon.conf with 2563 bytes
17:08:30.156 FuConfig             trying to load config values from /var/etc/fwupd/daemon.conf
17:08:30.156 FuConfig             using autodetected max archive size 4.3 GB
17:08:30.156 FuRemoteList         loading /usr/share/fwupd/metainfo
17:08:30.156 XbSilo               attempting to load /var/cache/fwupd/metainfo.xmlb
17:08:30.156 XbSilo               file: 270eb25b-8503-976d-827a-c1d5bd02113f, current:270eb25b-8503-976d-827a-c1d5bd02113f, cached: (null)
17:08:30.156 XbSilo               loading silo with file contents
17:08:30.156 FuRemoteList         loading remote from /etc/fwupd/remotes.d/dell-esrt.conf
17:08:30.156 FuRemoteList         loading remote from /etc/fwupd/remotes.d/vendor-directory.conf
17:08:30.156 FuRemoteList         loading remote from /etc/fwupd/remotes.d/lvfs-testing.conf
17:08:30.156 FuRemoteList         loading remote from /etc/fwupd/remotes.d/lvfs.conf
17:08:30.156 FuRemoteList         loading remote from /etc/fwupd/remotes.d/vendor.conf
17:08:30.156 FuRemoteList         ordering lvfs-testing=lvfs+1
17:08:30.156 FuRemoteList         ignoring unfound remote fwupd
17:08:30.156 Jcat                 ignoring GPG-KEY-Linux-Vendor-Firmware-Service as not PKCS-7 certificate
17:08:30.156 Jcat                 ignoring GPG-KEY-Linux-Foundation-Firmware as not PKCS-7 certificate
17:08:30.156 Jcat                 trying to load certificate from /etc/pki/fwupd/LVFS-CA.pem
17:08:30.156 Jcat                 reading /etc/pki/fwupd/LVFS-CA.pem with 1679 bytes
17:08:30.157 Jcat                 loaded 1 certificates
17:08:30.157 Jcat                 ignoring GPG-KEY-Linux-Vendor-Firmware-Service as not PKCS-7 certificate
17:08:30.157 Jcat                 trying to load certificate from /etc/pki/fwupd-metadata/LVFS-CA.pem
17:08:30.157 Jcat                 reading /etc/pki/fwupd-metadata/LVFS-CA.pem with 1679 bytes
17:08:30.157 Jcat                 loaded 1 certificates
17:08:30.157 Jcat                 ignoring GPG-KEY-Linux-Foundation-Metadata as not PKCS-7 certificate
17:08:30.157 Jcat                 reading /var/lib/fwupd/pki/secret.key with 2455 bytes
17:08:30.157 Jcat                 reading /var/lib/fwupd/pki/client.pem with 1383 bytes
17:08:30.157 FuEngine             client certificate now exists: nothing to do
17:08:30.157 FuHistory            trying to open database '/var/lib/fwupd/pending.db'
17:08:30.157 FuHistory            got schema version of 8
17:08:30.160 FuPlugin             load(/usr/lib/x86_64-linux-gnu/fwupd-1.8.12/libfu_plugin_modem_manager.so)
17:08:30.161 FuContext            added udev subsystem watch of platform
17:08:30.161 FuContext            added udev subsystem watch of pci
17:08:30.161 FuContext            added udev subsystem watch of block
17:08:30.161 GLib                 setenv()/putenv() are not thread-safe and should not be used after threads are created
17:08:30.161 FuPluginDell         Using libsmbios 2.4
17:08:30.161 FuContext            added udev subsystem watch of i2c
17:08:30.161 FuContext            added udev subsystem watch of i2c-dev
17:08:30.161 FuContext            added udev subsystem watch of hidraw
17:08:30.161 FuContext            added udev subsystem watch of gpio
17:08:30.161 FuContext            added udev subsystem watch of mei
17:08:30.161 FuContext            added udev subsystem watch of iommu
17:08:30.161 FuContext            added udev subsystem watch of video4linux
17:08:30.161 FuContext            added udev subsystem watch of msr
17:08:30.161 FuContext            added udev subsystem watch of mtd
17:08:30.161 FuContext            added udev subsystem watch of nvme
17:08:30.162 FuContext            added udev subsystem watch of drm
17:08:30.162 FuContext            added udev subsystem watch of drm_dp_aux_dev
17:08:30.162 FuContext            added udev subsystem watch of serio
17:08:30.162 FuPluginTest         init
17:08:30.162 FuContext            added udev subsystem watch of thunderbolt
17:08:30.162 FuContext            added udev subsystem watch of tpm
17:08:30.162 FuIdle               setting timeout to 7200s
17:08:30.162 FuQuirks             loading quirks from /usr/share/fwupd/quirks.d
17:08:30.162 FuQuirks             loading quirks from /var/lib/fwupd/quirks.d
17:08:30.162 XbSilo               attempting to load /var/cache/fwupd/quirks.xmlb
17:08:30.162 XbSilo               file: ac7e1fd3-457f-6d98-2b4c-179526944fc4, current:ac7e1fd3-457f-6d98-2b4c-179526944fc4, cached: (null)
17:08:30.162 XbSilo               loading silo with file contents
17:08:30.163 FuContext            loading HwId overrides from /etc/fwupd/daemon.conf
17:08:30.163 FuContext            not loading HwId overrides from /var/lib/fwupd/daemon.conf
17:08:30.163 FuContext            smbios property Manufacturer=GIGABYTE
[... 689 lines cut ...]
17:08:45.432 FuEngine             400b0094-fe57-5a67-b6d8-dee4a7c44c50 was not found: indexed string '400b0094-fe57-5a67-b6d8-dee4a7c44c50' was unfound
[... cut ...]
17:08:45.432 FuEngine             b80c9795-9df5-53a1-a51c-45e1c75c02c6 was not found: indexed string 'b80c9795-9df5-53a1-a51c-45e1c75c02c6' was unfound
17:08:45.432 FuEngine             f55bb2b1-385d-54d7-a7d1-db33a96b4b08 was not found: indexed string 'f55bb2b1-385d-54d7-a7d1-db33a96b4b08' was unfound
17:08:45.432 FuEngine             writing motd target /var/cache/fwupd/motd.d/85-fwupd
```

</details>

Perhaps whatever change fixed this should be backported to libxmlb in Debian 12, since fwupd is nearly broken on this system without it.

superm1, 2023-09-27:
> libxmlb 0.3.14 seems to more or less fix the issue. There's still a slight delay, but it's much smaller than before, and the timeout no longer occurs:

Glad to hear, thanks!

> Perhaps whatever change fixed this should be backported to libxmlb in Debian 12, since fwupd is nearly broken on this system without it.

That's a bug for Debian not for upstream fwupd.