fwupd/fwupd #6225: "Not enough data was provided by the platform to make an HSI calculation" on NixOS

Jackaed, 2023-09-29:
**Describe the bug**
When using NixOS, when running `fwupdmgr security` or `fwupdtool security`, no information is output. This remains true when running `fwupdmgr security --force` or `fwupdtool security --force`.

**Output of `fwupdmgr --version`**
```
compile   org.freedesktop.fwupd         1.9.5
compile   com.hughsie.libxmlb           0.3.10
compile   com.hughsie.libjcat           0.1.14
runtime   org.freedesktop.fwupd-efi     1.3
compile   org.freedesktop.gusb          0.4.6
runtime   com.hughsie.libjcat           0.1.14
runtime   com.dell.libsmbios            2.4
runtime   org.freedesktop.gusb          0.4.6
runtime   org.kernel                    6.5.5
runtime   org.freedesktop.fwupd         1.9.5
```

Installed from nixpkgs unstable in /etc/nixos/configuration.nix with the line `services.fwupd.enable = true`

Output of `fwupdmgr get-devices --show-all-devices`: https://pastebin.com/uRtaQtHF

Is this expected behaviour on an immutable system? If so, even if some information cannot be gathered, would it still not be better to print out the information that it can gather, such as secure boot, the TPM, etc.

hughsie, 2023-09-29:
What does "sudo fwupdtool security -vv --force' print?

Jackaed, 2023-09-29:
https://pastebin.com/EP55guhv

superm1, 2023-09-29:
It looks to me this is because lockdown is disabled in kernel config or detected wrongly.  

```
14:19:00.217 FuPlugin             startup(linux_lockdown)
14:19:00.217 FuEngine             disabling plugin because: failed to startup using linux_lockdown: Kernel doesn't offer lockdown support.
```

Do you have it intentionally disabled?

Jackaed, 2023-09-29:
I don't have secure boot enabled as of right now (enabling it is a bit of a process on nixos and I've just changed machine), so yes, it's not enabled. I haven't changed anything in terms of kernel parameters though, I don't know why it says the kernel doesn't offer lockdown support.

Jackaed, 2023-09-29:
Edit: have now setup secure boot. Note that lanzaboote (the tool that allows you to enable secure boot on nixos) does not support kernel lockdown, see https://github.com/nix-community/lanzaboote/issues/88

superm1, 2023-09-29:
Secure boot and lockdown are two different things. This is tied to lockdown. If lockdown is not supported on NixOS this makes perfect sense your experience.

superm1, 2023-09-29:
Can you please share your kernel config?
Let's see if it's compiled without it.

If it's compiled with it - is the sysfs path to it wrong for NixOS perhaps?

Jackaed, 2023-09-29:
My only kernel-related setting in my configuration.nix is the following `boot.kernelPackages = pkgs.linuxPackages_latest;`, alongside booting with the kernel parameter `workqueue.power_efficient=true`.

superm1, 2023-09-29:
I mean the config used to build the kernel. Whatever goes into the build process.

Jackaed, 2023-09-29:
The kernel config in `/proc/config.gz` is: https://pastebin.com/KGLBZefi

superm1, 2023-09-29:
> The kernel config in `/proc/config.gz` is: https://pastebin.com/KGLBZefi

It's compiled without `CONFIG_SECURITY_LOCKDOWN_LSM`, so this is the root cause.

Can you get that enabled please?

Jackaed, 2023-10-02:
How come this compiler flag is required for fwupd to run? Surely this should just mean that the kernel lockdown security output is "disabled" and everything else prints as normal? This shouldn't prevent all the other information from being printed.

superm1, 2023-10-02:
> We can't give you an HSI score as we don't have enough information -- that said we should be able to display the other attrs.

We intentionally stopped doing that because people kept saying "well all these things are green and good, why can't you give me a score?"

I suppose what we could do here is if the kernel is missing lockdown support then we just "fail" the lockdown check.  You'll get a score.

Jackaed, 2023-10-02:
> We intentionally stopped doing that because people kept saying "well all these things are green and good, why can't you give me a score?"

I understand this from the perspective of preventing people from making stupid issues, but it seems a little weird to actively remove functionality just to prevent people from doing that. Perhaps reintroduce the other attributes and print a slightly more verbose message explaining why an HSI score can't be given? Either that, or add a flag like "--print-incomplete-attrs" or something to that effect to allow people who know what they're doing to still get the results.

Jackaed, 2023-10-02:
Also, adding the following to my configuration.nix did produce a working fwupd output:

```nix
  boot.kernelPatches = [ {
    name = "kernel-lockdown";
    patch = null;
    extraConfig = ''
      SECURITY_LOCKDOWN_LSM y
      MODULE_SIG y
    '';
  }];
```

Although this does obviously force nixos to compile the kernel from scratch, which isn't exactly quick to do.

superm1, 2023-10-02:
I've submitted a PR to change fwupd to allow no lockdown to give you a score, can you try that?

I do think that you should also request a change to the policy for default kernel configuration in Nix though.  Lockdown is a security feature you should be able to enable.

superm1, 2023-10-02:
> As of right now, nixos does not even officially support secure boot, instead you have to utilise an alternate bootloader (lanzaboote) to make it work. This alternate bootloader does not support kernel lockdown as of right now, see https://github.com/nix-community/lanzaboote/issues/88.

I suggest not to conflate secure boot and lockdown; they're totally independent.  Some distros link them together by out of tree patches, but that's not an upstream kernel behavior.

>  I don't know the full technical details on this, but on the surface it seems like a pretty stupid decision to remove the functionality from the kernel.

Yeah I suggest you change this.

closed 2023-10-02 by superm1