fwupd/fwupd #6227: 'get-devices' seems to say that a given update was and was not successful (_sic_)

LinuxOnTheDesktop, 2023-09-30:
**Describe the bug**
What is one to make of the following?

```
$ fwupdmgr get-devices
20KH001UKP
│
├─Embedded Controller:
│     Device ID:          2292ae5236790b47884e37cf162dcf23bfcd1c60
│     Summary:            UEFI ESRT device
│     Current version:    0.1.22
│     Minimum Version:    0.1.22
│     Vendor:             Lenovo (DMI:LENOVO)
│     Update State:       Success
│     Update Error:       Cannot install update when not on AC power
```

**Steps to Reproduce**
See above.

**Expected behavior**
The program should make it clear whether or not it is reporting a failed update.

**fwupd version information**

```
$ fwupdmgr --version
runtime   org.freedesktop.fwupd         1.7.9
runtime   com.dell.libsmbios            2.4
compile   org.freedesktop.gusb          0.3.10
compile   com.hughsie.libjcat           0.1.9
runtime   org.kernel                    6.2.0-33-generic
compile   org.freedesktop.fwupd         1.7.9
runtime   org.freedesktop.gusb          0.3.10
```

Please note how you installed it (`apt`, `dnf`, `pacman`, source, etc): apt

<details>

<summary>**fwupd device information**</summary>

Please provide the output of the fwupd devices recognized in your system.

```shell
fwupdmgr get-devices --show-all-devices
20KH001UKP
│
├─Core™ i7-8550U CPU @ 1.80GHz:
│     Device ID:          4bde70ba4e39b28f9eab1628f9dd6e6244c03027
│     Current version:    0x000000f4
│     Vendor:             Intel
│     GUIDs:              b9a2dd81-159e-5537-a7db-e7101d164d3f
│                         30249f37-d140-5d3e-9319-186b1bd5cac3
│                         561403e8-143a-5071-ab09-bf5e4c146983
│                         54bdc429-0179-5019-9cb6-8a343324c97c
│     Device Flags:       • Internal device
│   
├─Embedded Controller:
│     Device ID:          2292ae5236790b47884e37cf162dcf23bfcd1c60
│     Summary:            UEFI ESRT device
│     Current version:    0.1.22
│     Minimum Version:    0.1.22
│     Vendor:             Lenovo (DMI:LENOVO)
│     Update State:       Success
│     Update Error:       Cannot install update when not on AC power
│     GUID:               3babca5f-b2bf-4f4b-a72e-2bdc84eb4019
│     Device Flags:       • Internal device
│                         • System requires external power source
│                         • Supported on remote server
│                         • Needs a reboot after installation
│                         • Device is usable for the duration of the update
│                         • Updatable
│   
├─Intel Management Engine:
│     Device ID:          349bb341230b1a86e5effe7dfe4337e1590227bd
│     Summary:            UEFI ESRT device
│     Current version:    184.94.4494
│     Minimum Version:    0.0.1
│     Vendor:             Lenovo (DMI:LENOVO)
│     Update State:       Success
│     Update Error:       Cannot install update when not on AC power
│     GUID:               9c9d9769-32fa-4841-b550-ea998e754e99
│     Device Flags:       • Internal device
│                         • System requires external power source
│                         • Supported on remote server
│                         • Needs a reboot after installation
│                         • Device is usable for the duration of the update
│                         • Updatable
│   
├─LENSE20512GMSP34MEAT2TA:
│     Device ID:          04e17fcf7d3de91da49a163ffe4907855c3648be
│     Summary:            NVM Express solid state drive
│     Current version:    2.8.8341
│     Vendor:             Lenovo (NVME:0x17AA)
│     Update Error:       Cannot install update when not on AC power
│     GUIDs:              22eade43-1bc6-5eb2-ae69-b7d59beedeb8
│                         762c02f3-333f-5f89-a0c6-289659ae3cb6
│                         963522d7-bc73-51fa-a70c-cf615f42bcf6
│                         b0cd8903-04e1-51f7-a69a-f08b77e4a6ad
│                         ef93a17b-97f6-550e-a5bc-32c22ff6cdf8
│     Device Flags:       • Internal device
│                         • System requires external power source
│                         • Supported on remote server
│                         • Needs a reboot after installation
│                         • Device is usable for the duration of the update
│                         • Updatable
│   
├─SD/MMC CRW:
│     Device ID:          c185849ef3cd93ed1a4a04ca4f3658b85e281ce1
│     Summary:            SCSI device
│     Current version:    1.00
│     Vendor:             Generic- (SCSI:Generic)
│     GUIDs:              566e2bd5-c984-5dd9-932b-ab55be37a20a
│                         5bf7d01c-7e2b-58dd-84ec-986ffde3a056
│   
├─System Firmware:
│ │   Device ID:          a45df35ac0e948ee180fe216a5f703f32dda163f
│ │   Summary:            UEFI ESRT device
│ │   Current version:    0.1.59
│ │   Minimum Version:    0.0.1
│ │   Vendor:             Lenovo (DMI:LENOVO)
│ │   Update State:       Success
│ │   Update Error:       Cannot install update when not on AC power
│ │   GUIDs:              a4b51dca-8f97-4310-8821-3330f83c9135
│ │                       230c8b18-8d9b-53ec-838b-6cfc0383493a
│ │   Device Flags:       • Internal device
│ │                       • System requires external power source
│ │                       • Supported on remote server
│ │                       • Needs a reboot after installation
│ │                       • Cryptographic hash verification is available
│ │                       • Device is usable for the duration of the update
│ │                       • Updatable
│ │ 
│ └─UEFI dbx:
│       Device ID:        362301da643102b9f38477387e2193e57abaa590
│       Summary:          UEFI revocation database
│       Current version:  423
│       Minimum Version:  423
│       Vendor:           UEFI:Linux Foundation
│       Install Duration: 1 second
│       GUIDs:            14503b3d-73ce-5d06-8137-77c68972a341
│                         5971a208-da00-5fce-b5f5-1234342f9cf7
│                         c6682ade-b5ec-57c4-b687-676351208742
│                         f8ba2887-9411-5c36-9cee-88995bb39731
│       Device Flags:     • Internal device
│                         • Updatable
│                         • Needs a reboot after installation
│                         • Only version upgrades are allowed
│                         • Signed Payload
│     
├─TPM:
│     Device ID:          c6a80ac3a22083423992a3cb15018989f37834d6
│     Current version:    71.12.17568.4100
│     Vendor:             ST Microelectronics (TPM:STM)
│     GUIDs:              ff71992e-52f7-5eea-94ef-883e56e034c6
│                         84df3581-f896-54d2-bd1a-372602f04c32
│                         bfaed10a-bbc1-525b-a329-35da2f63e918
│                         70b7b833-7e1a-550a-a291-b94a12d0f319
│                         06f005e9-cb62-5d1a-82d9-13c534c53c48
│     Device Flags:       • Internal device
│   
├─UEFI Device Firmware:
│     Device ID:          f95c9218acd12697af946874bfe4239587209232
│     Summary:            UEFI ESRT device
│     Current version:    0.71.12
│     Minimum Version:    0.71.12
│     Vendor:             Lenovo (DMI:LENOVO)
│     Update State:       Success
│     Update Error:       Cannot install update when not on AC power
│     GUID:               cea87551-1701-43fb-afbc-6e8ce9728345
│     Device Flags:       • Internal device
│                         • System requires external power source
│                         • Supported on remote server
│                         • Needs a reboot after installation
│                         • Device is usable for the duration of the update
│                         • Updatable
│   
└─UHD Graphics 620:
      Device ID:          5792b48846ce271fab11c4a545f7a3df0d36e00a
      Current version:    07
      Vendor:             Intel Corporation (PCI:0x8086)
      GUIDs:              2920038a-3772-52ef-8f3d-92c2fb1ac115
                          0284ca6a-a235-5433-b190-1de21c753494
                          fed2efa4-7045-55a1-b4fc-b29283d59fe5
                          8b72a10c-1279-5f8e-a28a-34fb11a58240
      Device Flags:       • Internal device
                          • Cryptographic hash verification is available

```

</details>

**Additional questions**

- Operating system and version: Mint 21.2 Cinnamon
- Have you tried rebooting? Yes
- Is this a regression? I don't know

hughsie, 2023-10-01:
> │     Update State:       Success

This refers to the last update you applied. Maybe the title is misleading -- other ideas welcome.

LinuxOnTheDesktop, 2023-10-01:
If the last update was a success, then what - or when - is the _error_ that, additionally, the output reports?

hughsie, 2023-10-02:
> If the last update was a success, then what - or when - is the error that, additionally, the output reports?

If you update to a modern fwupd version you get much better output:

[image]

I think we solved this one quite some time ago, sorry.

closed 2023-10-02 by hughsie

superm1, 2023-10-02:
It's a problem that prevents an update from running.  If you try to run `fwupdmgr update` it won't let you try to run the update until you solve the problem.

superm1, 2023-10-02:
Can you please update to a current version of fwupd and experience how this all works?  Everything is clear from my perspective.

LinuxOnTheDesktop, 2023-10-10:
I think that a newer version of fwup would generate similar - similarly problematic output. 

I have tried hard to explain the problem - a problem that seems to me easy to be easy to grasp. As I say, the problem is this: fwupd seems to say that one and the same update succeeded and did not succeed. Do you understand that formulation? If you do, is the difficulty that you have that you do not see how the formulation fits the behaviour of fwupd?

hughsie, 2023-10-10:
So you want to change "`Update Status`" to a shorter version of "`Last Performed Update Status`"?

LinuxOnTheDesktop, 2023-10-21:
> So you want to change "Update Status" to a shorter version of "Last Performed Update Status"?

I do not think so (and you did not answer the questions that I posed; I posed those questions because I thought that, if you answered them, then we might understand each other). I want to resolve the seeming contradiction between the following two lines of output.

`Update State:       Success`

`Update Error:       Cannot install update when not on AC power`

Still: just now, and belatedly, I took up the suggestion of running a newer version of fwup. I did so by installing the flatpak. Running `get-devices` via the flatpak produced, _inter alia_, the following output.

```
─System Firmware:
│ │   Device ID:          a45df35ac0e948ee180fe216a5f703f32dda163f
│ │   Summary:            UEFI ESRT device
│ │   Current version:    0.1.61
│ │   Minimum Version:    0.0.1
│ │   Vendor:             Lenovo (DMI:LENOVO)
│ │   Update State:       Success
│ │   GUIDs:              a4b51dca-8f97-4310-8821-3330f83c9135
│ │                       230c8b18-8d9b-53ec-838b-6cfc0383493a ← main-system-firmware
│ │   Device Flags:       • Internal device
│ │                       • Updatable
│ │                       • System requires external power source
│ │                       • Supported on remote server
│ │                       • Needs a reboot after installation
│ │                       • Cryptographic hash verification is available
│ │                       • Device is usable for the duration of the update

```
So, with the newer version of fwup, _at least at present_, fwup does not seem to both assert and deny that any attempted update on my system was both a success and a failure. (Still: the program did report the following: `18:59:08.935 FuPluginIntelMe      failed to get public key using /fpf/OemCred: generic failure [0xb]` - but never mind.)